evcc-io/evcc #14542: Erweiterte Anzeige warum gerade nicht geladen wird.

HaareistNase, 2024-06-24:
[image]

Meine Frau und ich sitzen zur Zeit manchmal ratlos vor EVCC und wundern uns, warum das Laden vom Überschuss nicht startet.
Ich würde mich über eine erweiterte Anzeige freuen an der Stelle wo sowieso der Status angezeigt wird.
Also z.b. anstatt "Warte auf Fahrzeug" oder "Stoppe, weil zu wenig Überschuss" auch noch ein:

Min-Preisgrenze nicht erreicht, nicht genug Überschuss vorhanden, Schalter steht auf Aus ... etc. was es für Gründe gibt.

Eventuell sind wir ja nur zu ungeduldig und warten nicht lange genug. Aber es ist halt komisch vor der Anzeige zu sitzen und sich zu fragen warum es nicht losgeht, obwohl gerade unglaublich viel Strom vom Dach ins öffentliche Netz wandert.

Das wäre total toll und ein cooles Feature für die Transparenz der EVCC-Methoden.

HaareistNase, 2024-06-24:
Hihi, DAS ist zu viel für meine Frau.
Die stellt zur Zeit panisch an allen Schaltern, solange bis EVCC den Wagen lädt anstatt Vertrauen zu haben.
Deshalb sind wir ja nur auf die Idee gekommen, ob es möglich wäre da etwas mehr Info anzuzeigen.

StefanSchoof, 2024-06-24:
Ähnliches Issue #12566

naltatis, 2024-06-24:
@HaareistNase generell probieren wir ja schon in der Statusmeldung, immer die wichtigste Information anzuzeigen. Ich würd hier ungern den gesamten Zustand in Textform ausklappen. Damit wird vermutlich niemand richtig glücklich.

Lass uns hier lieber über konkrete Situationen reden, in denen der aktuelle Status nicht weiterhilft. In der Tat haben wir heute schon Fälle, wo der Text nicht optimal ist. Bspw. übersteuert heute ein aktiver Ladeplan die PV-Timer. Siehe auch https://github.com/evcc-io/evcc/issues/14408

Das werden wir auf jeden Fall so umbauen, dass in solchen Fällen beide Informationen sichtbar bleiben. Momentan gibts immer nur genau eine Information.

HaareistNase, 2024-06-24:
@naltatis 
Da schau ich dann das nächste mal, wenn die Überschussladung nicht startet genauer ins Log WARUM sie nicht startet, und wünsche mir das als Statusmeldung.
Aber z.b. das OFFENSICHTLICHE (das man so gerne übersieht, weil man blind ist) z.b. : Schalter ist auf AUS (klar, der Schalter ist direkt daneben, aber wenn man Betriebsblind ist, übersieht man das tatsächlich).