evcc-io/evcc #14550: Ladevorgang startet nicht mehr zuverlässig

cerwin86, 2024-06-24:
### Describe the bug

Seit gefühlt 4-6 Wochen ist der Ladevorgang per evcc sehr instabil geworden. Wenn man das Auto ansteckt, steht es auf "Ladebereit, warten auf Fahrzeug". Nichts passiert. Reboot des evcc Containers und es startet in Sekunden.
Wenn dann die Sonne weg war etc. dann kommt es auch häufiger vor, dass es wieder auf "Ladebereit, warten auf Fahrzeug" steht. Auch hier hilft ein restart des containers sofort.
Sehr merkwürdig irgendwie. Die Logs werfen mir auch keine Fehler aus. S.u.

Hat das schon jmd. beobachtet?

(ich musste leider die API Responses von Zaptec rauscutten, waren zu lang)

### Steps to reproduce

1. Auto anschließen
2. Parameter für PV prüfen (genug Dampf vom Dach bzw. Akku voll genug gemäß Voreinstellung
3. nichts passiert
4. 15 Min warten
5. nichts passiert
6. Container restart
7. lädt
...

### Configuration details

```yaml
log: trace
levels:
  site: info
  lp-1: debug
  lp-2: debug
  cache: error
  db: error
meters:
  - name: grid
    type: template
    template: sungrow-hybrid
    usage: grid
    modbus: tcpip
    id: 1
    host: 1x
    port: 502
  - name: my_pv
    type: template
    template: sungrow-hybrid
    usage: pv
    modbus: tcpip
    id: 1
    host: 1x
    port: 502
  - name: my_battery
    type: template
    template: sungrow-hybrid
    usage: battery
    modbus: tcpip
    id: 1
    host: 1x
    port: 502
  - name: wp_pwr
    type: custom
    power:
      source: mqtt
      topic: WP-Meter/status/em:0
      jq: if .total_act_power < 100 then 0 else .total_act_power end
chargers:
  - name: zaptec-links
    type: template
    template: zaptec
    user: xxx
    password: xxx
  - name: WP_SG
    type: custom
[... 530 lines cut ...]
[lp-2  ] DEBUG 2024/06/24 19:10:34 charge power: 0W
[lp-1  ] DEBUG 2024/06/24 19:10:35 !! session: chargeRater.chargedEnergy=3.5 - chargedAtStartup=1.1
[lp-1  ] DEBUG 2024/06/24 19:10:35 charger status: B
[lp-1  ] INFO 2024/06/24 19:10:35 stop charging <-
[lp-1  ] DEBUG 2024/06/24 19:10:35 !! session: chargedEnergy=2.3 > chargedEnergy=0.0
[lp-1  ] DEBUG 2024/06/24 19:10:35 !! session: start=0.000 stop=0.000 chargedEnergy=2.340
[lp-1  ] DEBUG 2024/06/24 19:10:35 soc estimated: 62.48% (vehicle: 61.00%)
[lp-1  ] DEBUG 2024/06/24 19:10:35 vehicle soc: 62%
[lp-1  ] DEBUG 2024/06/24 19:10:35 vehicle range: 111km
[zaptec] TRACE 2024/06/24 19:10:35 GET https://api.zaptec.com/api/chargers/xxxx/state

[lp-1  ] DEBUG 2024/06/24 19:10:35 !! active phases: 3p = min(3p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/06/24 19:10:35 !! active phases: 3p = min(3p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/06/24 19:10:35 pv charge current: 0.575A = 0A + 0.575A (-397W @ 3p)
[lp-1  ] DEBUG 2024/06/24 19:10:35 pv enable timer reset
[lp-1  ] DEBUG 2024/06/24 19:10:35 pv timer inactive
```

### What type of operating system are you running?

Linux

### Version

0.127.2

andig, 2024-06-24:
> Seit gefühlt 4-6 Wochen ist der Ladevorgang per evcc sehr instabil geworden

Was war da? Hilft ein Downgrade auf die alte Version?

cerwin86, 2024-06-25:
In der Log ist vor und nach Reboot. (ca. 18:36) ich sehe keinen unterschied, außer dass danach sofort charging start passiert.

die 4-6 Wochen sind natürlich nur gefühlt, können auch 8 Wochen sein.
Aber ja ich probiere heute Abend mal ne Version ~4 Monate zurück. Mal sehen. Dann ist auch das Auto wieder da 😂

cerwin86, 2024-06-25:
Erstes Ergebnis 0.124.5 startete sofort einen Ladevorgang. Beobachte es weiter und gehe Major für Major hoch. 🤞

andig, 2024-06-25:
Super Find! Bisect hilft, dann sollten wir die Logfiles vergleichen. Mir fällt kein guter Grund für Unterschiede ein- da hat sich eigentlich wenig/nix geändert.

cerwin86, 2024-06-25:
Zu früh gefreut…
Hat gestartet und geladen aber nach der ersten Pause ist wieder Ende. Restart hat wieder sofort geholfen. 

Gibt es noch andere Logs außer
log: trace
levels:
  site: info
  lp-1: debug
  lp-2: debug
  cache: error
  db: error
?

cerwin86, 2024-06-28:
Ne es bleibt merkwürdig. Auch mit dieser älteren Version. 
Laden startet nicht 
Oder 
Laden setzt nicht mehr fort
Oder
Laden läuft mit Vollgas, obwohl PV Modus

alle Varianten ohne Meldungsicon oben rechts.

jedes mal hilft immer zuverlässig ein restart des Containers.

In jedem dieser Fälle war es aber ein Batterieunterstütztes Laden. Gab es hier erwähnenswerte changes?

andig, 2024-06-29:
Ich denke es liegt am Zaptec, erinnere mich das schonmal gehört zu haben. Müssen wir näher rein leuchten.

cerwin86, 2024-06-29:
Will ich auch nicht ausschließen, ja.
Kann ich noch andere / tiefere Logs ziehen?

andig, 2024-06-29:
Die Logs sind eigentlich ok, allerdings scheint der Status hier zu fehlen:

> GET https://api.zaptec.com/api/chargers/xxxx/state

Wo ist die Antwort hin?

cerwin86, 2024-06-29:
Die musste ich cutten, da das issue Formular so viel Text nicht erlaubt. Poste morgen was zurück kommt

cerwin86, 2024-07-03:
Moin again :)
also nach weiterem geteste sehe ich, dass manchmal einfach die API Benachrichtigungen an Zaptec ausbleiben

Eben wieder den Fall gehabt, dass der Ladevorgang trotz PV auf Vollgas lief.
Die Logs zeigten, einen fleißigen Abruf des Charger Status alle 30sek. Zusätzlich loggte evcc 11kw Ladeleistung.

Nach einem Reboot (5 min 11kw laden vergehen lassen) erfolgten direkt 2 API Calls
Command 507 (charging start) -> ok in dem fall doppelt gemoppelt
und direkt darauf der fehlende Update max Charge current. der vor restart nicht passiert ist.

Vor reboot:
`[lp-1  ] INFO 2024/07/03 11:11:13 start charging ->
[lp-1  ] DEBUG 2024/07/03 11:11:13 wake-up timer: stop
[lp-1  ] DEBUG 2024/07/03 11:11:13 soc estimated: 81.39% (vehicle: 80.00%)
[lp-1  ] DEBUG 2024/07/03 11:11:13 vehicle soc: 81%
[lp-1  ] DEBUG 2024/07/03 11:11:13 vehicle range: 144km
[lp-1  ] DEBUG 2024/07/03 11:11:13 !! active phases: 3p = min(3p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/07/03 11:11:13 !! active phases: 3p = min(3p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/07/03 11:11:13 pv charge current: min 6A > 0A (7883W @ 3p, battery: true)
[lp-1  ] DEBUG 2024/07/03 11:11:42 charge power: 10969W
[lp-1  ] DEBUG 2024/07/03 11:11:42 charge currents: [15.3 15.2 15.3]A
[lp-1  ] DEBUG 2024/07/03 11:12:12 charge power: 10969W
[lp-1  ] DEBUG 2024/07/03 11:12:12 charge currents: [15.3 15.2 15.3]A
[lp-1  ] DEBUG 2024/07/03 11:12:13 detected active phases: 3p
[lp-1  ] DEBUG 2024/07/03 11:12:13 !! session: chargeRater.chargedEnergy=1.1 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/07/03 11:12:13 charger status: C
[lp-1  ] DEBUG 2024/07/03 11:12:13 soc estimated: 81.74% (vehicle: 80.00%)
[lp-1  ] DEBUG 2024/07/03 11:12:13 vehicle soc: 82%
[lp-1  ] DEBUG 2024/07/03 11:12:13 vehicle range: 144km
[lp-1  ] DEBUG 2024/07/03 11:12:13 !! active phases: 3p = min(3p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/07/03 11:12:13 !! active phases: 3p = min(3p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/07/03 11:12:13 pv charge current: min 6A > 0A (7923W @ 3p, battery: true)
[lp-1  ] DEBUG 2024/07/03 11:12:42 charge power: 10969W
[lp-1  ] DEBUG 2024/07/03 11:12:42 charge currents: [15.3 15.2 15.3]A
[lp-1  ] DEBUG 2024/07/03 11:13:12 charge power: 10969W
[lp-1  ] DEBUG 2024/07/03 11:13:12 charge currents: [15.3 15.2 15.3]A
[lp-1  ] DEBUG 2024/07/03 11:13:13 detected active phases: 3p
[lp-1  ] DEBUG 2024/07/03 11:13:13 !! session: chargeRater.chargedEnergy=1.3 - chargedAtStartup=0.0
[zaptec] TRACE 2024/07/03 11:11:42 GET https://api.zaptec.com/api/chargers/46181390-a597-4272-9fc8-f38edd7bf1bb/state
[zaptec] TRACE 2024/07/03 11:11:42 [{"ChargerId":"46181390-a597-4272-9fc8-f38edd7bf1bb","StateId":-2,"Timestamp":"2024-06-29T14:57:20.437","ValueAsString":"1"},{"ChargerId":"46181390-a597-4272-9fc8-f38edd7bf1bb","StateId":-1,"Timestamp":"2024-07-03T09:10:17.627"},{"ChargerId":"46181390-a597-4272-9fc8-f38edd7bf1bb","StateId":1,"Timestamp":"2023-06-28T13:42:53.993","ValueAsString":"0"},{"ChargerId":"46181390-a597-4272-9fc8-f38edd7bf1bb","StateId":100,"Timestamp":"2024-06-23T09:05:05.793","ValueAsString":"{\"CommunicationModes\":[\"LTE\",\"WiFi\"],\"DeviceType\":\"Go\",\"MeterCalibrated\":false,\"OcppVersions\":[\"V1_6\"],\"SchemaVersion\":\"2.1\"}"},{"ChargerId":"46181390-a597-4272-9fc8-f38edd7bf1bb","StateId":120,"Timestamp":"2023-07-27T06:46:51.08","ValueAsString":"1"},{"ChargerId":"46181390-a597-4272-9fc8-f38edd7bf1bb","StateId":145,"Timestamp":"2023-06-28T13:48:17.133","ValueAsString":"3600"},{"ChargerId":"46181390-a597-4272-9fc8-f38edd7bf1bb","StateId":147,"Timestamp":"2024-07-03T09:10:17.267","ValueAsString":"180"},{"ChargerId":"46181390-a597-4272-9fc8-f38edd7bf1bb","StateId":150,"Timestamp":"2024-06-04T16:01:53.297","ValueAsString":"LTE"},{"ChargerId":"46181390-a597-4272-9fc8-f38edd7bf1bb","StateId":151,"Timestamp":"2024-02-24T15:30:41.103","ValueAsString":"0"},{"ChargerId":"46181390-a597-4272-9fc8-f38edd7bf1bb","StateId":153,"Timestamp":"2023-07-25T15:19:42.34","ValueAsString":"0.137"},{"ChargerId":"46181390-a597-4272-9fc8-f38edd7bf1bb","StateId":201,"Timestamp":"2024-07-03T08:57:22.51","ValueAsString":"34.525"},{"ChargerId":"46181390-a597-4272-9fc8-f38edd7bf1bb","StateId":202,"Timestamp":"2024-07-03T08:57:22.51","ValueAsString":"27.966"},{"ChargerId":"46181390-a597-4272-9fc8-f38edd7bf1bb","StateId":204,"Timestamp":"2024-07-03T08:57:22.513","ValueAsString":"28.343"},{"ChargerId":"46181390-a597-4272-9fc8-f38edd7bf1bb","StateId":205,"Timestamp":"2024-07-01T22:31:06.713","ValueAsString":"39.188"},{"ChargerId":"46181390-a597-4272-9fc8-f38edd7bf1bb","StateId":206,"Timestamp":"2024-07-03T08:57:22.513","ValueAsString":"26.699"},{"ChargerId":"46181390-a597-4272-9fc8-f38edd7bf1bb","StateId":207,"Timestamp":"2024-07-03T08:57:22.513","ValueAsString":"27.666"},{"ChargerId":"46181390-a597-4272-9fc8-f38edd7bf1bb","StateId":220,"Timestamp":"2024-07-02T10:47:18.17","ValueAsString":"40.224"},{"ChargerId":"46181390-a597-4272-9fc8-f38edd7bf1bb","StateId":221,"Timestamp":"2024-07-02T10:47:18.17","ValueAsString":"39.188"},{"ChargerId":"46181390-a597-4272-9fc8-f38edd7bf1bb","StateId":270,"Timestamp":"2024-07-03T08:57:22.51","ValueAsString":"33.712"},{"ChargerId":"46181390-a597-4272-9fc8-f38edd7bf1bb","StateId":501,"Timestamp":"2024-07-03T07:40:00.86","ValueAsString":"237.418"},{"ChargerId":"46181390-a597-4272-9fc8-f38edd7bf1bb","StateId":502,"Timestamp":"2024-07-03T07:40:00.863","ValueAsString":"238.0 [... cut ...] 556.209,\\\"RI\\\":\\\"1-0:1.8.0\\\",\\\"RU\\\":\\\"kWh\\\",\\\"RT\\\":\\\"AC\\\",\\\"ST\\\":\\\"G\\\"},{\\\"TM\\\":\\\"2024-07-02T09:00:00,000+00:00 R\\\",\\\"TX\\\":\\\"T\\\",\\\"RV\\\":1557.819,\\\"RI\\\":\\\"1-0:1.8.0\\\",\\\"RU\\\":\\\"kWh\\\",\\\"RT\\\":\\\"AC\\\",\\\"ST\\\":\\\"G\\\"},{\\\"TM\\\":\\\"2024-07-02T10:47:18,000+00:00 R\\\",\\\"TX\\\":\\\"E\\\",\\\"RV\\\":1557.819,\\\"RI\\\":\\\"1-0:1.8.0\\\",\\\"RU\\\":\\\"kWh\\\",\\\"RT\\\":\\\"AC\\\",\\\"ST\\\":\\\"G\\\"}]}\"}"},{"Charger
`

andig, 2024-07-03:
> also nach weiterem geteste sehe ich, dass manchmal einfach die API Benachrichtigungen an Zaptec ausbleiben

In dem Fall könnte man Phasen/Strom auch nachlesen wenn wir wissen wo die im API stehen.

andig, 2024-07-06:
> Die Logs zeigten
> @cerwin86 magst Du das mal bitte lesbar als Codeblock formatieren?

ping

PumpyChuck, 2024-07-06:
Ja das kann ich so auch bestätigen. Grad jetzt lade ich mein Auto und er scheint auf 3.7kW stecken zu bleiben.

[image]

Auch in der Grafik sieht man deutlich, dass der Ladevorhang konstant auf 3.7kW bleibt obwohl mehr geladen werden könnte.

[image]

Meiner Meinung nach hat es was mit der Priority Einstellung zu tun. Wen ich das hier über die Swagger Schnittstelle auf 3 setze und es zu meiner Zaptec Box sende, dann wird auch mit drei Phasen geladen. Ich dachte das wurde auch so im Code implementiert. Aber irgendwie klappt das noch nicht richtig.

[image]

Wenn ich die Priority auf 3 setze dann wird zwar auch mit drei Phasen geladen allerdings gibt es dann im UI diese Fehlermeldung.

[image]

[trace_log_zaptec_30min_060724._2txt.txt](https://github.com/user-attachments/files/16115502/trace_log_zaptec_30min_060724._2txt.txt)

andig, 2024-07-06:
@PumpyChuck ich habe leider nicht die Zeit, tausende von Logfiles auf potentielle Fehler durchzusuchen. Deshalb- wenn Du Hilfe erwartest- bitte konkret:
- zeitlich wo ist der Fehler?
- was zeigt das (ansonsten vollständige) Logfile?
- was hättest Du stattdessen erwartet

Die Erwartung, dass ich mir die Informationen aus Screenshots und sonstwas zusammen suche kann ich nicht erfüllen.

PumpyChuck, 2024-07-06:
Sorry war nicht meine Absicht. 

Der Fehler ist über eine längere Zeit sicherlich 30 Minuten. Ich habe nun aus dem Log den State hier dargestellt mit den Angaben.

[image]

Nur der Status 501 und 507 haben einen Wert sprich es wird nur 1-phasig geladen. 

501-503 --> Output voltage phase 1-3 in volts
507-509 --> Output current phase 1-3 in amperes. 

513 --> Total instant charge power in watts

Der Status 513 gibt hier 3.7kW, an obwohl zu diesem Zeitpunkt genügend Überschuss vorhanden ist. (ca. 4kW). Aber nicht nur grad bei dieser Messung, sondern auch fast die ganzen 30 Minuten zuvor (sorry aber siehe Grafana Grafik).

{"ChargerId":"xxx-xxx-xxx-xxx-xxxx","StateId":501,"Timestamp":"2024-07-06T09:21:14.173","ValueAsString":"237.9799"},
{"ChargerId":"xxx-xxx-xxx-xxx-xxxx","StateId":502,"Timestamp":"2024-07-06T09:21:14.173","ValueAsString":"1.1098"},
{"ChargerId":"xxx-xxx-xxx-xxx-xxxx","StateId":503,"Timestamp":"2024-07-06T09:21:14.173","ValueAsString":"0.9952"},

{"ChargerId":"xxx-xxx-xxx-xxx-xxxx","StateId":507,"Timestamp":"2024-07-06T09:21:14.173","ValueAsString":"15.6404"},
{"ChargerId":"xxx-xxx-xxx-xxx-xxxx","StateId":508,"Timestamp":"2024-07-06T09:21:14.173","ValueAsString":"0.1265"},
{"ChargerId":"xxx-xxx-xxx-xxx-xxxx","StateId":509,"Timestamp":"2024-07-06T09:21:14.173","ValueAsString":"0.078"},

{"ChargerId":"xxx-xxx-xxx-xxx-xxxx","StateId":513,"Timestamp":"2024-07-06T09:21:14.173","ValueAsString":"3688.9863"},

[image]

Ich würde zu diesem Zeitpunkt erwarten, dass er 3-phasig lädt und den restlichen Übrschuss nutzt für das Laden des E-Autos.

Gruss

andig, 2024-07-06:
@PumpyChuck ich muss es jetzt nochmal sagen. Screenshots und Fließtext sind völlig unnütz wenn es dazu kein Logfile gibt. Soweit ich sehe hast du das jetzt gelöscht. Es braucht: aussagekräftiges Logfile plus relevanten Zeitraum plus Fehlerbeschreibung. Jeweils nur eins der drei Dinge zu haben ist massiv frustierend und zeitraubend. Ich kann NICHT hellsehen!!!

andig, 2024-07-09:
> Eben wieder den Fall gehabt, dass der Ladevorgang trotz PV auf Vollgas lief.

Dein Log ist nutzlos. evcc denke die Box läuft auf 6A, tut sie  aber nicht. Wie sie in den Zustand kommt ist mangels Log nicht erkennbar.

Ich mache hier zu bis es ein verwertbares Logfile gibt.

closed 2024-07-09 by andig

cerwin86, 2024-07-09:
hmmm was genau hilft dir in den Logs? Habe die so kurz gehalten, da ich als ich den issue posten wollte nicht mehr als 64k zeichen posten durfte. was allein 3 Antworten von Zaptec sind :(

Das heißt in dieser Zeile sagt evcc die WB sollte eig. auf 6A laufen
[lp-1  ] DEBUG 2024/07/03 11:11:13 pv charge current: min 6A > 0A (7883W @ 3p, battery: true)

In der Folgezeile kommt dann quasi das unerwartete Vollgas Ergebnis vom Meter zurück. Evcc sagt hää ich habe doch 6A gesagt, right? Dann wühle ich weiter in den Logs wenn ich den Fall wieder reproduzieren kann.
[lp-1  ] DEBUG 2024/07/03 11:11:42 charge power: 10969W
[lp-1  ] DEBUG 2024/07/03 11:11:42 charge currents: [15.3 15.2 15.3]A
[lp-1  ] DEBUG 2024/07/03 11:12:12 charge power: 10969W
[lp-1  ] DEBUG 2024/07/03 11:12:12 charge currents: [15.3 15.2 15.3]A

Ah und in der Zaptec State Antwort liefert Zaptec sogar, auf den Werten 507,508,509 (Current Phase 1-3) jeweils grob 6A aktuelle Leistung zurück. Wert 510 (max Charge current) steht auch auf 6A. Dann kommt die falsche Info definitiv von Zaptec.
Das heißt der Reboot triggert einfach nur ein normales komme was da wolle erneutes setzen über die API weil frischer start. hat Quasi nix mit dem aktuelle Status zu tun.

Ich beobachte weiter Danke!

andig, 2024-07-16:
> hmmm was genau hilft dir in den Logs? Habe die so kurz gehalten, da ich als ich den issue posten wollte nicht mehr als 64k zeichen posten durfte. was allein 3 Antworten von Zaptec sind :(

Es braucht den relevanten Teil *vor* dem Fehler. Hier können auch Dateien hoch geladen werden 👍🏻 

> In der Folgezeile kommt dann quasi das unerwartete Vollgas Ergebnis vom Meter zurück. 

evcc versucht auch solche Ereignisse einzufangen, macht das im Moment allerdings nur auf Basis der Phasenströme (die es dann als zu hoch erkennt). Die Logik ist: erst gesetzten Strom überprüfen, falls es die Möglichkeit nicht gibt dann gemessenen Strom überprüfen. Bei der Zaptec ist `api.CurrentGetter` nicht implementiert, wohl aber `api.PhaseCurrents`. Anhand dessen müsste evcc eigentlich merken, dass da etwas nicht stimmt (15A statt 6A). Das scheint bei dir nicht zu funktionieren.

Es braucht weiterhin mal ein ausführliches Log mit genug Kontext *vor* dem Fehler.