evcc-io/evcc #14556: Fail to boot after upgrade from 0.127.2 to 0.127.3

jivesinger, 2024-06-25:
### Describe the bug

Running HAOS addon 0.127.2 with no issues
Upgrade to HAOS addon 0.127.3

Container fails to start with error:

`FATAL 2024/06/25 15:52:23 cannot create vehicle 'ev3': cannot create vehicle type 'template': cannot create vehicle type 'custom': decoding failed due to the following error(s): '' has invalid keys: vin`

From documentation at [https://docs.evcc.io/en/docs/devices/vehicles#mg2mqtt](https://docs.evcc.io/en/docs/devices/vehicles#mg2mqtt)

```
vehicles:
  - name: my_car
    type: template
    template: mg2mqtt
    title: # Will be displayed in the user interface (optional)
    user: # User account (e.g. email address, user id, etc.)
    vin: W... # Required if you own multiple vehicles of the same brand
    capacity: 50 # Battery capacity in kWh (optional)
    timeout: 1h # optional 
```

Key 'vin' looks to be optional as I don't own multiple vehicles of the same brand.
Commenting out key 'vin' allows evcc to start, but after plugging in the vehicle evcc does not recoginize that the car is connected.
Car status remains 'Disconnected'

Log evcc addon 0.127.3 after removing key 'vin'.
Car was plugged in immeditatley after charge mode was changed to minpv

```
[ocpp ] TRACE 2024/06/25 16:17:12 sent JSON message to 1: [3,"ngf8k3hyrjwvixlc6o3deu9w",{"currentTime":"2024-06-25T06:17:12Z"}]
[lp-1 ] DEBUG 2024/06/25 16:17:36 set charge mode: minpv
[ocpp ] TRACE 2024/06/25 16:18:12 received JSON message from 1: [2,"rq67os8fh060gb6pdgm6ql44","Heartbeat",{}]
[ocpp ] TRACE 2024/06/25 16:18:12 sent JSON message to 1: [3,"rq67os8fh060gb6pdgm6ql44",{"currentTime":"2024-06-25T06:18:12Z"}]
[ocpp ] TRACE 2024/06/25 16:18:38 received JSON message from 1: [2,"zdtttvq79tpq492fcdz237n4","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Preparing","timestamp":"2024-06-25T06:18:37Z"}]
[ocpp ] TRACE 2024/06/25 16:18:38 sent JSON message to 1: [3,"zdtttvq79tpq492fcdz237n4",{}]
[ocpp ] TRACE 2024/06/25 16:19:38 received JSON message from 1: [2,"zmaqnp1zzgpk6p91f26sys2f","Heartbeat",{}]
[ocpp ] TRACE 2024/06/25 16:19:38 sent JSON message to 1: [3,"zmaqnp1zzgpk6p91f26sys2f",{"currentTime":"2024-06-25T06:19:38Z"}]
[ocpp ] TRACE 2024/06/25 16:20:38 received JSON message from 1: [2,"csqs3psnpygm1r4zo125uliz","Heartbeat",{}]
[ocpp ] TRACE 2024/06/25 16:20:38 sent JSON message to 1: [3,"csqs3psnpygm1r4zo125uliz",{"currentTime":"2024-06-25T06:20:38Z"}]
[ocpp ] TRACE 2024/06/25 16:21:38 received JSON message from 1: [2,"bgjayjex7e6b8ueoas6un5rj","Heartbeat",{}]
[ocpp ] TRACE 2024/06/25 16:21:38 sent JSON message to 1: [3,"bgjayjex7e6b8ueoas6un5rj",{"currentTime":"2024-06-25T06:21:38Z"}]
[ocpp ] TRACE 2024/06/25 16:22:38 received JSON message from 1: [2,"1v6e1yp1s6arzz2ah36s5pyn","Heartbeat",{}]
[ocpp ] TRACE 2024/06/25 16:22:38 sent JSON message to 1: [3,"1v6e1yp1s6arzz2ah36s5pyn",{"currentTime":"2024-06-25T06:22:38Z"}]
[ocpp ] TRACE 2024/06/25 16:23:38 received JSON message from 1: [2,"vjqkh40m3latau5na3qwlg2t","Heartbeat",{}]
[ocpp ] TRACE 2024/06/25 16:23:38 sent JSON message to 1: [3,"vjqkh40m3latau5na3qwlg2t",{"currentTime":"2024-06-25T06:23:38Z"}]
```

Interestingly, after rolling back to 0.127.2, if I comment out key 'vin', the car is again never recognised as connected, so this behaviour is not new and key 'vin' is NOT optional ?
 

### Steps to reproduce

1. Upgrade from HAOS addon 0.127.2 to HAOS addon 0.127.3
2. Issue resolves on rollback to 0.127.2
3.
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: homeassistant.local
[... 272 lines cut ...]
  #   tags: <tags>
```

### Log details

```text
[main ] INFO 2024/06/25 15:52:05 evcc 0.127.3
[main ] INFO 2024/06/25 15:52:05 using config file: /config/evcc.yaml
[db ] INFO 2024/06/25 15:52:05 using sqlite database: /config/evcc.db
[mqtt ] INFO 2024/06/25 15:52:05 connecting evcc-1462954333 at tcp://localhost:1883
[mqtt ] DEBUG 2024/06/25 15:52:05 tcp://localhost:1883 connected
[main ] INFO 2024/06/25 15:52:05 listening at :7070
[ocpp-1] DEBUG 2024/06/25 15:52:05 waiting for chargepoint: 5m0s
[ocpp ] INFO 2024/06/25 15:52:22 charge point connected, registering: 1
[main ] FATAL 2024/06/25 15:52:23 cannot create vehicle 'ev3': cannot create vehicle type 'template': cannot create vehicle type 'custom': decoding failed due to the following error(s): '' has invalid keys: vin
[main ] FATAL 2024/06/25 15:52:23 will attempt restart in: 15m0s
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.127.3

andig, 2024-06-25:
Vin is required. I‘ve broken this when cleaning up in https://github.com/evcc-io/evcc/pull/14452, sorry!

andig, 2024-06-25:
Thank you for the report. Broke this in https://github.com/evcc-io/evcc/pull/14452.

closed 2024-06-26 by andig