evcc-io/evcc #14558: SMA Speedwire Integration

MarSlz, 2024-06-25:
### Describe the bug

Das Auslesen des Wechselrichters (SMA Sunny Tripower 10.0 SE (Firmeware: 
3.04.16.R) und Sunny Home Manager 2.0 (Firmware: 2.14.13.R)  funktioniert nicht richtig.

Batterieladen und Netzeinspeisung ist immer der gleiche wert, Batterie Ladezustand wird nicht angezeigt.
Es gibt die fehlermeldung pv Power ist negativ.

[image]

### Steps to reproduce

1. Einbinden der SMA Wechselrichter Batterie und PV über speedwire

### Configuration details

```yaml
- name: grid_rega
    type: template
    template: sma-home-manager
    usage: grid
    host: 192.168.177.78 # IP-Adresse oder Hostname 
    
  - name: sma_pv
    type: template
    template: sma-inverter-speedwire
    usage: pv
    host: 192.168.177.78 # IP-Adresse oder Hostname
    #password: # Passwort für Benutzergruppe Benutzer (optional) 
    
  - name: sma_battery
    type: template
    template: sma-inverter-speedwire
    usage: battery
    host: 192.168.177.78 # IP-Adresse oder Hostname
    #password: # Passwort für Benutzergruppe Benutzer (optional)

site:
  title: Regaweg # display name for UI
  meters:
    grid:  # grid meter
      grid_Rega
    pv:
      - hoymiles # list of pv inverters/ meters
      - sma_pv
    battery:
      - sma_battery # list of battery meters
```

### Log details

```text
[site ] WARN 2024/06/25 10:06:36 pv 2 power: -994W is negative - check configuration if sign is correct
[site ] DEBUG 2024/06/25 10:06:36 pv power: 427W
[site ] DEBUG 2024/06/25 10:06:36 battery soc: 0%
[site ] DEBUG 2024/06/25 10:06:36 battery power: -994W
[site ] DEBUG 2024/06/25 10:06:36 grid meter: -994W
[site ] DEBUG 2024/06/25 10:06:36 grid powers: [-178 -515 -301]W
[site ] DEBUG 2024/06/25 10:06:36 grid currents: [-1.48 -2.58 -1.67]A
[site ] DEBUG 2024/06/25 10:06:36 battery has priority at soc 0% (< 70%)
[site ] DEBUG 2024/06/25 10:06:36 site power: 6W
[site ] DEBUG 2024/06/25 10:06:56 ----
[site ] WARN 2024/06/25 10:06:56 pv 2 power: -965W is negative - check configuration if sign is correct
[site ] DEBUG 2024/06/25 10:06:56 pv power: 427W
[site ] DEBUG 2024/06/25 10:06:56 battery soc: 0%
[site ] DEBUG 2024/06/25 10:06:56 battery power: -965W
[site ] DEBUG 2024/06/25 10:06:56 grid meter: -965W
[site ] DEBUG 2024/06/25 10:06:56 grid powers: [-177 -508 -280]W
[site ] DEBUG 2024/06/25 10:06:56 grid currents: [-1.52 -2.57 -1.62]A
[site ] DEBUG 2024/06/25 10:06:56 battery has priority at soc 0% (< 70%)
[site ] DEBUG 2024/06/25 10:06:56 site power: 35W
[site ] DEBUG 2024/06/25 10:07:06 ----
[site ] WARN 2024/06/25 10:07:06 pv 2 power: -990W is negative - check configuration if sign is correct
[site ] DEBUG 2024/06/25 10:07:06 pv power: 427W
[site ] DEBUG 2024/06/25 10:07:06 battery soc: 0%
[site ] DEBUG 2024/06/25 10:07:06 battery power: -990W
[site ] DEBUG 2024/06/25 10:07:06 grid meter: -990W
[site ] DEBUG 2024/06/25 10:07:06 grid powers: [-198 -515 -277]W
[site ] DEBUG 2024/06/25 10:07:06 grid currents: [-1.53 -2.59 -1.6]A
[site ] DEBUG 2024/06/25 10:07:06 battery has priority at soc 0% (< 70%)
[site ] DEBUG 2024/06/25 10:07:06 site power: 10W
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.126.5

VolkerK62, 2024-06-25:
SHM und SMA-Inverter haben die gleiche IP???

MarSlz, 2024-06-25:
Da war ich mir auch unsicher...,
wenn ich allerdings die IP des Hybridwechselrichters eingebe, startet evcc nicht.

Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/06/25 11:00:12 evcc 0.126.5
[main  ] INFO 2024/06/25 11:00:12 using config file: /config/evcc.yaml
[main  ] INFO 2024/06/25 11:00:12 starting ui and api at :7070
[mqtt  ] INFO 2024/06/25 11:00:17 connecting evcc-639200904 at tcp://localhost:1883
[main  ] FATAL 2024/06/25 11:00:17 cannot create meter 'sma_pv': cannot create meter type 'template': cannot create meter type 'sma': login failed
[main  ] FATAL 2024/06/25 11:00:17 will attempt restart in: 5m0s