evcc-io/evcc #14563: SMA Hybrid-WR, max. AC-Leistung wird nicht berücksichtigt

buennerbernd, 2024-06-25:
Ich habe einen SMA STP 10.0, Panels mit 13 kW Peak und BYD HVS 7.7 kWh.

Config:

```
interval: 20s

site:
...
  residualPower: 100
  maxGridSupplyWhileBatteryCharging: 50

vehicles:
...
  features: ["coarsecurrent"]
```

Wenn die PV-Leistung über 10kW ist und die Hausbatterie noch nicht voll ist, dann habe ich die meiste Zeit etwa 300W Netzbezug.
Ich denke, es liegt daran, dass evcc keine Ahnung hat, was die maximale AC-Leistung des Wechselrichters ist und diese nicht mit in die Steuerung einfließt.
Ich vermute, evcc tastet sich mit Hilfe des Parameters maxGridSupplyWhileBatteryCharging an diese Grenze heran, liegt aber Prinzip-bedingt meistens drüber.
Mein Vorschlag ist:

1. Die maximale AC-Leistung des WR vom Modell ableiten oder konfigurierbar zu machen.
2. Die Steuerung in evcc so anzupassen, dass die aktuelle AC-Leistung nicht die maximale AC-Leistung übertrifft.

Generell würde ich die Möglichkeit begrüßen, eine maximale AC-Leistung des WR angeben zu können. Durch ungünstiges Timing der Messwerte der ganzen Zähler kommen nicht selten Werte zustande, die mehr verfügbare Leistung suggerieren. Dann schießt evcc mit der Ladeleistung über das Ziel hinaus.

Viele Grüße, Stefan.

mdkeil, 2024-06-25:
Also die AC-Leistung des WR benötigt evcc für die Regelung nicht.. einzig die Werte des Gridmeters sind doch auschlaggebend.. alles andere ist nur fürs UI.

buennerbernd, 2024-06-25:
Ich weiß, dass die AC-Leistung bisher nicht verwendet wird.
Es wird aber zusätzlich zum Gridmeter die Ladeleistung der Hausbatterie verwendet. Hierbei kommt es aber immer wieder zu unnötigem Netzbezug, der sich vermeiden ließe, wenn die AC-Leistung berücksichtigt werden würde.
Daher mein Vorschlag.