evcc-io/evcc #14573: Tesla Ladestand nicht abrufbar

damarges, 2024-06-26:
### Describe the bug

Leider habe ich nach wie vor dasselbe Problem mit EVCC in Verbindung mit meinem Model Y.
Heute morgen hatte mein Auto ein Software Update installiert, unmittelbar nach Abschluss des Updates konnte ich zum ersten(!) und einzigen (!) Mal kurz eine Ladestand des Autos direkt in EVCC erkennen. Dies verschwand nach wenigen Momenten aber wieder.

Ich bin aktuell auf Version 0.127.3, habe alle Tesla Tokens entfernt und auch nochmal komplett neu eingerichtet. Es hilft alles nichts.

[image]

### Steps to reproduce

1. configure evcc as described in config and creating tesla token as described

### Configuration details

```yaml
[main  ] INFO 2024/06/26 10:26:51 evcc 0.127.3 (3d1e9698)
[main  ] INFO 2024/06/26 10:26:51 using config file: /etc/evcc.yaml
[db    ] INFO 2024/06/26 10:26:52 using sqlite database: /root/.evcc/evcc.db

Fehlermeldung:

    cannot create meter 'grid1': cannot create meter type 'template': cannot create meter type 'goodwe-wifi': listen udp 0.0.0.0:8899: bind: address already in use

Konfiguration (/etc/evcc.yaml):

network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

plant: *****
interval: 10s # control cycle interval

# new one time sponsor token
sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: goodwe-wifi
  usage: grid
  uri: 192.168.0.23
  name: grid1
- type: template
  template: goodwe-wifi
  usage: pv
  uri: 192.168.0.23
  name: pv2
- type: template
  template: goodwe-wifi
  usage: battery
  uri: 192.168.0.23
  name: battery3

chargers:
- type: template
  template: go-e-v3
  host: 192.168.0.8
  name: wallbox5

loadpoints:
- title: Go-E Charger Gemini (11kW)
  charger: wallbox5
  mode: off
  vehicle: ev4

vehicles:
- type: template
  template: "tesla-command"
  title: Model Y
  accessToken: *****
  refreshToken: *****
  vin: *****
  capacity: 60
  name: ev4

site:
  title: Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

Version: `0.127.3 (3d1e9698)`
```

### Log details

```text
~/docker/evcc# docker logs evcc
[main  ] INFO 2024/06/26 10:25:30 evcc 0.127.3 (3d1e9698)
[main  ] INFO 2024/06/26 10:25:30 using config file: /etc/evcc.yaml
[db    ] INFO 2024/06/26 10:25:30 using sqlite database: /root/.evcc/evcc.db
[main  ] INFO 2024/06/26 10:25:30 listening at :7070
[main  ] ERROR 2024/06/26 10:25:40 creating vehicle ev4 failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla': Get "https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/users/region": net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)
[site  ] WARN 2024/06/26 10:25:40 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site  ] INFO 2024/06/26 10:25:40 site config:
[site  ] INFO 2024/06/26 10:25:40   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/06/26 10:25:40     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/06/26 10:25:40     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/06/26 10:25:40     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2024/06/26 10:25:40   vehicles:
[site  ] INFO 2024/06/26 10:25:40     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/06/26 10:25:40 loadpoint 1:
[lp-1  ] INFO 2024/06/26 10:25:40   mode:        off
[lp-1  ] INFO 2024/06/26 10:25:40   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2024/06/26 10:25:40   meters:      charge ✓
[lp-1  ] INFO 2024/06/26 10:25:40     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2024/06/26 10:25:40 vehicle updated: unknown -> Model Y
[site  ] WARN 2024/06/26 10:25:40 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[lp-1  ] INFO 2024/06/26 10:25:40 car connected
[lp-1  ] INFO 2024/06/26 10:25:40 start charging ->
```

### What type of operating system are you running?

Docker container

### Version

evcc version 0.127.3 (3d1e9698)

andig, 2024-06-26:
Der Tesla Server antwortet nicht. Da hilft erstmal nur Geduld.

damarges, 2024-06-26:
Das tut er aber seit März schon nicht... :( und ich weiß nicht ob ich der Einzige mit dem Problem bin?

damarges, 2024-06-26:
> Der Tesla Server antwortet nicht. Da hilft erstmal nur Geduld.

Hallo Andi, es tut mir leid dich/euch ständig mit diesem Fehler behelligen zu müssen, aber es wird langsam lästig. Denn alles according documentation eingerichtet. Zig neue Tokens erzeugt, Datenbanken gelöscht, from scratch alles angefangen. Auch mal ohne docker nativ installiert. Es ändert leider alles nicht am Outcome.

Was hat es denn mit "tesla" vs "tesla-command" Template auf sich? Und warum erscheint in den Logs immer nur der Fehler zum Template "tesla" auch wenn ich "tesla-command" nutze und vorher die Datenbank gelöscht habe usw. ?

Wie kann ich helfen das Problem weiter einzugrenzen? detailliertere Logs? Wenn ja wie?
ping auf tesla.com aus dem Docker Container heraus funktioniert ohne probleme. DNS Namensauflösung ist also auch nicht der entscheidende Faktor

andig, 2024-06-26:
> Was hat es denn mit "tesla" vs "tesla-command" Template auf sich?

Nichts. Siehe Doku- tesla-command gibts nicht mehr. 

Zu deinem Problem gabs schon ein Issue bei Tesla wenn ich mich richtig erinnere. Da können wir auch nur warten…

damarges, 2024-06-26:
Nur, warum funktioniert es bei einem User und beim Anderen nicht?
Was mich stutzig gemacht hat ist, dass evcc versucht die API für Nordamerika aufzurufen.

[https://developer.tesla.com/docs/fleet-api#setup](url)

Diese Seite gibt explizite URLs für Europa an. Habe versucht mit einem nginx und einem Proxy Pass die Anfragen auf die EU URL umzuleiten, im log von evcc taucht aber nach wie vor der (fehlschlagende) Request an die Nordamerika URL auf

andig, 2024-06-26:
Ich wiederhole mich. Wir hatten das Thema schon. Hier nochmal der Link: https://github.com/teslamotors/vehicle-command/issues/200

Zitat:

> This one appears to be either a backend issue, although some more investigation may be required. Please see https://developer.tesla.com/docs/fleet-api#help-and-support.

Sclüssel hast du sicher schon ausprobiert?!

damarges, 2024-06-26:
Den Thread dazu kannte ich noch nicht. Danke dir. Probiere ich und werde mich rückmelden

andig, 2024-06-26:
Dann hilft nur Tesla Support zu kontaktieren.

damarges, 2024-06-26:
Darf ich noch einmal einhaken, wo kommt diese API Url her? Warum versucht er zur Nordamerika-API zu verbinden? Ich vermute hierbei den Fehler für den fehlgeschlagenen Verbindungsaufbau

andig, 2024-06-26:
Weil das so ist…

damarges, 2024-06-26:
Andi, es tut mir leid, wahrscheinlich haben wir uns heute gegenseitig auf dem falschen Fuß erwischt. Ich möchte ja nur gern dazu beitragen, die Software weiter zu verbessern bzw. von Bugs zu befreien. Fühle mich etwas machtlos gerade.

andig, 2024-06-26:
Tesla schreibt, dass es wohl ein Backendissue ist und man sich an den Fleet API Support wenden sollte. Das können wir dann auch nur so stehen lassen. Es ist ein Problem bei deinem Fahrzeug, nicht bei der Implementierung…

damarges, 2024-06-26:
> Tesla schreibt, dass es wohl ein Backendissue ist und man sich an den Fleet API Support wenden sollte. Das können wir dann auch nur so stehen lassen. Es ist ein Problem bei deinem Fahrzeug, nicht bei der Implementierung…

Hatte nicht gesehen, dass dies Tesla geschrieben hat sondern hatte es für die Vermutung eines Users gehalten. Ich habe den Tesla Support angeschrieben. Mal schauen ob/was zurück kommt.

andig, 2024-06-26:
sethterashima ist vom Tesla Api Team

phrozen77, 2024-06-28:
Kurzer Kommentar dazu von meiner Seite:

hier funktioniert die Fleet API - ich bin mir relativ sicher, dass evcc auch die richtige URL nimmt…

Ein Unterschied könnte der Zugriffslevel sein - die neue API unterscheidet ja zwischen Driver und Owner Access - ich bin mir ziemlich sicher, das ich beim einrichten der Virtual Keys usw. nur Driver Access hatte - evtl. lädst du mal einen zweiten Benutzer ein und gibst über den dann evcc Zugriff?

damarges, 2024-06-28:
> Kurzer Kommentar dazu von meiner Seite:
>
> hier funktioniert die Fleet API - ich bin mir relativ sicher, dass evcc auch die richtige URL nimmt…
>
> Ein Unterschied könnte der Zugriffslevel sein - die [...]

Probiere ich heute Abend mal und gebe Feedback. Danke für den Tipp

Breakyman, 2024-06-28:
@damarges ob das funktioniert würde mich auch sehr interessieren, habe das gleich eProblem über lvcc und home assistant