evcc-io/evcc #14578: Switching between different strategies for PV surplus charging - "do not charge from grid" and "do not feed to grid"

eweri, 2024-06-26:
**Is your feature request related to a problem? Please describe.**
I have a solarsystem with only 3,7kW peak power and in germany we get a lot less money for power we feed to grid then we have to pay for power we draw from grid. Because at the moment PV mode starts loading when at least 1,4kW feed to grid are reached, because for my not so powerful solarsystem has to be really good weather to reach this level of power. 

**Describe the solution you'd like**
The current strategy for PV mode is "do not charge from grid", i would prefer a policy in PV mode "do not feed to grid".
So we need two thresholds that define a surplus power level at which loading is started and stopped even when this means to draw power from grid. But charging should only start when there is a surplus from solarsystem.  

Here again what I would like to achieve:

1. When the PV generates the house consumption plus a value X surplus, charging should start. In order to achieve the charging power of 1.4kW, the rest should then be drawn from the grid.
2. If the PV generates more than the house consumption plus minimum charging, the charging should then follow the generation and no longer draw energy from the grid.
3. If the PV generation falls below the domestic consumption plus a value X, the charging should stop.

The goal is to load as much solarenergy (even on cloudy days) into the car before it is feed to the grid.
A mode switch for "PV" to change strategy would be great.

Example:
Home needs 350W, PV generates 1200W currently 850W are feeded to the grid, no charging.
I would like this policy:
Home needs 350W, PV generates 1200W (not even enough surplus for 1 phase charging), car is charging with minimum 1400W, 550W taken from grid
Home needs 350W, PV generates 2200W (enough surplus for 1 phase charging), car is charging with minimum 1850W, 0W taken from grid.

**Describe alternatives you've considered**
I tried with residualpower, enable threshold and disable threshold  - they did not do exactly what i want. 
residualpower draws power from grid even when surplus is enough for 1 phase charging
thresholds are calculated from gridpower not from pv surplus power. See the attached picture.

**Additional context**
I tried with enable threshold 500W and disable threshold 950W and because my PHEV (Hyundai Tucson) charging power was only 841W the charging did not stop when surplus was 0W. see picture
[image]

andig, 2024-06-26:
Yes, thats in the backlog. There‘s a lengthy discussion on charging modes also covering this (can‘t find the link right now).

VolkerK62, 2024-06-26:
should be this?  https://github.com/evcc-io/evcc/discussions/3530

An-Wi, 2024-06-28:
I am also very familiar with this problem from the last winter months!
Experimenting with Planner the evening before (in combination with setting "Max. Current" to 6A) as work-a-round helped to start loading with minimum in the morning during sunrise, but all assuming that there will be no clouds the next day... => no really sensible solution.

Now I use "thresholds" and it works fine for me:
<pre>
loadpoints:
    enable: # pv mode enable behavior
      delay: 2m         # threshold must be exceeded for this long
      threshold: -800   # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 5m         # threshold must be exceeded for this long
      threshold: 500    # maximum import power (W)
    guardDuration: 10m  # switch charger contactor not more often than this (default 5m)

site:
  residualPower: 0
</pre>
@eweri Related to your examples:
Home needs 350W, PV generates 1150W: currently 800W are feeded to the grid => **start charging**
Home needs 350W, PV generates 1200W: car is charging with minimum 1400W, **550W taken from grid**
Home needs 350W, PV generates 2200W: car is charging with 1850W, **0W taken from grid**

Did you notice the negative sign at "enable threshold"?

andig, 2024-06-29:
I'm locking this here as the majority of the discussion is in https://github.com/evcc-io/evcc/discussions/3530. Leaving issue as reminder of the backlog item.

closed 2026-09-02 by andig