evcc-io/evcc #14579: RFID identification via OCPP

efschu, 2024-06-26:
**Is your feature request related to a problem? Please describe.**
RFID identification via OCPP

**Describe the solution you'd like**
work as with other chargers

**Describe alternatives you've considered**
the information it needs is already in evcc, and i made a script to work around meanwhile
https://github.com/evcc-io/evcc/discussions/14559#discussioncomment-9883834

**Additional context**
ocpp.go file tells me

```
// // Identify implements the api.Identifier interface
// Unless charger uses vehicle ID as idTag in authorize.req it is not possible to implement this in ocpp1.6
// func (c *OCPP) Identify() (string, error) {
// 	return "", errors.New("not implemented")
// }
```
but it should be workaroundable ;)

andig, 2024-06-26:
Which OCPP capability should supply the identifier? I assume you‘d want the id tag? Drawback is that this only seems available, once the transaction actually starts, not when the vehicle connects or an RFID is being presented?

efschu, 2024-06-27:
Transaction start and stop. So I plug in the cable to car and charger, then I "identify" myself with RFID @charger, then the tagid is transferred via ocpp to evcc.

Works like it should.

For sure it cannot be implemented by "asking" the charger. It's just transferred once. But this should be enough, shouldn't it?

efschu, 2024-06-28:
So, as we do have chargers with two connectors, it does not work if the second connector is used.

In the case someone already started a charge the second connector rfid reader sends "evcc" as idTag instead of the rfid id.

I think I will skip this and buy the proprietary solution :(

andig, 2024-06-28:
> So, as we do have chargers with two connectors, it does not work if the second connector is used.

That should be a lesser problem. Seems this would generally only work if the charger starts the transaction, right? If evcc starts it it will/ has to send its own id tag?

andig, 2024-06-30:
@efschu log bitte mal den OCPP Traffic mit Deiner RFID Karte mit, damit wir die Reihenfolge der Nachrichten sehen (`ocpp: trace` oder noch einfacher über das UI).

benesolar, 2024-07-02:
Ich bin zwar nicht der OP aber hoffe das hilft auch:

```
[ocpp ] TRACE 2024/06/23 12:12:20 received JSON message from Ladestation: [2, "1d2bd940-3149-11ef-9265-000a1484c32e", "Authorize", {"connectorId": 2, "idTag": "0584350940cxxx"}]
[ocpp ] TRACE 2024/06/23 12:12:20 sent JSON message to Ladestation: [3,"1d2bd940-3149-11ef-9265-000a1484c32e",{"idTagInfo":{"status":"Accepted"}}]
[ocpp ] TRACE 2024/06/23 12:12:23 received JSON message from Ladestation: [2, "1ebb3b48-3149-11ef-9265-000a1484c32e", "StatusNotification", {"status": "Preparing", "connectorId": 2, "errorCode": "NoError"}]
[ocpp ] TRACE 2024/06/23 12:12:23 sent JSON message to Ladestation: [3,"1ebb3b48-3149-11ef-9265-000a1484c32e",{}]
[ocpp ] TRACE 2024/06/23 12:12:37 received JSON message from Ladestation: [2, "26fe1c8a-3149-11ef-9265-000a1484c32e", "StatusNotification", {"status": "SuspendedEVSE", "connectorId": 2, "errorCode": "NoError"}]
[ocpp ] TRACE 2024/06/23 12:12:37 sent JSON message to Ladestation: [3,"26fe1c8a-3149-11ef-9265-000a1484c32e",{}]
[ocpp ] TRACE 2024/06/23 12:12:37 received JSON message from Ladestation: [2, "2787994c-3149-11ef-9265-000a1484c32e", "StartTransaction", {"connectorId": 2, "meterStart": 21505, "idTag": "0584350940cxxx", "timestamp": "2024-06-23T10:12:35.000Z"}]
[ocpp ] TRACE 2024/06/23 12:12:37 sent JSON message to Ladestation: [3,"2787994c-3149-11ef-9265-000a1484c32e",{"idTagInfo":{"status":"Accepted"},"transactionId":5}]
[ocpp ] TRACE 2024/06/23 12:12:41 received JSON message from Ladestation: [2, "29b905de-3149-11ef-9265-000a1484c32e", "StatusNotification", {"status": "SuspendedEV", "connectorId": 2, "errorCode": "NoError"}]
[ocpp ] TRACE 2024/06/23 12:12:41 sent JSON message to Ladestation: [3,"29b905de-3149-11ef-9265-000a1484c32e",{}]
[ocpp ] TRACE 2024/06/23 12:12:43 received JSON message from Ladestation: [2, "2b169658-3149-11ef-9265-000a1484c32e", "StatusNotification", {"status": "Charging", "connectorId": 2, "errorCode": "NoError"}]
[ocpp ] TRACE 2024/06/23 12:12:43 sent JSON message to Ladestation: [3,"2b169658-3149-11ef-9265-000a1484c32e",{}]
[ocpp ] TRACE 2024/06/23 12:13:08 sent JSON message to Ladestation: [2,"316583366","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":2}]
[ocpp ] TRACE 2024/06/23 12:13:08 received JSON message from Ladestation: [3, "316583366", {"status": "Accepted"}]
[ocpp ] TRACE 2024/06/23 12:13:08 TriggerMessage MeterValues for Ladestation: Accepted
[ocpp ] TRACE 2024/06/23 12:13:19 received JSON message from Ladestation: [2, "4072cae4-3149-11ef-9265-000a1484c32e", "MeterValues", {"meterValue": [{"timestamp": "2024-06-23T10:13:33.000Z", "sampledValue": [{"measurand": "Energy.Active.Import.Register", "unit": "Wh", "value": "21528"}]}], "connectorId": 2}]
[ocpp ] TRACE 2024/06/23 12:13:19 sent JSON message to Ladestation: [3,"4072cae4-3149-11ef-9265-000a1484c32e",{}]
[ocpp ] TRACE 2024/06/23 12:13:21 sent JSON message to Ladestation: [2,"770795250","RemoteStopTransaction",{"transactionId":5}]
[ocpp ] TRACE 2024/06/23 12:13:23 received JSON message from Ladestation: [3, "770795250", {"status": "Accepted"}]
[ocpp ] TRACE 2024/06/23 12:13:24 received JSON message from Ladestation: [2, "432e55e6-3149-11ef-9265-000a1484c32e", "StatusNotification", {"status": "SuspendedEVSE", "connectorId": 2, "errorCode": "NoError"}]
[ocpp ] TRACE 2024/06/23 12:13:24 sent JSON message to Ladestation: [3,"432e55e6-3149-11ef-9265-000a1484c32e",{}]
[ocpp ] TRACE 2024/06/23 12:13:31 received JSON message from Ladestation: [2, "4758834e-3149-11ef-9265-000a1484c32e", "StatusNotification", {"status": "Finishing", "connectorId": 2, "errorCode": "NoError"}]
[ocpp ] TRACE 2024/06/23 12:13:31 sent JSON message to Ladestation: [3,"4758834e-3149-11ef-9265-000a1484c32e",{}]
[ocpp ] TRACE 2024/06/23 12:14:31 received JSON message from Ladestation: [2, "6b33113a-3149-11ef-9265-000a1484c32e", "Heartbeat", {}]
[ocpp ] TRACE 2024/06/23 12:14:31 sent JSON message to Ladestation: [3,"6b33113a-3149-11ef-9265-000a1484c32e",{"currentTime":"2024-06-23T10:14:31Z"}]
[ocpp ] TRACE 2024/06/23 12:14:35 sent JSON message to Ladestation: [2,"2172881737","SetChargingProfile",{"connectorId":2,"csChargingProfiles":{"chargingProfileId":1,"transactionId":5,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":9.9}]}}}]
[ocpp ] TRACE 2024/06/23 12:14:36 received JSON message from Ladestation: [3, "2172881737", {"status": "Rejected"}]
```

Log von einer Mennekes Smart Ladesäule (2 Ladepunkte).
SetChargingProfile wenn keine Ladevorgang aktiv wir immer Rejected.

andig, 2024-07-02:
Interessant!

Hier ist eine gute Beschreibung: https://www.ampcontrol.io/ocpp-guide/how-to-start-an-ocpp-charging-session-with-starttransaction. Was mir nicht klar war: `Authorize` und `(Remote)StartTransaction` sind separate Requests. Damit sollte es genügen, dass `idTag` aus `Authorize` als Identifier zu verwenden bist das Fahrzeug wieder disconnected wird.

Dann bleiben aber zwei Fragen:

- Mit welchem Tag soll evcc den Ladevorgang starten? Weiter `evcc` oder dem erhaltenen Tag?
- Was passiert ohne RFID? Gibts dann kein `Authorize`? Welches Tag sollte `evcc` dann verwenden?

@benesolar könntest Du den Case ohne RFID nochmal testen?

premultiply, 2024-07-02:
> Mit welchem Tag soll evcc den Ladevorgang starten? Weiter evcc oder dem erhaltenen Tag?

Hier würde ich immer das zuvor erhaltene Tag verwenden um ggf. weitere lokale Freigaben, Profile, Statistiken o.ä. in der Box zu ermöglichen.

>Was passiert ohne RFID? Gibts dann kein Authorize? Welches Tag sollte evcc dann verwenden?

Mindestens in `StartTransaction.req` der Box ist `idTag` "required" und kommt daher immer mit. D.h. selbst bei Autostart (Free Mode) ohne Karte oder lokaler Whitelist-Freigabe sollte dort immer ein Wert mitgesendet werden den wir verarbeiten und ggf. für `RemoteStartTransaction` wiederverwenden können. Im Zweifelsfall ist dies ein Leerstring.

andig, 2024-07-02:
Leider zeigt das verlinkte Issue, dass die `connectorId` nicht zum Standard gehört. Bei multi-connector Boxen wissen wir also nicht, wo das Auto angeschlossen ist :(

benesolar, 2024-07-02:
Bei dieser Station muss ich eine RFID vorhalten, damit der verbaute Shutter den Anschluss mechanisch freigibt.

Hier hätte ich noch einen zweiten log wo EVCC mit "evcc" als idtag geantwortet hat:

```
[ocpp ] TRACE 2024/06/23 11:51:32 received JSON message from Ladestation: [2, "354148ce-3146-11ef-9265-000a1484c32e", "Authorize", {"connectorId": 2, "idTag": "043768e29a7xxx"}]
[ocpp ] TRACE 2024/06/23 11:51:32 sent JSON message to Ladestation: [3,"354148ce-3146-11ef-9265-000a1484c32e",{"idTagInfo":{"status":"Accepted"}}]
[ocpp ] TRACE 2024/06/23 11:51:35 received JSON message from Ladestation: [2, "36d793e6-3146-11ef-9265-000a1484c32e", "StatusNotification", {"status": "Preparing", "connectorId": 2, "errorCode": "NoError"}]
[ocpp ] TRACE 2024/06/23 11:51:35 sent JSON message to Ladestation: [3,"36d793e6-3146-11ef-9265-000a1484c32e",{}]
[ocpp ] TRACE 2024/06/23 11:51:45 received JSON message from Ladestation: [2, "3d0a5d0c-3146-11ef-9265-000a1484c32e", "StatusNotification", {"status": "SuspendedEVSE", "connectorId": 2, "errorCode": "NoError"}]
[ocpp ] TRACE 2024/06/23 11:51:45 sent JSON message to Ladestation: [3,"3d0a5d0c-3146-11ef-9265-000a1484c32e",{}]
[ocpp ] TRACE 2024/06/23 11:51:45 sent JSON message to Ladestation: [2,"550368086","RemoteStartTransaction",{"connectorId":2,"idTag":"evcc","chargingProfile":{"chargingProfileId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}]
[ocpp ] TRACE 2024/06/23 11:51:46 received JSON message from Ladestation: [2, "3d67dd38-3146-11ef-9265-000a1484c32e", "StartTransaction", {"connectorId": 2, "meterStart": 19661, "idTag": "043768e29a7xxx", "timestamp": "2024-06-23T09:51:47.000Z"}]
[ocpp ] TRACE 2024/06/23 11:51:46 sent JSON message to Ladestation: [3,"3d67dd38-3146-11ef-9265-000a1484c32e",{"idTagInfo":{"status":"Accepted"},"transactionId":4}]
[ocpp ] TRACE 2024/06/23 11:51:47 received JSON message from Ladestation: [3, "550368086", {"status": "Rejected"}]
[ocpp ] TRACE 2024/06/23 11:51:51 received JSON message from Ladestation: [2, "406b3642-3146-11ef-9265-000a1484c32e", "StatusNotification", {"status": "SuspendedEV", "connectorId": 2, "errorCode": "NoError"}]
[ocpp ] TRACE 2024/06/23 11:51:51 sent JSON message to Ladestation: [3,"406b3642-3146-11ef-9265-000a1484c32e",{}]
[ocpp ] TRACE 2024/06/23 11:51:52 received JSON message from Ladestation: [2, "415cac34-3146-11ef-9265-000a1484c32e", "StatusNotification", {"status": "Charging", "connectorId": 2, "errorCode": "NoError"}]
[ocpp ] TRACE 2024/06/23 11:51:52 sent JSON message to Ladestation: [3,"415cac34-3146-11ef-9265-000a1484c32e",{}]
[ocpp ] TRACE 2024/06/23 11:51:57 sent JSON message to Ladestation: [2,"3528859847","SetChargingProfile",{"connectorId":2,"csChargingProfiles":{"chargingProfileId":1,"transactionId":4,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":14}]}}}]
[ocpp ] TRACE 2024/06/23 11:51:58 received JSON message from Ladestation: [3, "3528859847", {"status": "Accepted"}]
[ocpp ] TRACE 2024/06/23 11:52:39 sent JSON message to Ladestation: [2,"2917458629","SetChargingProfile",{"connectorId":2,"csChargingProfiles":{"chargingProfileId":1,"transactionId":4,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":7.1}]}}}]
[ocpp ] TRACE 2024/06/23 11:52:40 received JSON message from Ladestation: [3, "2917458629", {"status": "Accepted"}]
[ocpp ] TRACE 2024/06/23 11:52:52 received JSON message from Ladestation: [2, "64e25b22-3146-11ef-9265-000a1484c32e", "MeterValues", {"meterValue": [{"timestamp": "2024-06-23T09:53:06.000Z", "sampledValue": [{"measurand": "Energy.Active.Import.Register", "unit": "Wh", "value": "19739"}]}], "connectorId": 2, "transactionId": 4}]
[ocpp ] TRACE 2024/06/23 11:52:52 sent JSON message to Ladestation: [3,"64e25b22-3146-11ef-9265-000a1484c32e",{}]
[ocpp ] TRACE 2024/06/23 11:52:57 sent JSON message to Ladestation: [2,"459988207","SetChargingProfile",{"connectorId":2,"csChargingProfiles":{"chargingProfileId":1,"transactionId":4,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":7.8}]}}}]
[ocpp ] TRACE 2024/06/23 11:52:58 received JSON message from Ladestation: [3, "459988207", {"status": "Accepted"}]
[ocpp ] TRACE 2024/06/23 11:53:15 sent JSON message to Ladestation: [2,"3730260534","SetChargingProfile",{"connectorId":2,"csChargingProfiles":{"chargingProfileId":1,"transactionId":4,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":9.2}]}}}]
[ocpp ] TRACE 2024/06/23 11:53:16 received JSON message from Ladestation: [3, "3730260534", {"status": "Accepted"}]
[ocpp ] TRACE 2024/06/23 11:53:33 sent JSON message to Ladestation: [2,"1419272302","SetChargingProfile",{"connectorId":2,"csChargingProfiles":{"chargingProfileId":1,"transactionId":4,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":10.1}]}}}]
[ocpp ] TRACE 2024/06/23 11:53:34 received JSON message from Ladestation: [3, "1419272302", {"status": "Accepted"}] 
```

premultiply, 2024-07-02:
@andig Vorschlag: Mal angenommen wir würden `Authorize` weiterhin wie bisher immer blind mit Accepted beantworten (das ist in der kommerziellen Welt die Vorabprüfung ans Backend ob die Karte grundsätzlich irgendwo bekannt und zugelassen ist) und den empfangen `idTag`-Wert dort mangels `connectorId` weiterhin erstmal ignorieren dann könnten wir aber das `idTag` aus `StartTransaction` nutzen, denn dort ist die `connectorId` verpflichtend enthalten.
Denn erst zu diesem Zeitpunkt teilt die Box ja mit welche (vorauthentifizierte) Kartenkennung tatsächlich an welchem Anschluss zum Laden verwendet wird.

andig, 2024-07-03:
Ja, die Idee hatte ich auch. Dann müssten wir aber bitte mal einen Trace sehen da in dem Fall der Flow mit `RemoteStartTransaction` anfängt bei dem evcc schon das `idtag` mit gibt. Was schickt der CP dann im `StartTransaction`?

premultiply, 2024-07-03:
Ich stelle mir gerade die Frage ob wenn https://github.com/evcc-io/evcc/pull/12861 fertig ist überhaupt noch ein Bedarf für einen RemoteStartTransaction-Flow (im kommerziellen Umfeld "App-Start") besteht?
Denn dann würde die Session _immer_ lokal von der Box initiiert (RFID oder Autostart mit fester ID) und von evcc nur ggf. pausiert.

andig, 2024-07-03:
Mindestens nach Disconnect brauchts den. Ich sehe auch keine Grund, warum die Box durch diese Änderung das selbst tun sollte.

premultiply, 2024-07-03:
Die Session startet bei OCPP nach dem Verbinden des Fahrzeugs entweder
- nach Freigabe durch lokale RFID-Authentifizierung am CS oder
- per lokalem (Auto)start durch die Box oder 
- per RemoteStartTransaction durch das CS (Usecase: App-Start).

Für unsere Anwendung sind nur die ersten beiden Fälle relevant und sauber zu handhaben. Aus sich von evcc sind diese identisch.

Will man RFID nutzen bleibt nur der erste Weg.
Für kartenlosen Betrieb muss man die Box auf Autostart/Free Mode setzen.

Weg 3 erscheint mir für unsere Zwecke immer nutzlos und macht die Sache verwirrend.

andig, 2024-07-03:
> nach Freigabe durch lokale RFID-Authentifizierung am CS oder

Die Authentifizierung startet keine Session. Das ist ein unabhängiger Prozess, siehe verlinkte Doku.

> per lokalem (Auto)start durch die Box oder

Der CP fragt beim CS an

> per RemoteStartTransaction durch das CS (Usecase: App-Start).

Das CS fragt beim CP an

> Für unsere Anwendung sind nur die ersten beiden Fälle relevant und sauber zu handhaben. 

Der erste Fall ist keiner weil er nichts mit Start einer Session zu tun hat

andig, 2024-07-03:
Ich bräuchte weiter https://github.com/evcc-io/evcc/issues/14579#issuecomment-2205164245 um das in der aktuellen Systematik lösen zu können.

andig, 2024-07-06:
Could you try https://github.com/evcc-io/evcc/pull/14733? For Docker users it's available using the `ocpp` tag.

closed 2024-07-07 by andig