evcc-io/evcc #14598: Verbindung zum Server verloren

jekruege, 2024-06-27:
### Describe the bug

Leider werden die Versionen immer schlechter bezüglich einer dauerhaften Anzeige über ein Display im Haus. 

Es kommt nun wie schon seit einigen Versionen die Meldung Verbindung zum Server verloren. Bisher hatte man dann aber doch alles aktuell im Blick. Nur die Meldung war bisher unsinnig. 
[image]
Inzwischen ist ein Schleier darüber gelegt. Wer kommt nur auf solche Ideen?

### Steps to reproduce

Tauch nach einer Weile von alleine auf

### Configuration details

```yaml
Spielt keine Rolle. Ist bei jeder Konfiguration so.
```

### Log details

```text
Ist bei jedem so.
```

### What type of operating system are you running?

Linux

andig, 2024-06-27:
@naltatis könntest du das mal anschauen?

naltatis, 2024-06-27:
Naja, das sieht so aus als ob wirklich die Verbindung zum Server nicht verfügbar ist. Das heißt, die UI zeigt alte Daten an und Klicks in der Oberfläche können nicht mehr die gewünschte Aktion ausführen. Was wäre denn dein gewünschtes Verhalten?

naltatis, 2024-06-27:
Ist das bei dir denn eine permanenter Zustand? Kommt und geht das? Bitte poste mal einen Screenshot von den Browser-Devtools (Konsole & Netwerk-Tab). Selbst wenn die Websocketverbindung, warum auch immer, mal abreißt sollte sofort ein Reconnect versucht werden.

StefanSchoof, 2024-06-27:
Was für ein Browser ist das? Passiert das auch in einem Desktop Browser?

Könnte mir vorstellen das der Browser die Websocket Verbindung auf Grund von irgendeinem Stromspar versuch killt.

Ist dein Intervall besonders groß? So das irgendeine Komponente denkt das es zum timeout gekommen ist.

jekruege, 2024-06-28:
Vielen Dank schon mal fürs kümmern. 

Das ist ein alte iPad mit Version 12.5.7. Zumindest hat es nicht wirklich irgendwo ein Menü oder ich finde es nicht. Es hat noch ein paar experimentelle Safari Einstellungen. Da habe ich jetzt mal mehr angestellt. 

Der Fehler tritt gefühlt nie nachts auf, sondern eher nachmittags/abends. Vielleicht bei der Anzeige vom Entladen vom Speicher? Ich habe da aber technisch keine Ahnung. 

Es kam vor ein paar Monaten neu hinzu. Bisher stand aber immer nur Verbindung zum Server verloren aber es ging trotzdem alles. Neu ist dieser Nebelschleier. 

[image]

An-Wi, 2024-06-28:
Mir ist dieses Verhalten letzte oder vorletzte Woche (erstmalig) auch aufgefallen, allerdings tritt es nun seit Version 0.127.3, die ich am Dienstag installiert habe, (gefühlt) gar nicht mehr auf?!?

Folgende Konstellation bei mir...
evcc operating system: _Windows 10_
client: _Windows 10_
browser: _Google Chrome - Version 126.0.6478.127 (Offizieller Build) (64-Bit)_
Stromsparmodus am PC selbst kann ich ausschließen.

Da in meinem Chrome Browser immer mehrere Tabs offen sind, läuft evcc gern mal "im Hintergrund". Das Verhalten trat bei mir bisher nur auf, wenn der Tab nicht im Vordergrund war und fiel mir somit erst auf, als ich den Tab wieder aktiviert hatte.
Zunächst hatte ich es auf ein Update vom Chrome Browser geschoben, dass dieser Tabs im Hintergrund nun auf "Sparflamme" fährt, aber genauer fassen konnte ich es noch nicht.
Die Browser Console hatte ich beim Auftreten zwar geöffnet, aber leider keinen Screenshot gemacht, weil keine Fehlermeldung zu sehen war. Ich denke, dass zuletzt `websocket disconneting` sowie `we are offline` stand und das nachfolgende `websocket connected` bzw. `we are online` fehlte. Sobald es wieder auftreten sollte, kommt der Screenshot.

adobe-hills, 2024-06-28:
Bei kommt die Meldung seit der Version 0.127.3. Ich hatte gestern die neue Version wegen meines Problems mit der Pulsar WB installiert, zuvor hatte ich die Meldung noch nie.  Das iPhone hat die neueste Softwareversion auf dem iPhone. Safari Browser, EVCC auf Homebildschirm (wie PWA) gelegt.

naltatis, 2024-06-28:
Wie temporär/permanent sind diese Zustände bei euch? Flackert das kurz auf, behebt es sich von selbst, ladet ihr dann neu?

An-Wi, 2024-06-28:
Bei mir (Google Chrome PC-Version): Behebt sich nicht von selbst - oder ich war zu ungeduldig mit 1min Warten.
Flackert nicht.
Seite manuell neu geladen.

naltatis, 2024-06-28:
> Seite manuell neu geladen.

Ok, das ist schlecht. Dann funktioniert der automatische Reconnect in der Situation scheinbar nicht. Lokal konnte ich das bislang noch nicht nachstellen. Wenn du hier Details aus den Devtools (Konsole und `ws`-Eintrag im Network-Tab) sammeln könntest, wäre das super.

An-Wi, 2024-06-28:
Mach ich gern - bin damit vertraut. 
Konsole ist kein Problem - die loggt automatisch mit. 
Für Network muss ich dran denken, das DevTool vor dem Starten zu öffnen, sonst wird der `ws`-Eintrag nicht geloggt.

Aber wie schon erwähnt, hatte ich diese Woche keine Probleme mehr damit - kann also eine Weile dauern, bis ich hier Details liefern könnte.

TobiasNeuschulz, 2024-06-30:
Moin, habe ähnliches Problem auf einem Raspberry PI mit TouchDisplay im Kiosk Modus. Ich lasse den Raspberry nicht 24x7 laufen, sondern nur am Tage und wenn es notwendig ist. Manchmal nach dem Starten hatte ich auch die Meldung "Keine Verbindung zum Server" - evcc Webseite war aber aktuell und auch sogar funktional. Jetzt habe ich öfter den Schleier und kann im Kiosk Modus nichts machen, außer dem raspberry neu zu starten. Meistens läuft es dann wieder.

CarstenPV, 2024-07-03:
Hallo erstmal, Ja dieses Fehlerbild habe ich auch, kommt normalerweise dann automatisch wieder. Gerade wollte ich die Logs einsehen, äh da sagt er "Login fehlgeschlagen: Passwort ist ungültig". Im Terminal nach dem 2ten mal "sudo systemctl restart evcc", war er dann wieder da. Das Passwort habe ich gestern neu eingegeben, das wird nicht akzeptiert. Ist das gleiche wie vom SUDO (geht das überhaupt, dann?!)
Ansonsten Tibber gestern integriert, funzt einwandfrei. Supi

andig, 2024-07-03:
@CarstenPV bitte beim Thema bleiben. Danke.

jekruege, 2024-07-04:
Bei mir tritt es alle 2-3 Tage auf im Moment bei bewölktem Wetter. Hinter dem grauen Schleier sieht man, dass das richtige angezeigt wird. Eventuell wenn der gelbe Balken kommt? Rückspeisung ins Netz?

andig, 2024-07-04:
Der "Schleier" kommt wenn der Browser die Verbindung verloren hat. Ob die Websocket Verbindung steht ist im Browser/Entwicklerkonsole/Netzwerk sichtbar. Die Frage ist: stimmt die Anzeige nicht, oder ist die Verbindung wirklich unterbrochen?

jekruege, 2024-07-04:
Das hier spricht gegen eine verlorene Verbindung: "Hinter dem grauen Schleier sieht man, dass das richtige angezeigt wird". Man sieht hinterm Schleier, dass sich alle paar Sekunden etwas verändert.

An-Wi, 2024-07-04:
Bei mir tritt es seit gut 1 Woche im PC-Browser (Google Chrome) gar nicht mehr auf. Das war in etwa seit ich auf 0.127.3 aktualisiert hatte - aber ich würde auch ein Google Chrome Update als Ursache nicht ausschließen wollen.
Seit heute früh läuft bei die Version 0.128.0 genauso fehlerfrei. Gelegentlich kommt zwar der graue Schleier, wenn der Tab inaktiv ist, aber beim Reaktivieren ist innerhalb 1 Sekunde die Verbindung wieder da, der graue Schleier verschwindet sofort und alle Werte aktualisieren sich.
Das gleiche, fehlerfreie Verhalten habe ich am Wochenende beim Experimentieren im WLAN mit meinem iPhone auch nachstellen können: WLAN weg => 5G an => grauer Schleier => WLAN wieder da => graue Schleier verschwindet (so wie es sein soll)

jekruege, 2024-07-05:
Bitte nicht kapern. Es geht hier im fest installierte Displays (die man in der Regel nicht bedient) und nicht um einen PC. Ich bin auch gerade auf die 0.128.0 gewechselt. Ich behalte es im Auge.

naltatis, 2024-07-05:
So, Fehler gefunden. Erklärung siehe PR. ☝️
Danke für die Hinweise.

jekruege, 2024-07-11:
Leider heute schon wieder mit aktueller Version. 

[image]

andig, 2024-07-11:
…und im Hintergrund aktualisiert es weiter die Anzeige?

naltatis, 2024-07-11:
Welche Version verwendest du? Mit dem Nightly von heute Nacht gabs noch mal eine Änderung am Reconnectverhalten.

jekruege, 2024-07-12:
0.128.2

andig, 2024-07-21:
@naltatis könnten wir hier probehalber mehr Logging logging einbauen und das Logging z.b. optional in dem disconnected overlay anzeigen? Vor dem Release wieder raus werfen?

naltatis, 2024-07-21:
In 0.128.3 ist das neue Reconnect-Verhalten mit drin (s.o.). Ich gehe davon aus, dass das das Problem behebt. @jekruege magst du mal upgraden und schauen ob es immer noch bei dir auftritt?

jekruege, 2024-07-21:
Danke! Ich bin jedoch derzeit im Urlaub. Ich melde mich danach.

closed 2024-07-21 by andig

jekruege, 2024-08-19:
Ich habe es jetzt 2 Wochen beobachtet. Der Fehler tritt nicht mehr auf. Super, danke nochmal!

naltatis, 2024-08-19:
@jekruege  Super. Danke für deine Rückmeldung.