evcc-io/evcc #14604: Cannot downgrade to older Version

wittmannchris, 2024-06-28:
### Describe the bug

According https://docs.evcc.io/docs/installation/linux , the downgrade works with `  sudo apt install evcc=x.xxx.x # Versionsnummer`

`sudo apt install evcc=0.127.2 # Versionsnummer Reading package lists... Done Building dependency tree... Done Reading state information... Done Package evcc is not available, but is referred to by another package. This may mean that the package is missing, has been obsoleted, or is only available from another source`

Installed is nightly 

### Steps to reproduce

1.
2.
3.
...
NA

### Configuration details

```yaml
Not needed
```

### Log details

```text
Not necessary
```

### What type of operating system are you running?

Linux

### Version

0.127.3

0x3d13f, 2024-06-28:
Stable und nightly sind verschiedene Repos. Daher musst Du für die Installation einer stabilen Version nach Verwendung der nightly auch das Repo auf stable umstellen.

wittmannchris, 2024-06-28:
> Stable und nightly sind verschiedene Repos. Daher musst Du für die Installation einer stabilen Version nach Verwendung der nightly auch das Repo auf stable umstellen.

Ich würde gerne auf das Nightly 0.127.2 umstellen. Nichts auf das letzte stable. ist das dann nicht möglich?

StefanSchoof, 2024-06-28:
Es gibt kein "Nightly 0.127.2", nur z.B. ein "0.127.2+1719195631", siehe https://cloudsmith.io/~evcc/packages/

Nighty wird jeder Version mit aus dem neusten Code gebaut. Mit dem nächsten Release sind dann alle Funktionen in dem Release.

0x3d13f, 2024-06-28:
Alle verfügbaren Versionen (und deren Bezeichnung) kannst Du Dir mit `apt list -a evcc` anzeigen lassen.

wittmannchris, 2024-06-28:
Danke

closed 2024-06-28 by wittmannchris