evcc-io/evcc #14606: Test Token expired 27.6.2024 - new one?

Flirina, 2024-06-28:
### Describe the bug

Hi,
i just installed evcc yesterday evening and was just only once able to start the system with the provided test token. after a reboot it says: 

[main  ] FATAL 2024/06/28 07:52:15 sponsortoken: rpc error: code = Unknown desc = token has invalid claims: token is expired
[main  ] FATAL 2024/06/28 07:52:15 will attempt restart in: 15m0s

i guess its related to the test token, which expired yesterday. any chance to get a new test token?
for sure its planned to get a purchased token soon.

# trial token, valid until 2024-06-27
sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJldmNjLmlvIiwic3ViIjoidHJpYWwiLCJleHAiOjE3MTk1MTEyMDAsImlhdCI6MTcxNjkxOTIwMCwic3BlIjp0cnVlLCJzcmMiOiJtYSJ9.WAeeK8L-ZzaUUs92R6wm2pLboeCw6jyG0KjN-p6qIn8

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
-
```

### Log details

```text
[main  ] FATAL 2024/06/28 07:52:15 sponsortoken: rpc error: code = Unknown desc = token has invalid claims: token is expired
[main  ] FATAL 2024/06/28 07:52:15 will attempt restart in: 15m0s

# trial token, valid until 2024-06-27
sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJldmNjLmlvIiwic3ViIjoidHJpYWwiLCJleHAiOjE3MTk1MTEyMDAsImlhdCI6MTcxNjkxOTIwMCwic3BlIjp0cnVlLCJzcmMiOiJtYSJ9.WAeeK8L-ZzaUUs92R6wm2pLboeCw6jyG0KjN-p6qIn8
```

### What type of operating system are you running?

HomeAssistant Add-on

naltatis, 2024-06-28:
Danke für den Hinweis. Token ist erneuert.

Flirina, 2024-06-28:
thx, aber wo?
auf der sponsorship site steht noch der alte?
[image]

oder gilt der alte wieder?

naltatis, 2024-06-28:
Sorry, in den falschen Branch gepusht. Nun ists auch korrekt deployt 😄
https://docs.evcc.io/docs/sponsorship