evcc-io/evcc #14608: cannot create vehicle type template - Volkswagen

wittmannchris, 2024-06-28:
### Describe the bug

After update to 0.127.3 , the VW temlatete cannot created.

Maybe an error because of two cars in obe account 

Downgrade to 0.127.2 (latest) solved the issue for the moment

### Steps to reproduce

1.Update tp 0.127.3
2.Could not find car
3.
...

### Configuration details

```yaml
vehicles:
- type: template
  template: vw
  title: ID3
  icon: car
  user: xxx
  password: xxx
  vin: xxx
  capacity: 58
  phases: 3
  cache: 20m
  mode: pv
  minCurrent: 6
  maxCurrent: 16
  timeout: 10s
  name: ev5
```

### Log details

```text
[main ] ERROR 2024/06/27 21:59:02 creating vehicle ev5 failed: cannot create vehicle type 'template': cannot create vehicle type 'vw': cannot find vehicle, got: [WVXXXXXXXX WVXXXXXXX]

sudo evcc -l debug vehicle [main  ] INFO 2024/06/27 22:01:25 evcc 0.127.3 (c04f52f8) [main  ] INFO 2024/06/27 22:01:25 using config file: /etc/evcc.yaml [db    ] INFO 2024/06/27 22:01:26 using sqlite database: /root/.evcc/evcc.db [main  ] ERROR 2024/06/27 22:01:28 creating vehicle ev5 failed: cannot create vehicle type 'template': cannot create vehicle type 'vw': cannot find vehicle, got: [WVXXXXXX WVXXXXXXX] .Soc:     vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'vw': cannot find vehicle, got: [WVXXX WVWXXX] Capacity: 58.0kWh Features: [Offline Retryable]
```

### What type of operating system are you running?

Linux

### Version

0.127.3 (latest)

oberschmuh, 2024-06-28:
try to use template: id instead of template: vw

Henne-mds, 2024-06-28:
same here. template: id

- type: template
  template: id 
  title: ID.7  
  user: ** 
  password: ***  
  vin: ***
  capacity: 77  
  phases: 3  
  icon: car  
  cache: 15m  
  mode: pv

digitalonkel, 2024-06-28:
Same with Cupra Born, see separate bug report

andig, 2024-06-28:
Could you share your login with info@evcc.io for testing? It seems strange this should affect the VW/ID integration and I'd like to verify whats wrong.

Henne-mds, 2024-06-28:
Du brauchst die my.volkswagen.de Login-Daten, richtig? Ich würde dafür mein PW ändern. Reicht es bis Sonntag Abend?Von unterwegs gesendet

andig, 2024-06-28:
Perfekt!

con313, 2024-06-28:
Hast du dich zuletzt mal bei Volkswagen ID angemeldet? In den letzten 1-2 Wochen gab es dort ein neues Pop-Up zu „Datenschutz-Richtlinien“, bis zur Bestätigung blieb mein Fahrzeug offline.

andig, 2024-06-28:
Das hat nix damit zu tun:

    cannot find vehicle, got: [WVXXXXXXXX WVXXXXXXX]

wittmannchris, 2024-06-29:
> Das hat nix damit zu tun. Hier fehlt einfach die VIN als Unterscheidungsmerkmal:
> 
> ```
> cannot find vehicle, got: [WVXXXXXXXX WVXXXXXXX]
> ```

Habe hier nur meine VINs für die Öffentlichkeit unkenntlich gemacht. Kann dir gerne das Template per PN senden. 

> Hast du dich zuletzt mal bei Volkswagen ID angemeldet? In den letzten 1-2 Wochen gab es dort ein neues Pop-Up zu „Datenschutz-Richtlinien“, bis zur Bestätigung blieb mein Fahrzeug offline.

Ja alles geprüft und akzeptiert. Mit der 0.127.2 läuft ja wieder alles. Sobald ich auf 0.127.3 Update findet er kein Fahrzeug mehr.

Henne-mds, 2024-06-29:
***@***.***: 9811E#ccVon unterwegs gesendet

andig, 2024-06-29:
Ihr braucht nicht zu raten ;). Es liegt aus irgendeinem Grund am Template und ich brauche ein Testobjekt.

closed 2024-06-29 by andig