evcc-io/evcc #14609: ocpp charge power timeout but actual the data is transferred

efschu, 2024-06-28:
### Describe the bug

I have dual connector chargers connected via ocpp.

After plugging second connector charge power is not read anymore. Not on first connector (where it works until second charge is started) nor second.

### Steps to reproduce

Start charging at one connector. Then start charging at other one.

### Configuration details

```yaml
chargers:
- type: template
  template: ocpp
  timeout: 2m
  name: EVB-xxxx-1
  stationid: EVB-xxxx
  connector: 1
- type: template
  template: ocpp
  timeout: 2m
  name: EVB-xxxx-2
  stationid: EVB-xxxx
  connector: 2

loadpoints:
- title: Wido01
  charger: EVB-xxxx-1
  circuit: Widocircuit
  mode: off
  mincurrent: 6
  maxcurrent: 16
- title: Wido02
  charger: EVB-xxxx-2
  circuit: Widocircuit
  mode: off
  mincurrent: 6
  maxcurrent: 16
```

### Log details

```text
[lp-1 ] ERROR 2024/06/28 11:10:25 charge power: timeout 

[ocpp ] TRACE 2024/06/28 11:10:19 received JSON message from EVB-xxxx: [2,"2e113a7b-df11-4650-b78c-46cb35fdca42","MeterValues",{"connectorId":1,"transactionId":2,"meterValue":[{"timestamp":"2024-06-28T08:21:14Z","sampledValue":[{"value":"15.9","measurand":"Current.Import","phase":"L1","unit":"A"},{"value":"15.8","measurand":"Current.Import","phase":"L2","unit":"A"},{"value":"15.0","measurand":"Current.Import","phase":"L3","unit":"A"},{"value":"19977","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"10235","measurand":"Power.Active.Import","unit":"W"},{"value":"214","measurand":"Voltage","phase":"L1","unit":"V"},{"value":"218","measurand":"Voltage","phase":"L2","unit":"V"},{"value":"223","measurand":"Voltage","phase":"L3","unit":"V"}]}]}]
[ocpp ] TRACE 2024/06/28 11:10:19 sent JSON message to EVB-xxxx: [3,"2e113a7b-df11-4650-b78c-46cb35fdca42",{}]

[lp-2 ] ERROR 2024/06/28 11:10:26 charge power: timeout 

[ocpp ] TRACE 2024/06/28 11:10:24 received JSON message from EVB-xxxx: [2,"2e113a80-df11-4650-b78c-46cb3e649860","MeterValues",{"connectorId":2,"transactionId":2,"meterValue":[{"timestamp":"2024-06-28T08:21:14Z","sampledValue":[{"value":"15.5","measurand":"Current.Import","phase":"L1","unit":"A"},{"value":"15.9","measurand":"Current.Import","phase":"L2","unit":"A"},{"value":"15.1","measurand":"Current.Import","phase":"L3","unit":"A"},{"value":"74717","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"10185","measurand":"Power.Active.Import","unit":"W"},{"value":"218","measurand":"Voltage","phase":"L1","unit":"V"},{"value":"213","measurand":"Voltage","phase":"L2","unit":"V"},{"value":"224","measurand":"Voltage","phase":"L3","unit":"V"}]}]}]
[ocpp ] TRACE 2024/06/28 11:10:24 sent JSON message to EVB-xxxx: [3,"2e113a80-df11-4650-b78c-46cb3e649860",{}]
```

### What type of operating system are you running?

Linux

### Version

0.127.3

efschu, 2024-06-28:
Even more weired, cus sometimes it works

```
[lp-1 ] ERROR 2024/06/28 11:27:55 charge power: timeout
[lp-1 ] DEBUG 2024/06/28 11:28:25 charge power: 10257W
[lp-1 ] DEBUG 2024/06/28 11:28:55 charge power: 10257W
[lp-1 ] DEBUG 2024/06/28 11:28:55 !! session: chargeRater.chargedEnergy=4.2 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/06/28 11:28:55 charge total import: 21.187kWh
[lp-1 ] DEBUG 2024/06/28 11:28:55 charger status: C
[lp-1 ] DEBUG 2024/06/28 11:28:55 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/06/28 11:29:25 charge power: 10257W
[lp-1 ] DEBUG 2024/06/28 11:29:55 charge power: 10257W
[lp-1 ] ERROR 2024/06/28 11:30:26 charge power: timeout 

[ocpp ] TRACE 2024/06/28 11:28:02 received JSON message from EVB-xxxx: [2,"2e113ea1-df11-4650-b78c-46cbcb63cefc","MeterValues",{"connectorId":1,"transactionId":2,"meterValue":[{"timestamp":"2024-06-28T08:28:18Z","sampledValue":[{"value":"15.8","measurand":"Current.Import","phase":"L1","unit":"A"},{"value":"15.7","measurand":"Current.Import","phase":"L2","unit":"A"},{"value":"14.9","measurand":"Current.Import","phase":"L3","unit":"A"},{"value":"21187","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"10257","measurand":"Power.Active.Import","unit":"W"},{"value":"216","measurand":"Voltage","phase":"L1","unit":"V"},{"value":"219","measurand":"Voltage","phase":"L2","unit":"V"},{"value":"226","measurand":"Voltage","phase":"L3","unit":"V"}]}]}]
[ocpp ] TRACE 2024/06/28 11:28:12 received JSON message from EVB-xxxx: [2,"2e113eab-df11-4650-b78c-46cb142d6938","MeterValues",{"connectorId":1,"transactionId":2,"meterValue":[{"timestamp":"2024-06-28T08:28:22Z","sampledValue":[{"value":"15.8","measurand":"Current.Import","phase":"L1","unit":"A"},{"value":"15.7","measurand":"Current.Import","phase":"L2","unit":"A"},{"value":"14.9","measurand":"Current.Import","phase":"L3","unit":"A"},{"value":"21199","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"10257","measurand":"Power.Active.Import","unit":"W"},{"value":"216","measurand":"Voltage","phase":"L1","unit":"V"},{"value":"219","measurand":"Voltage","phase":"L2","unit":"V"},{"value":"226","measurand":"Voltage","phase":"L3","unit":"V"}]}]}]
[ocpp ] TRACE 2024/06/28 11:28:26 received JSON message from EVB-xxxx: [2,"2e113eba-df11-4650-b78c-46cb0c57d191","MeterValues",{"connectorId":2,"transactionId":2,"meterValue":[{"timestamp":"2024-06-28T08:28:27Z","sampledValue":[{"value":"15.5","measurand":"Current.Import","phase":"L1","unit":"A"},{"value":"15.9","measurand":"Current.Import","phase":"L2","unit":"A"},{"value":"15.0","measurand":"Current.Import","phase":"L3","unit":"A"},{"value":"75945","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"10257","measurand":"Power.Active.Import","unit":"W"},{"value":"220","measurand":"Voltage","phase":"L1","unit":"V"},{"value":"216","measurand":"Voltage","phase":"L2","unit":"V"},{"value":"226","measurand":"Voltage","phase":"L3","unit":"V"}]}]}]
[ocpp ] TRACE 2024/06/28 11:28:37 received JSON message from EVB-xxxx: [2,"2e113ec4-df11-4650-b78c-46cb41236ccd","MeterValues",{"connectorId":2,"transactionId":2,"meterValue":[{"timestamp":"2024-06-28T08:28:33Z","sampledValue":[{"value":"15.5","measurand":"Current.Import","phase":"L1","unit":"A"},{"value":"15.9","measurand":"Current.Import","phase":"L2","unit":"A"},{"value":"15.0","measurand":"Current.Import","phase":"L3","unit":"A"},{"value":"75957","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"10257","measurand":"Power.Active.Import","unit":"W"},{"value":"220","measurand":"Voltage","phase":"L1","unit":"V"},{"value":"216","measurand":"Voltage","phase":"L2","unit":"V"},{"value":"226","measurand":"Voltage","phase":"L3","unit":"V"}]}]}]
[ocpp ] TRACE 2024/06/28 11:28:47 received JSON message from EVB-xxxx: [2,"2e113ece-df11-4650-b78c-46cb0bef0609","MeterValues",{"connectorId":2,"transactionId":2,"meterValue":[{"timestamp":"2024-06-28T08:28:35Z","sampledValue":[{"value":"15.5","measurand":"Current.Import","phase":"L1","unit":"A"},{"value":"15.9","measurand":"Current.Import","phase":"L2","unit":"A"},{"value":"15.0","measurand":"Current.Import","phase":"L3","unit":"A"},{"value":"75968","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"10257","measurand":"Power.Active.Import","unit":"W"},{"value":"220","measurand":"Voltage","phase":"L1","unit":"V"},{"value":"216","measurand":"Voltage","phase":"L2","unit":"V"},{"value":"226","measurand":"Voltage","phase":"L3","unit":"V"}]}]}] 
```

and then stops working again

```
[lp-1 ] ERROR 2024/06/28 11:36:25 charge power: timeout
[lp-1 ] ERROR 2024/06/28 11:36:55 charge power: timeout
[lp-1 ] ERROR 2024/06/28 11:36:56 charge rater: charge meter error timeout
[lp-1 ] ERROR 2024/06/28 11:36:56 charge total import: timeout
[lp-1 ] DEBUG 2024/06/28 11:36:56 charger status: C
[lp-1 ] DEBUG 2024/06/28 11:36:56 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] ERROR 2024/06/28 11:37:25 charge power: timeout
[lp-1 ] ERROR 2024/06/28 11:37:55 charge power: timeout 

[ocpp ] TRACE 2024/06/28 11:37:51 received JSON message from EVB-xxxx: [2,"2e1140f0-df11-4650-b78c-46cbfceb7ab9","MeterValues",{"connectorId":1,"transactionId":2,"meterValue":[{"timestamp":"2024-06-28T08:32:15Z","sampledValue":[{"value":"13.2","measurand":"Current.Import","phase":"L1","unit":"A"},{"value":"13.0","measurand":"Current.Import","phase":"L2","unit":"A"},{"value":"12.7","measurand":"Current.Import","phase":"L3","unit":"A"},{"value":"21773","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"8656","measurand":"Power.Active.Import","unit":"W"},{"value":"217","measurand":"Voltage","phase":"L1","unit":"V"},{"value":"221","measurand":"Voltage","phase":"L2","unit":"V"},{"value":"226","measurand":"Voltage","phase":"L3","unit":"V"}]}]}]
[ocpp ] TRACE 2024/06/28 11:37:51 sent JSON message to EVB-xxxx: [3,"2e1140f0-df11-4650-b78c-46cbfceb7ab9",{}] 
```

andig, 2024-06-28:
Please add full log. 10 lines are insufficient.

andig, 2024-06-30:
Wir erwarten alle 2 Minuten einen Wert:

```go
// isMeterTimeout checks if meter values are outdated.
// Must only be called while holding lock.
func (conn *Connector) isMeterTimeout() bool {
	return conn.timeout > 0 && conn.clock.Since(conn.meterUpdated) > conn.timeout
}

var _ api.Meter = (*Connector)(nil)

func (conn *Connector) CurrentPower() (float64, error) {
	if !conn.cp.Connected() {
		return 0, api.ErrTimeout
	}

	conn.mu.Lock()
	defer conn.mu.Unlock()

	// zero value on timeout when not charging
	if conn.isMeterTimeout() {
		if conn.txnId != 0 {
			return 0, api.ErrTimeout
		}

		return 0, nil
	}

	if m, ok := conn.measurements[types.MeasurandPowerActiveImport]; ok {
		f, err := strconv.ParseFloat(m.Value, 64)
		return scale(f, m.Unit), err
	}

	return 0, api.ErrNotAvailable
}
```

anscheinend gabs bei Dir keinen. Daher der Fehler. Wenns mehr Logfile gibt sollte sich das bestätigen.

closed 2024-06-30 by andig

premultiply, 2024-07-01:
Mal ganz davon abgesehen gibt es für die seltsamen EvBox-Boxen extra ein Template:
https://docs.evcc.io/docs/devices/chargers#evbox-elvi