evcc-io/evcc #14610: v0.127.3 Cupra Born keine Verbindung mehr

digitalonkel, 2024-06-28:
### Describe the bug

Seit der Version 0.123.7 kann keine Verbindung mehr hergestellt werden.
Beim Prüfen der Einstellungen erscheint folgende Fehlermeldung:

cannot create vehicle type 'template': cannot create vehicle type 'cupra': missing id_token

### Steps to reproduce

1. create car
2.
3.
...

### Configuration details

```yaml
Einstellungen über UI
```

### Log details

```text
cannot create vehicle type 'template': cannot create vehicle type 'cupra': missing id_token
```

### What type of operating system are you running?

Linux

StefanSchoof, 2024-06-28:
Gab da bei einigen einen AGB oä auf der Website (nicht App) die abzuarbeiten ist.

pschlein, 2024-06-28:
Hi
ich hatte auch gerade das Problem mit einem Seat Mii.
Es gibt keine Web App mehr wo dies gesetzt werden kann, aber ich habe einen Workaround gefunden.

- App starten
- Profil
- "Ihre Daten"
- "Persönliche Daten bearbeiten"

Dies öffnet einen Browser -> einloggen -> AGBs akzeptieren

Dann sollte es wieder funktionieren (zumindest funktionierte das bei mir)

digitalonkel, 2024-06-28:
Ja, kann ich bestätigen. Funktioniert.

closed 2024-06-28 by digitalonkel

mhibu, 2024-07-10:
Habe seit Neustart mit V0.128.1 keine Verbindung mehr zum Seat Mii. AGBs habe ich aber bereits vor kurzem zugestimmt.
Ich habe gesehen, dass in einem geschlossen Issue [ein](https://github.com/evcc-io/evcc/issues/14753) andere Seat Mii auch keine Verbindung mehr hat. Was ist los?

bekomme folgende Fehlermeldung:

cannot create vehicle type 'template': cannot create vehicle type 'seat': missing id_token