evcc-io/evcc #14618: Falsche Anzeige Ladestand im Home

skipperkm, 2024-06-29:
### Describe the bug

Hi,

anbei zwei Screenshots. Der Ladestand wird unterschiedlich im Home und in der Konfiguration angezeigt. In der Konfiguration ist der richtige zu sehen.

[image]
[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
chargers:
  - name: wb1
    type: template
    template: cfos
    host: 172.16.98.74

loadpoints:
  - title: Carport
    charger: wb1
    mode: off
    vehicle: mgzsev
    soc:
      poll:
        mode: connected
        interval: 1m
      estimate: true

vehicles:
  - name: mgzsev
    type: template
#    template: mg
    template: mg2mqtt
    title: MGZSEV # Wird in der Benutzeroberfläche angezeigt (optional)
    user: VVVV@VVVV.de # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
#    password: xxxxxxxxxxx # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    vin: LSXXXX # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind
    capacity: 70 # Akkukapazität in kWh (optional)
    timeout: 1h
    mode: pv
    mincurrent: 6
    maxcurrent: 16
```

### Log details

```text
[lp-1 ] DEBUG 2024/06/29 08:39:02 charge power: 5368W
[site ] DEBUG 2024/06/29 08:39:02 pv power: 5095W
[site ] DEBUG 2024/06/29 08:39:02 battery soc: 98%
[site ] DEBUG 2024/06/29 08:39:02 battery power: -197W
[site ] DEBUG 2024/06/29 08:39:02 grid meter: -2W
[site ] DEBUG 2024/06/29 08:39:02 site power: -99W
[lp-1 ] DEBUG 2024/06/29 08:39:02 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/06/29 08:39:02 !! session: chargeRater.chargedEnergy=3.3 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/06/29 08:39:02 charger status: C
[lp-1 ] DEBUG 2024/06/29 08:39:02 soc estimated: 88.93% (vehicle: 84.70%)
[lp-1 ] DEBUG 2024/06/29 08:39:02 vehicle soc: 89%
[lp-1 ] DEBUG 2024/06/29 08:39:02 vehicle range: 396km
[lp-1 ] DEBUG 2024/06/29 08:39:02 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/06/29 08:39:02 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/06/29 08:39:02 pv charge current: 7.92A = 7.78A + 0.143A (-99W @ 3p)
[lp-1 ] DEBUG 2024/06/29 08:39:02 max charge current: 7.92A
[lp-1 ] DEBUG 2024/06/29 08:39:02 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
```

### What type of operating system are you running?

Linux

### Version

0.127.2

skipperkm, 2024-06-29:
Hi,

ich noch mal, nur am Rande: es wäre schön, wenn bei automatischer Phasenumschaltung angezeigt würde wie der aktuelle status ist. Entweder 1- oder 3-phasig oder (schicker) die Ströme der 3 Phasen

schönes WE

andig, 2024-06-29:
Auf Home shcreibt der Estimator den Ladestand fort. Wenn dich das stört kannst du 

    estimate: false zum deaktivieren nutzen.

closed 2024-06-29 by andig

skipperkm, 2024-06-29:
Hi,

Danke für die Info 

Weis nicht ob Du das gelesen hast :

nur am Rande: es wäre schön, wenn bei automatischer Phasenumschaltung angezeigt würde wie der aktuelle status ist. Entweder 1- oder 3-phasig oder (schicker) die Ströme der 3 Phasen

Als nice to have 🙂

andig, 2024-06-29:
Das hat mit diesem Issue nix zu tun und wird bereits angezeigt. Gerne Discussion dazu.

VolkerK62, 2024-06-29:
@skipperkm  die kleinen Balken unter der Ladeleistung zeigen die Anzahl der Phasen an

[image]

skipperkm, 2024-06-29:
Danke , hätte ich auch drauf kommen können