evcc-io/evcc #14624: Charged energy wrong (warp2 pro)

andreas-bulling, 2024-06-29:
### Describe the bug

Since a couple of days (not sure which evcc or warp2 update I might have done to trigger this I am afraid) some charging sessions are suddenly not properly shown in evcc - but correctly in the warp2 charger web interface:

[image]

[image]

This only happens for guest vehicles so far. And only once the vehicle is unplugged. The charged energy was shown correctly in the evcc interface.

I have not changed the configuration - the setup has been running flawlessly before for months.

### Steps to reproduce

1. install evcc in HA
2. install warp2 pro charger
3. connect both following the official evcc configuration for this charger
...

### Configuration details

```yaml
evcc 0.127.3 running within HomeAssistant
warp2 charger pro running firmware 2.4.0+66558a84 (created 28/05/2024 09:40:52)
```

### Log details

```text
lp-1 ] DEBUG 2024/06/29 16:54:28 !! session: chargeRater.chargedEnergy=1147.4 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/06/29 16:54:38 !! session: chargeRater.chargedEnergy=1147.4 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/06/29 16:54:48 !! session: chargeRater.chargedEnergy=1147.4 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/06/29 16:54:58 !! session: chargeRater.chargedEnergy=1147.4 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/06/29 16:55:08 !! session: chargeRater.chargedEnergy=1147.4 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/06/29 16:55:18 !! session: chargeRater.chargedEnergy=1147.4 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/06/29 16:55:28 !! session: chargeRater.chargedEnergy=1147.4 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/06/29 16:55:38 !! session: chargeRater.chargedEnergy=1147.4 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/06/29 16:55:48 !! session: chargeRater.chargedEnergy=1147.4 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/06/29 16:55:58 !! session: chargeRater.chargedEnergy=1147.4 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/06/29 16:56:08 !! session: chargeRater.chargedEnergy=1147.4 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/06/29 16:56:18 !! session: chargeRater.chargedEnergy=1147.4 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/06/29 16:56:28 !! session: chargeRater.chargedEnergy=1147.4 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/06/29 16:56:38 !! session: chargeRater.chargedEnergy=1147.4 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/06/29 16:56:48 !! session: chargeRater.chargedEnergy=1147.4 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/06/29 16:56:58 !! session: chargeRater.chargedEnergy=1147.4 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/06/29 16:57:08 !! session: chargeRater.chargedEnergy=1147.4 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/06/29 16:57:18 !! session: chargeRater.chargedEnergy=1147.4 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/06/29 16:57:28 !! session: chargeRater.chargedEnergy=1147.4 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/06/29 16:57:38 !! session: chargeRater.chargedEnergy=1147.4 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/06/29 16:57:48 !! session: chargeRater.chargedEnergy=1147.4 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/06/29 16:57:58 !! session: chargeRater.chargedEnergy=1147.4 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/06/29 16:58:08 !! session: chargeRater.chargedEnergy=1147.4 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/06/29 16:58:18 !! session: chargeRater.chargedEnergy=1147.4 - chargedAtStartup=0.0
```

### What type of operating system are you running?

HomeAssistant Add-on

andig, 2024-06-29:
> some charging sessions are suddenly not properly shown in evcc - but correctly in the warp2 charger web interface:

Was heisst das konkret?

closed 2024-06-29 by andig

andreas-bulling, 2024-06-29:
Wie kann ich das vollständige Log bekommen? Muss ich da eine neue Charging Session abwarten/machen?
 
Ich habe alles kopiert was mir im Webinterface angezeigt wird. Auch CSV Export liefert leider nur einen Teillog...

andreas-bulling, 2024-06-29:
[image]

andig, 2024-07-01:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

andreas-bulling, 2024-07-02:
I know but I haven't figured out how I can generate a detailed log file when running evcc as an add-on within HomeAssistant.

andig, 2024-07-02:
@VolkerK62 ich sehe gerade, dass wir noch gar nicht den UI-Weg dokumentiert haben?
@andreas-bulling einfach Hamburger-Menü -> Logs?

andreas-bulling, 2024-07-02:
Das habe ich ja gemacht - da sehe ich (nur) den in meiner initialen Nachricht geposteten Auszug. Der Log ist offensichtlic nicht vollständig bzw. geht nicht ausreichend weit zurück.

Ist das normal oder sollte ich dort einen vollständigen/umfassenden Log finden, der Tage/Wochen zurückgeht?

andig, 2024-07-02:
Da gibts nur die letzten 10k Zeilen.

andreas-bulling, 2024-07-02:
Tja, dann weiß ich wie gesagt leider nicht, wie ich in HomeAssistant mit evcc als Add-On an den vollständigen Log komme. Würde ich gerne, weil dieser Bug wirklich nervig ist...

andig, 2024-07-02:
Einfach ohne HA?