evcc-io/evcc #14625: evcc produces Network Errors GET when changing the value for "Regelverhalten → Residualleistung"

SolarPowerEV, 2024-06-29:
### Describe the bug

When you change the value for Residualleistung, evcc afterwards produces between one (desktop browser) and an endless string (mobile browser) of "Network Errors GET".
This only ends when you close the mobile browser window and open evcc in a new one.

The problem seems to be more severe when a car is charging but also occurs when no car is charging.

[image]

### Steps to reproduce

1. Open evcc on your mobile (Android?) browser
2. Go to Konfiguration → Regelverhalten → Residualleistung
3. Change the value (for example from 200 W to 300 W)
4. Click on "Speichern"
5. Click on "Home" to get back to the evcc mainscreen
6. Wait a minute to see a variety of error messages pop up (see screenshots above (0.127.2) and below (0.127.3))
...
[image]

### Configuration details

```yaml
nothing too much out of the ordinary
```

### Log details

```text
[main ] INFO 2024/06/29 20:19:27 evcc 0.127.3
[main ] INFO 2024/06/29 20:19:27 using config file: /etc/evcc.yaml
[db ] INFO 2024/06/29 20:19:27 using sqlite database: /root/.evcc/evcc.db
[main ] INFO 2024/06/29 20:19:27 listening at :7070
[alfen ] DEBUG 2024/06/29 20:19:28 detected 3p alfen
[alfen ] DEBUG 2024/06/29 20:19:28 detected 3p alfen
[alfen ] DEBUG 2024/06/29 20:19:28 detected 3p alfen
[lp-1 ] DEBUG 2024/06/29 20:19:29 charge total import: 2019.032kWh
[lp-2 ] DEBUG 2024/06/29 20:19:30 charge total import: 4801.415kWh
[lp-3 ] DEBUG 2024/06/29 20:19:30 charge total import: 168.873kWh
[site ] INFO 2024/06/29 20:19:30 site config:
[site ] INFO 2024/06/29 20:19:30 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2024/06/29 20:19:30 grid: power ✓ energy ✓ currents ✓
[site ] INFO 2024/06/29 20:19:30 pv 1: power ✓ energy ✓ currents ✗
[site ] INFO 2024/06/29 20:19:30 pv 2: power ✓ energy ✓ currents ✗
[site ] INFO 2024/06/29 20:19:30 battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site ] INFO 2024/06/29 20:19:30 vehicles:
[site ] INFO 2024/06/29 20:19:30 vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[site ] INFO 2024/06/29 20:19:30 vehicle 2: range ✓ finish ✗ status ✓ climate ✗ wakeup ✗
[site ] INFO 2024/06/29 20:19:30 vehicle 3: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1 ] INFO 2024/06/29 20:19:30 loadpoint 1:
[lp-1 ] INFO 2024/06/29 20:19:30 mode: pv
[lp-1 ] INFO 2024/06/29 20:19:30 charger: power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1 ] INFO 2024/06/29 20:19:30 meters: charge ✓
[lp-1 ] INFO 2024/06/29 20:19:30 charge: power ✓ energy ✓ currents ✓
[lp-2 ] INFO 2024/06/29 20:19:30 loadpoint 2:
[lp-2 ] INFO 2024/06/29 20:19:30 mode: pv
[lp-2 ] INFO 2024/06/29 20:19:30 charger: power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-2 ] INFO 2024/06/29 20:19:30 meters: charge ✓
[lp-2 ] INFO 2024/06/29 20:19:30 charge: power ✓ energy ✓ currents ✓
[lp-3 ] INFO 2024/06/29 20:19:30 loadpoint 3:
[lp-3 ] INFO 2024/06/29 20:19:30 mode: pv
[lp-3 ] INFO 2024/06/29 20:19:30 charger: power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-3 ] INFO 2024/06/29 20:19:30 meters: charge ✓
[lp-3 ] INFO 2024/06/29 20:19:30 charge: power ✓ energy ✓ currents ✓
[site ] DEBUG 2024/06/29 20:19:30 set buffer soc: 80
[site ] DEBUG 2024/06/29 20:19:30 set buffer start soc: 0
[site ] DEBUG 2024/06/29 20:19:30 set priority soc: 20
[site ] DEBUG 2024/06/29 20:19:30 set battery discharge control: false
[site ] DEBUG 2024/06/29 20:19:30 set residual power: 100
[lp-1 ] DEBUG 2024/06/29 20:19:30 !! active phases: 3p = min(0p measured 0p vehicle 0p physical 0p charger)
[... 440 lines cut ...]
[site ] DEBUG 2024/06/29 20:43:29 grid meter: 6W
[site ] DEBUG 2024/06/29 20:44:18 grid currents: [1.05 -0.83 1.41]A
[site ] DEBUG 2024/06/29 20:44:35 site power: 854W
[lp-1 ] DEBUG 2024/06/29 20:44:36 charge total import: 2019.032kWh
[lp-1 ] DEBUG 2024/06/29 20:44:36 charger status: A
[site ] DEBUG 2024/06/29 20:44:36 ----
[lp-1 ] DEBUG 2024/06/29 20:44:36 charge power: 0W
[lp-1 ] DEBUG 2024/06/29 20:44:36 charge currents: [0 0 0]A
[lp-2 ] DEBUG 2024/06/29 20:44:36 charge power: 0W
[lp-2 ] DEBUG 2024/06/29 20:44:36 charge currents: [0 0 0]A
[lp-3 ] DEBUG 2024/06/29 20:44:36 charge power: 0W
[lp-3 ] DEBUG 2024/06/29 20:44:37 charge currents: [0 0 0]A
[site ] DEBUG 2024/06/29 20:45:28 pv power: 161W
[site ] DEBUG 2024/06/29 20:45:47 battery soc: 89%
[site ] DEBUG 2024/06/29 20:45:47 battery power: 2047W
[site ] DEBUG 2024/06/29 20:45:53 grid meter: 5W
```

### What type of operating system are you running?

Docker container

### Version

0.127.2 and 0.127.3

andig, 2024-06-30:
@SolarPowerEV check browsers developer console, network tab.

SolarPowerEV, 2024-06-30:
Sorry, I don't understand. For what purpose should I check the dev console of the browser? Do you need more data to find the bug?
How do I start that console on Chrome on my mobile phone? There is no menu "Weitere Tools" nor do I have a keyboard to press F12 or CTRL+SHIFT+J

andig, 2024-06-30:
> evcc afterwards produces between one (desktop browser) and an endless string (mobile browser) of "Network Errors GET".

Use desktop browser...

SolarPowerEV, 2024-06-30:
And then do what in the developer console? Look for error messages? Copy & paste the entire window into this thread?
And does that mean you cannot reproduce the error on your computers & browsers? 
Because then it would probably be easier that you're just checking the dev console for whatever you're looking for.

andig, 2024-06-30:
You would check which requests are executed to see whats wrong. But seems this isn‘t gonna happen. Closing.

closed 2024-06-30 by andig

SolarPowerEV, 2024-07-01:
Sorry that I am not a GO dev and hence can't predict what you exactly need and therefore had to ask for more details. 

I don't think the problem is just occuring on my computers and mobile phones, so I figured you might be interested in getting rid of the bug to improve the general quality of evcc. 

I can live with the problem, since I found the work around to close the browser after changing the setting and start again. If you don't care about your project producing error messages for no good reason, then so be it.

naltatis, 2024-07-05:
@SolarPowerEV I just tried to reproduce it but did not get the error you've described. Tried desktop safari and chrome (android & macos). I dont get an error in any of these scenarios. My assumption is, that this somehow depends on another setting (UI configured devices, ...) you've made. Would be helpful if you could provide a screenshot of the network tab of your browser console.

andig, 2024-07-05:
> but did not get the error you've described

Neither did I and hence- since no reproducer- closed.

SolarPowerEV, 2024-07-05:
I was able to reproduce the error immediately again. Switching from Firefox to Chrome didn't make a difference. 
Does that screenshot help / suffice?

[image]

SolarPowerEV, 2024-07-05:
Looks indeed more like something related to my devices than related to the browser used. The errors in windows "Meldungen" appeared immediately after saving. Even backwards in time. Like I saved the new value, left the "Konfiguration" window, and immediately saw the red warning triangle left of the hamburger icon. I clicked on it and it already had three messages in it. One from 20 seconds ago, from 10 seconds ago and one frome 1 or 2 seconds ago. Weird.

andig, 2024-07-05:
Könntest Du bei einem der gefailten mal bitte drauf klicken und schauen was für ein Request da gemacht werden sollte (URL, Inhalt)?

naltatis, 2024-07-05:
Ja, eine URL zu sehen wäre gut. Und schau bitte mal ob diese Fehler nicht vielleicht sogar auch auftreten, wenn du gar nichts auf der Config-Seite machst sondern einfach nur diese Seite besuchtst und dann wieder zurück auf Home gehst. Die `/status` Requests werden für jedes konfigurierte Device ausgeführt und das in regelmäßigen Abständen. Das erklärt auch, dass du unterschiedlich viele Meldungen siehst. Das sollte mit der Verweildauer auf der Konfig-Seite korrelieren.

Eine Sache die ich mir schon mal mitnehme. Das Fehler-Icon/Notifications oben in der Navi brauchen wir auch auf der Config-Seite. Aktuell wird das nur bei Home ausgespielt. Im Hintergrund werden aber alle Notifications gesammelt.

SolarPowerEV, 2024-07-05:
Also wenn ich nur auf die Konfigurationssseite gehen und sonst nichts mache und dann zurück auf die Home-Seite gehe (also das Haus anklicke), treten keine Fehler auf. Wenn das so wäre, wäre mir das Problem auch schon viel früher aufgefallen.

Wenn ich auf die Unterkonfigurationsseite "Regelverhalten" gehe und dort nichts verändere und dann mit "Abbrechen" bzw. dem "X" in der rechten oberen Ecke zurückgehe, tritt allerdings ebenfalls ein Fehler auf. Allerdings nur einmal und auch nicht immer (selbst mit Abspeichern). Es scheint nachts weniger Fehler zu geben, als tagsüber und während eines aktiven Ladevorgangs mehr Fehler als ohne einen Ladevorgang:

[image]

Das kam nach dem Verändern des "Residualleistungs"-Wertes und Abspeichern:
[image]

Und jetzt gerade ohne Abspeichern sogar wieder eine Kaskade von Fehlern:
[image]

Es scheint also in der Tat zu reichen, die Unterkonfigurationsseite aufzurufen. Werte verändern und abspeichern oder nicht, scheint keine Rolle zu spielen. Und wann nur ein Fehler oder gleich mehrere kommen, kann ich noch nicht ganz erkennen.

@andig : wo genau soll ich draufklicken? Nichts auf dem Popup "Meldungen" ist anklickbar, abgesehen von den beiden Buttons "Vollständiges Log angesehen" und "Meldungen entfernen" am unteren Rand.

andig, 2024-07-06:
> wo genau soll ich draufklicken

Entwicklerkonsole, einen der roten Requests. Ich möchte die Details dazu sehen.

SolarPowerEV, 2024-07-06:
Eine erste zusätzliche Erkenntnis: im Chrome kommen nur die kürzeren "Network Error. GET" Meldungen (egal ob Desktop oder Handy). Im Firefox kommen die langen Fehlermeldungen. 

Hilft der Screenshot weiter?
[image]

Oder der?
[image]

Wenn nicht, müsst Ihr mir bitte ganz genau sagen, auf welchen (Unter-)tab ich noch klicken soll. Ich bin kein Frontend-Dev und habe daher keine Ahnung, nach was genau Ihr sucht.

andig, 2024-07-06:
Man sieht, dass der Browser einen ganz normalen Request sieht/bekommt. Das deutet für mich auf ein wir auch immer geartetes lokales Problem bei Dir hin:

[image]

Bitte noch einmal bei diesem Request auf "Preview" klicken.

andig, 2024-07-06:
Und: mal bitte auf der Konsole ein 

    curl -v <url>

SolarPowerEV, 2024-07-06:
Nur kurz dazu: das Problem tritt auf, egal ob ich als Browser den Firefox oder Chrome unter Windows nutze oder Chrome auf meinem Mobiltelefon (Android) verwende, um evcc aufzurufen.
 
Umgekehrt tritt das Problem wiederum nur auf, wenn ich in evcc ins Untermenü Konfiguration → Regelverhalten gegangen bin. Sonst läuft alles. 

Wenn es also nur an meiner Hardware oder meinem Setup liegen würde, müsste das Problem ja immer auftreten. Irgendwie scheint das UI von evcc also schon eine Rolle zu spielen. 

Bei Anklicken des Reiters "Preview" kommt nur das hier:
[image]

Das war auch bei den letzten Malen so. Deswegen habe ich da keinen Screenshot von gemacht, weil ich dachte, der eine Satz hilft eh nicht weiter.

Bei der Konsole komme ich leider nicht weiter. Welche URL willst Du genau haben?
Oder reichen Dir diese Fehlermeldungen?

[image]

[image]

andig, 2024-07-06:
> Welche URL willst Du genau haben?

Die, die nicht geht.

andig, 2024-07-06:
@naltatis irgendwie habe ich das Gefühl, dass das ein `panic` im http Handler sein könnte und der Request abbricht. 
@SolarPowerEV deshalb wäre `curl` gut.

naltatis, 2024-07-06:
@andig wenn ich den `net:: ERR_EMPTY_RESPONSE` richtig verstehen kommt hier gar keine Antwort vom Server weil da der Request fatal kaputt geht. Da hilft uns vmtl. auch `curl` nicht. Der wird auch nicht mehr bekommen als der Browser.

Spannend ist allerdings, dass diese Fehler nicht im Applikationslog auftauchen. @SolarPowerEV kannst du das noch mal verifizieren? Du hattest ja ganz oben schon ein Log gepostet. So sieht das auch immer noch aus, oder?

Konkret zur Konfiguration. Der Fehler tritt bei der Abfrage der Status-Informationen der Geräte `pv-master`, `pv-slave` und `battery` auf. @SolarPowerEV magst du mal deine evcc.yaml posten?

SolarPowerEV, 2024-07-06:
> > Welche URL willst Du genau haben?
> 
> Die, die nicht geht.

Ich habe diverse URLs ausprobiert (siehe in den beiden letzten oben von mir geposteten Screenshots rechts unten). Da kamen aber nur komische Fehlermeldungen von der Browser-Konsole zurück, dass http: oder amadeus: ein unexpected identifier seien oder kein ":" akzeptiert würde.

SolarPowerEV, 2024-07-07:
Den Fehler bekam ich gerade zweimal nicht nachgestellt, obwohl ich beidesmal über 1 Minute in dem Konfig-Fenster gewartet habe. Wie schon gesagt tritt der Fehler tagsüber, wenn die Sonne scheint, eher auf und wenn gerade ein Ladevorgang läuft, dann noch eher. 

Die evcc.yaml zu posten ist immer ein ziemlicher Aufwand, wegen der ganzen Tokens und Passwörter darin und ich habe dann immer Sorge, etwas übersehen zu haben und unfreiwillig öffentlich zu stellen. Aber ok.

naltatis, 2024-07-07:
Ja, das passt zu deiner Beschreibung. Auf der Config-Seite fragen wir alle 10s den Zustand aller Geräte ab. Kann durchaus sein, dass dein Huawei Gerät damit nicht klar kommt. Lass uns das so machen wie du vorgeschlagen hast. Wie betrachten dieses Thema hier erstmal als erledigt und schauen uns das noch mal an, wenn Config UI ausgerollt ist und andere das Problem ggf. auch haben. Danke fürs melden.

SolarPowerEV, 2024-07-09:
> Ja, das passt zu deiner Beschreibung. Auf der Config-Seite fragen wir alle 10s den Zustand aller Geräte ab.

Eine kurze Frage hätte ich dazu noch: spricht denn etwas dagegen, auch in der Config-Seite einfach das übliche Interval gemäß den Einstellungen im Regelverhalten weiterlaufen zu lassen? Also in meinem Falle die 45 Sekunden? Das würde das Thema doch für alle lösen, deren Wechselrichter- und Gridmeter-Setup keine 10 Sekunden zulässt.

naltatis, 2024-07-09:
@SolarPowerEV Ja, spricht nichts gegen. Inzwischen hätten wir den Interval-Wert an der Stelle, wo wir das Polling machen auch verfügbar. Ich pass das bei nächster Gelegenheit mit an.