evcc-io/evcc #14627: wrong logging of charging sessions and charging limit after fast unplug/plugin

eweri, 2024-06-29:
### Describe the bug

I have two cars, on PHEV and one BEV. I have often had the impression that something is wrong with the charging protocol. 
And today I finally have the evidence.
One of my cars was loading with SoC set to 90% and I came home and unpluged this car and started to load the other car I just got home with. After a few hours I looked into evcc how charging is going and I saw that my Tuscon with a 13,8kWh batterie capacity has loaded 28,8kWh and still going and the SoC was set to 90% (no SoC-Limit is set for this car). Because my Wallbox (Tinkerforge WARP3) logs charging sessions too, i compared it with the log sessions from evcc. The running session is not log in session log, but my wallbox shows a new charging session.
Wallbox session: e-up 29.06.2024 10:07 - 15,057 kWh 06:35:35
Wallbox current session: Tuscon 29.06.2024 16:43 - 6,887 kWh 06:16:45 - still running
evcc current session: Tucson - 29,3kWh 17:38h - still running
As you can see in the log of evcc at 16:43 the car was changed but no new charging sessions was started.

The two cars are differentiated via NFC and activated at the wallbox

I think it has to do with time between unpluging the first car and pluging the next car. Interval is set to 30s.

I never discover this effect, when there are minutes or hours between unplugin one car and plugin another car.

### Steps to reproduce

1. unplug charging car
2. plugin next car (i think this has to be done in a short time)
3. present NFC dongle to the wallbox
4. after some time charging starts
5. open evcc web gui: the new car is shown, but charging start with the energy the first car has loaded
6. look at charging sessions, no session for the unplugged car
7.
...

### Configuration details

```yaml
main  ] INFO 2024/06/29 23:13:29 evcc 0.127.3
[main  ] INFO 2024/06/29 23:13:29 using config file: /home/eweri/evcc.yaml
[db    ] INFO 2024/06/29 23:13:30 using sqlite database: /home/eweri/.evcc/evcc.db
[warp  ] INFO 2024/06/29 23:13:31 connecting evcc-1005429029 at tcp://192.168.51.10:1883

Konfiguration (/home/eweri/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.3296 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.085 # [currency]/kWh

log: info
levels:
  cache: error
  ev1: 	debug
  ev2: debug
  guest_car: info
  lp-1: info
  mqtt: info

interval: 30s # control cycle interval

meters:
- name: grid1
  type: template
  template: sma-energy-meter 
  usage: grid  
  host: 192.168.51.169 
- name: pv2
  type: template
  template: sma-inverter-speedwire 
  usage: pv  
  host: 192.168.51.15  
  password: *****

chargers:
#- name: wallbox1
#  type: template
#  template: tinkerforge-warp 
#  host: 192.168.51.10  
#  port: 1883  
#  topic: warp2/29JZ  
- name: wallbox2
  type: template
  template: tinkerforge-warp3
  host: 192.168.51.10
  port: 1883
  topic: warp3/2at5

loadpoints:
#- title: WARP2 (abgebaut)
#  charger: wallbox1
#  mode: off
- title: WARP3 vorm Haus
  charger: wallbox2
  mode: off

vehicles:
- name: ev1
  type: template
  template: hyundai 
  title: Tucson-XXXX  
  icon: tractor
  user: *****
  password: *****
  vin: *****
  language: de  
  phases: 1
  capacity: 13.8
  mode: minpv
  identifiers:
  - 04:53:30:22:BC:73:80
- name: ev2
  type: template
  template: id
  title: e-up-XXXXX
  icon: car
  user: *****
  password: *****
  vin: *****
  phases: 2
  capacity: 32.3
  mode: minpv
  identifiers:
  - 04:68:1C:22:BC:73:80
- name: guest_car
  type: template
  template: offline
  title: Gastfahrzeug 
  icon: car
  mode: minpv
  identifiers:
  - 04:69:96:72:3B:74:80

site:
  title: Bei Ewerings
  meters:
    grid: grid1
    pv:
    - pv2

Version: `0.127.3`
```

### Log details

```text
Jun 29 14:56:49 sun-watch evcc[6628]: [lp-1  ] INFO 2024/06/29 14:56:49 stop charging <-
Jun 29 14:58:49 sun-watch evcc[6628]: [lp-1  ] INFO 2024/06/29 14:58:49 start charging ->
Jun 29 16:43:49 sun-watch evcc[6628]: [lp-1  ] INFO 2024/06/29 16:43:49 vehicle updated: e-up-XXXXX -> Tucson-XXXX
Jun 29 22:24:50 sun-watch evcc[6628]: [lp-1  ] WARN 2024/06/29 22:24:50 vehicle soc: unexpected status: 401 (Unauthorized) (ignored by estimator)
Jun 29 22:24:50 sun-watch evcc[6628]: [lp-1  ] ERROR 2024/06/29 22:24:50 vehicle soc limit: unexpected status: 401 (Unauthorized)
Jun 29 22:24:50 sun-watch evcc[6628]: [lp-1  ] ERROR 2024/06/29 22:24:50 vehicle range: unexpected status: 401 (Unauthorized)
Jun 29 22:27:19 sun-watch evcc[6628]: [lp-1  ] WARN 2024/06/29 22:27:19 vehicle soc: timeout (ignored by estimator)
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.127.3

eweri, 2024-06-29:
Sorry that was not my current config - here comes my current config:

[main  ] INFO 2024/06/29 23:50:50 evcc 0.127.3
[main  ] INFO 2024/06/29 23:50:50 using config file: /home/eweri/evcc.yaml
[db    ] INFO 2024/06/29 23:50:51 using sqlite database: /home/eweri/.evcc/evcc.db
[warp  ] INFO 2024/06/29 23:50:52 connecting evcc-1082578823 at tcp://192.168.51.10:1883

Konfiguration (/home/eweri/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.3296 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.085 # [currency]/kWh

log: info
levels:
  cache: error
  ev1: info
  ev2: info
  guest_car: info
  lp-1: info
  mqtt: info

interval: 30s # control cycle interval

meters:
- name: grid1
  type: template
  template: sma-energy-meter 
  usage: grid  
  host: 192.168.51.169 
- name: pv2
  type: template
  template: sma-inverter-speedwire 
  usage: pv  
  host: 192.168.51.15  
  password: *****

chargers:
#- name: wallbox1
#  type: template
#  template: tinkerforge-warp 
#  host: 192.168.51.10  
#  port: 1883  
#  topic: warp2/29JZ  
- name: wallbox2
  type: template
  template: tinkerforge-warp3
  host: 192.168.51.10
  port: 1883
  topic: warp3/2at5

loadpoints:
#- title: WARP2 (abgebaut)
#  charger: wallbox1
#  mode: off
- title: WARP3 vorm Haus
  enable:
# einschalten, wenn 3 Minuten lang mindestens 400W Ueberschuss eingespeist wurde
    delay: 3m
    threshold: -400
  disable:
# ausschalten, wenn 10 Minuten lang mehr als 1050W aus dem Netz bezogen  wurde
    delay: 10m
    threshold: 1050
  charger: wallbox2
  mode: off

vehicles:
- name: ev1
  type: template
  template: hyundai 
  title: Tucson-XXXX
  icon: tractor
  user: *****
  password: *****
  vin: *****
  language: de  
  phases: 1
  capacity: 13.8
  mode: pv
  identifiers:
  - 04:53:30:22:BC:73:80
- name: ev2
  type: template
  template: id
  title: e-up-XXXX
  icon: car
  user: *****
  password: *****
  vin: *****
  phases: 2
  capacity: 32.3
  mode: minpv
  identifiers:
  - 04:68:1C:22:BC:73:80
- name: guest_car
  type: template
  template: offline
  title: Gastfahrzeug 
  icon: car
  mode: minpv
  identifiers:
  - 04:69:96:72:3B:74:80

site:
  title: Bei Ewerings
#  residualPower: -690
  meters:
    grid: grid1
    pv:
    - pv2

Version: `0.127.3`

eweri, 2024-06-29:
Sorry - above is not my current config, i've attached it here
[config.txt](https://github.com/user-attachments/files/16042260/config.txt)

premultiply, 2024-06-30:
Try to reduce the `interval` as low as possible without getting into unstable grid metering values after charge current modifications.

Most systems allow for intervals in the range from 10 to 20 seconds. Some even lower.
Only some very slow systems like Huawei with dongle require long intervals.

andig, 2024-06-30:
> The two cars are differentiated via NFC and activated at the wallbox

We can only differenciate sessions if we see a status change. The wallbox will see that electrically, evcc only though API polling. If you go from B->A->B and evcc only sees B->B then there's nothing to detect.