evcc-io/evcc #14628: [Feature Request] Functionality to edit individual charging sessions

andreas-bulling, 2024-06-30:
It unfortunately happens (issue opened already) that the amount of energy charged is recorded incorrectly by evcc. These wrong (very high) energy values mess up the charging statistics.

It'd be great if, in addition to the ability to delete an entry, a simple editing interface was available. Sufficient for me would already be if I could edit/fix the amount of energy charged. If additional values could be edited that would also be nice of course.

naltatis, 2024-06-30:
Our primary goal should be to fix the root causes of wrong data (charged energy) being written to the database. In normal operation, there should be no need to manually adjust values, but I see why you might want this as an easier fixing mechanism when something goes wrong.

We could make some fields UI editable, but this opens up more opportunity for other inconsistencies. The price and solar_percentage field and the upcoming savings https://github.com/evcc-io/evcc/pull/13950 are logically linked. We could try to build heuristics to also adjust these values based on energy change, but this will not be accurate when you e.g. have a dynamic tariff.

My hope is, that this "wrong values due to bugs" use case is not widespread enough to justify this feature. Direct editing in your SQLite db always works for one-time hiccups.

Nevertheless, I'll keep this issue open to collect more opinions but limit the scope (title) to charging values. Editing start/end or loadpoints is out of scope for this issue.

SolarPower2024, 2024-06-30:
I also would appreciate this option.

Not because of general issues in the calculation, but in my case I e.g. made a simulation if a dynamic energy tariff would make sense. So I enabled the dynamic tariff and kept it running for couple of weeks. However, of course after my simulation my prices are not correct anymore.
I updated it directly in the database but what I noticed is that in that charge history the average value of the price per month is not updated, so I'm not so sure if everything really was updated correctly. I think that is what you mean that some fields are linked logically. So perhaps also not so easy.

To avoid manual adjustments directly in database for user, this feature could be helpful. For simulation as use case. (but not sure if this is wanted, anyway no priority)

andreas-bulling, 2024-06-30:
If I knew how to edit the sqlite database directly I could do that. I am running evcc as an HomeAssistant add-on and I have no idea where the database is stored/how to login into it and fix the wrong values..

SolarPower2024, 2024-06-30:
Ich switche mal schnell auf deutsch, da das eh nur indirekt mit dem Thema zu tun hat.
Wenn du keinen Speicherort verändert hast, kommst du leider nicht direkt auf die DB drauf. Als schnellen workaround: du kannst ein Backup vom Add-on machen und dann siehst du die DB darin. Bearbeiten, abspeichern und dann das Backup wieder herstellen.

Generell würde ich aber empfehlen die DB nicht im Add-on zu speichern, sondern den Speicherort ins HA config Verzeichnis zu verlegen, dann kommst du zb per Samba share immer direkt drauf. (nur regelmäßig Backup machen der DB dann nicht vergessen)

Für das Bearbeiten der DB gibt's auch auf Windows eigene Programme, dann kann man mit einer GUI arbeiten. Einfach mal nach Sqlite  DB Browser googeln.

andreas-bulling, 2024-06-30:
Gerade wollte ich schreiben, dass ich die Datenbank gefunden habe (per docker eine Shell aufmachen und dann in `/data/evcc.db`. Leider hat das image kein sqlite(3) wie es scheint, insofern muss ich wohl den von dir beschriebenen Weg gehen (über das Backup).

Frage ist nur: wenn ich den Speicherort dann dauerhaft ändern möchte, und das confnig-File entsprechend editiere, verliere ich dann alle alten Daten, d.h. wird im neuen Zielverzeichnis die DB neu angelegt?

SolarPower2024, 2024-06-30:
Zur ersten Frage: Wenn du das Verzeichnis im Add-on änderst, brauchst du nur die "alte" DB in das neue Verzeichnis kopieren und du hast alle Daten wie gehabt.

Zweite Frage: System - Backup - Backup erstellen - partielles Backup - evcc addon auswählen.

andreas-bulling, 2024-06-30:
vielen Dank. Sorry für die Beginner-Fragen, aber wie kopiere ich eine Datei aus einem docker container (hier die evcc.db) in das HA-Filesystem?

SolarPower2024, 2024-06-30:
Am besten lädst du dir das Addon Samba share runter, dann kommst du via IP Adresse ganz normal im Windows Explorer drauf.

andreas-bulling, 2024-06-30:
Backup: hat geklappt mit dem Editieren, nur hochladen des Backup-Files gibt mir einen "Bad request"...

SolarPower2024, 2024-06-30:
Schwierig zu sagen an was es genau liegt. Am einfachsten wäre es wohl den Speicherort der DB im addon zu ändern und die gerade von dir bearbeite DB an die Stelle zu kopieren.