evcc-io/evcc #14637: Fault with Fronius Wattpilot

Maddin2020, 2024-06-30:
### Describe the bug

Hello, I have a problem with the Fronius Wattpilot. I have the error charger enabled: could not find value of alw. I don´t know what i can do. The log you can see in the follow. The configuration is the same as before version. Then i have no problems.

### Steps to reproduce

[image]
[image]

### Configuration details

```yaml
see in the description
```

### Log details

```text
see in the description
```

### What type of operating system are you running?

Docker container

### Version

0.127.3

andig, 2024-06-30:
Unfortunately your report is missing a detailed log file and config. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2024-06-30 by andig

maijo366, 2024-07-27:
> ### Describe the bug
> 
> Hello, I have a problem with the Fronius Wattpilot. I have the error charger enabled: could not find value of alw. I don´t know what i can do. The log you can see in the follow. The configuration is the same as before version. Then i have no problems.
> ### Steps to reproduce
> 
> ![WhatsApp Image 2024-06-30 at 14 35 00](https://private-user-images.githubusercontent.com/53319531/344465552-20183960-ff52-4d9a-b150-e970ca10c4ff.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjIwOTE2NTMsIm5iZiI6MTcyMjA5MTM1MywicGF0aCI6Ii81MzMxOTUzMS8zNDQ0NjU1NTItMjAxODM5NjAtZmY1Mi00ZDlhLWIxNTAtZTk3MGNhMTBjNGZmLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcyN1QxNDQyMzNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zMjMyNDNjOTFjYzcxMGQ2NWU2NjRjNDJjMzVkOGJiNGRjMTIxMGZmZmE0YzBkOGJiMzk1M2FiNDgwNDE5MzVjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Be-k836jhq55DH6TjGhqph-dbAalLlYvpn06LiqUBgc) ![WhatsApp Image 2024-06-30 at 14 35 50](https://private-user-images.githubusercontent.com/53319531/344465553-976154ba-42a4-474b-b5ca-1fdb102359a6.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjIwOTE2NTMsIm5iZiI6MTcyMjA5MTM1MywicGF0aCI6Ii81MzMxOTUzMS8zNDQ0NjU1NTMtOTc2MTU0YmEtNDJhNC00NzRiLWI1Y2EtMWZkYjEwMjM1OWE2LmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcyN1QxNDQyMzNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02NWU1MDFjOTZjNGZiZDc4MTU2MmExYmMxMzE5ZDVhMGFlZjAxYmU3YmY4OGRmYmFjZmMyNWFkMjNkMWM1ZGJkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.JPeKwTMeAmlk03MwgW19D6zX3EEl-2omtFBZrcJz45k)
> ### Configuration details
> 
> ```yaml
> see in the description
> ```
> 
> ### Log details
> 
> ```
> see in the description
> ```
> 
> ### What type of operating system are you running?
> 
> Docker container
> ### Version
> 
> 0.127.3

Hello, is there a solution to the problem. I have the same error.
Homeassistant Addon V 0.128.4