evcc-io/evcc #14641: Error with Fronius Wattpilot - charger enabled: could not find value of alw

Maddin2020, 2024-06-30:
### Describe the bug

Hello, I have a problem with the Fronius Wattpilot. I have the error charger enabled: could not find value of alw. I don´t know what i can do. 
[image]
[image]

### Steps to reproduce

[evcc-20240630-162123-trace.log](https://github.com/user-attachments/files/16045005/evcc-20240630-162123-trace.log)
[evcc-20240630-162137-trace.log](https://github.com/user-attachments/files/16045006/evcc-20240630-162137-trace.log)

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: x

interval: 30s # control cycle interval

meters:
- type: template
  template: fronius-gen24
  usage: grid
  host: 192.168.xxx.xxx
  port: 502
  name: grid

- type: template
  template: fronius-gen24
  usage: pv
  host: 192.168.xxx.xxx
  port: 502
  name: pv

- type: template
  template: fronius-gen24
  usage: pv
  host: 192.168.xxx.xxx
  port: 502
  name: pv2

- type: template
  template: fronius-solarapi-v1
  usage: battery
  host: 192.168.xxx.xxx
  user: customer
  password: x
  name: battery
  capacity: 20.48

- name: hoymiles
  type: template
  template: hoymiles-opendtu
  usage: pv
  host: 192.168.xxx.xxx # IP address or hostname

chargers:

- type: template
  template: fronius-wattpilot
[... 601 lines cut ...]
[lp-2  ] DEBUG 2024/06/30 16:21:05 charge power: 0W
[lp-2  ] DEBUG 2024/06/30 16:21:05 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/06/30 16:21:10 charge power: 0W
[lp-2  ] DEBUG 2024/06/30 16:21:10 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/06/30 16:21:10 !! session: chargeRater.chargedEnergy=9.6 - chargedAtStartup=0.0
[lp-2  ] DEBUG 2024/06/30 16:21:10 charger status: A
[lp-2  ] DEBUG 2024/06/30 16:21:10 next soc poll remaining time: 3m50s
[lp-2  ] ERROR 2024/06/30 16:21:10 charger enabled: could not find value of alw
[lp-2  ] DEBUG 2024/06/30 16:21:15 charge power: 0W
[lp-2  ] DEBUG 2024/06/30 16:21:15 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/06/30 16:21:20 charge power: 0W
[lp-2  ] DEBUG 2024/06/30 16:21:20 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/06/30 16:21:20 !! session: chargeRater.chargedEnergy=9.6 - chargedAtStartup=0.0
[lp-2  ] DEBUG 2024/06/30 16:21:20 charger status: A
[lp-2  ] DEBUG 2024/06/30 16:21:20 next soc poll remaining time: 3m40s
[lp-2  ] ERROR 2024/06/30 16:21:20 charger enabled: could not find value of alw
```

### What type of operating system are you running?

Docker container

### Version

0.127.3

Maddin2020, 2024-06-30:
I hope now everythink you need is there.

andig, 2024-06-30:
@Maddin2020 could you please retry and set `wattpilot: trace` or add trace log from UI?

/cc @mabunixda message seems to come from inside the library?

Maddin2020, 2024-06-30:
[evcc-20240630-180428-trace.log](https://github.com/user-attachments/files/16045278/evcc-20240630-180428-trace.log)
Here the complete Log. If i restart the Container i have no error but then the fault will come back after a time

andig, 2024-07-01:
Unfortunately, log doesn't show any Wattpilot. Since we have a similar topic in https://github.com/evcc-io/evcc/discussions/14629#discussioncomment-9921974, lets continue there.

closed 2024-07-01 by andig