evcc-io/evcc #14644: GO-E -  charger out of sync: expected disabled, got enabled

wittmannchris, 2024-06-30:
### Describe the bug

Update to 0.127.3 brings charger out of sync: expected disabled, got enabled in log. 

Downgrade to 0.127.2 (latest) solve the issue

### Steps to reproduce

1.Update to 0.127.3
2.Log
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

tariffs:
  currency: EUR # (default EUR)
  grid:
    # either
    type: tibber
    token: "*"

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.122 # [currency]/kWh

influx:
  url: http://192.168.178.47:8086
  database: evcc_log
  token: *
  org: *

log: debug
levels:
  cache: error

# unique installation id
plant: *

interval: 20s # control cycle interval

sponsortoken: *

meters:
- type: template
  template: sma-home-manager
  usage: grid
  host: 192.168.178.90
  name: grid1
- type: template
  template: sma-inverter-modbus
  id: 3
  host: 192.168.178.91
  port: 502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: sma-inverter-modbus
  id: 3
  host: 192.168.178.92
  port: 502
  usage: pv
  modbus: tcpip
  name: pv3
- type: template
  template: sma-sbs-modbus
  id: 3
  host: 192.168.178.93
  port: 502
  usage: battery
  modbus: tcpip
  capacity: 7.161
  watchdog: 60s
  name: battery4

chargers:
- type: template
  template: go-e-v3
  host: 192.168.178.98
  name: wallbox6

vehicles:
- type: template
  template: vw
  title: ID3
  user: *
  password: *
  vin: *6
  capacity: 58
  phases: 3
  cache: 20m
  mode: pv
  minCurrent: 6
  maxCurrent: 16
  timeout: 10s
  name: ev5

loadpoints:
- title: Garage
  charger: wallbox6
  vehicle: ev5
  mode: pv

site:
  title: Daham
  meters:
    grid: grid1
    pv:
    - pv2
    - pv3
    battery:
    - battery4
  residualpower: 200

hems:
  type: sma
  AllowControl: false
```

### Log details

```text
charger out of sync: expected disabled, got enabled
```

### What type of operating system are you running?

Linux

### Version

0.127.3 (latest)

andig, 2024-06-30:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

wittmannchris, 2024-06-30:
was going back to .127.2 ! 

Logfile now online...

andig, 2024-06-30:
Ah, that's helpful, thanks. Sounds like known issue already fixed in https://github.com/evcc-io/evcc/pull/14639. Could you please try new nightly version in ~15 minutes?

wittmannchris, 2024-06-30:
> Ah, that's helpful, thanks. Sounds like known issue already fixed in #14639. Could you please try new nightly version in ~15 minutes?

One hour approx. Kids need a bath and bed ;)

wittmannchris, 2024-06-30:
seems to be fixed 👍

andig, 2024-06-30:
Super, danke 🙏

closed 2024-06-30 by andig

wittmannchris, 2024-06-30:
Gerne