evcc-io/evcc #14658: Implement start/stop charge support for fiat

FraBoCH, 2024-07-01:
**Is your feature request related to a problem? Please describe.**
When charging a Fiat on TWC3 (or other non-controllable charger), evcc cannot start or stop charge.

**Describe the solution you'd like**
The API provide a start charge action => we could use that.
https://github.com/TA2k/ioBroker.fiat managed to implement changing the charging schedules. My proposition is that evcc manage 1 of the 3 schedules available to simulate a stop request. 
In practice:
- when starting charge: update schedule with start time = current time (to be consistent)
- when in mode pv and not enough solar: update schedule with end time = current time + disable duration (few minutes generally)
- when manually switching to off: update schedule with end time = current time

**Describe alternatives you've considered**
Manually editing the start schedule every day according to solar forecast + automation in HA to send notifications to me & my wife when we need to force start or stop charge manually by physically pressing the button on the car. Yes, I’m doing that, my wife not so much ;-)

andig, 2024-07-01:
This is currently blocked on the fact that controlling Fiat requires PIN access and PIN access via evcc doesn't work for an unknown reason. I've given up after spending two full days on debugging. If you can get PIN access to work this would be easy to add.

FraBoCH, 2024-07-01:
Ok, that’s bad new. I’ll try to dig into this pin issue, but if TA2k has it working, it means there is a solution ;-)

SolarPowerEV, 2024-07-01:
As a FIAT 500e owner (but not being a developer) I can only add that the PIN issue with evcc is weird. 

If evcc accesses the FIAT server without a PIN, it will get only the last saved SoC back. Which can be a day or more old and completely different from the actual SoC. But the data will come quickly.

If evcc accesses the FIAT server with a valid PIN, either one of the following two things will happen:
1. If the car is driving or charging at that same moment the PIN will trigger a refresh and the server will send back the very latest SoC from a few seconds ago.
So far everything works as one would expect it. 

2. If the car is not driving or charging, the evcc request with the PIN will take a long time and then produce a few errors. So no value for `SoC` is reported back and also no values for `range` or `odometer`

[image]

andig, 2024-07-01:
I always thought we would get an HTTP 403? The long timeout is just retries happening. That could be avoided and instead `api.ErrMustRetry` returned. An http trace log could give clarity.

SolarPowerEV, 2024-07-01:
The 403 was returned until a few weeks (or months, I don't recall exactly) ago. 
Then it was replaced by a 503 message instead (or maybe it was first the 503 and then the 403? I could dig through my old screenshots if it's of any importance).
And then some time ago the "i/o timeout" and "must retry" came back instead.

FraBoCH, 2024-07-03:
Thanks for the additional information. As I see it, it would be enough to have it working when the vehicle is plugged in. I also think we need to mimic the behavior of the Fiat app who requires the pin only when sending commands. I don’t think a « deep refresh » is needed for a proper solar charging to work.

andig, 2024-07-03:
> I don’t think a « deep refresh » is needed for a proper solar charging to work.

That means what? It is required to get an updated soc.

FraBoCH, 2024-07-03:
> > I don’t think a « deep refresh » is needed for a proper solar charging to work.
> 
> That means what? It is required to get an updated soc.

Just a thought for now ;-) But it looks I’m getting the soc from a status call (https://channels.sdpr-01.fcagcv.com/v2/accounts/****/vehicles/***/status) and that I may have it updated when the car is charging. At least in the Fiat app, I don’t need to enter my pin to see the soc change when charging. It may not be Uber-fresh data, but it may be good enough for evcc to stop near the limit soc configured.
Anyway, will try to make progress on that in the next days in my fork and I will propose a pull request as soon as I have something.

SolarPowerEV, 2024-07-06:
In my experience the SoC without the PIN tends to deviate more and more over time. Like if you start charging at 30%, an hour later the SoC estimated by evcc is at 55%, whereas the unupdated SoC of the FIAT server is still at 30% and if you then check the app using the PIN code (or trigger a refresh with evcc on the command line using an evcc.yaml with a PIN in it) the actual SoC is already at 59%. 
Now you might think 4% isn't too much, but for me it makes a difference if I've set the charging limit to 90% and evcc stops charging only when it thinks (estimates) 90% have been reached, but in reality the car is already very close to 100% SoC.

andig, 2024-07-06:
Can we please return to topic, which is "using pin does error". If anybody does not want to use the pin that's fine and not discussed here.

FraBoCH, 2024-07-12:
Strangely enough, I have no pin error and managed to send with success a start charge command which worked !
Could this be due to the activation of services from Fiat ? I myself have activated (purchased) the "My car" and "My Remote" services.
I'll go on with tackling the schedule update. Fingers crossed.

SolarPowerEV, 2024-07-13:
Are you sure the car wasn't charging or driving when you tried to access it with the PIN?
Because in those cases I don't get any errors as well. Only when the car is parked and not connected to a charger do the errors WITH a PIN occur. 

Regarding the services: Aren't they automatically activated in the first three years for everyone? 
Anyhow, according to the Dashbopard of MyUconnect, I am still subscribed to these Uconnect Services until February 2026: 
"My Assistant", "My Car", Q"My eCharge", "My Navigation" and "My Remote" 

So that can't be a factor for the PIN error. I wish you luck with the schedule update!

FraBoCH, 2024-07-13:
The car was connected but not charging. And actually, it's when the car is connected to the charger managed by EVCC that it needs to start or stop the charge.
Anyway, I have a first working implementation taking over the first schedule available in the API to start & stop the charge. A pull request is coming soon ;-)

andig, 2024-07-13:
I'm not sure if we want to go this way. It requires a _lot_ of API to enable a rare TWC3 scenario. Putting on hold.

FraBoCH, 2024-07-13:
> I'm not sure if we want to go this way. It requires a _lot_ of API to enable a rare TWC3 scenario. Putting on hold.

I don’t see the drawback. I only added 1 api call to set the schedules and this is working great to start or stop the charging on a TWC3. I don’t know if this is a rare scenario but it is mine. I still believe it’s worth making the TWC3 work as much as possible with more vehicles. 

However, if you are worried about the number of API calls and see possible improvements to reduce them, I would be happy to discuss optimization in the PR, but I would need some guidance as this is my first experience in go ;-)

FraBoCH, 2024-07-16:
Hi @andig. Any new thoughts on that ? 
While waiting for Tesla to implemented OCPP or a real control API on the TWC3 (let's hope), my PR would in practice allow solar charging for my Fiat. I'm really looking forward to be able to stop managing manually my schedules in the Fiat app and rely on EVCC supervision. What can I do to see this new feature make its way to the next release ?

andig, 2024-07-17:
We have no plans to add this. The implementation is sound. However, it cannot provide a full solution (no current control), adds a very rare scenario (Twc with Fiat) and will likely be replaced by a better solution using Ocpp in the near future. Given this, we‘d line to avoid the additional api surface.

closed 2024-07-17 by andig

FraBoCH, 2024-07-17:
Ok. I can’t say I’m not disappointed but I understand your decision. 
I really hope it will soon be possible to actually control TWC3 and will definitely be one of the first to test with my odd « fleet » of Tesla MY and Fiat 500.
In the meantime, I’ll continue running my local build to start/stop with a fixed current. Still better than doing it manually.