evcc-io/evcc #14661: Indiv. current settings for 1p/3p charging

bjoernwuest, 2024-07-02:
Feature request: enable individual min and max current settings for one phase and three phase charging.

Motivation:
* in 1p, I would like to set a min of 10A to reduce charging losses (e.g. basic energy consumption of vehicle in relation to charging power)
* in 3p, I accept a min of 6A to enable PV surplus charging more early

Request:
Enable individual min and max current setting for one phase and three phase charging.

Alternatives:
* setting min current to 6A => would result in high charging loss in 1p mode but enable early start of PV surplus charging in 3p mode (i.e. 4.1kW)
* setting min current to 10A => would result in later start of PV surplus charging in 3p mode (i.e. 6.9kW) but reduce loss "ration" in 1p mode

sebnaf, 2024-07-07:
I would like to think this even further:

split this setting to pv & fast mode.

Using "pv", I always want to use 1p max - living right on the edge to 3p I do see 1p-3p switching 20 times a day. I want to fix this to 1p.

Using "fast", I always want to use 3p.

maatinh, 2024-07-12:
I think similar request has been discussed in the past in the context of some vehicles requiring different currents like my Zoe. It does charge with 7A@1P but needs at least 10A@3P.

maf-soft, 2024-08-07:
My Tesla is only able to charge max 11kW, but for one phase the range is more than 16A, which is good to reduce phase switching. So this is my reason to want individual limits (which I need to use in the vehicle config). Currently I set maxcurrent to 25A, but this makes evcc think it can calculate with 17kW...

ChrisGitHub42, 2024-08-19:
this is absolute needed...

as in Switzerland, due to regulation, when charing at 1-Phase, you are only allowed to charge with up to 8A, while charging at 3-Phase, there is no legal limit (you can use, what the car, charger, house-socket) allows you...

so currently you cannot operate this legally in Switzerland expect limiting the whole charging to 8A, what makes it completly useless....

ThiloGa, 2024-09-22:
Another reason for this feature is the scheduled loading:

1P@32A / 3P@16A / automatic switching between 1p and 3p results in starting too late, cause calculation of charging time is wrong.

michaelheiml, 2024-11-19:
+1 from me as I have requested this in the past and would love to see this for our Zoes..

Elchkopp, 2024-12-05:
My box officially registered with 11kW. But with limiting to 16A on all phases there is a gap between 1p/16A and 3p/6A and as a result, the box is often switching from 1p to 3p and back in PV-mode (too much for 1p/16A, but not enough for 3p/6A). 

With the indiv. current seeting 1p/20A (20A due to "Schieflast") and 3p/16A the gap will be closed and 
excessive switching is avoided.

The workaround set the configuration limit to 20A and limit the load management to 11kW (maxPower: 11000), the result is the same - BUT as it was written before the calculation of charging time is wrong.

andig, 2025-04-30:
It might be feasible to do this now as a config-ui-only feature.

Current API only knows global min/max settings. Can we leave this as-is and have an additional `3p` API or should we refactor the current api to accept and return 2 values for 1p, 3p?

If we go for 2 values we should also update any published keys like `effectiveMinCurrent` to contain 2 values (as slice?). If we introduce additional api like `(Get|Set)(Min|Max)Current3p` we could instead add additional publishing keys.

General rule of thumb for 1-value API: if 3p is not set, API will treat deviating 3p as not defined. Once 3p is set, existing (1p) API will or will not override 3p value- to be decided.

bjoernwuest, 2025-04-30:
> It might be feasible to do this now as a config-ui-only feature.
>
> Current API only knows global min/max settings. Can we leave this as-is and have an additional `3p` API or should we refactor the [...]

Not an expert on evcc but I would recommend a results such as:
```
{ "1p": {
  "min": ...,
  "max": ...
}, "3p": {
  "min": ...,
  "max": ...
}
```

This would make it obvious what to do. When setting the value, the same syntax could be used, while where any value is missing (either a min, or 1p/3p), it would just update the values provided. Yet, not sure how the current API looks like and this would break the API?

encbladexp, 2025-07-12:
> I don't understand why vehicle-specific settings are available via the loadpoint and not directly through the vehicle.

Isn't a loadpoint telling a car "I can to 1p/3p with X amps"? So from a technical point of view, its a loadpoint setting, not a vehicle setting.

Am I wrong?

marcelGoerentz, 2025-07-12:
> > I don't understand why vehicle-specific settings are available via the loadpoint and not directly through the vehicle.
> 
> Isn't a loadpoint telling a car "I can to 1p/3p with X amps"? So from a technical point of view, its a loadpoint setting, not a vehicle setting.
> 
> Am I wrong?

I think it's different since the installation, charger and vehicle max current can differ. So technically we have 3 values there. And the user may want to set a different value that is within these boundaries. So basically there are 4 values, with three of them giving the boundaries for the 4th.

premultiply, 2025-07-16:
Setup a load management circuit for the charger with values for current limit (1p) _and_ power limit (3p) should fix that in many cases.

mfuchs1984, 2025-07-17:
But this won't help concerning charge plans, right? Since they don't take load management limits into consideration.

poml88, 2025-08-30:
+1. I would also like to set different min Max Amps for 1p and 3p. Very useful for Zoes…

StefanSchoof, 2025-09-07:
Please do not just post +1. You can click thumbs up on the first post.

Posting just +1 will send a mail to everyone in this thread.

maatinh, 2025-12-16:
I don't think this should be closed as resolved by #25303. From my understanding this change adds a maximum Power. Some vehicles require different MINIMUM currents to charge correctly. As mentioned many times a Zoe starts charging OK with 7A@1p but wouldnt start charging with 7A@3P. It requires 10A or 11A at 3 Phases to start charging.

lterfloth, 2026-02-01:
Having a seperate timer for the phase switching (especially switching down) would be quite nice, too. Currently it is simply the same as the enable timer in the PV mode. That does not suffice for my case

D4Ci0, 2026-03-22:
Running into a problem:

Cupra Tavascan can charge 1p up to 20A or 3p 16A in my setup.

Automatic phase switching wants to switch from 1p 20A to 3p because available power is higher than 3p switch treshhold: 

`available power 4709W > 4140W min 3p threshold `
`start phase scale3p timer `

Problem: Surplus is only really ~109W, because car is charging with 4,6kW (20A 1p), switching to 3p is not benefitting in this situation.

Custom treshold and delay for 3p switching would be ideal I think.

Is there something on the roadmap?