evcc-io/evcc #14673: en+ ocpp hijacking logic

andig, 2024-07-02:
### Discussed in https://github.com/evcc-io/evcc/discussions/14602

<div type='discussions-op-text'>

<sup>Originally posted by **jeffborg** June 28, 2024</sup>
Hi

I have a en+ wallbox and it is sending the transactionId on metervalues even though the transaction has stopped

 `[2,"134","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-06-28T01:47:54Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"32.0","context":"Sample.Periodic"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"0"},{"measurand":"Voltage","unit":"V","phase":"L1","value":"234.4","context":"Sample.Periodic"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"1100.0","context":"Sample.Periodic"},{"measurand":"Power.Active.Import","unit":"W","value":"0","context":"Sample.Periodic"}]}],"transactionId":1}]`
 
The problem is here https://github.com/evcc-io/evcc/blob/c04f52f8518e0ef8e5edf8bbaa8fc518da68c706/charger/ocpp/connector_core.go#L56-L57 is assuming this is a live working transaction.

The only way to remove this transactionid from the wallbox is to turn it on and off again, then you get the MeterValues without it.

I can get more reliable operation by simply turning off meter values for the box, but I'm thinking this hijacking should only occur if evcc thinks the charger is enabled and if the import W value is > 0

At least then if evcc get's it wrong you can stop and start in evcc and it will work. Once the transaction id is set internally the only way to get charging working again is to power cycle the wallbox and restart evcc at the same time.
</div>

andig, 2024-07-02:
@jeffborg does this happen only after startup (when we want to hijack a transaction)?

jeffborg, 2024-07-02:
@andig the only way to get it working is to power cycle the evse (so it doesn't send txnid) and restart evcc (so to clear the transaction id stoped). Then there is no txnid sent it's blank on the meter values and everything works. 

Once this box has a transaction started no matter the state of started or stopped it will always send the txnid in the meter values. Doesn't matter if it disconnects and reconnects

andig, 2024-07-06:
> does this happen only after startup (when we want to hijack a transaction)?

ping @jeffborg 

> I don't know is there some message in the ocpp standard which we can use to check if the transaction id we are about to assume is ok is valid or not? or query the current status of it. stopped or started.

No... could you share an ocpp trace log of the evcc start? I'm uncertain how we would find out if there actually is a running transaction or not- maybe the messages give a hint.

jeffborg, 2024-07-07:
I haven't had a chance to get the trace output as I'm home late and if the charger <-> evcc communication breaks up I just need to get it working. 

Given the autostart change in #14733 I don't think this will work as my unit has no setting to set to just always be on. so will still require an initial start transaction upon startup of the unit to get going.

What I am thinking @andig is maybe if a StopTransaction returns rejected then assume transaction stopped. If StartTransaction returns rejected then there is already a running transaction, and if SetChargingProfile returns rejected then a StartTrasaction will be required to be sent to get things going and then the SetCharginingProfile to be resent.

premultiply, 2024-07-08:
See https://github.com/evcc-io/evcc/pull/14733#issuecomment-2213054603

andig, 2024-07-08:
> Given the autostart change in https://github.com/evcc-io/evcc/pull/14733 I don't think this will work as my unit has no setting to set to just always be on. so will still require an initial start transaction upon startup of the unit to get going.

It would be great if you could test this anyway. Out of curiosity: If your box doesn't have autostart- how would you charge without evcc?

jeffborg, 2024-07-08:
@andig Can't do it, if I put the charger into "plug and play" mode vs Network it never connects to the defined ocpp server, I was previously using chargehq so always had ocpp in use with the unit.

andig, 2024-07-08:
Oh, you just have to love those arbitrary OCPP implementations...

jeffborg, 2024-07-09:
I think I may have worked out the pattern here, I'm running a evcc in k8s and evcc got restarted due to code server being updated - forgot about this but it explains a lot.

- Midnight - 9am working fine charged as per charging plan
- 9:10am updated code server so evcc got restarted
- 12pm I noticed charging no longer working, so I'm guessing the evse stops the session when disconnected but still is sending the transactionId even though it's stopped

restarted evcc again here is the ocpp trace logs

```
[ocpp-1] DEBUG 2024/07/09 12:02:15 waiting for chargepoint: 5m0s
[ocpp ] INFO 2024/07/09 12:02:19 charge point connected, registering: SN10052307203612
[ocpp ] TRACE 2024/07/09 12:02:19 sent SN10052307203612: [2,"2431342368","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp ] TRACE 2024/07/09 12:02:19 recv SN10052307203612: [2,"2711","DataTransfer",{"vendorId":"EN+","messageId":"gatewayInfo","data":"{\"SN\":\"SN10052307203612\",\"fwVer\":\"V43.0.552\",\"gateCode\":107,\"measureType\":1,\"disconnectReason\":0,\"maxCurr\":320,\"phase\":1,\"sdCardStatus\":0}"}]
[ocpp ] TRACE 2024/07/09 12:02:19 sent SN10052307203612: [3,"2711",{"status":"Accepted"}]
[ocpp ] TRACE 2024/07/09 12:02:21 recv SN10052307203612: [3,"2431342368",{"status":"Accepted"}]
[ocpp ] TRACE 2024/07/09 12:02:21 recv SN10052307203612: [2,"2712","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-07-09T02:02:19Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"32.0","context":"Trigger"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"0.0"},{"measurand":"Voltage","unit":"V","phase":"L1","value":"242.8","context":"Trigger"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"0.0","context":"Trigger"},{"measurand":"Power.Active.Import","unit":"W","value":"0.0","context":"Trigger"}]}],"transactionId":3}]
[ocpp ] TRACE 2024/07/09 12:02:21 sent SN10052307203612: [2,"3540832731","ChangeConfiguration",{"key":"MeterValueSampleInterval","value":"10"}]
[ocpp-1] DEBUG 2024/07/09 12:02:21 hijacking transaction: 3
[ocpp ] TRACE 2024/07/09 12:02:21 TriggerMessage MeterValues for SN10052307203612: Accepted
[ocpp ] TRACE 2024/07/09 12:02:21 sent SN10052307203612: [3,"2712",{}]
[ocpp ] TRACE 2024/07/09 12:02:21 recv SN10052307203612: [3,"3540832731",{"status":"Accepted"}]
[ocpp-1] DEBUG 2024/07/09 12:02:21 enabling meter watchdog
[ocpp ] TRACE 2024/07/09 12:02:29 recv SN10052307203612: [2,"2713","Heartbeat",{}]
[ocpp ] TRACE 2024/07/09 12:02:29 sent SN10052307203612: [3,"2713",{"currentTime":"2024-07-09T02:02:29Z"}]
[ocpp ] TRACE 2024/07/09 12:02:31 sent SN10052307203612: [2,"3720629883","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp ] TRACE 2024/07/09 12:02:32 recv SN10052307203612: [3,"3720629883",{"status":"Accepted"}]
[ocpp ] TRACE 2024/07/09 12:02:32 TriggerMessage MeterValues for SN10052307203612: Accepted
[ocpp ] TRACE 2024/07/09 12:02:32 recv SN10052307203612: [2,"2714","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-07-09T02:02:31Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"32.0","context":"Trigger"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"0.0"},{"measurand":"Voltage","unit":"V","phase":"L1","value":"245.8","context":"Trigger"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"10798.0","context":"Trigger"},{"measurand":"Power.Active.Import","unit":"W","value":"0.0","context":"Trigger"}]}],"transactionId":3}]
[ocpp ] TRACE 2024/07/09 12:02:32 sent SN10052307203612: [3,"2714",{}]
[ocpp ] TRACE 2024/07/09 12:02:51 sent SN10052307203612: [2,"1243541157","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp ] TRACE 2024/07/09 12:02:52 recv SN10052307203612: [3,"1243541157",{"status":"Accepted"}]
[ocpp ] TRACE 2024/07/09 12:02:52 TriggerMessage MeterValues for SN10052307203612: Accepted
[ocpp ] TRACE 2024/07/09 12:02:53 recv SN10052307203612: [2,"2715","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-07-09T02:02:51Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"32.0","context":"Trigger"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"0.0"},{"measurand":"Voltage","unit":"V","phase":"L1","value":"245.8","context":"Trigger"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"10798.0","context":"Trigger"},{"measurand":"Power.Active.Import","unit":"W","value":"0.0","context":"Trigger"}]}],"transactionId":3}]
[ocpp ] TRACE 2024/07/09 12:02:53 sent SN10052307203612: [3,"2715",{}]
[ocpp ] TRACE 2024/07/09 12:03:11 sent SN10052307203612: [2,"2022665267","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp ] TRACE 2024/07/09 12:03:12 recv SN10052307203612: [3,"2022665267",{"status":"Accepted"}]
[ocpp ] TRACE 2024/07/09 12:03:12 TriggerMessage MeterValues for SN10052307203612: Accepted
[ocpp ] TRACE 2024/07/09 12:03:13 recv SN10052307203612: [2,"2716","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-07-09T02:03:11Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"32.0","context":"Trigger"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"0.0"},{"measurand":"Voltage","unit":"V","phase":"L1","value":"245.8","context":"Trigger"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"10798.0","context":"Trigger"},{"measurand":"Power.Active.Import","unit":"W","value":"0.0","context":"Trigger"}]}],"transactionId" [... cut ...] unit":"V","phase":"L1","value":"245.8","context":"Trigger"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"0.0","context":"Trigger"},{"measurand":"Power.Active.Import","unit":"W","value":"0.0","context":"Trigger"}]}],"transactionId":3}]
[ocpp ] TRACE 2024/07/09 12:11:13 sent SN10052307203612: [3,"2749",{}]
[ocpp ] TRACE 2024/07/09 12:11:13 TriggerMessage MeterValues for SN10052307203612: Accepted 
```

I'll need to powercycle the evse & restart evcc to get things working agian

andig, 2024-07-09:
It seems this is a pretty unusual deployment scenario (though not wrong). Given how rare that is and having no good idea what a good logic would be for _not_ taking over the txn I tend to close as wontfix. Thoughts?

closed 2024-07-11 by andig