evcc-io/evcc #14675: Kilometerstand in der Ladehistorie anzeigen

dog-rgb, 2024-07-02:
**Is your feature request related to a problem? Please describe.**
Wenn ich die Ladevorgänge in Spritmonitor überführen möchte, benötige ich den jeweiligen km-stand zum Ladezeitpunkt. Derzeit ist dieser nur über andere Apps (zum Beispiel des Herstellers) oder direkt im Auto auslesbar.

**Describe the solution you'd like**
Der Kilometerstand sollte als Spalte in der Ladehistorie auswählbar sein.

**Describe alternatives you've considered**
Wie oben geschrieben kann ich den Kilometerstand über andere Tools auslesen, oder über einen csv-Export. Das sind aber alles zusätzliche Schritte die zusätzliche Zeit kosten.

**Additional context**
N/A

VolkerK62, 2024-07-03:
Ein Klick auf die Ladesession und du hast die Details incl KM-Stand.

closed 2024-07-03 by andig

dog-rgb, 2024-07-03:
Für mich war es nicht offensichtlich das die einzelnen Einträge angeklickt werden können. Ist für mich ein Workaround, aber besser als ohne.
Ich brauche zum Beispiel nicht die Spalte Preis/kWh. Dafür würde ich halt gerne den km-Stand sehen.