evcc-io/evcc #14684: After upgrading to 0.128.0 getting "cannot create charger type 'template': invalid key: ip"

mmohrwinkel, 2024-07-04:
### Describe the bug

After upgrading to 0.128.0 getting "cannot create charger type 'template': invalid key: ip". 

Evcc is running inside homeassistant on a Raspberry Pi, Charger is a VW ID Charger Pro and it is still reachable under that ip. The IP is fixed both in the Charger and in the DHCP.

### Steps to reproduce

1. Have working evcc pre 0.128.0
2. Upgrade via homeassistant ui
3. won't start
...

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 8e1f7bb65365a8d35e05f23d8936e4dfe3f0868b2c179eedaf1cf6a6b56bb5a4

interval: 30s # control cycle interval

meters:
- type: template
  template: sma-home-manager 
  usage: grid  
  host: 192.168.178.88  
  name: grid1

chargers:
- type: template
  template: ellipro 
  ski: 5633 6FB7 6B1D EC17 AE86 59E1 9F3C 6F01 3327 2C0A  
  ip: 192.168.178.143  
  name: wallbox3

vehicles:
- type: template
  template: vw 
  title: ID4  
  user: $USER
  password: $PASSWORD 
  vin: $VIN 
  capacity: 77  
  timeout: 10s  
  name: ev2

loadpoints:
- title: Garage
  charger: wallbox3
  mode: pv

site:
  title: Mein Zuhause
  meters:
    grid: grid1

hems:
  type: sma
  AllowControl: false
  

eebus:
  certificate:
      private: |
          -----BEGIN EC PRIVATE KEY-----
          $pkey
          -----END EC PRIVATE KEY-----
      public: |
          -----BEGIN CERTIFICATE-----
          $key
          -----END CERTIFICATE-----
```

### Log details

```text
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/07/04 08:28:43 evcc 0.128.0
[main  ] INFO 2024/07/04 08:28:43 using config file: /config/evcc.yaml
[db    ] INFO 2024/07/04 08:28:44 using sqlite database: /data/evcc.db
[eebus ] INFO 2024/07/04 08:28:44 Local SKI:  612af39c5e43457adf3fef70896d8b6ddbeb613d
[eebus ] DEBUG 2024/07/04 08:28:44 starting websocket server on :4712
[eebus ] DEBUG 2024/07/04 08:28:44 mdns: announce
[eebus ] DEBUG 2024/07/04 08:28:44 mdns: using zeroconf
[eebus ] DEBUG 2024/07/04 08:28:44 mdns: start search
[main  ] INFO 2024/07/04 08:28:44 listening at :7070
[eebus ] DEBUG 2024/07/04 08:28:44 ski: 56336fb76b1dec17ae8659e19f3c6f0133272c0a name: Elli-Wallbox-2149A1UTUO brand: Elli model: Wallbox typ: Wallbox identifier: Elli-Wallbox-2149A1UTUO register: false host: wallbox-2149A1UTUO.local. port: 4712 addresses: [192.168.178.143]
[eebus ] DEBUG 2024/07/04 08:28:44 ski: 0e38a190d49670ad7d0aac7bbe23b84727a95eb9 name: SEMPSHIPGW3016090715 brand: SMA model: Sunny Home Manager 2.0 typ: Energy Manager identifier: SEMPSHIPGW register: false host: SMA3016090715.local. port: 4712 addresses: [192.168.178.88]
[main  ] FATAL 2024/07/04 08:28:46 cannot create charger 'wallbox3': cannot create charger type 'template': invalid key: ip
[main  ] FATAL 2024/07/04 08:28:46 will attempt restart in: 15m0s
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.128.0

VolkerK62, 2024-07-04:
ist schon erkannt und gefixed 
https://github.com/evcc-io/evcc/pull/14662#issuecomment-2207291082

Nimm beim Charger die Zeile einfach raus  
`  ip: 192.168.178.143 `

mmohrwinkel, 2024-07-04:
Danke für die schnelle Hilfe! In Unkenntnis des eebus protokolls kam es mir ziemlich wild vor dass es ohne IP funktioniert, aber ohne die Zeile klappt es tatsächlich direkt wieder :) Danke!

closed 2024-07-04 by mmohrwinkel