evcc-io/evcc #14685: invalid message template for connect & disconnect

TobiasHuber1980, 2024-07-04:
### Describe the bug

bei connect & disconnect kommt folgender fehler:
`[push ] ERROR 2024/07/04 09:19:24 invalid message template for connect: wanted: [vehicleTitle], got:.... siehe unten`

kann vielleicht noch mit #13856 in Verbindung gebracht werden.

### Steps to reproduce

1. ./.
2. ./.
3. ./.
...

### Configuration details

```yaml
events:
  start: # charge start event
    title: Charge started
    msg: ${vehicleTitle} started charging
  stop: # charge stop event
    title: Charge finished
    msg: ${vehicleTitle} finished charging ${chargedEnergy:%.1fk}kWh
  connect: # vehicle connect event
    title: Car connected
    msg: ${vehicleTitle} connected on Wallbox ${title}
  disconnect: # vehicle connected event
    title: Car disconnected
    msg: ${vehicleTitle} disconnected
```

### Log details

```text
[push ] ERROR 2024/07/04 09:19:24 invalid message template for connect: wanted: [vehicleTitle], got: [residualPower chargeTotalImport sessionCo2PerKWh chargeDuration pvPower enabled phasesConfigured prioritySoc effectivePlanSoc tariffPriceHome charging minCurrent effectivePlanTime sessionPrice pvAction chargerFeatureIntegratedDevice vehicleDetectionActive statistics influx network battery chargerFeatureHeating vehicleOdometer auth limitSoc siteTitle sessionSolarPercentage smartCostActive connectedDuration bufferSoc phaseRemaining smartCostType vehicleName vehicleSoc effectiveMinCurrent tariffFeedIn limitEnergy vehicles effectivePriority vehicleRange gridConfigured chargerIcon sessionPricePerKWh priority chargePower chargerPhysicalPhases loadpoint maxCurrent phasesEnabled fatal pvEnergy connected sessionEnergy vehicleLimitSoc mqtt circuits tariffPriceLoadpoints interval planEnergy chargedEnergy pv effectiveLimitSoc version batteryMode maxGridSupplyWhileBatteryCharging homePower bufferStartSoc chargeCurrent vehicleClimaterActive greenShareLoadpoints currency phasesActive mode effectiveMaxCurrent planOverrun planProjectedStart pvRemaining sponsor phaseAction gridPower title disableThreshold enableThreshold chargerPhases1p3p batteryDischargeControl planTime chargeCurrents greenShareHome tariffGrid]
```

### What type of operating system are you running?

Linux

### Version

0.128.0

andig, 2024-07-04:
Was sagt /api/state in dem Moment?

TobiasHuber1980, 2024-07-04:
Ich habe versucht den "richtigen Zeitpunkt" zu erwischen.

[connect_1.txt](https://github.com/user-attachments/files/16096398/connect_1.txt)
[connect_2.txt](https://github.com/user-attachments/files/16096399/connect_2.txt)
[disconnect_1.txt](https://github.com/user-attachments/files/16096400/disconnect_1.txt)
[disconnect_2.txt](https://github.com/user-attachments/files/16096401/disconnect_2.txt)
[disconnect_3.txt](https://github.com/user-attachments/files/16096402/disconnect_3.txt)
[während_ladevorgang.txt](https://github.com/user-attachments/files/16096403/wahrend_ladevorgang.txt)

TobiasHuber1980, 2024-07-04:
bei connect:
`invalid message template for connect: wanted: [vehicleTitle], got: [chargerIcon tariffPriceLoadpoints phasesActive influx maxGridSupplyWhileBatteryCharging connected charging vehicleName planOverrun disableThreshold vehicles effectiveMaxCurrent chargerFeatureIntegratedDevice limitEnergy enabled statistics currency version greenShareHome effectiveMinCurrent vehicleRange smartCostActive phasesEnabled vehicleClimaterActive bufferSoc bufferStartSoc planEnergy planTime effectivePlanSoc chargeTotalImport auth phaseAction batteryDischargeControl mode title siteTitle greenShareLoadpoints sponsor loadpoint maxCurrent tariffPriceHome minCurrent fatal chargeRemainingDuration mqtt gridConfigured chargerFeatureHeating vehicleLimitSoc smartCostType pvAction sessionSolarPercentage sessionPricePerKWh chargerPhysicalPhases homePower phaseRemaining chargeDuration pvPower sessionPrice limitSoc gridPower pv circuits priority vehicleOdometer chargeCurrents chargedEnergy connectedDuration planProjectedStart effectivePlanTime phasesConfigured tariffGrid residualPower vehicleSoc chargeRemainingEnergy pvRemaining chargePower batteryMode network sessionCo2PerKWh prioritySoc effectiveLimitSoc effectivePriority sessionEnergy chargerPhases1p3p tariffFeedIn enableThreshold pvEnergy vehicleDetectionActive chargeCurrent interval battery]`

bei disconnect:
`invalid message template for disconnect: wanted: [vehicleTitle], got: [connectedDuration charging planProjectedStart tariffGrid phasesActive effectivePriority phaseAction pvPower sessionPrice version sessionPricePerKWh sessionEnergy bufferSoc sponsor chargerIcon connected tariffPriceHome pvRemaining pvEnergy chargerPhases1p3p sessionCo2PerKWh planOverrun tariffPriceLoadpoints vehicleOdometer enableThreshold chargerPhysicalPhases chargeRemainingDuration chargeCurrent chargedEnergy circuits chargeDuration chargeRemainingEnergy title influx loadpoint effectiveLimitSoc effectivePlanTime chargePower statistics smartCostType effectiveMinCurrent effectiveMaxCurrent phasesEnabled limitEnergy vehicleClimaterActive vehicleSoc tariffFeedIn gridConfigured siteTitle chargerFeatureHeating chargerFeatureIntegratedDevice vehicleName chargeCurrents chargeTotalImport fatal battery batteryDischargeControl gridPower pv enabled phasesConfigured planEnergy mqtt residualPower priority vehicleLimitSoc greenShareLoadpoints prioritySoc pvAction planTime maxCurrent disableThreshold currency mode vehicleRange smartCostActive batteryMode network interval limitSoc auth vehicles homePower maxGridSupplyWhileBatteryCharging sessionSolarPercentage greenShareHome effectivePlanSoc minCurrent vehicleDetectionActive phaseRemaining bufferStartSoc]`

andig, 2024-07-06:
Du hast ja mehrere Ladepunkte. Zu welchem gehört das Problem beim connect?

TobiasHuber1980, 2024-07-06:
in desem fall war es lp-1.

zuhause mit nur einem ladepunkt ist das gleiche verhalten. `${vehicleTitle}` bei conncect/disconnect und (glaube ich start/stop). battery soc update mit `${vehicleTitle}` funktioniert.

andig, 2024-07-06:
Also:

    "vehicleName": "car1",
    "title": "VW ID.5 (Tobias Huber)",

dann sollte der Titel auch verfügbar sein- im Code ist das drin. Du kannst nochmal `cache: trace` setzen und so einen Fall reproduzieren- vielleicht sehen wir dann etwas. Gerne Lp1.

TobiasHuber1980, 2024-07-06:
Trace-Log:
[evcc-20240706-115040-trace.log](https://github.com/user-attachments/files/16115544/evcc-20240706-115040-trace.log)

andig, 2024-07-06:
Wo siehst Du trace?

TobiasHuber1980, 2024-07-07:
@hrpv kannst du den fehler bei dir mit der konfig nachstellen?
also konkret `${vehicleTitle}` bei `connect` und `disconnect`

```
events:
  start: # charge start event
    title: Charge started
    msg: ${vehicleTitle} started charging
  stop: # charge stop event
    title: Charge finished
    msg: ${vehicleTitle} finished charging ${chargedEnergy:%.1fk}kWh
  connect: # vehicle connect event
    title: Car connected
    msg: ${vehicleTitle} connected on Wallbox ${title}
  disconnect: # vehicle connected event
    title: Car disconnected
    msg: ${vehicleTitle} disconnected
```

TobiasHuber1980, 2024-07-07:
> Wo siehst Du trace?
yaml log levels gesetzt:
```
log: debug
levels:
  messaging: trace
  cache: trace
```
Im Log im UI findet  jedoch an keiner Stelle `[cache ] TRACE`. Keine Ahnung warum!?

api/state erkennt korrekt:
```"vehicleName": "db:1",```
und
```
"db:1": {
"title": "VW ID.5",
}
```

[evcc-20240707-195416-trace.log](https://github.com/user-attachments/files/16120209/evcc-20240707-195416-trace.log)
[api_state.txt](https://github.com/user-attachments/files/16120213/api_state.txt)

habe noch https://github.com/evcc-io/evcc/issues/12862 und https://github.com/evcc-io/evcc/pull/10668 dazu gefunden, ich kann aber nicht sagen, ob das damit zusammenhängen könnte.

closed 2024-07-19 by github-actions[bot]