evcc-io/evcc #14686: Elli Pro Wallbox zeigt bei 0.128 keine Leistung und unendliche Dauer an

TobiasSt78, 2024-07-04:
### Describe the bug

Laden funktioniert noch beim letzten Versionssprung, allerdings wird die Leistung nicht mehr angezeigt. => 0 W

### Steps to reproduce

1. Update von 0.127.x auf 0.128
2. auto Laden über PV
3.
...

### Configuration details

```yaml
Nur IP rausgenommen
```

### Log details

```text
Debian
```

### What type of operating system are you running?

Linux

### Version

0.128

TobiasSt78, 2024-07-04:
[image]

pebebe, 2024-07-04:
Bei mir das gleiche, Restdauer wird mir gar nicht angeboten (ich habe normal auf Ladedauer stehen, diese passt).
Ablauf der gleiche wie oben, Update und die IP deaktiviert.

closed 2024-07-04 by andig

andig, 2024-07-04:
@DerAndereAndi müssten wir in `currentPower` nicht `api.ErrNotAvailable` anstatt des "nackten" EEbus Fehlers zurück geben?