evcc-io/evcc #14689: Charger logic error: phases mismatch (got 1, expected 3)

Philipp-X1, 2024-07-04:
### Describe the bug

After updating from 0.127.3 to 0.128.0 I am getting constantly this error: 
[lp-1 ] WARN 2024/07/04 13:04:56 charger logic error: phases mismatch (got 1, expected 3)

Everything was working fine with 0.127.3. 
Charging still works as far as I see at the moment.

My car can only charge 1 phase.
EVC is connected with 3 phases.

Do I need to change some configuration now? 
If you need more informations let me know.

Thank you! :)

### Steps to reproduce

Normal Charing process. Plug in and Charge in PV Mode.

### Configuration details

```yaml
chargers:
  - name: Solax_X3-EVC-11K
    type: template
    template: solax
    modbus: tcpip
    id: 70
    host: ipAdress
    port: 502

# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for documentation see https://docs.evcc.io/docs/devices/vehicles
vehicles:
  - name: multivan
    type: template
    template: id
    title: Multivan # Wird in der Benutzeroberfläche angezeigt (optional)
    user: ***
    password: ***
    #    vin: # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)
    capacity: 10.5 # Akkukapazität in kWh (optional)
    timeout: 60s # optional

# site describes the EVU connection, PV and home battery
site:
  title: Unser Haus # display name for UI
  meters:
    grid: my_grid # grid meter
    pv:
      - my_pv # list of pv inverters/ meters
    battery:
      - my_battery # list of battery meters
  residualPower: 100 # additional household usage margin
  maxGridSupplyWhileBatteryCharging: 0 # ignore battery charging if AC consumption is above this value

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
  - title: Carport # display name for UI
    charger: Solax_X3-EVC-11K # charger
    #meter: charge # charge meter
    mode: pv # default charge mode to apply when vehicle is disconnected; use "off" to disable by default if charger is publicly available

    # remaining settings are experts-only and best left at default values
    priority: 0 # relative priority for concurrent charging in PV mode with multiple loadpoints (higher values have higher priority)
    soc:
      # polling defines usage of the vehicle APIs
      # Modifying the default settings it NOT recommended. It MAY deplete your vehicle's battery
      # or lead to vehicle manufacturer banning you from API use. USE AT YOUR OWN RISK.
      poll:
        # poll mode defines under which condition the vehicle API is called:
        #   charging: update vehicle ONLY when charging (this is the recommended default)
        #   connected: update vehicle when connected (not only charging), interval defines how often
        #   always: always update vehicle regardless of connection state, interval defines how often (only supported for single vehicle)
        mode: charging
        # poll interval defines how often the vehicle API may be polled if NOT charging
        interval: 60m
      estimate: true # set false to disable interpolating between api updates (not recommended)
    enable: # pv mode enable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 3m # threshold must be exceeded for this long
      threshold: 0 # maximum import power (W)
```

### Log details

```text
[lp-1 ] INFO 2024/07/04 13:01:56 start charging ->
[lp-1 ] DEBUG 2024/07/04 13:01:56 wake-up timer: stop
[lp-1 ] DEBUG 2024/07/04 13:01:56 soc estimated: 40.00% (vehicle: 40.00%)
[lp-1 ] DEBUG 2024/07/04 13:01:56 vehicle soc: 40%
[lp-1 ] DEBUG 2024/07/04 13:01:56 vehicle range: 16km
[lp-1 ] DEBUG 2024/07/04 13:01:56 !! active phases: 1p = min(1p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/07/04 13:01:56 pv charge current: 9.8A = 6A + 3.8A (-874W @ 1p)
[lp-1 ] DEBUG 2024/07/04 13:01:56 pv timer reset
[lp-1 ] DEBUG 2024/07/04 13:01:56 pv timer inactive
[lp-1 ] DEBUG 2024/07/04 13:01:56 max charge current: 9.8A
[site ] DEBUG 2024/07/04 13:02:25 ----
[lp-1 ] DEBUG 2024/07/04 13:02:25 charge power: 2295W
[lp-1 ] DEBUG 2024/07/04 13:02:25 charge currents: [9.72 0 0]A
[site ] DEBUG 2024/07/04 13:02:26 pv power: 3509W
[site ] DEBUG 2024/07/04 13:02:26 battery soc: 93%
[site ] DEBUG 2024/07/04 13:02:26 battery power: -258W
[site ] DEBUG 2024/07/04 13:02:26 grid meter: -0W
[site ] DEBUG 2024/07/04 13:02:26 site power: -158W
[lp-1 ] DEBUG 2024/07/04 13:02:26 charge voltages: [237 239 238]V
[lp-1 ] DEBUG 2024/07/04 13:02:26 detected connected phases: 3p
[lp-1 ] DEBUG 2024/07/04 13:02:26 detected active phases: 1p
[lp-1 ] DEBUG 2024/07/04 13:02:26 !! session: chargeRater.chargedEnergy=2.8 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/07/04 13:02:26 charge total import: 2303.800kWh
[lp-1 ] DEBUG 2024/07/04 13:02:27 charger status: C
[lp-1 ] DEBUG 2024/07/04 13:02:27 soc estimated: 40.00% (vehicle: 40.00%)
[lp-1 ] DEBUG 2024/07/04 13:02:27 vehicle soc: 40%
[lp-1 ] DEBUG 2024/07/04 13:02:27 vehicle range: 16km
[lp-1 ] DEBUG 2024/07/04 13:02:27 !! active phases: 1p = min(1p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/07/04 13:02:27 pv charge current: 10.5A = 9.8A + 0.687A (-158W @ 1p)
[lp-1 ] DEBUG 2024/07/04 13:02:27 max charge current: 10.5A
[site ] DEBUG 2024/07/04 13:02:55 ----
[lp-1 ] DEBUG 2024/07/04 13:02:55 charge power: 2415W
[lp-1 ] DEBUG 2024/07/04 13:02:55 charge currents: [10.2 0 0]A
[site ] DEBUG 2024/07/04 13:02:56 pv power: 3691W
[site ] DEBUG 2024/07/04 13:02:56 battery soc: 93%
[site ] DEBUG 2024/07/04 13:02:56 battery power: -290W
[site ] DEBUG 2024/07/04 13:02:56 grid meter: -0W
[site ] DEBUG 2024/07/04 13:02:56 site power: -190W
[lp-1 ] DEBUG 2024/07/04 13:02:56 charge voltages: [238 240 239]V
[lp-1 ] DEBUG 2024/07/04 13:02:56 detected connected phases: 3p
[lp-1 ] DEBUG 2024/07/04 13:02:56 detected active phases: 1p
[lp-1 ] DEBUG 2024/07/04 13:02:56 !! session: chargeRater.chargedEnergy=2.9 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/07/04 13:02:56 charge total import: 2303.900kWh
[lp-1 ] DEBUG 2024/07/04 13:02:56 charger status: C
[lp-1 ] DEBUG 2024/07/04 13:02:56 soc estimated: 41.05% (vehicle: 40.00%)
[lp-1 ] DEBUG 2024/07/04 13:02:56 vehicle soc: 41%
[lp-1 ] DEBUG 2024/07/04 13:02:56 vehicle range: 16km
[lp-1 ] WARN 2024/07/04 13:02:56 charger logic error: phases mismatch (got 1, expected 3)
[lp-1 ] DEBUG 2024/07/04 13:02:56 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1 ] DEBUG 2024/07/04 13:02:56 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1 ] DEBUG 2024/07/04 13:02:56 pv charge current: 11.3A = 10.5A + 0.826A (-190W @ 1p)
[lp-1 ] DEBUG 2024/07/04 13:02:56 max charge current: 11.3A
[site ] DEBUG 2024/07/04 13:03:25 ----
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.128.0

Philipp-X1, 2024-07-04:
Nachtrag:
Im PV Modus wird die Ladung nicht gestartet trotz genügend Überschuss.

Scheint so als wartet er auf genug Überschuss für drei Phasen statt für eine?

```
[site ] DEBUG 2024/07/04 14:32:25 ----
[lp-1 ] DEBUG 2024/07/04 14:32:25 charge power: 0W
[lp-1 ] DEBUG 2024/07/04 14:32:25 charge currents: [0 0 0]A
[site ] DEBUG 2024/07/04 14:32:26 pv power: 2912W
[site ] DEBUG 2024/07/04 14:32:26 battery soc: 97%
[site ] DEBUG 2024/07/04 14:32:26 battery power: -2044W
[site ] DEBUG 2024/07/04 14:32:26 grid meter: -0W
[site ] DEBUG 2024/07/04 14:32:26 site power: -1944W
[lp-1 ] DEBUG 2024/07/04 14:32:26 charge voltages: [237 237 238]V
[lp-1 ] DEBUG 2024/07/04 14:32:26 detected connected phases: 3p
[lp-1 ] DEBUG 2024/07/04 14:32:26 !! session: chargeRater.chargedEnergy=4.2 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/07/04 14:32:26 charge total import: 2305.200kWh
[lp-1 ] DEBUG 2024/07/04 14:32:26 charger status: B
[lp-1 ] DEBUG 2024/07/04 14:32:26 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/07/04 14:32:26 pv charge current: 2.82A = 0A + 2.82A (-1944W @ 3p)
```

closed 2024-07-04 by andig

andig, 2025-05-31:
> Wenn ich aber das Auto (enyaq iv 60) angesteckt habe und "schnell" laden auswähle oder auch eine Ladeplanung einstelle, funktioniert die Umschaltung nicht.

Ist in deinem Log nicht zu sehen. Da fehlt ausreichender Kontext.

MetaStack-Exchange, 2025-05-31:
Entschuldigung, da war mein Handy schneller als ich, hab's korrigiert...