evcc-io/evcc #14711: Solis Hybrid ab 0.128.0

DaPetra, 2024-07-05:
### Describe the bug

Ab Version 0.128.0 wird bei mir die Ladung/Entladung vom Hausakku, der über den Solis-Hybrid Umrichter kommt verkehrt dargestellt.
Obwohl der Akku geladen wird, zeigt er in der Web UI entladen an.

### Steps to reproduce

1.
2.
3.
...
Zurück auf 0.127.3 alles richtig!

### Configuration details

```yaml
Keine Änderung
```

### Log details

```text
Kein Log!
```

### What type of operating system are you running?

Linux

### Version

0.128.0 und 0.128.1

andig, 2024-07-05:
Liegt vmtl. an https://github.com/evcc-io/evcc/pull/14423. Wir bräuchten bitte die Ausgabe von

    evcc meter --log trace

...am liebsten in alter und neuer Version.

premultiply, 2024-07-05:
Ist die Firmware auf dem Solis-Hybrid-WR aktuell?

DaPetra, 2024-07-09:
Der Solis Hybrid ist über GU installiert, mit dem evcc meter --log trace wird er gar nicht angezeigt. Ob er die neuste Firmware hat kann ich nicht sagen, bin Anwender kein Programmierer.

DaPetra, 2024-07-10:
Ich habe es mal mit der neuen 0.128.2 getestet. Der Fehler ist noch da. Bin wieder zurück auf 0.127.3 habe aber von beiden Versionen das Logfile gespeichert. In der 0.127.3 kommt der Wert vom Akku mit Minus an, bei 0.128.2 ohne Minus.
[01273evcc-20240710-074925-trace.log](https://github.com/user-attachments/files/16155822/01273evcc-20240710-074925-trace.log)
[image]
[01282evcc-20240710-074601-trace.log](https://github.com/user-attachments/files/16155823/01282evcc-20240710-074601-trace.log)
[image]

andig, 2024-07-10:
@DaPetra es braucht bitte ein

    evcc meter --log trace,db:error

jeweils vom laden und entladen mit der neuen Version. Hat die Leistung da unterschiedliche Vorzeichen? Gerne im Vergleich auch mit der alten Version. Keine Screenshots aus evcc!

DaPetra, 2024-07-12:
@andig wie schon geschrieben, der Solis Hybrid ist über die GU integriert und in der log nicht enthalten.
Ich kann nur die Log Datei über die Weboberfläche anbieten, wenn der Akku entlädt.

DaPetra, 2024-07-12:
Hier beim entladen der 0.128.2 ist es ein positiver Wert.
[01282evcc-20240712-073114-trace.log](https://github.com/user-attachments/files/16189070/01282evcc-20240712-073114-trace.log)

DaPetra, 2024-07-12:
@andig hier mal die logfiles von der Weboberfläche, beide Versionen und und laden und entladen.
[01273_entladen-evcc-20240712-074242-trace.log](https://github.com/user-attachments/files/16189273/01273_entladen-evcc-20240712-074242-trace.log)
[01273_laden-evcc-20240712-075443-trace.log](https://github.com/user-attachments/files/16189274/01273_laden-evcc-20240712-075443-trace.log)
[01282_entladen-evcc-20240712-074703-trace.log](https://github.com/user-attachments/files/16189275/01282_entladen-evcc-20240712-074703-trace.log)
[01282_laden-evcc-20240712-075141-trace.log](https://github.com/user-attachments/files/16189276/01282_laden-evcc-20240712-075141-trace.log)

andig, 2024-07-12:
Gleiche Frage wie oben: Hat die (Batterie)Leistung da unterschiedliche Vorzeichen? Dann sind die einfach falsch und müssen umgedreht werden.

> der Solis Hybrid ist über die GU integriert und in der log nicht enthalten.

Das wäre ein Bug. Auch Geräte aus der Ui sollten in 

    evcc meter

auftauchen.

DaPetra, 2024-07-12:
@andig  Also ich sehe den Hybrid nicht!
[log.txt](https://github.com/user-attachments/files/16189332/log.txt)

andig, 2024-07-12:
Die sagtest 0.128.0. In Deinem Log steht:

> [main  ] INFO 2024/07/12 08:03:44 evcc 0.127.3

Wie passt das zusammen?

VolkerK62, 2024-07-12:
>Und viel wichtiger: verwenden Deine beiden Aufrufe überhaupt die gleichen Datenbank???

Vermutlich nicht, wenn die DB nicht in der yaml angegeben ist.

DaPetra, 2024-07-12:
@andig  ich springe für deine Anfragen zwischen 0.127.3 und der aktuellen 0.128.2 hin und her. Die Log.txt habe ich mit der für mich funktionierenden 0.127.3 erzeugt. Mit dem evcc meter wie du es oben gewünscht hast. Es sollte egal sein welche Version, da der Hybrid in keiner von beiden sichtbar ist.
Da sich ja nur die evcc Datei ändert sollte doch immer die gleiche Datenbank aufgerufen werden, zumal ja auf der Webobefläche alles gleich bleibt.

DaPetra, 2024-07-12:
@andig  
[main  ] INFO 2024/07/12 07:45:54 evcc 0.128.2
[main  ] INFO 2024/07/12 07:45:54 using config file: /etc/evcc.yaml
[db    ] INFO 2024/07/12 07:45:54 using sqlite database: /var/lib/evcc/evcc.db

[main  ] INFO 2024/07/12 07:39:23 evcc 0.127.3
[main  ] INFO 2024/07/12 07:39:23 using config file: /etc/evcc.yaml
[db    ] INFO 2024/07/12 07:39:23 using sqlite database: /var/lib/evcc/evcc.db

Wenn ich die Logfiles richtig interpretiere, solle es schon die gleich Datenbank sein!

VolkerK62, 2024-07-12:
@DaPetra  die Logs aus dem "Betrieb" heraus, greifen auf die gleiche DB zu.
Aber, wenn du `evcc meter` aufrufst, sollte dort auch angezeigt werden, welche DB verwendet wird. 
Das ist in deiner "log.txt" nicht ersichtlich. 
Das wird eine andere sein und deshalb ist dort der Hybrid nicht aufgeführt.

DaPetra, 2024-07-12:
@VolkerK62 kann ich das irgendwie kontrollieren, in der log wird es nicht angezeigt. Nur die gleiche using config file: /etc/evcc.yaml steht drin.

VolkerK62, 2024-07-12:
Hmm, das ist seltsam, Keine Ahnung, warum.
Ergänze mal in deiner yaml das folgende
```
database:
  type: sqlite
  dsn: /var/lib/evcc/evcc.db
```
Dann wird immer nur auf diese DB zugegriffen,
Und dann nochmal `evcc meter`.

DaPetra, 2024-07-12:
@VolkerK62 
Jetzt sind ein paar Einträge dazu gekommen:

db:1
----
[modbus] TRACE 2024/07/12 09:24:14 modbus: send 00 01 00 00 00 06 01 04 81 7d 00 02
[modbus] TRACE 2024/07/12 09:24:14 modbus: recv 00 01 00 00 00 07 01 04 04 00 00 03 6b
[modbus] TRACE 2024/07/12 09:24:14 modbus: send 00 02 00 00 00 06 01 04 81 6f 00 01
[modbus] TRACE 2024/07/12 09:24:14 modbus: recv 00 02 00 00 00 05 01 04 02 00 00
[modbus] TRACE 2024/07/12 09:24:14 modbus: send 00 03 00 00 00 06 01 04 81 8d 00 02
[modbus] TRACE 2024/07/12 09:24:14 modbus: recv 00 03 00 00 00 07 01 04 04 00 00 05 fb
[modbus] TRACE 2024/07/12 09:24:14 modbus: send 00 04 00 00 00 06 01 04 81 73 00 01
[modbus] TRACE 2024/07/12 09:24:15 modbus: recv 00 04 00 00 00 05 01 04 02 00 29
Power:    -875W
Energy:   1531.0kWh
Soc:      41%
Capacity: 10.0kWh

db:2
----
[modbus] TRACE 2024/07/12 09:24:15 modbus: send 00 05 00 00 00 06 01 04 81 21 00 02
[modbus] TRACE 2024/07/12 09:24:15 modbus: recv 00 05 00 00 00 07 01 04 04 00 00 02 6a
[modbus] TRACE 2024/07/12 09:24:15 modbus: send 00 06 00 00 00 06 01 04 81 05 00 02
[modbus] TRACE 2024/07/12 09:24:15 modbus: recv 00 06 00 00 00 07 01 04 04 00 00 0b 09
Power:  618W
Energy: 2825.0kWh

db:3
----
[tasmota] TRACE 2024/07/12 09:24:15 GET http://192.168.0.35/cm?cmnd=Status+8&password=&user=
[tasmota] TRACE 2024/07/12 09:24:15 {"StatusSNS":{"Time":"2024-07-12T09:24:15","SML":{"Total_in":11029,"Total_out":24376,"Power_curr":3,"V_L1":229.4,"V_L2":230.2,"V_L3":229.2}}}
Power:  3W
Energy: 11029.0kWh

VolkerK62, 2024-07-12:
@DaPetra  hast du mal die genaue Typ Bezeichnung des WR und die Firmware-Version?

DaPetra, 2024-07-12:
Solis S5-EH1P5K-L steht auf dem Typenschild. Firmware habe ich im Menü einen Punkt mit Software Ver. : 3D0037 Model No.: 3104 gefunden.

cs33lm, 2024-07-13:
Ich konnte beim Nachbarn nur auf die App schauen, dort habe ich keine Software Version gefunden, nur eine Inverter-Version 0F003C-01000. 

Hier meine Logs (trace + log des Containers), mir ist dabei noch ein Wert ins Auge gesprungen:
```
[lp-1  ] DEBUG 2024/07/13 20:01:33 charge start energy: -2.12e+06kWh
```

0.127.3:
```
[main  ] INFO 2024/07/13 20:01:12 evcc 0.127.3
[main  ] INFO 2024/07/13 20:01:12 using config file: /etc/evcc.yaml
grid
----
[modbus] TRACE 2024/07/13 20:01:12 modbus: send 00 01 00 00 00 06 01 04 81 ef 00 02
[modbus] TRACE 2024/07/13 20:01:13 modbus: recv 00 01 00 00 00 07 01 04 04 ff ff ff d2
[modbus] TRACE 2024/07/13 20:01:13 modbus: send 00 02 00 00 00 06 01 04 82 03 00 02
[modbus] TRACE 2024/07/13 20:01:13 modbus: recv 00 02 00 00 00 07 01 04 04 00 03 d0 cf
[modbus] TRACE 2024/07/13 20:01:13 modbus: send 00 03 00 00 00 06 01 04 81 e4 00 01
[modbus] TRACE 2024/07/13 20:01:13 modbus: recv 00 03 00 00 00 05 01 04 02 00 79
[modbus] TRACE 2024/07/13 20:01:13 modbus: send 00 04 00 00 00 06 01 04 81 e6 00 01
[modbus] TRACE 2024/07/13 20:01:13 modbus: recv 00 04 00 00 00 05 01 04 02 00 73
[modbus] TRACE 2024/07/13 20:01:13 modbus: send 00 05 00 00 00 06 01 04 81 e8 00 01
[modbus] TRACE 2024/07/13 20:01:13 modbus: recv 00 05 00 00 00 05 01 04 02 00 81
[modbus] TRACE 2024/07/13 20:01:13 modbus: send 00 06 00 00 00 06 01 04 81 e9 00 02
[modbus] TRACE 2024/07/13 20:01:14 modbus: recv 00 06 00 00 00 07 01 04 04 ff ff ff e4
[modbus] TRACE 2024/07/13 20:01:14 modbus: send 00 07 00 00 00 06 01 04 81 eb 00 02
[modbus] TRACE 2024/07/13 20:01:14 modbus: recv 00 07 00 00 00 07 01 04 04 00 00 00 28
[modbus] TRACE 2024/07/13 20:01:14 modbus: send 00 08 00 00 00 06 01 04 81 ed 00 02
[modbus] TRACE 2024/07/13 20:01:14 modbus: recv 00 08 00 00 00 07 01 04 04 ff ff ff d0
Power:          46W
Energy:         2500.6kWh
Current L1..L3: 1.21A 1.15A 1.29A
Power L1..L3:   28W -40W 48W

pv
--
[modbus] TRACE 2024/07/13 20:01:14 modbus: send 00 09 00 00 00 06 01 04 81 21 00 02
[modbus] TRACE 2024/07/13 20:01:14 modbus: recv 00 09 00 00 00 07 01 04 04 00 00 03 96
[modbus] TRACE 2024/07/13 20:01:14 modbus: send 00 0a 00 00 00 06 01 04 81 05 00 02
[modbus] TRACE 2024/07/13 20:01:14 modbus: recv 00 0a 00 00 00 07 01 04 04 00 00 1b 9b
Power:  918W
Energy: 7067.0kWh

battery
-------
[modbus] TRACE 2024/07/13 20:01:14 modbus: send 00 0b 00 00 00 06 01 04 81 7d 00 02
[modbus] TRACE 2024/07/13 20:01:15 modbus: recv 00 0b 00 00 00 07 01 04 04 00 00 02 db
[modbus] TRACE 2024/07/13 20:01:15 modbus: send 00 0c 00 00 00 06 01 04 81 6f 00 01
[modbus] TRACE 2024/07/13 20:01:15 modbus: recv 00 0c 00 00 00 05 01 04 02 00 00
[modbus] TRACE 2024/07/13 20:01:15 modbus: send 00 0d 00 00 00 06 01 04 81 8d 00 02
[modbus] TRACE 2024/07/13 20:01:15 modbus: recv 00 0d 00 00 00 07 01 04 04 00 00 05 84
[modbus] TRACE 2024/07/13 20:01:15 modbus: send 00 0e 00 00 00 06 01 04 81 73 00 01
[modbus] TRACE 2024/07/13 20:01:15 modbus: recv 00 0e 00 00 00 05 01 04 02 00 5a
Power:  -731W
Energy: 1412.0kWh
Soc:    90%

[main  ] INFO 2024/07/13 20:01:24 evcc 0.127.3
[main  ] INFO 2024/07/13 20:01:24 using config file: /etc/evcc.yaml
[main  ] INFO 2024/07/13 20:01:25 listening at :7070
[lp-1  ] WARN 2024/07/13 20:01:27 poll mode '{connected 1h0m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[lp-1  ] DEBUG 2024/07/13 20:01:27 charge total import: -2123656.500kWh
[site  ] INFO 2024/07/13 20:01:27 site config:
[site  ] INFO 2024/07/13 20:01:27   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/07/13 20:01:27     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/07/13 20:01:27     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/07/13 20:01:27     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2024/07/13 20:01:27   vehicles:
[site  ] INFO 2024/07/13 20:01:27     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[lp-1  ] INFO 2024/07/13 20:01:27 loadpoint 1:
[... 125 lines cut ...]
[lp-1  ] DEBUG 2024/07/13 19:58:37 charge start energy: -2.12e+06kWh
[lp-1  ] INFO 2024/07/13 19:58:37 car connected
[lp-1  ] DEBUG 2024/07/13 19:58:37 pv timer elapse
[lp-1  ] DEBUG 2024/07/13 19:58:37 pv timer inactive
[lp-1  ] DEBUG 2024/07/13 19:58:37 charge total import: -2123656.500kWh
[lp-1  ] DEBUG 2024/07/13 19:58:37 vehicle status: A
[lp-1  ] DEBUG 2024/07/13 19:58:37 vehicle soc: 66%
[lp-1  ] DEBUG 2024/07/13 19:58:37 vehicle soc limit: 100%
[lp-1  ] DEBUG 2024/07/13 19:58:37 vehicle range: 358km
[site  ] DEBUG 2024/07/13 19:59:03 ----
[lp-1  ] DEBUG 2024/07/13 19:59:03 charge power: 0W
[lp-1  ] DEBUG 2024/07/13 19:59:03 charge currents: [0.03 0 0]A
[site  ] DEBUG 2024/07/13 19:59:04 pv power: 1091W
[site  ] DEBUG 2024/07/13 19:59:04 battery soc: 89%
[site  ] DEBUG 2024/07/13 19:59:04 battery power: 811W
[site  ] DEBUG 2024/07/13 19:59:04 grid meter: -99W
[site  ] DEBUG 2024/07/13 19:59:05 grid powers: [-0 -99 -0]W
[site  ] DEBUG 2024/07/13 19:59:06 grid currents: [1.2 -1.19 1.27]A
[site  ] DEBUG 2024/07/13 19:59:06 site power: 712W
[lp-1  ] DEBUG 2024/07/13 19:59:06 vehicle odometer: 184km
[lp-1  ] DEBUG 2024/07/13 19:59:06 charge voltages: [0.3 0 0]V
[lp-1  ] DEBUG 2024/07/13 19:59:06 charge total import: -2123656.500kWh
[lp-1  ] DEBUG 2024/07/13 19:59:06 charger status: B
[lp-1  ] DEBUG 2024/07/13 19:59:06 next soc poll remaining time: 59m30s
```

VolkerK62, 2024-07-13:
@cs33lm  danke.
vermutlich ist in beiden Fällen die Batterie geladen worden. 
0.127.3 = Power:  -731W
0.128.2 = Power:  1098W
Dann wäre der Wert aus 0.128.2 falsch. 

Leider fehlt der Vergleich, was angezeigt wird, wenn sie entladen wird.

cs33lm, 2024-07-13:
Korrekt, pv power war um 20 Uhr noch ~1kW. Hier die entsprechenden Logs mit pv power 0:

0.127.3
```
[main  ] INFO 2024/07/13 22:28:41 evcc 0.127.3
[main  ] INFO 2024/07/13 22:28:41 using config file: /etc/evcc.yaml
grid
----
[modbus] TRACE 2024/07/13 22:28:41 modbus: send 00 01 00 00 00 06 01 04 81 ef 00 02
[modbus] TRACE 2024/07/13 22:28:41 modbus: recv 00 01 00 00 00 07 01 04 04 00 00 00 4f
[modbus] TRACE 2024/07/13 22:28:41 modbus: send 00 02 00 00 00 06 01 04 82 03 00 02
[modbus] TRACE 2024/07/13 22:28:41 modbus: recv 00 02 00 00 00 07 01 04 04 00 03 d0 f4
[modbus] TRACE 2024/07/13 22:28:41 modbus: send 00 03 00 00 00 06 01 04 81 e4 00 01
[modbus] TRACE 2024/07/13 22:28:42 modbus: recv 00 03 00 00 00 05 01 04 02 00 6f
[modbus] TRACE 2024/07/13 22:28:42 modbus: send 00 04 00 00 00 06 01 04 81 e6 00 01
[modbus] TRACE 2024/07/13 22:28:42 modbus: recv 00 04 00 00 00 05 01 04 02 00 84
[modbus] TRACE 2024/07/13 22:28:42 modbus: send 00 05 00 00 00 06 01 04 81 e8 00 01
[modbus] TRACE 2024/07/13 22:28:42 modbus: recv 00 05 00 00 00 05 01 04 02 00 c2
[modbus] TRACE 2024/07/13 22:28:42 modbus: send 00 06 00 00 00 06 01 04 81 e9 00 02
[modbus] TRACE 2024/07/13 22:28:42 modbus: recv 00 06 00 00 00 07 01 04 04 00 00 00 2a
[modbus] TRACE 2024/07/13 22:28:42 modbus: send 00 07 00 00 00 06 01 04 81 eb 00 02
[modbus] TRACE 2024/07/13 22:28:42 modbus: recv 00 07 00 00 00 07 01 04 04 00 00 00 9d
[modbus] TRACE 2024/07/13 22:28:42 modbus: send 00 08 00 00 00 06 01 04 81 ed 00 02
[modbus] TRACE 2024/07/13 22:28:43 modbus: recv 00 08 00 00 00 07 01 04 04 ff ff ff 88
Power:          -79W
Energy:         2501.0kWh
Current L1..L3: 1.11A 1.32A 1.94A
Power L1..L3:   -42W -157W 120W

pv
--
[modbus] TRACE 2024/07/13 22:28:43 modbus: send 00 09 00 00 00 06 01 04 81 21 00 02
[modbus] TRACE 2024/07/13 22:28:43 modbus: recv 00 09 00 00 00 07 01 04 04 00 00 00 00
[modbus] TRACE 2024/07/13 22:28:43 modbus: send 00 0a 00 00 00 06 01 04 81 05 00 02
[modbus] TRACE 2024/07/13 22:28:43 modbus: recv 00 0a 00 00 00 07 01 04 04 00 00 1b 9b
Power:  0W
Energy: 7067.0kWh

battery
-------
[modbus] TRACE 2024/07/13 22:28:43 modbus: send 00 0b 00 00 00 06 01 04 81 7d 00 02
[modbus] TRACE 2024/07/13 22:28:43 modbus: recv 00 0b 00 00 00 07 01 04 04 ff ff fd fd
[modbus] TRACE 2024/07/13 22:28:43 modbus: send 00 0c 00 00 00 06 01 04 81 6f 00 01
[modbus] TRACE 2024/07/13 22:28:43 modbus: recv 00 0c 00 00 00 05 01 04 02 00 01
[modbus] TRACE 2024/07/13 22:28:43 modbus: send 00 0d 00 00 00 06 01 04 81 8d 00 02
[modbus] TRACE 2024/07/13 22:28:44 modbus: recv 00 0d 00 00 00 07 01 04 04 00 00 05 84
[modbus] TRACE 2024/07/13 22:28:44 modbus: send 00 0e 00 00 00 06 01 04 81 73 00 01
[modbus] TRACE 2024/07/13 22:28:44 modbus: recv 00 0e 00 00 00 05 01 04 02 00 5b
Power:  4294966781W
Energy: 1412.0kWh
Soc:    91%

[main  ] INFO 2024/07/13 22:28:46 evcc 0.127.3
[main  ] INFO 2024/07/13 22:28:46 using config file: /etc/evcc.yaml
[main  ] INFO 2024/07/13 22:28:47 listening at :7070
[lp-1  ] WARN 2024/07/13 22:28:50 poll mode '{connected 1h0m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[lp-1  ] DEBUG 2024/07/13 22:28:50 charge total import: -2123656.500kWh
[site  ] INFO 2024/07/13 22:28:50 site config:
[site  ] INFO 2024/07/13 22:28:50   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/07/13 22:28:50     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/07/13 22:28:50     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/07/13 22:28:50     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2024/07/13 22:28:50   vehicles:
[site  ] INFO 2024/07/13 22:28:50     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[lp-1  ] INFO 2024/07/13 22:28:50 loadpoint 1:
[lp-1  ] INFO 2024/07/13 22:28:50   mode:        pv
[lp-1  ] INFO 2024/07/13 22:28:50   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/07/13 22:28:50   meters:      charge ✓
[lp-1  ] INFO 2024/07/13 22:28:50     charge:    power ✓ energy ✓ currents ✓
[site  ] DEBUG 2024/07/13 22:28:50 set buffer soc: 100
[... 88 lines cut ...]
[lp-1  ] INFO 2024/07/13 22:30:34     charge:    power ✓ energy ✓ currents ✓
[site  ] DEBUG 2024/07/13 22:30:34 set buffer soc: 100
[site  ] DEBUG 2024/07/13 22:30:34 set priority soc: 100
[lp-1  ] DEBUG 2024/07/13 22:30:34 set smart cost limit: 0.4
[lp-1  ] DEBUG 2024/07/13 22:30:34 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/13 22:30:34 phase timer inactive
[lp-1  ] DEBUG 2024/07/13 22:30:34 pv timer inactive
[lp-1  ] INFO 2024/07/13 22:30:34 vehicle updated: unknown -> ID7
[lp-1  ] DEBUG 2024/07/13 22:30:34 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[site  ] DEBUG 2024/07/13 22:30:34 ----
[lp-1  ] DEBUG 2024/07/13 22:30:34 charge power: 0W
[lp-1  ] DEBUG 2024/07/13 22:30:34 charge currents: [0.07 0 0]A
[site  ] DEBUG 2024/07/13 22:30:35 pv power: 0W
[site  ] DEBUG 2024/07/13 22:30:35 battery soc: 90%
[site  ] DEBUG 2024/07/13 22:30:35 battery power: -673W
[site  ] DEBUG 2024/07/13 22:30:36 grid meter: -71W
[site  ] DEBUG 2024/07/13 22:30:36 grid powers: [-88 -201 203]W
[site  ] DEBUG 2024/07/13 22:30:37 grid currents: [-1.18 -1.44 1.8]A
[site  ] DEBUG 2024/07/13 22:30:37 battery has priority at soc 90% (< 100%)
[site  ] DEBUG 2024/07/13 22:30:37 site power: 29W
[lp-1  ] DEBUG 2024/07/13 22:30:37 vehicle odometer: 184km
[lp-1  ] DEBUG 2024/07/13 22:30:37 charge voltages: [0.1 0 0]V
[lp-1  ] DEBUG 2024/07/13 22:30:37 charge total import: -2123656.500kWh
[lp-1  ] DEBUG 2024/07/13 22:30:37 charger status: A
```

VolkerK62, 2024-07-14:
Vielen Dank.
Beim entladen
0.127.3. = Power:  4294966781W
0.128.2. = Power:  -653W

Also reicht es in deinem Fall, in 0.128.2 einfach das Vorzeichen umzudrehen.

andig, 2024-07-14:
@DaPetra @cs33lm 

> Zur Referenz: das ursprüngliche Problem war ja https://github.com/evcc-io/evcc/discussions/14153#discussioncomment-9802892. Irgendwie müssen wir beide Anforderungen differenzieren. Mein Vorschlag wäre, dass ihr bitte den Solis Support um aktuelle Modbus Doku für Eure WR oder ein Firmware Update anfragt. Die müssen sind irgendwie unterscheiden.

DaPetra, 2024-07-15:
@andig Passt schon! Ich bleibe einfach bei der für mich funktionierenden Version. Ich bin Benutzer und kein Technik Freak, ich möchte eine funktionierende Reglung und habe kein Interesse an Firmwareupdates und Modbusdokumentationen. Bis 0.127.3 hat es ja funktioniert, scheinbar für beide Wechselrichter und danach nicht mehr.

VolkerK62, 2024-07-15:
@DaPetra  es wird demnächst 2 Templates geben (siehe verlinkter PR). 
Dann kannst du auch wieder aktualisieren. Musst dann nur den Template Namen ändern.

DaPetra, 2024-07-16:
@VolkerK62 Ok, vielen Dank. Das ist hier kein Selbstläufer und eigentlich für nur Benutzer ziemlich schwierig. Habe es meinem Elektriker gezeigt, weil ich dacht es wäre cool wenn er evcc seinen Kunden anbieten könnte. Der hat gleich abgewunken, Bastellösung für Techniknerd sagt er. Würde er nie seinen Kunden anbieten, weil er einfach das Risiko nicht eingehen kann, dass auf einmal nix mehr geht. 
Ich komme durch euch immer mehr in die Thematik rein und es macht teilweise sogar ein wenig Spaß. Unterstützung von meinem Elektriker darf ich allerdings nicht erwarten.

VolkerK62, 2024-07-16:
Ja, es ist noch nicht out-of-the-box, aber "Bastellösung für Techniknerd"  ist sehr stark übertrieben.

andig, 2024-07-16:
Es ist halt die Qual der Wahl. Unsummen für geschlossene Systeme mit bescheidenster Funktionalität bezahlen oder eben ein wenig Initiative investieren ;)

closed 2024-07-16 by andig

Jok3r89, 2024-10-16:
> @DaPetra es wird demnächst 2 Templates geben (siehe verlinkter PR). Dann kannst du auch wieder aktualisieren. Musst dann nur den Template Namen ändern.

Gibt es dazu schon ein Template, ich hab das gleiche Problem. Grüße

andig, 2024-10-17:
Welches Problem? Die Lösung hier gibts seit 4 Monaten! Bitte mach eine neue Diskussion dazu auf.