evcc-io/evcc #14715: EM2GO Home: mA Regelung funktioniert nicht

njd1893, 2024-07-05:
### Describe the bug

Die mA-Regelung bei der EM2GO Home Wallbox funktioniert nicht. Es werden nur volle Ampere-Schritte übernommen und von der Box umgesetzt, obwohl evcc korrekt die Werte per Modbus sendet, zumindest soweit ich das verstehe (bin kein Modbus Experte und klicke mir das gerade alles aus verschiedenen Quellen zusammen). 

Beispielhafter Auszug aus dem Log (beschnitten aufs Wesentliche)

<html>

<body>

TYPE| LEVEL | DATE | CONTENT
-- | -- | -- | --
[lp-1  ] | DEBUG | 2024/07/05 11:40:14 | !! active phases: 1p = min(1p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] | DEBUG | 2024/07/05 11:40:14 | pv charge current: 11.8A = 11A + 0.754A (-173W @ 1p)
[em2go ] | TRACE | 2024/07/05 11:40:14 | modbus: send 7c 0c 00 00 00 09 ff 10 00 5b 00 01 02 00 75
[em2go ] | TRACE | 2024/07/05 11:40:14 | modbus: recv 7c 0c 00 00 00 06 ff 10 00 5b 00 01
[lp-1  ] | DEBUG | 2024/07/05 11:40:14 | max charge current: 11.8A
-- | -- | -- | --
[em2go ] | TRACE | 2024/07/05 11:40:25 | modbus: send 7c 0e 00 00 00 06 ff 03 00 06 00 01
[em2go ] | TRACE | 2024/07/05 11:40:25 | modbus: recv 7c 0e 00 00 00 05 ff 03 02 00 69
[em2go ] | TRACE | 2024/07/05 11:40:25 | modbus: send 7c 0f 00 00 00 06 ff 03 00 08 00 01
[em2go ] | TRACE | 2024/07/05 11:40:25 | modbus: recv 7c 0f 00 00 00 05 ff 03 02 00 00
[em2go ] | TRACE | 2024/07/05 11:40:25 | modbus: send 7c 10 00 00 00 06 ff 03 00 0a 00 01
[em2go ] | TRACE | 2024/07/05 11:40:25 | modbus: recv 7c 10 00 00 00 05 ff 03 02 00 00
[lp-1  ] | DEBUG | 2024/07/05 11:40:25 | charge currents: [10.5 0 0]A
[lp-1  ] | DEBUG | 2024/07/05 11:40:28 | detected active phases: 1p
-- | -- | -- | --
[em2go ] | TRACE | 2024/07/05 11:40:29 | modbus: send 7c 16 00 00 00 06 ff 03 00 5b 00 01
[em2go ] | TRACE | 2024/07/05 11:40:29 | modbus: recv 7c 16 00 00 00 05 ff 03 02 00 6e
[lp-1  ] | WARN | 2024/07/05 11:40:29 | charger logic error: current mismatch (got 11A, expected 11.8A)

</body>

</html>

Im ersten Block wird der Strom auf 11,8A berechnet und mit 11,7A (117dec entspricht 0x75)  ins Strombegrenzungsregister 0x5B geschrieben. Die Box bestätigt den Schreibvorgang.
Im zweiten Block werden die aktuellen Ladeströme der Phasen gelesen. Rückmeldung ist, wie zu sehen für Phase 1 10,5A (0,5A Offset sind bei meiner Box offensichtlich normal, also bitte den Wert als 10A annehmen ;))
Im dritten Block wird dann der Stromlimitwert aus Register 0x5B zurück gelesen. Die Box liefert aber anstatt 0x75 nur 0x6E zurück, was 110dec bzw. 11A entspricht.

Link zur Modbus Registerbeschreibung in der groß drin steht, dass das Strombegrenzungsregister 0,1A akzeptiert
[image]
https://www.em2go.de/download/25609/2289

Vielleicht übersehe ich ja was offensichtliches im Modbus Telegramm, aber im Moment bin ich der Meinung, dass die Box nur volle Ampere-Schritte akzeptiert. Müsste dann in der Doku als Feature rausgenommen werden.

Gesamtes Log ungefiltert im Anhang
[evcc-20240705-120938-trace.log](https://github.com/user-attachments/files/16109311/evcc-20240705-120938-trace.log)

### Steps to reproduce

1. max charge current darf keine natürliche Zahl sein
2. fertig

### Configuration details

```yaml
.
```

### Log details

```text
.
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

v0.128.1

RenatusRo, 2024-07-05:
oder es ist der skoda?
https://github.com/evcc-io/evcc/discussions/10487

andig, 2024-07-05:
> Die mA-Regelung bei der EM2GO Home Wallbox funktioniert nicht.

Aktuelle Version oder auch 0.127.0?