evcc-io/evcc #14721: Smart #1 - Laden beenden/PV laden funktioniert nicht

romestylez, 2024-07-05:
### Describe the bug

evcc (neuste Version) läuft im Docker auf einem Synology Nas mit einer EVBox ELVI + Smart #1 (49 kWh)

Die Daten wie Einspeisung, Verbrauch etc. passen.

Jedoch startet er nur im "Schnellmodus" den Ladevorgang aber auch nicht immer.

[image]

Im PV und Min+PV Modus gehts nie. 

Die Kommunikation scheint aber zu gehen:

`/app # evcc -l debug vehicle                                                                                                                                
[main  ] INFO 2024/07/05 17:20:36 evcc 0.127.3                                                                                                              
[main  ] INFO 2024/07/05 17:20:36 using config file: /app/evcc.yaml                                                                                         
[db    ] INFO 2024/07/05 17:20:36 using sqlite database: /root/.evcc/evcc.db                                                                                
Soc:            51%                                                                                                                                         
Capacity:       49.0kWh                                                                                                                                     
Range:          119km                                                                                                                                       
Odometer:       705km                                                                                                                                       
Climate active: false                                                                                                                                       
Position:       XXXXXXX,XXXXXX                                                                                                      
Features:       []         `

Ebenso beendet er den Ladevorgang nie. Egal ob im PV-Modus befindlich (er erkennt zwar, wenn zu wenig Einspeisung vorhanden ist, will dann abschalten, macht es aber nicht und macht dann einfach mit "im Ladezustand" weiter. Auch manuell geht es nicht über den "AUS" Button.

### Steps to reproduce

evcc im Docker installiere und nach der initialen Einrichtung los legen.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 7782b6589e52d27137bdd11fdb1716dad5486f5ca4e8b088546f3dfd44a0cecd

interval: 30s # control cycle interval

meters:
- type: template
  template: powerfox-poweropti 
  usage: grid  
  user: XXXXXXXX@aol.com  
  password: XXXXXXXX  
  name: grid1
- type: template
  template: sma-inverter-modbus 
  id: 3  
  host: 192.168.188.65  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: solarman 
  usage: pv  
  host: 192.168.188.139  
  user: admin  
  password: admin  
  name: pv3

chargers:
- type: template
  template: elvi 
  timeout: 2m  
  meter: false  
  name: wallbox5

vehicles:
- type: template
  template: smart-hello 
  title: Smart  
  user: XXXXXXXX@aol.com  
  password: XXXXXXXX 
  vin: XXXXXXXX  
  capacity: 49  
  name: ev4

loadpoints:
- title: Carport
  charger: wallbox5
  mode: now

site:
  title: Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    - pv3
```

### Log details

```text
[site ] DEBUG 2024/07/05 13:23:30 site power: -2196W
[lp-1 ] DEBUG 2024/07/05 13:23:30 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/07/05 13:23:30 !! session: chargeRater.chargedEnergy=10.3 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/07/05 13:23:30 charger status: B
[lp-1 ] ERROR 2024/07/05 13:23:30 charger disable: Rejected
[site ] DEBUG 2024/07/05 13:24:00 ----
[lp-1 ] DEBUG 2024/07/05 13:24:00 charge power: 0W
[site ] DEBUG 2024/07/05 13:24:00 pv power: 1721W
[site ] DEBUG 2024/07/05 13:24:00 grid meter: -2218W
[site ] DEBUG 2024/07/05 13:24:00 site power: -2218W
[lp-1 ] DEBUG 2024/07/05 13:24:00 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/07/05 13:24:00 !! session: chargeRater.chargedEnergy=10.3 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/07/05 13:24:00 charger status: B
[lp-1 ] ERROR 2024/07/05 13:24:00 charger disable: Rejected
[site ] DEBUG 2024/07/05 13:24:30 ----
[lp-1 ] DEBUG 2024/07/05 13:24:30 charge power: 0W
[site ] DEBUG 2024/07/05 13:24:30 pv power: 1492W
[site ] DEBUG 2024/07/05 13:24:30 grid meter: -2128W
[site ] DEBUG 2024/07/05 13:24:30 site power: -2128W
[lp-1 ] DEBUG 2024/07/05 13:24:30 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/07/05 13:24:30 !! session: chargeRater.chargedEnergy=10.3 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/07/05 13:24:30 charger status: B
[lp-1 ] ERROR 2024/07/05 13:24:30 charger disable: Rejected
[site ] DEBUG 2024/07/05 13:25:00 ----
[lp-1 ] DEBUG 2024/07/05 13:25:00 charge power: 0W
[site ] DEBUG 2024/07/05 13:25:00 pv power: 2978W
[site ] DEBUG 2024/07/05 13:25:00 grid meter: -921W
[site ] DEBUG 2024/07/05 13:25:00 site power: -921W
[lp-1 ] DEBUG 2024/07/05 13:25:00 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/07/05 13:25:00 !! session: chargeRater.chargedEnergy=10.3 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/07/05 13:25:00 charger status: B
[lp-1 ] ERROR 2024/07/05 13:25:00 charger disable: Rejected
```

### What type of operating system are you running?

Docker container

### Version

evcc version 0.127.3

romestylez, 2024-07-05:
Komisch finde ich auch folgendes

`/app # evcc -l debug charger                                                                                                                                
[main  ] INFO 2024/07/05 17:21:39 evcc 0.127.3                                                                                                              
[main  ] INFO 2024/07/05 17:21:39 using config file: /app/evcc.yaml                                                                                         
[db    ] INFO 2024/07/05 17:21:40 using sqlite database: /root/.evcc/evcc.db                                                                                
[ocpp-1] DEBUG 2024/07/05 17:21:40 waiting for chargepoint: 5m0s                                                                                            
[main  ] FATAL 2024/07/05 17:26:40 cannot create charger 'wallbox5': cannot create charger type 'template': cannot create charger type 'ocpp': timeout `

In der Wallbox ist per ws:\\ mein Nas hinterlegt und gestern Abend bzw. heute Morgen konnte ich zumindest noch über die Option "Schnell" das Laden starten beim beenden kam dann

`[lp-1 ] DEBUG 2024/07/05 13:25:00 charge power: 0W
[site ] DEBUG 2024/07/05 13:25:00 pv power: 2978W
[site ] DEBUG 2024/07/05 13:25:00 grid meter: -921W
[site ] DEBUG 2024/07/05 13:25:00 site power: -921W
[lp-1 ] DEBUG 2024/07/05 13:25:00 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/07/05 13:25:00 !! session: chargeRater.chargedEnergy=10.3 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/07/05 13:25:00 charger status: B
[lp-1 ] ERROR 2024/07/05 13:25:00 charger disable: Rejected`

andig, 2024-07-05:
Bitte:
- es braucht ein ausreichend vollständiges Log. Der Fehler `[lp-1 ] ERROR 2024/07/05 13:23:30 charger disable: Rejected` kommt irgendwo her- was war vorher?
- und: bitte formatiere Deine Postings. Das Kauderwelsch ist leider unlesbar.

romestylez, 2024-07-05:
Habs erst jetzt bemerkt das es unformatiert war sorry. Habe die Config angepasst nun ist die Verbindung zum Charger da. Ich poste gleich das gesamte log.

romestylez, 2024-07-05:
`[main  ] INFO 2024/07/05 17:37:33 evcc 0.127.3
[main  ] INFO 2024/07/05 17:37:33 using config file: /app/evcc.yaml
[db    ] INFO 2024/07/05 17:37:33 using sqlite database: /root/.evcc/evcc.db
[main  ] INFO 2024/07/05 17:37:33 listening at :7070
[ocpp-1] DEBUG 2024/07/05 17:37:33 waiting for chargepoint: 5m0s
[ocpp  ] INFO 2024/07/05 17:37:34 charge point connected, registering: EVB-P21302733
[site  ] INFO 2024/07/05 17:38:36 site config:
[site  ] INFO 2024/07/05 17:38:36   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2024/07/05 17:38:36     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/07/05 17:38:36     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/07/05 17:38:36     pv 2:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/07/05 17:38:36   vehicles:
[site  ] INFO 2024/07/05 17:38:36     vehicle 1: range ✓ finish ✗ status ✗ climate ✓ wakeup ✗
[lp-1  ] INFO 2024/07/05 17:38:36 loadpoint 1:
[lp-1  ] INFO 2024/07/05 17:38:36   mode:        pv
[lp-1  ] INFO 2024/07/05 17:38:36   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/07/05 17:38:36   meters:      charge ✗
[lp-1  ] DEBUG 2024/07/05 17:38:36 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/07/05 17:38:36 phase timer inactive
[lp-1  ] DEBUG 2024/07/05 17:38:36 pv timer inactive
[lp-1  ] INFO 2024/07/05 17:38:36 vehicle updated: unknown -> Smart
[lp-1  ] DEBUG 2024/07/05 17:38:36 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[site  ] DEBUG 2024/07/05 17:38:36 ----
[lp-1  ] DEBUG 2024/07/05 17:38:36 charge power: 0W
[main  ] INFO 2024/07/05 17:38:36 new version available: 0.128.1
[site  ] DEBUG 2024/07/05 17:38:36 pv power: 432W
[site  ] DEBUG 2024/07/05 17:38:36 grid meter: -9W
[site  ] DEBUG 2024/07/05 17:38:36 site power: -9W
[lp-1  ] DEBUG 2024/07/05 17:38:36 vehicle odometer: 705km
[lp-1  ] DEBUG 2024/07/05 17:38:36 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/07/05 17:38:36 charger status: B
[lp-1  ] INFO 2024/07/05 17:38:36 car connected
[lp-1  ] DEBUG 2024/07/05 17:38:36 pv timer elapse
[lp-1  ] DEBUG 2024/07/05 17:38:36 pv timer inactive
[lp-1  ] DEBUG 2024/07/05 17:38:36 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/07/05 17:38:36 vehicle soc: 51%
[lp-1  ] DEBUG 2024/07/05 17:38:36 vehicle range: 119km
[lp-1  ] DEBUG 2024/07/05 17:38:36 max charge current: 16A
[lp-1  ] DEBUG 2024/07/05 17:38:36 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] ERROR 2024/07/05 17:38:37 charger enable: Rejected
[site  ] DEBUG 2024/07/05 17:39:06 ----
[lp-1  ] DEBUG 2024/07/05 17:39:06 charge power: 0W
[site  ] DEBUG 2024/07/05 17:39:06 pv power: 434W
[site  ] DEBUG 2024/07/05 17:39:06 grid meter: -19W
[site  ] DEBUG 2024/07/05 17:39:06 site power: -19W
[lp-1  ] DEBUG 2024/07/05 17:39:06 vehicle odometer: 705km
[lp-1  ] DEBUG 2024/07/05 17:39:06 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/07/05 17:39:06 charger status: B
[lp-1  ] ERROR 2024/07/05 17:39:07 charger enable: Rejected
[site  ] DEBUG 2024/07/05 17:39:36 ----
[lp-1  ] DEBUG 2024/07/05 17:39:36 charge power: 0W
[site  ] DEBUG 2024/07/05 17:39:36 pv power: 444W
[site  ] DEBUG 2024/07/05 17:39:36 grid meter: -35W
[site  ] DEBUG 2024/07/05 17:39:36 site power: -35W
[lp-1  ] DEBUG 2024/07/05 17:39:36 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/07/05 17:39:36 charger status: B
[lp-1  ] ERROR 2024/07/05 17:39:36 charger enable: Rejected
[lp-1  ] DEBUG 2024/07/05 17:39:46 set charge mode: pv
[site  ] DEBUG 2024/07/05 17:39:46 ----
[lp-1  ] DEBUG 2024/07/05 17:39:46 charge power: 0W
[site  ] DEBUG 2024/07/05 17:39:46 pv power: 447W
[site  ] DEBUG 2024/07/05 17:39:46 grid meter: -41W
[site  ] DEBUG 2024/07/05 17:39:46 site power: -41W
[lp-1  ] DEBUG 2024/07/05 17:39:46 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/07/05 17:39:46 charger status: B
[lp-1  ] DEBUG 2024/07/05 17:39:46 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/07/05 17:39:46 pv charge current: 0.0594A = 0A + 0.0594A (-41W @ 3p)
[lp-1  ] DEBUG 2024/07/05 17:39:46 pv enable timer reset
[lp-1  ] DEBUG 2024/07/05 17:39:46 pv timer inactive
[lp-1  ] DEBUG 2024/07/05 17:39:49 set charge mode: now
[... 16 lines cut ...]
[site  ] DEBUG 2024/07/05 17:40:36 ----
[lp-1  ] DEBUG 2024/07/05 17:40:36 charge power: 0W
[site  ] DEBUG 2024/07/05 17:40:36 pv power: 471W
[site  ] DEBUG 2024/07/05 17:40:36 grid meter: -121W
[site  ] DEBUG 2024/07/05 17:40:36 site power: -121W
[lp-1  ] DEBUG 2024/07/05 17:40:36 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/07/05 17:40:36 charger status: B
[lp-1  ] ERROR 2024/07/05 17:40:36 charger enable: Rejected
[site  ] DEBUG 2024/07/05 17:41:06 ----
[lp-1  ] DEBUG 2024/07/05 17:41:06 charge power: 0W
[site  ] DEBUG 2024/07/05 17:41:06 pv power: 480W
[site  ] DEBUG 2024/07/05 17:41:06 grid meter: -157W
[site  ] DEBUG 2024/07/05 17:41:06 site power: -157W
[lp-1  ] DEBUG 2024/07/05 17:41:06 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/07/05 17:41:06 charger status: B
[lp-1  ] ERROR 2024/07/05 17:41:06 charger enable: Rejected
[site  ] DEBUG 2024/07/05 17:41:36 ----
[lp-1  ] DEBUG 2024/07/05 17:41:36 charge power: 0W
[site  ] DEBUG 2024/07/05 17:41:36 pv power: 495W
[site  ] DEBUG 2024/07/05 17:41:36 grid meter: -77W
[site  ] DEBUG 2024/07/05 17:41:36 site power: -77W
[lp-1  ] DEBUG 2024/07/05 17:41:36 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/07/05 17:41:36 charger status: B
[lp-1  ] ERROR 2024/07/05 17:41:36 charger enable: Rejected
`

romestylez, 2024-07-05:
[log.txt](https://github.com/user-attachments/files/16111629/log.txt)
Hier das Log

andig, 2024-07-06:
Da hilft nur warten oder IP Adresse wechseln oder Support kontaktieren.

andig, 2024-07-06:
Deine WB sagt schon ab Start

    [lp-1  ] ERROR 2024/07/05 17:38:37 charger enable: Rejected

Vielleicht sieht man im trace log warum.

romestylez, 2024-07-06:
Kann ich einfach nebenher ein "evcc -l debug charger" machen oder was genau ?

romestylez, 2024-07-07:
Also ich habe noch mal etwas die Config angepasst und die Wallbox kontrolliert. Die Wallbox ist per ping erreichbar und das auch dauerhaft. Wenn ich evcc mit trace starte sehe ich folgendes was für mich sehr gut aussieht.

> [ocpp ] DEBUG 2024/07/07 13:22:31 charge point connected: EVB-XXXXXXXXX
[ocpp ] TRACE 2024/07/07 13:22:32 sent JSON message to EVB-XXXXXXXXX: [2,"3773121425","GetConfiguration",{}]
[ocpp ] TRACE 2024/07/07 13:22:33 received JSON message from EVB-XXXXXXXXX: [3,"3773121425",{"configurationKey":[{"key":"AllowOfflineTxForUnknownId","readonly":false,"value":"false"},{"key":"AuthorizationCacheEnabled","readonly":false,"value":"false"},{"key":"AuthorizationKey","readonly":false,"value":"FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"},{"key":"AuthorizeRemoteTxRequests","readonly":false,"value":"true"},{"key":"ChargeProfileMaxStackLevel","readonly":true,"value":"6"},{"key":"ChargingScheduleAllowedChargingRateUnit","readonly":true,"value":"Current,Power"},{"key":"ChargingScheduleMaxPeriods","readonly":true,"value":"25"},{"key":"ClockAlignedDataInterval","readonly":false,"value":"0"},{"key":"ConnectionTimeOut","readonly":false,"value":"60"},{"key":"ConnectorPhaseRotation","readonly":false,"value":"0.Unknown,1.RST"},{"key":"ConnectorPhaseRotationMaxLength","readonly":true,"value":"21"},{"key":"ConnectorSwitch3to1PhaseSupported","readonly":true,"value":"false"},{"key":"DefaultChargingDelay","readonly":false,"value":"0"},{"key":"GetConfigurationMaxKeys","readonly":true,"value":"1"},{"key":"HeartbeatInterval","readonly":false,"value":"60"},{"key":"LightIntensity","readonly":false,"value":"5"},{"key":"LocalAuthListEnabled","readonly":false,"value":"true"},{"key":"LocalAuthListMaxLength","readonly":true,"value":"500"},{"key":"LocalAuthorizeOffline","readonly":false,"value":"true"},{"key":"LocalPreAuthorize","readonly":false,"value":"true"},{"key":"MaxChargingProfilesInstalled","readonly":true,"value":"66"},{"key":"MessageTimeout","readonly":false,"value":"60"},{"key":"MeterValuesAlignedData","readonly":false,"value":"Energy.Active.Import.Register"},{"key":"MeterValuesAlignedDataMaxLength","readonly":true,"value":"11"},{"key":"MeterValueSampleInterval","readonly":false,"value":"4"},{"key":"MeterValuesSampledData","readonly":false,"value":"Current.Import,Energy.Active.Import.Register,Power.Active.Import,Voltage"},{"key":"MeterValuesSampledDataMaxLength","readonly":true,"value":"11"},{"key":"NumberOfConnectors","readonly":true,"value":"1"},{"key":"ReserveConnectorZeroSupported","readonly":true,"value":"true"},{"key":"ResetRetries","readonly":false,"value":"0"},{"key":"SendLocalListMaxLength","readonly":true,"value":"25"},{"key":"StopTransactionMaxMeterValues","readonly":true,"value":"0"},{"key":"StopTransactionOnEVSideDisconnect","readonly":false,"value":"true"},{"key":"StopTransactionOnInvalidId","readonly":false,"value":"false"},{"key":"StopTxnAlignedData","readonly":false,"value":""},{"key":"StopTxnAlignedDataMaxLength","readonly":true,"value":"0"},{"key":"StopTxnSampledData","readonly":false,"value":""},{"key":"StopTxnSampledDataMaxLength","readonly":true,"value":"0"},{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,FirmwareManagement,LocalAuthListManagement,SmartCharging,RemoteTrigger,Reservation"},{"key":"SupportedFeatureProfilesMaxLength","readonly":true,"value":"6"},{"key":"SupportedFileTransferProtocols","readonly":true,"value":"FTP"},{"key":"TransactionMessageAttempts","readonly":false,"value":"1"},{"key":"TransactionMessageRetryInterval","readonly":false,"value":"60"},{"key":"UnlockConnectorOnEVSideDisconnect","readonly":false,"value":"false"},{"key":"WebSocketPingInterval","readonly":false,"value":"60"},{"key":"evb_APNName","readonly":false,"value":"m2m.tele2.com"},{"key":"evb_APNPass","readonly":false,"value":"**********"},{"key":"evb_APNUser","readonly":false,"value":""},{"key":"evb_AutoStart","readonly":false,"value":"042379BA7E6B85"},{"key":"evb_BootInfo","readonly":true,"value":"EV-BOX,G4E-WBO,21302733,P0424B0424v0.220706_W6.0.0-050,,"},{"key":"evb_ConnectionInfo","readonly":false,"value":"false"},{"key":"evb_ConnectorCableLock","readonly":false,"value":"1.false"},{"key":"evb_ConnectorCcid","readonly":true,"value":"1.CCID V2 Trip EU"},{"key":"evb_ConnectorCcidAC","readonly":false,"value":"1.0"},{"key":"evb_ConnectorInput1","readonly":false,"value":"1.NotConnected"},{"key":"evb_ConnectorInput2","readonly":false,"value":"1.NotConnected"},{"key":"evb_ConnectorInternalTiltSensor","readonly":false,"value":"1.true"},{"key":"evb_ConnectorKWhMeterAddress","readonly":false,"value":"1.0"},{"key":"evb_ConnectorList","readonly":false,"value":"false"},{"key":"evb_ConnectorListOrder","readonly":false,"value":"1.21304698"},{"key":"evb_ConnectorPhaseRotationAfterPowerMeter","readonly":false,"value":"1.false"},{"key":"evb_ConnectorPulsePerKWh","readonly":false,"value":"1.1000"},{"key":"evb_ConnectorRelay","readonly":false,"value":"1.16A+32A"},{"key":"evb_DateTime","readonly":false,"value":"2024-07-07T11:22:33Z"},{"key":"evb_DebugPort","readonly":false,"value":"false"},{"key":"evb_DebugValues","readonly":false,"value":"true"},{"key":"evb_DefaultChargingDelay","readonly":false,"value":"0"},{"key":"evb_ELaadFunction","readonly":false,"value":"false"},{"key":"evb_EVBFunction","readonly":false,"value":"true"},{"key":"evb_GPSValues","readonly":false,"value":"true"},{"key":"evb_LastFirmwareUpdate","readonly":true,"value":"2024-07-03T17:06:41Z,Installed"},{"key":"evb_LastSmartGridModuleMessage" [... cut ...]
Ein "evcc -l trace charger" läuft aber nach 5 Minuten auf ein Timeout

> [EVB-P21302733-1] DEBUG 2024/07/07 13:25:42 waiting for chargepoint: 5m0s                                                                             
> [main  ] FATAL 2024/07/07 13:30:42 cannot create charger 'evbox': cannot create charger type 'template': cannot create charger type 'ocpp': timeout   

Ich verstehe nicht so ganz wieso er im trace alles richtig anzeigt und diese Meldungen "received JSON" usw auch regelmäßig kommen aber ein "evcc -l trace charger" nicht funktioniert.

andig, 2024-07-07:
Was hat das jetzt wieder mit dem ursächlichen `Rejected` Fehler zu tun? Mach doch mal bitte ein Log, das auch den Fehler zeigt.

romestylez, 2024-07-07:
Was das mit dem Fehler zu tun hat ? Das ich die Wallbox ja scheinbar per evcc nicht erreichen kann. Ich habe oben das gesamte Log bereit angehangen das lief halt unter "debug" da ist nix passiert vorher.

andig, 2024-07-07:
Siehe https://github.com/evcc-io/evcc/issues/14721#issuecomment-2211083371. Der Fehler ist die Ursache.

romestylez, 2024-07-07:
Ich sehe aktuell keinen rejected - aber eventuell weil der Smart nicht angeschlossen ist. Ich wollte erst mal sicher gehen das die Kommunikation mit der Wallbox klappt.

romestylez, 2024-07-07:
Hier ein Log ich hoffe das ist vollständig
[evcc-20240707-195616-trace.log](https://github.com/user-attachments/files/16120207/evcc-20240707-195616-trace.log)

Die aktuelle Config habe ich im ersten Beitrag editiert.

romestylez, 2024-07-07:
Ich habe mal folgendes Template für den charger genutzt. 

`chargers:                                                                                                                                             
  - name: evbox
    type: template
    template: elvi
    timeout: 2m # optional
    meter: true`

Ich sehe außen an der Wallbox das die Orange LED leuchtet und das sehe ich auch im Log die Kommunikation scheint also zu stehen.

[evcc-20240707-202956-trace.log](https://github.com/user-attachments/files/16120311/evcc-20240707-202956-trace.log)

Wenn ich den Ladevorgang manuell starte sehe ich auch im Log das er lädt und die LED "Blue" ist. Am Ende sieht man dann das irgendwann die LED orange ist nachdem ich den Ladevorgang beendet habe und wenn ich auf "aus" klicke kommt "Rejected".

andig, 2024-07-07:
Du hast beim Start schon eine laufende Session (`"transactionId":1`):

    [ocpp  ] TRACE 2024/07/07 19:52:59 recv EVB-P21302733: [2,"2e1d927b-df12-4665-b78d-40ffe4c245a7","MeterValues",{"connectorId":1,"transactionId":1,"meterValue":[{"timestamp":"2024-07-07T17:52:59Z","sampledValue":[{"value":"0.0","context":"Trigger","measurand":"Current.Import","phase":"L1","unit":"A"},{"value":"0.0","context":"Trigger","measurand":"Current.Import","phase":"L2","unit":"A"},{"value":"0.0","context":"Trigger","measurand":"Current.Import","phase":"L3","unit":"A"},{"value":"0","context":"Trigger","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"0","context":"Trigger","measurand":"Power.Active.Import","unit":"W"},{"value":"0","context":"Trigger","measurand":"Voltage","phase":"L1","unit":"V"},{"value":"0","context":"Trigger","measurand":"Voltage","phase":"L2","unit":"V"},{"value":"0","context":"Trigger","measurand":"Voltage","phase":"L3","unit":"V"}]}]}]

Das ist immer schwierig. Diese kann evcc nicht stoppen:

```
[ocpp  ] TRACE 2024/07/07 19:54:39 sent EVB-P21302733: [2,"2198157949","RemoteStopTransaction",{"transactionId":1}]
[ocpp  ] TRACE 2024/07/07 19:54:41 recv EVB-P21302733: [3,"2198157949",{"status":"Rejected"}]
```

Warum weiss nur Elvi- bitte Support kontaktieren.

Du könntest testen das Auto mal zu trennen während evcc läuft und schauen ob es dann geht. Zweite Möglichkeit wäre mal zu schauen, ob https://github.com/evcc-io/evcc/pull/14733 für dich funktioniert (ist allerdings noch ungetestet). Da verzichtet evcc darauf, die Session überhaupt zu beenden.

romestylez, 2024-07-07:
Okay komisch das dort eine Session besteht. Habe das Auto nur angeklemmt. Mit dem neuen Parameter wäre die Charger Konfig dann so richtig ?

`chargers:                                                                                                                                             
  - name: evbox
    type: template
    autostart: true
    template: elvi
    timeout: 2m # optional
    meter: true 
`

andig, 2024-07-07:
Bitte:

```yaml
name: evbox
type: template
template: ocpp
```

sollte völlig reichen.

romestylez, 2024-07-07:
Ich habe heute erst mit dem stable geupdated. Ich schaue mal wie ich das nightly installiert bekomme.

Die drei Zeilen reichen weil dadurch automatisch autostart false ist wenn er nicht gesetzt ist ?

andig, 2024-07-07:
Ups- aktualisiert

romestylez, 2024-07-07:
Nightly läuft. Config wie von dir vorgeschlagen. Ich hab jetzt immer "timeouts" im log von der Wallbox wieso auch immer. Laden starten funktioniet nicht stoppen auch nicht aber sehe jetzt so auch kein "Rejected"

[evcc-20240707-210549-trace.log](https://github.com/user-attachments/files/16120398/evcc-20240707-210549-trace.log)

andig, 2024-07-07:
Setz mal bitte

    http: debug

damit man vom Rest etwas sehen kann. Verhalten schaue ich mir morgen an- das war tatsächlich noch nicht getestet, sorry.

romestylez, 2024-07-07:
Habe ich so eingestellt. Habe alles neugestartet nun scheine ich das Laden stoppen zu können aber nicht starten zu können da dort ab und an timeouts kommen.

Ich teste noch mal etwas. Es sieht aber besser aus !

romestylez, 2024-07-07:
Ladevorgang stoppen geht allerdings blinkt die Wallbox nach dem Stop dann "Orange". Hier wird der Vorgang gestoppt:

[lp-1 ] INFO 2024/07/07 21:38:09 stop charging <-

Was komisch aussieht nachdem man den Ladevorgang beendet hat erscheinen die Timeouts im Log:

[lp-1 ] ERROR 2024/07/07 21:40:09 charge power: timeout

Wenn man nun einen Ladevorgang starten will geht das nicht da man nur "Ladebereit. Warte auf Fahrzeug …" sieht. Vermutlich weil die Wallbox nicht mehr erreichbar ist. Sieht irgendwie so aus als würde der Ladevorgang "falsch" gestoppt und die WB ist danach in einem unbekannten Status was auch die blinkende LED erklären würde.

[Uploading evcc-20240707-214534-trace.log…]()

Wenn man den Stecker am Auto zieht und wieder reinsteckt lädt das Auto direkt wieder und den Ladevorgang kann ich dann auch stoppen.

andig, 2024-07-08:
Bzgl. Timeouts bitte noch hinzufügen:

```yaml
metervalues: Current.Import.L1,Current.Import.L2,Current.Import.L3,Energy.Active.Import.Register,Power.Active.Import,Voltage.L1,Voltage.L2,Voltage.L3
meterinterval: 5s
```

romestylez, 2024-07-08:
Im charger Abschnitt ?

romestylez, 2024-07-08:
[evcc-20240708-133804-trace.log](https://github.com/user-attachments/files/16127069/evcc-20240708-133804-trace.log)

Parameter habe ich hinzugefügt und dann mal den Ladevorgang gestartet. Nichts passiert und irgendwann kommt

> [lp-1 ] ERROR 2024/07/08 13:37:46 charge total import: timeout
[lp-1 ] ERROR 2024/07/08 13:37:46 charger status: timeout
[ocpp ] ERROR 2024/07/08 13:37:47 send TriggerMessage MeterValues for EVB-P21302733 failed: cannot send request 32720281, no client EVB-P21302733 exists
[ocpp ] ERROR 2024/07/08 13:37:52 send TriggerMessage MeterValues for EVB-P21302733 failed: cannot send request 695010861, no client EVB-P21302733 exists
[ocpp ] ERROR 2024/07/08 13:37:57 send TriggerMessage MeterValues for EVB-P21302733 failed: cannot send request 2348528088, no client EVB-P21302733 exists

andig, 2024-07-09:
Siehe https://github.com/evcc-io/evcc/issues/14721#issuecomment-2212543321. Ich mache hier zu bis es ein verwertbares Log gibt.

romestylez, 2024-07-09:
http: debug

das habe ich in der config gesetzt. Wo muss es denn genau rein ?

romestylez, 2024-07-19:
Hatte ich rausgefunden und baue das nach meinem Urlaub mal ein.

Nein das WiFi kann nicht schuld seins. Der Router hängt an der anderen Seite der Wand und die Wallbox ist mit 95% Signalstärke verbunden und das dauerhaft.

coderph0x, 2024-07-19:
`no client EVB-P21302733 exists` bedeutet aber, dass deine Box sich nicht verbunden hat.
Die Box neuzustarten hilft auch hin und wieder.