evcc-io/evcc #14731: 16A vs. 17A

Hofyyy, 2024-07-06:
### Describe the bug

EVCC gibt eine Warnung im UI aus das 16A eingestellt sind und aber 17A gezogen werden.
Handelt es sich hier um ein Rundungsproblem?
- Wallbe Pro (auf 11 KW) HW seitig eingestellt.
- Mit Tesla Y

### Steps to reproduce

1. Sofort laden mit max speed.
2.
3.
...

### Configuration details

```yaml
...
```

### Log details

```text
[lp-1  ] DEBUG 2024/07/06 12:56:57 charge power: 11492W
[lp-1  ] DEBUG 2024/07/06 12:56:57 charge currents: [16 16 17]A
[site  ] DEBUG 2024/07/06 12:56:58 pv power: 1483W
[site  ] DEBUG 2024/07/06 12:56:58 battery soc: 59%
[site  ] DEBUG 2024/07/06 12:56:58 battery power: 4W
[site  ] DEBUG 2024/07/06 12:56:58 grid meter: 10276W
[site  ] DEBUG 2024/07/06 12:56:58 site power: 10230W
```

### What type of operating system are you running?

Docker container

### Version

0.128.1

andig, 2024-07-06:
Joah... das liegt an https://github.com/evcc-io/evcc/pull/14690/files#diff-c456526d4bd9022a488ce32de208b40d86016e2c40b0c81a2dffe557c186377dR733. Wenn wir den Zähler nutzen wollen um zu schauen ob die WB den Befehl bekommen hat brauchts irgendeine Grenze. 

Wir können wieder auf >1A gehen?

Hofyyy, 2024-07-06:
[image]
Also die Wallbox scheint per Modbus TCP 17A auszugeben, manchmal, das toggelt da so rum.

Hofyyy, 2024-07-06:
Also eigentlich ein Fehler der Wallbox. Wenn sie auf 16A eingestellt ist sollte Sie wohl nicht 17A zurückgeben, oder gar ausgeben :-). Aber ja für die Benutzerfreundlichkeit, wenn man hier keine Fehlkonfiguration unterstellt, wäre wahrscheinlich eine Grenze von ">1" angenehmer für Warnungen.

closed 2024-07-06 by andig