evcc-io/evcc #14739: Feature request: Schedules for battery usage

meniku, 2024-07-06:
**Is your feature request related to a problem? Please describe.**
I want to use my house battery for supporting charging at day, but not in the evening or night

**Describe the solution you'd like**
Aklow to setup Plans for when to use the home battery for charging and when not 

**Describe alternatives you've considered**
No idea 

**Additional context**

meniku, 2024-07-06:
Found there is another issue about this already

closed 2024-07-06 by meniku

abuesen, 2024-07-07:
Hej @meniku.

Der andere thread ist inzwischen sehr lang. Ich habe eine Verständnisfrage zur Batterie, da ich die über Homeassistant steuere und nicht nachschauen kann. Interessiere mich aber für die Steuerung. 

Du hast anscheinend den automatischen Start aktiviert, wie in Deinem editieren Post ersichtlich war. Also wird auch nachts die Batterie genutzt. Kann man das abschalten? 
https://docs.evcc.io/docs/features/battery#automatischer-start

Was würdest Du dann genau per Zeit steuern wollen? 
Batterie allgemein nutzen ja/nein
Batterie für PV nutzen ja/nein
Automatisch starten ja/nein 

LG.

meniku, 2024-07-08:
* Mir persönlich würde ein allgemeines Batterie Nutzen Von-Bis reichen. 
* Etwas besser wäre es, wenn die Zeiten relativ zum Sonnenuntergang/Aufgang gemacht werden könnten.
* Am allerbesten wäre es vermutlich, wenn es Zeitpläne gäbe und es je Eintrag die kompletten Einstellmöglichkeiten unter "Hausbatterie". Das ist aber vermutlich recht aufwendig.

Übrigens gibt es hier eine Diskussion dazu: https://github.com/evcc-io/evcc/discussions/14648 , ist allerdings Englisch und dort wurde auf den anderen Thread verwiesen.
Ich bin noch recht neu hier und bin nicht sicher, ob die Entwickler die englischen Themen überhaupt anschauen, deswegen hatte ich den Kommentar im anderen deutschen Thread hinzugefügt und jetzt wieder entfernt.

Insgesamt ist mir das Thema nicht so super dringend, da die Parameter wohl via Rest-API gesetzt werden können. Damit lässt sich ja was basteln. Wäre natürlich schön, wenn evcc das schon von Haus aus liefern könnte.

premultiply, 2024-07-08:
Klassischer Fall (sehr individuell) für eine Hausautomationslösung.
evcc stellt dazu alle nötigen API-Schnittstellen bereit um die persönliche Wunschlösung zu realisieren.

meniku, 2024-07-09:
findest du es individuell? Der andere Thread zeigt ja, dass es zumindest schonmal zwei Menschen gibt, die nicht gerne mit halbvollem Hausspeicher in den Abend starten :-)
Naja vielleicht wird dieses `bufferStartSoc` auch einfach nicht so häufig genutzt