evcc-io/evcc #14740: Discovergy (SmartMeter PV) via UI die Zahlen negativ anzeigt -> Wanr-Meldung

autoGenerated24, 2024-07-07:
### Describe the bug

Ich habe ein SmartMeter von Discovergy per xml und einmal per UI konfiguriert. Per YAML funktioniert es problemlos. Wenn ich den Zähler via UI konfiguriere, wird eine Fehlermeldung angezeigt und der Zählerstand falsch (als negative Zahl dargestellt).

### Steps to reproduce

1. Konfiguriere ein SmartMeter von Discovergy via UI nicht via YAML
2.
3.
...

### Configuration details

```yaml
- name: grid_prod
  type: template
  template: discovergy
  usage: grid
  user: xxxx@xxxx.de
  password: xxxxxxxx
  meter: 1ESYxxxxxx
```

### Log details

```text
[discgy] TRACE 2024/07/07 10:51:28 {"time":1720342287828,"values":{"power":709320,"power3":230780,"energyOut":0,"power1":239940,"energy":330985089352000,"power2":238600}}
[site ] WARN 2024/07/07 10:51:28 pv 2 power: -709W is negative - check configuration if sign is correct
```

### What type of operating system are you running?

Linux

### Version

0.128.1

autoGenerated24, 2024-07-07:
[image]hier ist ein Bild dazu

andig, 2024-07-07:
Bitte 2x konfigurieren (je 1x) und dann

    evcc meter

autoGenerated24, 2024-07-07:
@andig  was soll mir das helfen? das gibt nur raus, dass beides gehen und funktionieren. Im Log sieht man, dass die Abfrage von Discovergy die richtige Zahl zurück liefert (daher funktioniert es auch via YAML)

naltatis, 2024-07-07:
> was soll mir das helfen?

Dann haben wir die Messwerte zum gleiche Zeitpunkt und durch den gleichen Code-Pfad.

andig, 2024-07-07:
Es braucht beide Logs. Dann können wir hier weiter machen.

autoGenerated24, 2024-07-07:
@andig  kann ich machen, habe alles umkonfiguriert, damit ich keine Fehlermeldung bekomme. Also ich brauche etwas Zeit....Warum hast du aber den Ticket geschlossen? So ist der Support bei evcc???

autoGenerated24, 2024-07-07:
[image]
@andig  bitte schön

autoGenerated24, 2024-07-07:
grid_prod Discovergy Produktionszähler
grind_main Discovergy Hauptzähler
beides via yaml konfiguriert
db:4 Discovergy Produktionszähler via UI konfiguriert

andig, 2024-07-07:
> So ist der Support bei evcc???

Aus dem Issuetemplate:

> NOTE Due to amount of issues raised we will close incomplete issues (unclear description what the issue actually is, missing log file, unsupported platform) without further comments.

andig, 2024-07-07:
Kannst du dein Discovergy Login mit info@evcc.io teilen?

autoGenerated24, 2024-07-07:
Meinst du e-mail/pwd? Ich kann das Pwd morgen ändern und dann zuschicken.

autoGenerated24, 2024-07-08:
@andig done

andig, 2024-07-09:
Du kannst deinen PV Zähler nicht als

    usage: grid

anlegen- denn dann ist es ein Netzübergabezähler. Hier muss `pv` stehen.

autoGenerated24, 2024-07-09:
@andig ok, das mag bei der yaml Konfiguration ein Fehler sein, Trotzdem wird an der UI die Zahl negativ dargestellt und dort wird der Zähler bei PV/Batterie hinzugefügt.

premultiply, 2024-07-09:
Ja, deswegen.

autoGenerated24, 2024-07-09:
[image]
ich habe es als usage:pv konfiguriert und bekomme ich nun eine Fehlermeldung

autoGenerated24, 2024-07-09:
Ein Produktionszähler kann nicht negativ sein, da es den Ertrag misst, was >0 sein sollte...

autoGenerated24, 2024-07-09:
[image]
mit useage:pv als grid_prod

andig, 2024-07-09:
Verhalten yaml/ui für pv ist jetzt identisch?

premultiply, 2024-07-09:
Das passt schon. Eine Produktionszähler des MSB hat bei Einspeisung in Richtung Netz per Definition negative Werte (A- bzw. OBIS 2.6.0).
Daher sollte `usage: pv` diesen eigentlich invertieren.

autoGenerated24, 2024-07-09:
@andig ja, identisch aber falsch und löst das Problem nicht. Wenn ich in yaml usage: pv nutze oder den Zähler via UI definiere, wird die Zahl negativ angezeigt (PV-Ertrag) und bekomme ich eine Fehlermeldung

autoGenerated24, 2024-07-09:
> Das passt schon. Eine Produktionszähler des MSB hat bei Einspeisung in Richtung Netz per Definition negative Werte (A- bzw. OBIS 2.6.0). Daher sollte `usage: pv` diesen eigentlich invertieren.

siehe den Trace-Eintrag in der Beschreibung. Discovergy liefert hier eine positive Zahl, daher wird das Invertieren falsch und der User bekommt eine Fehlermeldung

premultiply, 2024-07-09:
Bei `usage: grid` zeigt er positive Werte? Dann ist er wirklich falsch rum angeklemmt

autoGenerated24, 2024-07-09:
btw. warum sollte eine Produktionszähler (unidirektional) negative Werte leifern? Das macht nur bidirektionale Zähler

autoGenerated24, 2024-07-09:
> Bei `usage: grid` zeigt er positive Werte? Dann ist er wirklich falsch rum angeklemmt

ja zeigt er. Nein nicht falsch eingeklemmt, weil im Discovergy-Protal seit 5 Jahren alles richtig angezeigt und berechnet wird

premultiply, 2024-07-09:
Es sollte keinen "unidirektionalen" Zähler mehr geben.
Die Energieflussrichtigungen sind per Definition festgelegt.

Dann trag einfach `usage: grid` ein und dann sollte es passen.

autoGenerated24, 2024-07-09:
> Ein Produktionszähler des MSB hat bei Einspeisung in Richtung Netz per Definition negative Werte (A- bzw. OBIS 2.6.0). Daher sollte `usage: pv` diesen eigentlich invertieren.

**es ist keine Eispeisung, sondern einen Produktionszähler!!!!!** Er misst, wie viel die PV-Anlage als Ertrag produziert!!!

autoGenerated24, 2024-07-09:
> Es sollte keinen "unidirektionalen" Zähler mehr geben. Die Energieflussrichtigungen sind per Definition festgelegt.
> 
> Dann trag einfach `usage: grid` ein und dann sollte es passen.

es gibt aber noch. Ich habe zwei Zähler (wie oben beschrieben). Eine ist Uni und misst geeicht die Produktion/Ertrag. Der andere ist bidirektional und misst den Bezug/Einspeisung.

autoGenerated24, 2024-07-09:
also das Problem ist dann, dass evcc mit unidirektionaler Zähler nichts anfangen kann. usage:pv setzt eine Einspeisung voraus und usage:grid wäre für eine Produktionszähler (unidirektional9 richtig. Letzeres kann man aber via UI nicht konfigurieren, da dort immer usage:pv angelegt wird... Richtig?

premultiply, 2024-07-09:
> kann man aber via UI nicht konfigurieren, da dort immer usage:pv angelegt wird...

Ok, das ist dann der tatsächliche Kern dieses Issues.

autoGenerated24, 2024-07-09:
danke, ich weiß es nun auch Bescheid.

andig, 2024-07-09:
In https://github.com/evcc-io/evcc/commit/17bfca781261fe5b3e473171c68e03ef586c7ad2 wurde das umgedreht, Grund nicht mehr erkennbar. Zurück drehen macht jetzt wieder etwas Anderes kaputt. Insofern würde ich sagen WONTFIX- als `custom` Gerät anlegen oder in yaml eintragen.

closed 2024-07-09 by andig

autoGenerated24, 2024-07-10:
@andig ich galube die Idee war, dass man in der UI auch es als grid und nicht nur automatisch als pv anlegen kann, das würde mein Problem lösen. Es ist also eher eine Feature, dass man grids via UI erstellen kann. In der aktuelle Version ist die Funktion noch nicht vorhanden.

andig, 2024-07-10:
Schon klar- aber das Problem liegt ja darin, dass der PV Zähler seinen Zweck nicht erfüllt. Warum das so ist erschliesst sich mir nicht.

autoGenerated24, 2024-07-10:
@andig das sehe ich anders. der PV Zähler ist ein unidirektionaler Zähler, könnte auch ein KNX-Strommessgerät sein, ist eigentlich egal. Im Log sieht man, dass er eine positive Zahl zurück liefert, da er den Ertrag misst, also wie viel Strom produziert wurde. Deshalb die Zahle immer >0
Legt man diese als usage:pv an, wird die Zahl invertiert, da evcc immer davon ausgeht, dass Strommessung immer bidirektional erfolgt (also ich messe mit usage:pv nicht den Ertrag sondern nur die Einspeisung).
Für Ertragsmessung kann man ja kostel, etc wechselrichter nutzen. Bei Discovergy seid ihr davon ausgegangen, dass es immer eine bidirektionalär Zähler ist, was nicht der Fall (zumindest bei mir) ist.
Lösung:
Ihr müsst einfach erkennen, dass Discovergy nicht nur bidirektionaler Zähler hat und somit mir die Möglichkeit geben via UI den Zähler als usage:grid anzulegen. Da ich es per yaml kann, wäre es für mich nicetohave

andig, 2024-07-10:
Anders gesagt: die Lösung wäre 2 Typen Discovergy PV Zähler zu haben (oder diesen konfigurierbar zu machen), aber nicht beliebige Zähler als PV zuzulassen.

autoGenerated24, 2024-07-10:
@andig  jetzt sind wir an die gleiche Meinung :-)

PeterPablo, 2024-07-10:
Wie wäre es im UI eine Checkbox "Vorzeichen invertieren" anzubieten? Man würde es aus Usability-Sicht damit Anwendern erlauben auch falsch herum angeklemmte Zähler unkompliziert zu korrigieren. Der hier diskutierte Einwegzähler könnte als Sonderfall ebenfalls leichter schlagen werden, ohne weitere Templates zu erzeugen.

premultiply, 2024-07-10:
Nein, denn irgendwelche Schalter in der UI müssen auch immer ein technisches Futter und eine entsprechende API-Schnittstelle haben. Die Geräte in evcc, insbesondere in der UI basieren vollständig auf Templates.
Und es ist auch nicht mal mal eben damit getan nur das Vorzeichen der Leistung zu invertieren, da in der Regel auch noch weitere Werte ebenfalls falsch sind weil Sie z.B. dann aus ganz falschen Registern abgerufen werden. Bei MSB-Zählern siehe dazu z. B. die diversen OBIS Kennzahlen. "Wirkarbeit Netzbezug (+)" ist halt was anderes als "Wirkarbeit Netzeinspeisung (-)" usw.