evcc-io/evcc #14741: Aktivieren/Deaktivieren von Komponenten

autoGenerated24, 2024-07-07:
**Is your feature request related to a problem? Please describe.**
Eine der zwei SmartMeters (PV-Produktionsmessung) hat ein Problem, dass öfters die Netzverbindung verliert und muss neugestartet werden. Wenn dadurch keine Zahlen abgefragt werden können, wird ecc unbenutzbar. Den Fehler zu beheben (außer mir) kann keine zu Hause und so kann zb  meine Familie unser BEV nicht mehr laden (auch aus Netzbezug geht es nicht mehr)

**Describe the solution you'd like**
Die Möglichkeit haben, Komponente einfach zu deaktivieren, damit das System trotz des Fehlers mit den restlichen Komponenten (WallBox, BEV) funktioniert.

**Describe alternatives you've considered**
Ich kann den PV-SmartMeter und den PV-Wechselrichter zwar beides konfigurieren, werden leider dann aber die Ergebnisse automatisch zusammen addiert (was falsch ist). Hier wäre eine Funktion auch gut, dass ich den SmartMeter als Messstelle habe, wenn es nicht funktioniert wird die Messung via einen Backup-Komponente (PV-Wechselrichter) gezogen. Sollte keine der beiden funktionieren, führt zur Fehlermeldung.

**Additional context**
---

An-Wi, 2024-07-07:
In diesem Zusammenhang habe ich ein ähnliches Problem (Bug?) festgestellt.
Mein "SmartMeter"-Werte für die PV-Produktionsmessung kommen von einem Skript. Wenn dort keine Daten bzw. fehlerhafte "0 W" zurückkommen, dann funktioniert die Ladesteuerung noch problemlos, weil ich noch einen "realen" Smartmeter am Netzverknüpfungspunkt (Bezug/Einspeisung) habe, der richtige Werte liefert, aber die aufgezeichneten Daten gehen von 100% Netzbezug (0% Sonne) aus obwohl rein mit PV-Überschuss geladen wird! Liegt hier ein Berechnungs- oder Logikfehler vor?

[image]
Screenshot ist von 15:47 Uhr

### Log
<pre>
[site  ] DEBUG 2024/07/07 15:44:13 ----
[lp-1  ] DEBUG 2024/07/07 15:44:13 charge power: 0W
[lp-1  ] DEBUG 2024/07/07 15:44:13 charge currents: [0 0 0]A
[script] DEBUG 2024/07/07 15:44:13 /fupcService/pvCurrentPower.bat: 0
[site  ] DEBUG 2024/07/07 15:44:13 pv power: 0W
[site  ] DEBUG 2024/07/07 15:44:13 grid meter: -5055W
[site  ] DEBUG 2024/07/07 15:44:13 grid currents: [7.12 6.82 7]A
[site  ] DEBUG 2024/07/07 15:44:13 site power: -4855W
[lp-1  ] DEBUG 2024/07/07 15:44:13 charge total import: 5575.130kWh
[lp-1  ] DEBUG 2024/07/07 15:44:13 charger status: A
[lp-1  ] DEBUG 2024/07/07 15:44:13 next soc poll remaining time: 49m39s
[site  ] DEBUG 2024/07/07 15:44:33 ----
[lp-1  ] DEBUG 2024/07/07 15:44:33 charge power: 0W
[lp-1  ] DEBUG 2024/07/07 15:44:33 charge currents: [0 0 0]A
[script] DEBUG 2024/07/07 15:44:33 /fupcService/pvCurrentPower.bat: 0
[site  ] DEBUG 2024/07/07 15:44:33 pv power: 0W
[site  ] DEBUG 2024/07/07 15:44:33 grid meter: -5077W
[site  ] DEBUG 2024/07/07 15:44:33 grid currents: [7.03 6.91 7.09]A
[site  ] DEBUG 2024/07/07 15:44:33 site power: -4877W
[lp-1  ] DEBUG 2024/07/07 15:44:33 charge total import: 5575.130kWh
[lp-1  ] DEBUG 2024/07/07 15:44:33 charger status: B
[lp-1  ] INFO 2024/07/07 15:44:33 car connected
[lp-1  ] DEBUG 2024/07/07 15:44:33 pv timer elapse
[lp-1  ] DEBUG 2024/07/07 15:44:33 pv timer inactive
[lp-1  ] DEBUG 2024/07/07 15:44:33 charge total import: 5575.130kWh
[nissan] TRACE 2024/07/07 15:44:33 GET https://alliance-platform-caradapter-prod.apps.eu2.kamereon.io/car-adapter/v2/cars/***/battery-status
[nissan] TRACE 2024/07/07 15:44:33 {"id":"***","timestamp":"2024-07-07T13:44:28Z","batteryLevel":66,"batteryAutonomy":378,"batteryCapacity":87,"batteryAvailableEnergy":57,"plugStatus":1,"chargingStatus":0.3,"chargingRemainingTime":266}
[lp-1  ] DEBUG 2024/07/07 15:44:33 vehicle status: B
[lp-1  ] DEBUG 2024/07/07 15:44:33 vehicle soc: 66%
[lp-1  ] DEBUG 2024/07/07 15:44:33 vehicle range: 378km
[lp-1  ] DEBUG 2024/07/07 15:44:33 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/07 15:44:33 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/07 15:44:33 pv charge current: 21.2A = 0A + 21.2A (-4877W @ 1p)
[lp-1  ] DEBUG 2024/07/07 15:44:33 site power -4877W <= -1000W enable threshold
[lp-1  ] DEBUG 2024/07/07 15:44:33 pv enable in 0s
[lp-1  ] DEBUG 2024/07/07 15:44:33 pv enable timer elapsed
[lp-1  ] DEBUG 2024/07/07 15:44:33 max charge current: 6A
[lp-1  ] DEBUG 2024/07/07 15:44:33 charger enable
[lp-1  ] DEBUG 2024/07/07 15:44:33 wake-up timer: start
[site  ] DEBUG 2024/07/07 15:44:53 ----
[lp-1  ] DEBUG 2024/07/07 15:44:53 charge power: 1329W
[lp-1  ] DEBUG 2024/07/07 15:44:53 charge currents: [5.57 0 0]A
[script] DEBUG 2024/07/07 15:44:53 /fupcService/pvCurrentPower.bat: 0
[site  ] DEBUG 2024/07/07 15:44:53 pv power: 0W
[site  ] DEBUG 2024/07/07 15:44:53 grid meter: -3746W
[site  ] DEBUG 2024/07/07 15:44:53 grid currents: [1.7 6.91 7.1]A
[site  ] DEBUG 2024/07/07 15:44:53 site power: -3546W
[lp-1  ] DEBUG 2024/07/07 15:44:53 !! session: chargeRater.chargedEnergy=0.0 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/07/07 15:44:53 charge total import: 5575.132kWh
[lp-1  ] DEBUG 2024/07/07 15:44:53 charger status: C
[lp-1  ] INFO 2024/07/07 15:44:53 start charging ->
[lp-1  ] DEBUG 2024/07/07 15:44:53 wake-up timer: stop
[db    ] TRACE 2024/07/07 15:44:53 INSERT INTO `sessions` (`created`,`finished`,`loadpoint`,`identifier`,`vehicle`,`odometer`,`meter_start_kwh`,`meter_end_kwh`,`charged_kwh`,`charge_duration`,`solar_percentage`,`price`,`price_per_kwh`,`co2_per_kwh`) VALUES ("2024-07-07 15:44:53.289","0000-00-00 00:00:00","Einfahrt","","Ariya",NULL,5575.13,NULL,0,NULL,NULL,NULL,NULL,NULL) RETURNING `id` 1 <nil>
[lp-1  ] DEBUG 2024/07/07 15:44:53 charger vehicle id: P-543-Test                              
[lp-1  ] DEBUG 2024/07/07 15:44:53 soc estimated: 66.00% (vehicle: 66.00%)
[lp-1  ] DEBUG 2024/07/07 15:44:53 vehicle soc: 66%
[lp-1  ] DEBUG 2024/07/07 15:44:53 vehicle range: 378km
[lp-1  ] DEBUG 2024/07/07 15:44:53 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/07 15:44:53 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/07 15:44:53 pv charge current: 21.4A = 6A + 15.4A (-3546W @ 1p)
[lp-1  ] DEBUG 2024/07/07 15:44:53 pv timer reset
[lp-1  ] DEBUG 2024/07/07 15:44:53 pv timer inactive
[... 145 lines cut ...]
[site  ] DEBUG 2024/07/07 15:47:53 ----
[lp-1  ] DEBUG 2024/07/07 15:47:53 charge power: 3511W
[lp-1  ] DEBUG 2024/07/07 15:47:53 charge currents: [14.9 0 0]A
[script] DEBUG 2024/07/07 15:47:53 /fupcService/pvCurrentPower.bat: 0
[site  ] DEBUG 2024/07/07 15:47:53 pv power: 0W
[site  ] DEBUG 2024/07/07 15:47:53 grid meter: -686W
[... cut ...]
[lp-1  ] DEBUG 2024/07/07 15:47:53 !! active phases: 1p = min(1p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/07 15:47:53 !! active phases: 1p = min(1p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/07 15:47:53 pv charge current: 18.1A = 16A + 2.11A (-486W @ 1p)
[site  ] DEBUG 2024/07/07 15:48:13 ---- 
</pre>

### evcc.yaml (Auszug)
`site:`
`  residualPower: 200`
=> klar zu erkennen im Log bei `15:44:13` als Differenz zw. `grid meter` und `site power` als `charge power: 0W` war.

andig, 2024-07-09:
Closing as duplicate of https://github.com/evcc-io/evcc/issues/14496

closed 2024-07-09 by andig