evcc-io/evcc #14753: Seat Mii Electric "Offline"

markus80d, 2024-07-07:
### Describe the bug

Hi,
gefühlt seit dem letzten Update auf 128.1 ist mein Seat Mii Electric "Offline" und EVCC geht nun natürlich immer ins Mindestladen da der Akkustand fehlt. Bin aber nicht sicher ob es am EVCC-Update liegt oder ob VAG was geändert hat. Neustarts von EVCC haben ebenso nix gebracht wie das Auto ab-/anstecken. Wäre toll wenn ihr euch das mal anschauen könntet. An meiner Konfig hab ich nix geändert, also daran sollte es nicht liegen.
Danke und Gruß,
Markus

### Steps to reproduce

1. Auto anstecken
2. EVCC zeigt Mindestladen da Fahrzeug offline
3. EVCC lädt aber tatsächlich im PV-Mode
4. Auto lädt im PV-Mode bis es selber abschaltet

### Configuration details

```yaml
network:
  schema: http
  host: 192.168.33.76
  port: 7070

interval: 20s

log: info

chargers:
- name: garage
  type: template
  template: go-e-v3
  host: 192.168.178.69

loadpoints:
- title: Garage 
  charger: garage
  mode: pv
  vehicle: mii
#   phases: 0
  enable:
    delay: 5m
    threshold: -1000
  disable:
    delay: 5m
    threshold: 700
  guardduration: 5m
#   mincurrent: 6
#   maxcurrent: 16
#   resetOnDisconnect: true
  soc:
    poll:
    #   mode: connected
    #   interval: 120m
      mode: charging
      interval: 60m
    estimate: true

meters:
- name: grid
  type: template
  template: shelly-3em
  usage: grid
  host: 192.168.178.40

site:
  title: Laderegelung E-Auto
  meters:
    grid: grid
  residualPower: 0

vehicles:
  - name: mii
    type: template
    template: seat
    title: Seat Mii electric
    user: xxxxxx
    password: xxxxxx
    phases: 2
    capacity: 32
    # minSoC: 15
    # targetSoC: 87
    icon: car

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.2904
  feedin:
    type: fixed
    price: 0.145

sponsortoken: xxxxxx

telemetry: true
plant: xxxxx
```

### Log details

```text
xxxx
```

### What type of operating system are you running?

HomeAssistant Add-on

VolkerK62, 2024-07-07:
Im UI kannst du auch ein Log abrufen.
Siehe "Tip" in https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen

andig, 2024-07-07:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2024-07-07 by andig

markus80d, 2024-07-07:
[evcc-20240707-174204-debug.log.txt](https://github.com/user-attachments/files/16120096/evcc-20240707-174204-debug.log.txt)

markus80d, 2024-07-07:
Sorry, war zu spät mit dem Log. Die Kinder gingen vor :-). Hier das Debug-Log-File aus der UI.

markus80d, 2024-07-07:
Es wäre auch schön wenn mein Anliegen durch Andi nicht gleich wieder geschlossen worden wäre! Hat mich wieder mal etwas geärgert, echt. Ich hab oft schon gar keine Lust mehr einen Case aufzumachen, aber es hilft ja evtl. auch einigen anderen die vielleicht das gleiche Problem haben. Viele Grüße, Markus

andig, 2024-07-07:
> Hat mich wieder mal etwas geärgert, echt. 

Aus dem Issuetemplate:

> NOTE Due to amount of issues raised we will close incomplete issues (unclear description what the issue actually is, missing log file, unsupported platform) without further comments.

Da steht auch:

> NOTE if you're using HomeAssistant or Docker we ask you to reproduce the problem on plain Linux or Windows first.