evcc-io/evcc #14758: Cannot create vehicle type 'renault'

gimmel, 2024-07-07:
### Describe the bug

I added a Renault ZOE 2021 to evcc following the documentation. Unfortunately I get the following messages:
```
[main ] ERROR 2024/07/07 18:41:08 creating vehicle zoe failed: cannot create vehicle type 'template': cannot create vehicle type 'renault': vehicle is not connected to driver
[site ] WARN 2024/07/07 18:41:09 vehicle 'ZOE' does not support automatic detection
```
There are no errors in the communication with the server, everything seems to be ok. I also tried other services like ZoePHP and others, all of them can read the actual data, but evcc can’t. Here’s the output of ZoePHP:

| Renault EV | Update |
|----------|----------|
Mileage: | 20019 km
Connected: | No
Charging: | No
Charging schedule: | Inactive
Battery level: | 94 %
Range: | 304 km
Battery temperature: | °C
Outside temperature: | °C
Status update: | 07.07.2024 18:06
Start preconditioning
Charging schedule: on \| off
Start charging
Charging history

### Steps to reproduce

1. Add a ZOE to evcc
2. Restart

### Configuration details

```yaml
vehicles:
  - name: elmo
    title: Elmo
    type: template
    template: tesla-command
    icon: car
    ...
  - name: zoe
    title: ZOE
    type: template
    template: renault
    icon: car
    vin: 'VF1***'
    user: '***'
    password: '***'
    capacity: 52
```

### Log details

```text
[renault] TRACE 2024/07/07 18:41:07 GET https://renault-wrd-prod-1-euw1-myrapp-one.s3-eu-west-1.amazonaws.com/configuration/android/config_de_DE.json
[renault] TRACE 2024/07/07 18:41:07 { "instabug": { "enabled": false }, "visibility": { "sections": { "smart_route_planner": true, "pairing_ivi": true, "pairing_ze": true, "ots": false, "ots_retention": "86400" }, "subsections": {}, "views": { "profile_menu_alertes": true, "car_maintenance_tab_title": true, "car_offers_tab_title": true, "car_menu_status_pressure": false, "car_status_quotation": true, "help_menu_contact": true, "help_FAQ_tab_title": true, "help_ADVICES_tab_title": false, "help_notice_title": true, "profile_menu_infos": true, "profile_menu_infos_modify_email": false, "profile_menu_settings": true, "profile_menu_settings_delete_cta": true, "profile_menu_edition_region_layout": false, "carpage_tab_title_status": true, "navigation_myr_off": true, "navigation_myr": true, "navigation_menu_map_parking_btn": false, "navigation_menu_map_more_btn": false, "navigation_menu_poi_bottom_ctas": true, "navigation_menu_poi_details_sendto_cta": true, "help_myr": true, "profile_myr": true, "signup_firstname": true, "add_car_type_layout": false, "input_layout_add_car_reg": false, "profile": { "input_layout_civility": true, "input_layout_firstname": true, "input_layout_lastname": true, "input_layout_birthday": false, "input_layout_national_id": false, "input_layout_mobile_phone": true, "input_layout_landline_phone": false, "input_layout_address": true, "input_layout_postal_code": true, "input_layout_city": true, "profile_menu_infos_region_layout": false, "input_layout_marital_status": false, "input_layout_children_number": false } } }, "pages": { "car": { "cotation": "https://ankaufservice.renault.de/", "general_advise": [ { "picture": "", "maxAdviseNb": 2 } ], "acceleration_advise": [ { "picture": "", "maxAdviseNb": 2 } ], "switch_advise": [ { "picture": "", "maxAdviseNb": 2 } ], "anticipation_advise": [ { "picture": "", "maxAdviseNb": 2 } ], "eco_advice_elec_conduct": [ { "picture": "", "maxAdviseNb": 3 } ], "eco_advice_elec_tyre": [ { "picture": "", "maxAdviseNb": 1 } ], "eco_advice_elec_conso": [ { "picture": "", "maxAdviseNb": 4 } ] }, "offers": { "user": true, "car": true }, "rating_application": { "mail": "", "url": "https://www.renault.de/services/tipps-und-anleitungen/anfrage-an-renault.html" }, "faq": { "maxSectionNb": 3, "maxQuestionNb": 4, "picture": "", "response": [ { "idSection": "1", "idQuestion": "1", "urlType": "video", "url": "Eni138mjSpw" }, { "idSection": "1", "idQuestion": "2", "urlType": "video", "url": "KRkqPA0_5rk" } ], "contact": { "display": true, "urlType": "url", "url": "https://www.renault.de/kontakt.html" } }, "advices": { "maxSectionNb": 3, "maxQuestionNb": 3, "picture": "", "response": [ { "idSection": "1", "idQuestion": "1", "urlType": "video", "url": "Eni138mjSpw" }, { "idSection": "1", "idQuestion": "2", "urlType": "video", "url": "KRkqPA0_5rk" }, { "idSection": "1", "idQuestion": "3", "urlType": "video", "url": "ReUP5ErzceA" }, { "idSection": "2", "idQuestion": "1", "urlType": "video", "url": "Ur-JvJd8M90" }, { "idSection": "2", "idQuestion": "2", "urlType": "video", "url": "aK9j5urvogw" }, { "idSection": "2", "idQuestion": "3", "urlType": "video", "url": "-YSbVnkdaEI" } ], "contact": { "display": true, "urlType": "url", "url": "https://www.renault.de/kontakt.html" } } }, "regex": { "name": "^[\\p{L}\\s\\’'-]{2,255}$", "email": "^[[:alnum:]]([-_.]?[[:alnum:]])*@[[:alnum:]]([-.]?[[:alnum:]])*\\.([a-z]{2,4})$", "password": "^(?=.*[a-z])(?=.*[A-Z])(?=.*\\d)(?!.*\\s)[A-Za-z\\d\\W]{8,}", "phone": "^01([1-9][0-9]{5,10}?)$", "mobile_phone": "^01([1-9][0-9]{5,10}?)$", "vin": "^[a-zA-Z0-9]{17}$", "registration_number": "", "postal_code": "^\\d{5}$", "city": "^[a-zA-ZÀ-ÿ\\s-]{1,40}$" }, "contact": { "call_center": { "channels": [ { "id": 1, "type": "phone", "value": "02232737600" }, { "id": 2, "type": "url", "value": "https://www.renault.de/services/tipps-und-anleitungen/anfrage-an-renault.html" } ] }, "renault_assistance": { "phone_number": "01806365365" } }, "servers": { "kamareon": { "target": "https://alliance-platform-serviceadapter-staging.apps.prod.eu.kamereon.org", "apikey": "Z7P4xyNrTITzNZh52oObSOVfKns9XGLE" }, "wired": { "target": "https://api-wired-dev-1-euw1.wrd-aws.com", "apikey": "AHdOWFASWEPUVQVlhJWshsios0FqTG2E" }, "wiredValid": { "target": "https://api-wired-valid-1-euw1.wrd-aws.com", "apikey": "AHdOWFASWEPUVQVlhJWshsios0FqTG2E" }, "wiredProd": { "target": "https://api-wired-prod-1-euw1.wrd-aws.com", "apikey": "oF09WnKqvBDcrQzcW1rJNpjIuy7KdGaB" }, "gigya": { "target": "https://accounts.eu1.gigya.com", "apikey": "3_S0OWIrqeJ6mxOkXFT8i3TTDwW1IGKk2rIypZjGXi4Hh8vce2ohuERio1Ka5DBbUr" }, "gigyaValid": { "target": "https://accounts.eu1.gigya.com", "apikey": "3_GG7Z4vrxiUff04Vvk45ftlM54qS0-083IL0BfyAJBp5SHFTFO9iclY2fewoTBlIv" }, "gigyaProd": { "target": "https://accounts.eu1.gigya.com", "apikey": "3_7PLksOyBRkHv126x5WhHb-5pqC1qFR8pQjxSeLB6nhAnPERTUlwnYoznHSxwX668" } }, "booking": { "url": "https://anfrage.renault.
[renault] TRACE 2024/07/07 18:41:07 GET https://accounts.eu1.gigya.com/accounts.login?apiKey=3_7PLksOyBRkHv126x5WhHb-5pqC1qFR8pQjxSeLB6nhAnPERTUlwnYoznHSxwX668&loginID=***&password=***
[renault] TRACE 2024/07/07 18:41:07 {
 "callId": "cbae264d853a61edfb8e6f6ca10ded57",
 "errorCode": 0,
 "apiVersion": 2,
 "statusCode": 200,
 "statusReason": "OK",
 "time": "2024-07-07T16:41:07.631Z",
 "registeredTimestamp": 1718312154,
 "UID": "85f7d41b-6269-4b8b-9fea-546e03b4d381",
 "UIDSignature": "B9BQO9ETQlpv0eJA6x4Wy9vKsrA=",
 "signatureTimestamp": "1720370467",
 "created": "2024-06-13T20:55:54.552Z",
 "createdTimestamp": 1718312154,
[... 68 lines cut ...]
 "oldestDataUpdatedTimestamp": 1718312154552,
 "password": {
 "created": "2024-06-13T20:55:54.581Z"
 },
 "profile": {
 "email": "***"
 },
 "registered": "2024-06-13T20:55:54.642Z",
 "socialProviders": "site",
 "verified": "2024-06-13T20:56:22.923Z",
 "verifiedTimestamp": 1718312182923
 }
[renault] TRACE 2024/07/07 18:41:07 GET https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/persons/a9bd4429-e30f-470d-aa6b-7217deb533c9?country=DE
[renault] TRACE 2024/07/07 18:41:08 {"personId":"a9bd4429-e30f-470d-aa6b-7217deb533c9","type":"I","context":"RENAULTGROUP","country":"DE","civility":"2","firstName":"***","lastName":"***","idp":{"idpId":"85f7d41b-6269-4b8b-9fea-546e03b4d381","idpType":"GIGYA","idpStatus":"ACTIVE","login":"***","loginType":"EMAIL","lastLoginDate":"2024-07-07T15:39:38Z","termsConditionAcceptance":true,"termsConditionLastAcceptanceDate":"2024-06-13T20:55:54.828224055Z","registerData":{"purposes":[],"events":[]},"originUserId":"irn-69506_ope_pkjwt_x8fnryboygis","originApplicationName":"One_Nginx","registerBrand":"RENAULT"},"preferredDealers":[{"brand":"RENAULT","dealerId":"27681037_001","dealerName":"Lüdemann & Zankel AG","createdDate":"2024-07-05T14:13:13.618980739Z","lastModifiedDate":"2024-07-05T14:13:13.619070890Z","functionalCreationDate":"2024-07-05T14:13:13.618980739Z","functionalModificationDate":"2024-07-05T14:13:13.619070890Z"}],"emails":[{"emailType":"MAIN","emailValue":"***", [... cut ...] FITPK7%2FVOLCHA%2FREACTI%2FAEBS07%2FPRAHL%2FRRCAM&databaseId=81a762e0-4640-48af-a4bf-66832f080159&bookmarkSet=RSITE&bookmark=EXT_34_AV&profile=HELIOS_OWNERSERVICES_SMALL_V2"}]}],"yearsOfMaintenance":12,"connectivityTechnology":"RLINK1","easyConnectStore":false,"electrical":true,"rlinkStore":false,"deliveryDate":"2021-03-23","retrievedFromDhs":false,"engineEnergyType":"ELEC","radioCode":""}}]}
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

evcc 0.128.1

andig, 2024-07-07:
Duplikat zu https://github.com/evcc-io/evcc/issues/14395.

mariuszmol, 2024-07-07:
same issue like gimmel #14395 is not of help as HA is not used...To become part of evcc community, connection  to ZOE is mandatory. I hope for a straight forward solution. best regards Mariusz

andig, 2024-07-08:
As Renault customer- why not ask them to provide an api with proper support?

mariuszmol, 2024-07-08:
Guten Morgen, ich habe gesehen wir können auch auf deutsch schreiben. Bei Node Red hat es auf Anhieb geklappt die Daten abzurufen? Wegen der API werde ich nachfragen.

andig, 2024-07-08:
Bitte Fahrzeugzugangsdaten an info@evcc.io zum testen.

mariuszmol, 2024-07-08:
> Bitte Fahrzeugzugangsdaten an ***@***.*** ***@***.***> zum testen.

Hallo Andi,
hier meine Zugangsdaten zum Renault Zoe für Testzwecke:

User: 
***@***.*** ***@***.***>	
PW:
***@***.***

Danke und Gruß
Mariusz

closed 2024-07-09 by andig

gimmel, 2024-08-01:
Ich konnte das Problem lösen, indem ich einen neuen Renault-Account angelegt habe, das Fahrzeug dort auch verbunden habe und diese Zugangsdaten in evcc eingegeben habe. Vielleicht gab es ein Problem bei der Übermittlung von Sonderzeichen im Passwort, vielleicht waren es zu viele Requests auf den gleichen Account - keine Ahnung. Auf jeden Fall geht es jetzt.

Wichtigste Erkenntnis: man kann Renault-Fahrzeuge scheinbar mit beliebig vielen Renault-Accounts verbinden.

Weitere Renault-Tipps: das Paket "E-Tech Services" (36 Monate für 39,95€) sendet nur Batteriestand, Reichweite, Kilometerstand und Stecker- und Ladezustand. Wer sich wundert, warum die Position seines Autos irgendwo in der Walachei liegt: um die Position des Autos zu empfangen, braucht man das Paket "Smart Navigation" (12 Monate für 79,90 €), scheinbar bleibt die Position von dem Zeitpunkt erhalten, an dem das Paket abgelaufen ist (3 Jahre nach Kauf des Autos).

----------

I was able to solve the problem by creating a new Renault account, connecting the vehicle to it and entering this account data in evcc. Maybe there was a problem with the transmission of special characters in the password, maybe there were too many requests to the same account - I don't know. In any case, it works now.

The most important finding: you can apparently connect Renault vehicles to any number of Renault accounts.

Other Renault tips: the "E-Tech Services" package (36 months for €39.95) only sends the battery level, range, mileage and plug and charge status. Anyone wondering why the position of their car is somewhere in Wallachia: to receive the position of the car, you need the "Smart Navigation" package (12 months for €79.90), apparently the position is retained from the time the package expired (3 years after purchase of the car).