evcc-io/evcc #14763: Tronity auto config / config broken

sebnaf, 2024-07-07:
### Describe the bug

I have read the news that tronity has co-developed a new API with tesla:

Neuer Ansatz der Fahrzeugintegration für Tesla-Fahrzeuge

> Fahrzeuge senden jetzt Daten direkt an TRONITY, was einen schnelleren Datenzugriff alle 30 Sekunden ermöglicht. Dies verbessert die Aktualität und optimiert das Energiemanagement, wodurch unnötiger Batterieverlust vermieden wird. Tesla-Fahrer werden bald per E-Mail informiert.

For this reason I wanted to reactive my tronity vehicle using

```
vehicles:
- type: template
  template: tronity 
  title: 123
  clientid: 96a24fc6[]  
  clientsecret: decdce92[]  
  vin: LRW3E7[]  
  capacity: 72  
  name: tesla
```

However, some data is read followed by an error.

```
❯ ./evcc -c evcc0.124.4.yaml -l debug vehicle
[main  ] INFO 2024/07/07 21:46:21 evcc 0.124.4
[main  ] INFO 2024/07/07 21:46:21 using config file: evcc0.124.4.yaml
[main  ] ERROR 2024/07/07 21:46:21 creating vehicle db:1 failed: cannot create vehicle type 'template': cannot create vehicle type 'tronity': cannot get vehicles: Get "https://api.tronity.tech/tronity/vehicles": unexpected status: 401 (Unauthorized)
tesla
-----
Soc:           77%
Capacity:      72.0kWh
Charge status: A
Range:         346km
Odometer:      32705km
Features:      []

db:1
----
.Soc:     vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'tronity': cannot get vehicles: Get "https://api.tronity.tech/tronity/vehicles": unexpected status: 401 (Unauthorized)
Capacity: 72.0kWh
Features: [Offline]
```

### Steps to reproduce

1. create a new vehicle using the tesla template from the documentation OR
2. ./evcc configure and chose tronity

### Configuration details

```yaml
vehicles:
- type: template
  template: tronity 
  title: 123
  clientid: 96a24fc6[]  
  clientsecret: decdce92[]  
  vin: LRW3E7[]  
  capacity: 72  
  name: tesla
```

### Log details

```text
❯ ./evcc -c evcc0.124.4.yaml -l debug vehicle
[main  ] INFO 2024/07/07 21:46:21 evcc 0.124.4
[main  ] INFO 2024/07/07 21:46:21 using config file: evcc0.124.4.yaml
[main  ] ERROR 2024/07/07 21:46:21 creating vehicle db:1 failed: cannot create vehicle type 'template': cannot create vehicle type 'tronity': cannot get vehicles: Get "https://api.tronity.tech/tronity/vehicles": unexpected status: 401 (Unauthorized)
tesla
-----
Soc:           77%
Capacity:      72.0kWh
Charge status: A
Range:         346km
Odometer:      32705km
Features:      []

db:1
----
.Soc:     vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'tronity': cannot get vehicles: Get "https://api.tronity.tech/tronity/vehicles": unexpected status: 401 (Unauthorized)
Capacity: 72.0kWh
Features: [Offline]
```

### What type of operating system are you running?

Docker container

### Version

0.124.4

andig, 2024-07-07:
Da bräuchte es Zugangsdaten für info@evcc.io. Aber:

> Fahrzeuge senden jetzt Daten direkt an TRONITY, was einen schnelleren Datenzugriff alle 30 Sekunden ermöglicht

das bringt dir gar nix, denn dann muss evcc ja Tronity pollen?!

sebnaf, 2024-07-07:
ebenso mit "evcc version 0.128.1"

StefanSchoof, 2024-07-07:
Wenn du schreibst, du hast tronity reaktiviert, hast du das Secret neu erzeugt oder kannst du ausschließen, dass es einfach abgelaufen ist?

sebnaf, 2024-07-07:
> Wenn du schreibst, du hast tronity reaktiviert, hast du das Secret neu erzeugt oder kannst du ausschließen, dass es einfach abgelaufen ist?

Ich habe das sowohl das Secret als auch eine neue app in tronity erzeugt und testweise dem Fahrzeug alle Berechtigungen zugewiesen.

sebnaf, 2024-07-07:
> Da bräuchte es Zugangsdaten für [info@evcc.io](mailto:info@evcc.io). Aber:
> 
> > Fahrzeuge senden jetzt Daten direkt an TRONITY, was einen schnelleren Datenzugriff alle 30 Sekunden ermöglicht
> 
> das bringt dir gar nix, denn dann muss evcc ja Tronity pollen?!

Es hat schon einige Vorteile wie:
- Fahrzeug wird nie aufgeweckt wenn nicht notwendig
- Tronity pollt Tesla nicht mehr, Tesla sendet aktiv an Tronity

Somit sehe ich wieder (endlich) einige Vorteile, um Tronity aktiver zu nutzen.

sebnaf, 2024-07-07:
Info:

> Was ist neu und warum ist dies auch für Nicht-Tesla-Fahrer spannend?
> 
> In der aktuellen Implementierung werden die Fahrzeugdaten jeweils von den Servern der Fahrzeughersteller abgerufen. Dabei sendet das Fahrzeug die Daten an den Fahrzeughersteller, und TRONITY kann im Namen der Nutzer die Daten entsprechend abrufen und verarbeiten. 
> 
> Im Gegensatz dazu ist in der Zusammenarbeit mit Tesla nun eine Integration in TRONITY möglich, bei der das Fahrzeug die Daten direkt an TRONITY sendet! Dabei entscheidet das Fahrzeug, wann Daten gesendet werden, und es ist somit in der Regel ein Datenzugriff alle 30 Sekunden möglich 🤯 und nicht wie aktuell alle 3-5 Minuten.
> 
> Daneben gibt es viele weitere Vorteile. Zum Beispiel kann das Fahrzeug das eigene Energiemanagement berücksichtigen, wodurch kein unnötiger Batterieverlust in der Datenübertragung entsteht.
> 
> Wir freuen uns auf diese neue Innovation und sehen in den Dialogen mit den Fahrzeugherstellern, dass sich die Technologie und die Datenverarbeitung zielführend weiterentwickeln!

closed 2024-07-23 by github-actions[bot]