evcc-io/evcc #14772: Configuration broken after migration: `'[0]' has invalid keys: settings`

tiehfood, 2024-07-08:
### Describe the bug

Hi,

I used the command `evcc migrate` but now my configuration seems to be broken.
Everytime I try to start evcc it throws a error message

I tried using the "--ignore-db" parameter, but same result.
The word "settings" does not even appear in the config file.

Using latest version of evcc.

### Steps to reproduce

Use the migration feature

### Configuration details

```yaml
network:
  schema: http
  host: ***
  port: ***

database:
  type: sqlite
  dsn: ***

sponsortoken: ***

log: debug
levels:
  cache: error

plant: ***

interval: ***

modbusproxy:
  - port: ***
    uri: ***

meters:
  - type: template
    template: alpha-ess-smile
    id: ***
    host: ***
    port: ***
    usage: ***
    modbus: ***
    name: ***
  - type: template
    template: alpha-ess-smile
    id: ***
    host: ***
    port: ***
    usage: ***
    modbus: ***
    name: ***
  - type: template
    template: alpha-ess-smile
    id: ***
    host: ***
    port: ***
    usage: ***
    modbus: ***
    name: ***

chargers:
  - name: ***
    type: template
    template: twc3
    host: ***

loadpoints:
  - title: ***
    charger: ***
    mode: ***
    phases: ***
    mincurrent: ***
    maxcurrent: ***
    guardduration: ***
    vehicle: ***

site:
  title: ***
  meters:
    grid: ***
    pv:
      - ***
    battery:
      - ***
  residualPower: ***

vehicles:
  - name: ***
    type: template
    template: tesla
    title: ***
    accessToken: ***
    refreshToken: ***
```

### Log details

```text
[main  ] INFO 2024/07/08 07:43:34 listening at :7070
[main  ] FATAL 2024/07/08 07:43:34 decoding failed due to the following error(s):

'[0]' has invalid keys: settings
```
```

### What type of operating system are you running?

Linux

### Version

0.128.1

tiehfood, 2024-07-08:
What should I do with the sensitive information inside the db? Will this information help you? The yaml is above.
[image]
[image]
[image]
[image]

andig, 2024-07-08:
I need the db :(

tiehfood, 2024-07-08:
Ok, I removed the sensitive info from the db file. Everything else is untouched. Mail is send 😊

tiehfood, 2024-07-09:
Any news why I get that error? Can't use evcc since yesterday.

andig, 2024-07-09:
Workaround for time being:

    evcc  migrate --reset

closed 2024-07-09 by andig