evcc-io/evcc #14782: [site ] WARN 2024/07/08 19:18:54 smartCostActive: no matching rate

TobiasHuber1980, 2024-07-08:
### Describe the bug

[site ] WARN 2024/07/08 19:18:54 smartCostActive: no matching rate obwohl kein planner aktiviert.

### Steps to reproduce

1. evcc starten
2. warten
3. fehlermeldung kommt (orange)
...

### Configuration details

```yaml
./.
```

### Log details

```text
[site ] WARN 2024/07/08 19:18:14 smartCostActive: no matching rate
```

### What type of operating system are you running?

Linux

### Version

0.128.1+1720421358

andig, 2024-07-09:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

TobiasHuber1980, 2024-07-09:
na dann hier mit allen infos:

log:
```
[main ] INFO 2024/07/09 08:24:31 evcc 0.128.1 (eb15f37d)
[main ] INFO 2024/07/09 08:24:31 using config file: /etc/evcc.yaml
[db ] INFO 2024/07/09 08:24:32 using sqlite database: /var/lib/evcc/evcc.db
[main ] INFO 2024/07/09 08:24:32 listening at :7070
[lp-1 ] DEBUG 2024/07/09 08:24:35 charge total import: 1358.939kWh
[site ] WARN 2024/07/09 08:24:35 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site ] INFO 2024/07/09 08:24:35 site config:
[site ] INFO 2024/07/09 08:24:35 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2024/07/09 08:24:35 grid: power ✓ energy ✓ currents ✗
[site ] INFO 2024/07/09 08:24:35 pv 1: power ✓ energy ✗ currents ✗
[site ] INFO 2024/07/09 08:24:35 battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site ] INFO 2024/07/09 08:24:35 vehicles:
[site ] INFO 2024/07/09 08:24:35 vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site ] INFO 2024/07/09 08:24:35 vehicle 2: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site ] INFO 2024/07/09 08:24:35 vehicle 3: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[lp-1 ] INFO 2024/07/09 08:24:35 loadpoint 1:
[lp-1 ] INFO 2024/07/09 08:24:35 mode: pv
[lp-1 ] INFO 2024/07/09 08:24:35 charger: power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1 ] INFO 2024/07/09 08:24:35 meters: charge ✓
[lp-1 ] INFO 2024/07/09 08:24:35 charge: power ✓ energy ✓ currents ✓
[site ] DEBUG 2024/07/09 08:24:35 set buffer soc: 100
[site ] DEBUG 2024/07/09 08:24:35 set buffer start soc: 0
[site ] DEBUG 2024/07/09 08:24:35 set priority soc: 90
[site ] DEBUG 2024/07/09 08:24:35 set battery discharge control: false
[site ] DEBUG 2024/07/09 08:24:35 set residual power: 100
[lp-1 ] DEBUG 2024/07/09 08:24:35 !! active phases: 3p = min(0p measured 0p vehicle 0p physical 0p charger)
[lp-1 ] DEBUG 2024/07/09 08:24:35 phase timer inactive
[lp-1 ] DEBUG 2024/07/09 08:24:35 pv timer inactive
[site ] WARN 2024/07/09 08:24:35 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site ] DEBUG 2024/07/09 08:24:35 ----
[lp-1 ] DEBUG 2024/07/09 08:24:35 charge power: 0W
[lp-1 ] DEBUG 2024/07/09 08:24:35 charge currents: [0 0 0]A
[site ] WARN 2024/07/09 08:24:35 smartCostActive: no matching rate
[site ] DEBUG 2024/07/09 08:24:36 pv power: 1312W
[site ] DEBUG 2024/07/09 08:24:36 battery soc: 48%
[site ] DEBUG 2024/07/09 08:24:36 battery power: -1050W
[site ] DEBUG 2024/07/09 08:24:36 grid meter: -1W
[site ] DEBUG 2024/07/09 08:24:36 battery has priority at soc 48% (< 90%)
[site ] DEBUG 2024/07/09 08:24:36 site power: 99W
[lp-1 ] DEBUG 2024/07/09 08:24:36 charge total import: 1358.939kWh
[lp-1 ] DEBUG 2024/07/09 08:24:36 charger status: A
[lp-1 ] INFO 2024/07/09 08:24:36 car disconnected
[lp-1 ] DEBUG 2024/07/09 08:24:36 !! active phases: 3p = min(0p measured 0p vehicle 0p physical 0p charger)
[lp-1 ] DEBUG 2024/07/09 08:24:36 !! active phases: 3p = min(0p measured 0p vehicle 0p physical 0p charger)
[lp-1 ] DEBUG 2024/07/09 08:24:36 set session soc limit: 0
[lp-1 ] DEBUG 2024/07/09 08:24:36 set session energy limit: 0
[site ] DEBUG 2024/07/09 08:24:55 ----
[lp-1 ] DEBUG 2024/07/09 08:24:55 charge power: 0W
[lp-1 ] DEBUG 2024/07/09 08:24:55 charge currents: [0 0 0]A
[site ] WARN 2024/07/09 08:24:55 smartCostActive: no matching rate
[site ] DEBUG 2024/07/09 08:24:56 pv power: 1314W
[site ] DEBUG 2024/07/09 08:24:56 battery soc: 48%
[site ] DEBUG 2024/07/09 08:24:56 battery power: -1052W
[site ] DEBUG 2024/07/09 08:24:56 grid meter: -4W
[site ] DEBUG 2024/07/09 08:24:56 battery has priority at soc 48% (< 90%)
[site ] DEBUG 2024/07/09 08:24:56 site power: 96W
[lp-1 ] DEBUG 2024/07/09 08:24:56 charge total import: 1358.939kWh
[lp-1 ] DEBUG 2024/07/09 08:24:56 charger status: A
```

nur site:
```
[site ] WARN 2024/07/09 08:26:15 smartCostActive: no matching rate
[site ] DEBUG 2024/07/09 08:26:16 pv power: 1379W
[site ] DEBUG 2024/07/09 08:26:16 battery soc: 48%
[site ] DEBUG 2024/07/09 08:26:16 battery power: -1148W
[site ] DEBUG 2024/07/09 08:26:16 grid meter: -2W
[site ] DEBUG 2024/07/09 08:26:16 battery has priority at soc 48% (< 90%)
[site ] DEBUG 2024/07/09 08:26:16 site power: 98W
[site ] DEBUG 2024/07/09 08:26:35 ----
[site ] WARN 2024/07/09 08:26:35 smartCostActive: no matching rate
[site ] DEBUG 2024/07/09 08:26:36 pv power: 1375W
[site ] DEBUG 2024/07/09 08:26:36 battery soc: 48%
[site ] DEBUG 2024/07/09 08:26:36 battery power: -1160W
[site ] DEBUG 2024/07/09 08:26:36 grid meter: -3W
[site ] DEBUG 2024/07/09 08:26:36 battery has priority at soc 48% (< 90%)
[site ] DEBUG 2024/07/09 08:26:36 site power: 97W
[site ] DEBUG 2024/07/09 08:26:55 ----
[site ] WARN 2024/07/09 08:26:55 smartCostActive: no matching rate
[site ] DEBUG 2024/07/09 08:26:56 pv power: 1439W
[site ] DEBUG 2024/07/09 08:26:56 battery soc: 48%
[site ] DEBUG 2024/07/09 08:26:56 battery power: -1179W
[site ] DEBUG 2024/07/09 08:26:56 grid meter: 3W
[site ] DEBUG 2024/07/09 08:26:56 battery has priority at soc 48% (< 90%)
[site ] DEBUG 2024/07/09 08:26:56 site power: 103W
```

tariffs aus UI:
```
currency: EUR
grid:
    type: fixed
    price: 0.3094
feedin:
    type: fixed
    price: 0.08
```
yaml (rest ist im UI):
```
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  messaging: trace
  cache: trace

# unique installation id
plant: d5672f047ed940d74c94d2330b76a52cb3b956776b38565ed49d5b98aa23d0f0

interval: 30s # control cycle interval

sponsortoken: xxx

site:
  title: Zuhause
  meters:
    grid: grid1

loadpoints:
- title: Garage
  charger: openwb_pro
  mode: pv

chargers:
- name: openwb_pro
  type: template
  template: openwb-pro
  host: 192.168.178.11   

meters:
- type: template
  template: rct-power 
  usage: grid  
  host: 192.168.178.10  
  name: grid1
```

TobiasHuber1980, 2024-07-09:
ich glaube das hat hiermit zu tun.
https://github.com/evcc-io/evcc/commit/93c8b36f7fd0399af6c495393e1c7734a04af4f6#diff-7aa5baae0bd3243f85bfc6db64612f7450eb0170fa310ca2c93dc3968fc02e02

andig, 2024-07-09:
Ah, danke- Du hast ja auch gar keinen dynamischen Tarif. Das Puzzlestück fehlte mir.

naltatis, 2024-07-09:
Jo, ist mir gestern auch schon aufgefallen. Ich schau da mal rein.

PhiBeh, 2024-07-23:
Habe dasselbe "Problem" wie von TobiasHuber1980 beschrieben.
Erstmalig aufgetreten, nachdem ich heute tariffs um co2 (grünstromindex) ergänzt habe.
Config ist sehr ähnlich zu TobiasHuber1980's.
Version ist 0.128.4.
Wird dies mit https://github.com/evcc-io/evcc/pull/14795 korrigiert?
Falls ja, kommt die Korrektur in der nächsten Version?
P.S.: Bin GitHub Neuling und konnte mir die Fragen leider nicht selbst beantworten. Habt bitte Nachsicht.

TobiasHuber1980, 2024-07-23:
#14795 ist bereits in 0.128.4 enthalten. Also sollte kein Problem mehr sein. 
Zeig mal deine Konfiguration...

PhiBeh, 2024-07-23:
Danke für die kurzfristige Antwort!

Tatsächlich kann ich die Meldung jetzt nicht mehr reproduzieren.

Trotzdem, hier meine Config:

`
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: 1fc2ffb2035f0ca11d573eea622807fc99c90a96824ab31170871eb35f0b48c0

interval: 15s # control cycle interval

sponsortoken: xxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: kostal-ksem
  id: 71
  host: 192.168.178.6
  port: 502
  usage: grid
  modbus: tcpip
  name: grid

chargers:
- type: template
  template: go-e-v3
  host: 192.168.178.8
  name: wallbox

vehicles:
- type: template
#  template: tesla
  template: tesla-command
  title: EVE

  accessToken: xxx
  refreshToken: xxx
#  vin: xxx
#  capacity: 62
  name: eve

loadpoints:
- title: Wall-E
  charger: wallbox
  vehicle: eve
  mode: minpv
  phases: 0
  mincurrent: 6
  maxcurrent: 16
  enable:
    delay: 1m
  disable:
    delay: 1m
  guardduration: 3m

site:
  title: Zuhause
  meters:
    grid: grid
    pv:
    - pv

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.26 # [currency]/kWh since 23.07.2024
  feedin:
    type: fixed
    price: 0.08 # [currency]/kWh
  co2:
    type: grünstromindex
    zip: xxx # PLZ mit führender Null mit in "" setzen
`

PhiBeh, 2024-07-24:
Korrektur: Heute, mit PV Überschuss, kommt erneut die Meldung.

Hier noch die log:
[evcc-20240724-074016-debug.log](https://github.com/user-attachments/files/16357679/evcc-20240724-074016-debug.log)