evcc-io/evcc #14785: evcc: startup fails if not connected to internet

Pebberman, 2024-07-08:
### Describe the bug

Internet access is down. However, all systems in the home network work and can communicate. Starting evcc fails with the message:

Error during startup. Check your configuration and restart.
cannot create tariff type 'grünstromindex': Get "https://api.corrently.io/v2.0/gsi/prediction?zip=01234": dial tcp: lookup api.corrently.io: i/o timeout

Absence of the Internet should only produce a warning, not an error. 

### Steps to reproduce

1. configuration with tariff
2. disconnect internet from outside (but keep WLAN running)
3. start evcc 

### Configuration details

```yaml
tariffs: 
    currency: EUR # (default EUR) 
    grid:
    # static grid price
      type: fixed 
      price: 0.278 # [currency]/kWh 
    feedin:
    # rate for feeding excess (pv) energy to 
the grid
      type: fixed
      price: 0.08 # [currency]/kWh
```

### Log details

```text
Error during startup. Check your configuration and restart.
cannot create tariff type 'grünstromindex': Get "https://api.corrently.io/v2.0/gsi/prediction?zip=01234": dial tcp: lookup api.corrently.io: i/o timeout
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.128.1

andig, 2024-07-09:
What would determine if you can work without the tariff, i.e. if you don‘t have solar and only use Tibber?

Pebberman, 2024-07-09:
I have currently commented out the tariff. So evcc is running.

VolkerK62, 2024-07-09:
seems to be part of https://github.com/evcc-io/evcc/issues/14496 ?

andig, 2024-07-09:
> I have currently commented out the tariff. So evcc is running.

Sure. But why would you want it to run if you have e.g. Tibber and nothing else? Isn't a working tariff a precondition? Closing as duplicate of https://github.com/evcc-io/evcc/issues/14496.

closed 2024-07-09 by andig

VolkerK62, 2024-07-09:
I think not to start is not consistent to https://github.com/evcc-io/evcc/pull/14341

>Ist weder Grid/PV/DynamischerTarif vorhanden gibte es nur Aus/Schnell

>Hintergrund: Sobald es ein Grid Meter gibt, hat evcc alles, um PV-Steuerung zu machen. Heißt, wenn du jetzt Strom in deinem Haus erzeugen würdest (Balkon PV in die Steckdose). Würde das alles schon klappen, ohne dass du eine PV konfigurierst. Daher gehen wir, aktuell, bei einem vorhandenen Grid-Meter immer davon aus, dass es auch Einspeisung geben kann.