evcc-io/evcc #14786: Add NrgKick 2nd Gen (new API Released)

Ahhpaddy, 2024-07-08:
**Is your feature request related to a problem? Please describe.**

- Add NrgKick 2nd Gen (new API Released)
https://www.nrgkick.com/wp-content/uploads/2024/07/local_api_docu_simulate.html

should now possible to add.

wobu, 2024-07-19:
i started an implementation, which is untested yet but i have a running NRGKick gen2 charger with an activated JSON api, so i could double check the used types, values and endpoints

Ahhpaddy, 2024-07-20:
Hello I can test, I habe 2 Nrgkick 's
But I don't know how to setup my evcc.yaml
Can you maybe explain how to do that?

I'm on v0.128.2

wobu, 2024-07-20:
Hi Ahhpaddy,

if u want to test the implementation, currently you need to compile the project locally with my feature branch, because my implementation hasn't been merged and released with official EVCC yet.

Also you need to have an activated JSON API in your nrgkicks.

Configuration in Evcc.yaml would look like

```yaml
chargers:
  - name: nrgkick-gen2
    type: template
    template: nrgkick-gen2
    host: <ip or host name of your nrgkick-gen2> 
```

Ahhpaddy, 2024-07-20:
I already activated the API to get some values into Iobroker. 

How can I compile locally? 

Can you pls give me short Instructions?

Thank you

wobu, 2024-07-20:
you have to check out my https://github.com/wobu/evcc/tree/feature/nrgkick-gen2-charger

then follow the instruction in CONTRIBUTING.md.

If you use Windows its gets bit complicated. I couldn't use a working Make installation for windows so i had to execute the build commandy manually.

this needs to be done once:
```shell
npm ci
go install github.com/dmarkham/enumer
go install go.uber.org/mock/mockgen
```
then for compiling evcc after every code change
```shell
npm run build
go build
```

then u will get an executable of `evcc` which u can run

closed 2024-08-11 by andig

JornHermans, 2024-08-17:
I used this config template and it works perfect, using the nightly build!
```
chargers:
  - name: NRGKickG2
    type: template
    template: nrggen2
    host: 192.168.1.23
``` 
    
Change the host to you're own ip 😉