evcc-io/evcc #14793: Problem mit Elli-Box (EEBUS)

pencoe, 2024-07-09:
### Describe the bug

Seit 0.128.0 habe ich Probleme mit der Elli-Box: im Modus PV wird immer wieder das Laden mit Minimalleistung (2,8kW) kurz gestartet und wieder abgeschaltet wenn zu wenig Überschuss da ist. Ich hatte z.B. gestern Abend den Wagen angeschlossen und PV eingeschaltet, damit er heute Vormittag anfängt, sobald genug Sonne da ist. Er hat aber über Nacht schon den Speicher leer gezogen und dann auch Stadtstrom verbraucht.

Ich habe jetzt zunächst die 0.127.3 wieder installiert, die arbeitet fehlerfrei.

Zur Konfig: die Fahrzeuge habe ich inzwischen über die Weboberfläche konfiguriert und nicht mehr in der evcc.yaml, ebenso das EEBus-Cert.

Wie man im Log sieht gibt es ständig "charger out of sync" und "charger logic" Fehler, das EEBus-Modul selbst logt aber keine Fehler. Wenn ich das mit einem anderen Loglevel nochmal nachstellen soll sagt Bescheid, kann ich dann aber erst morgen tun.

### Steps to reproduce

1. Fahrzeug an eine Ellibox anschließen
2. PV-Modus wählen bei zu wenig Überschuss

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
#log: trace
levels:
  #main: trace
  cache: error
  #mqtt: info
  eebus: info
  #modbus: info

interval: 30s # control cycle interval

sponsortoken: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

telemetry: true

meters:
- name: pv
  type: template
  template: kostal-plenticore
  usage: pv
  modbus: tcpip
  id: 71
  host: 192.168.48.109 # Hostname
  port: 1502 # Port
- name: grid
  type: custom
  power:
    source: http
    uri:  http://192.168.48.92/cm?cmnd=Status%208
    jq:   .StatusSNS.ENERGY.Leistung
- name: AkkuBYD
  type: template
  template: kostal-plenticore
  usage: battery
  modbus: tcpip
  id: 71
  host: 192.168.48.109 # Hostname
  port: 1502 # Port
- name: wb3em
  type: template
  template: shelly-pro-3em
  usage: charge
  host: 192.168.48.148
chargers:
- type: template
  template: ellipro
[... 77 lines cut ...]
Jul 08 22:04:41 RPNagel3 evcc[1124156]: [lp-1  ] WARN 2024/07/08 22:04:41 charger out of sync: expected disabled, got enabled
Jul 08 22:05:11 RPNagel3 evcc[1124156]: [lp-1  ] INFO 2024/07/08 22:05:11 stop charging <-
Jul 08 22:05:11 RPNagel3 evcc[1124156]: [lp-1  ] WARN 2024/07/08 22:05:11 charger out of sync: expected enabled, got disabled
Jul 08 22:09:10 RPNagel3 evcc[1124156]: [lp-1  ] INFO 2024/07/08 22:09:10 start charging ->
Jul 08 22:09:10 RPNagel3 evcc[1124156]: [lp-1  ] WARN 2024/07/08 22:09:10 charger out of sync: expected disabled, got enabled
Jul 08 22:09:41 RPNagel3 evcc[1124156]: [lp-1  ] INFO 2024/07/08 22:09:41 stop charging <-
Jul 08 22:09:54 RPNagel3 evcc[1124156]: [lp-1  ] WARN 2024/07/08 22:09:54 charger out of sync: expected enabled, got disabled
Jul 08 22:10:11 RPNagel3 evcc[1124156]: [lp-1  ] INFO 2024/07/08 22:10:11 start charging ->
Jul 08 22:10:11 RPNagel3 evcc[1124156]: [lp-1  ] WARN 2024/07/08 22:10:11 charger out of sync: expected disabled, got enabled
Jul 08 22:10:40 RPNagel3 evcc[1124156]: [lp-1  ] WARN 2024/07/08 22:10:40 charger logic error: current mismatch (got 0A, expected 6A)
Jul 08 22:14:11 RPNagel3 evcc[1124156]: [lp-1  ] INFO 2024/07/08 22:14:10 stop charging <-
Jul 08 22:15:10 RPNagel3 evcc[1124156]: [lp-1  ] INFO 2024/07/08 22:15:10 start charging ->
Jul 08 22:15:10 RPNagel3 evcc[1124156]: [lp-1  ] WARN 2024/07/08 22:15:10 charger out of sync: expected disabled, got enabled
Jul 08 22:15:41 RPNagel3 evcc[1124156]: [lp-1  ] WARN 2024/07/08 22:15:41 charger logic error: current mismatch (got 0A, expected 6A)
Jul 08 22:18:40 RPNagel3 evcc[1124156]: [lp-1  ] INFO 2024/07/08 22:18:40 stop charging <-
Jul 08 22:25:40 RPNagel3 evcc[1124156]: [lp-1  ] INFO 2024/07/08 22:25:40 start charging ->
```

### What type of operating system are you running?

Linux

### Version

0.128.1

andig, 2024-07-09:
Sufu benutzen- ist ge

closed 2024-07-09 by andig

pencoe, 2024-07-09:
Ups, wollte auch noch zwei Grafen aus Grafana und Homeassistant anhängen: hier...
[image]
[image]