evcc-io/evcc #14797: Add support for Zendure batteries and inverters

linuxstony, 2024-07-09:
Would the documentation of the API from zendure be enough to implement support for their products?

https://github.com/Zendure/developer-device-data-report

I am by far no developer but it seems to me that you can set the parameters via MQTT.

andig, 2024-07-09:
Seems you can do this yourself using the MQTT plugin. See documentation, PR for adding a template welcome!

closed 2024-07-09 by andig

linuxstony, 2024-07-09:
I have exactly and literally zero knowledge when it comes to coding. But I would give it a try when you could push me to a guide how to create such a template. Just to know where to start.

andig, 2024-07-09:
See https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/ac-elwa-2.yaml and other templates for examples. Use the `render` part of the template to test the `custom` meter you need to build. Wrapping that in a "template" is only the last step once the config works.

andig, 2024-07-09:
Siehe auch https://github.com/evcc-io/evcc/discussions/14794

Raudi1, 2024-07-15:
Ich habe es mittlerweile hinbekommen meinen Zendure Hyper 2000 zumindest grundlegend in EVCC einzubinden, da Zendure die Daten vom Hyper 2000 seit heute auch endlich per MQTT verteilt.
Habe die Anleitung https://github.com/z-master42/solarflow/wiki/Einbindung-in-Home-Assistant befolgt um die Zendure Daten in meinen vorhandenen Mosqitto Broker(HA Addon) zu bekommen. EVCC war schon mit dem Broker verbunden. Habe dann eine custom Batterie angelegt und sehe auch Entladeleistung und SOC in EVCC. Lediglich ist die Übernahme der Werte nur sporadisch und ich bekomme ständig die Fehlermeldung `battery 1 power: ja72U0ha/Ddh6svgk/state invalid: '<nil>'`
Liegt wahrscheinlich daran, dass der relevante Wert leider nur alle paar Nachrichten geschickt wird. Hier mal ein Auszug aus den logs:
```
[mqtt  ] TRACE 2024/07/15 20:21:08 recv ja72U0ha/Ddh6svgk/state: '{"wifiState":true,"electricLevel":75,"solarInputPower":0,"buzzerSwitch":false,"sn":"xxxx","socSet":1000}'
[lp-1  ] DEBUG 2024/07/15 20:21:09 charge power: 0W
[lp-1  ] DEBUG 2024/07/15 20:21:09 charge currents: [0 0 0]A
[mqtt  ] TRACE 2024/07/15 20:21:09 recv ja72U0ha/Ddh6svgk/state: '{"hubState":0,"remainInputTime":59940,"packState":2,"sn":"xxxxx","remainOutTime":281}'
[site  ] ERROR 2024/07/15 20:21:10 battery 1 power: ja72U0ha/Ddh6svgk/state invalid: '<nil>'
[mqtt  ] TRACE 2024/07/15 20:21:10 recv ja72U0ha/Ddh6svgk/state: '{"heatState":0,"sn":"xxxx"}'
[mqtt  ] TRACE 2024/07/15 20:21:11 recv ja72U0ha/Ddh6svgk/state: '{"hyperTmp":3131,"sn":"xxxx"}'
[mqtt  ] TRACE 2024/07/15 20:21:12 recv ja72U0ha/Ddh6svgk/state: '{"inverseMaxPower":600,"inputLimit":116,"outputLimit":30,"acMode":1,"sn":"xxxx"}'
[mqtt  ] TRACE 2024/07/15 20:21:14 recv ja72U0ha/Ddh6svgk/state: '{"outputHomePower":261,"packInputPower":270,"electricLevel":75,"sn":"xxxx"}'
[mqtt  ] TRACE 2024/07/15 20:21:17 recv ja72U0ha/Ddh6svgk/state: '{"hyperTmp":3141,"sn":"xxxx"}'
[lp-1  ] DEBUG 2024/07/15 20:21:19 charge power: 0W
[lp-1  ] DEBUG 2024/07/15 20:21:19 charge currents: [0 0 0]A
[site  ] ERROR 2024/07/15 20:21:20 battery 1 power: ja72U0ha/Ddh6svgk/state invalid: '<nil>'
```
hier der relevante Teil meiner config:
```
  - name: zendure
    type: custom
    power:
      source: mqtt
      topic: ja72U0ha/Ddh6svgk/state
      jq: .outputHomePower
    soc:
      source: mqtt
      topic: ja72U0ha/Ddh6svgk/state
      jq: .electricLevel
    capacity: 2
```

Leider bin ich mit meinen MQTT und JSON Kenntnissen am Ende. Ich weiß auch nicht mangels Sonne heute Abend, ob outputHomePower auch einen negativen Wert liefert, wenn die Batterie über AC geladen wird. Und wie das Ganze aussieht, wenn noch PV Panels an den MPPTs hängen, weiß ich auch noch nicht.

Raudi1, 2024-07-16:
Habs jetzt hinbekommen, dass zumindest beim AC laden das korrekt angezeigt wird. Ist bestimmt nicht optimal und es kommen auch weiterhin die Fehlermeldungen, da ja nicht in jeder MQTT Meldung alle Parameter drin sind, aber es läuft zumindest grundlegend:
```

- name: zendure
    type: custom
    power:
      source: mqtt
      topic: ja72U0ha/Ddh6svgk/state
      jq: .outputHomePower // if .gridInputPower > 0 then -.gridInputPower else null end
      timeout: 5m
    soc:
      source: mqtt
      topic: ja72U0ha/Ddh6svgk/state
      jq: .electricLevel?
      timeout: 30m
    capacity: 2
```

linuxstony, 2024-08-16:
Dieses Wochenende kommmt mein System, ich hoffe ich kann dann auch was beisteuern, dass wir den Hyper2000 in evcc komplett mit eingebunden bekommen. Danke @Raudi1 für deine Mühe schon bisher !!

Raudi1, 2024-08-16:
> Dieses Wochenende kommmt mein System, ich hoffe ich kann dann auch was beisteuern, dass wir den Hyper2000 in evcc komplett mit eingebunden bekommen. Danke @Raudi1 für deine Mühe schon bisher !!

Momentan funktioniert das Ganze leider noch nicht so gut. Durch die fehlenden Daten zeigt EVCC ständig Fehlermeldungen an und scheint auch teilweise etwas zu hängen usw.

Raudi1, 2024-08-24:
So, ich habe jetzt wohl ein Lösung gefunden, die ganz gut funktioniert:
Mit [diesem ioBroker Adapter](https://github.com/nograx/ioBroker.zendure-solarflow) lässt sich alles problemlos abrufen und sogar steuern! Dort gebe ich dann die Werte per MQTT weiter.
Als nächstes steht dann die Steuerung der Batterie an, wenn ich Zeit dafür habe. Ich kann allerdings in der Dokumentation keine Infos dazu finden, wie man das bei einem custom Akku macht.
Die Überschuss Funktion fürs AC laden vom Hyper 2000 ist leider etwas doof programmiert. Die lässt immer 150W ins Netz. Angeblich um das Relais zu schützen. Aber es führt natürlich dazu, dass bei unter 150W Überschuss das Relais schaltet. In Sofern ist lediglich der Punkt etwas verlagert, aber wirklich bringen tut es nichts. Und gerade bei einer Balkonsolar ist es natürlich schon ärgerlich so verhältnismäßig viel zu verschenken. Manuell lässt es sich sowohl beim laden, als auch beim entladen von 30W bis 1200W in 1W schritten steuern.

linuxstony, 2024-08-24:
Mist - jetzt bräuche es sowas für Homeassitant :) ioBroker nutze ich nicht :(

Raudi1, 2024-08-24:
@linuxstony Geht mir genauso. Es ist aber ziemlich simpel die relevanten Dinge von iobroker per MQTT weiter zu leiten. Dann kann man damit in Home Assistant auch arbeiten.
Ich bin auf dem globalen Server.

linuxstony, 2024-08-24:
Dann muss ich mein System nochmal in den Fabrikzustand packen und von vorne beginnen. 
Der EU-Server bring hier gar nichts an Daten.  Danke Dir.

Raudi1, 2024-08-24:
@linuxstony Bei mir scheint es jetzt wieder zu gehen. Sehe ich mir morgen nochmal an. Bin nicht mehr am Rechner.

linuxstony, 2024-09-05:
Bin jetzt auch auf dem Globalen Server. Wenigstens die Solarwerte werden anständig regelmäßig aktualisiert. 

Ich überlege jetzt die ganze Zeit (der MQTT-Support von Zendure wird ja offensichtlich nicht besser), die Werte stumpf über eine Messteckdose zu erfassen und über das "calc" plugin zu errechnen, ob halt Strom in den Akku geht oder nicht. 

Solar habe ich über 

- name: pv2
  type: custom
  power: # power (W)
    source: mqtt
    topic: DISnXQ5W/Q5mkHULJ/state 
    jq: .solarInputPower?
    timeout: 5s 

eingebunden bekommen. 

Nachdem ich deinen Battery-Part adaptiert habe, zeigt evcc kein Grid-Verbrauch mehr an - aber das bekomme ich schon noch hin. 

@andig hoffe du hast nichts dagegen, dass hier weiter geschrieben wird? Oder kannst/willst Du das in die Diskussionsecke verschieben?

Raudi1, 2024-09-05:
Ich nutze den [ioBroker plugin.](https://github.com/nograx/ioBroker.zendure-solarflow)
Der stellt dann die Werte per MQTT in vernünftiger Art bereit. Funktioniert super.

andig, 2024-10-18:
Gibts einen letzten Stand der SolarFlow Konfiguration ohne Nutzung externer Komponenten? Welche Probleme gabs damit now?

Raudi1, 2024-10-18:
> Gibts einen letzten Stand der SolarFlow Konfiguration ohne Nutzung externer Komponenten? Welche Probleme gabs damit now?

Der Zendure MQTT Server (Cloud) liefert seine Werte alle in einem State topic. Dort sind dann die Sensoren in einer JSON. Es werden aber immer nur ein paar von denen geliefert. Das sieht dann z.B. so aus:

```
18.10.2024 20:35:50(-11.28 seconds)
{"sn":"EE1XXXXXXXXXXXX","remainOutTime":180}

18.10.2024 20:35:45(-5.11 seconds)
{"sn":"EE1XXXXXXXXXXXX","remainOutTime":191}

18.10.2024 20:35:44(-0.99 seconds)
{"electricLevel":46,"packData":[{"socLevel":46," ,"sn": "CO4XXXXXXXXXXX"} ],"sn": "EE1XXXXXXXXXXXX"}

18.10.2024 20:35:40(-4.08 seconds)
{"sn":"EE1XXXXXXXXXXXX","remainOutTime":208}

18.10.2024 20:35:36(-3.21 seconds)
{"outputHomePower":268,"packInputPower":284,"electricLevel":47,"sn":"EE1XXXXXXXXXXXX"}

18.10.2024 20:35:34(-2.06 seconds)
{"sn":"EE1XXXXXXXXXXXX","remainOutTime":205}

...

18.10.2024 20:34:35(-6.14 seconds)
 { "outputHomePower": 286, "packInputPower": 303, "electricLevel": 47, "sn": "EE1XXXXXXXXXXXX", "remainOutTime": 201}
```

Die für uns relevanten Daten wie SOC(electricLevel) werden z.B. nur ca. 1x pro Minute gesendet. Das ist leider zumindest in meinen Augen viel zu langsam für eine vernünftige Steuerung.
Der ioBroker Adapter kann aber genau wie die APP auch Echtzeitwerte bekommen und auch alles steuern.

andig, 2024-10-19:
Danke. Weiss man wie der das macht?

Raudi1, 2024-10-19:
Ist open source: 
https://github.com/nograx/ioBroker.zendure-solarflow
Was ggfs. auch interessant sein könnte ist 
https://github.com/reinhard-brandstaedter/solarflow

andig, 2024-11-04:
> Die für uns relevanten Daten wie SOC(electricLevel) werden z.B. nur ca. 1x pro Minute gesendet. Das ist leider zumindest in meinen Augen viel zu langsam für eine vernünftige Steuerung.

Für soc reicht das völlig Leistung wäre eher grenzwertig. 

Habt ihr schon raus gefunden, ob/wie sich Leistungsaufnahme und Abgabe über MQTT steuern lassen? Ich sehe command_topics im Broker, aber die Nutzung ist unklar.

Raudi1, 2024-11-04:
Vielleicht hilft das hier weiter: https://github.com/reinhard-brandstaedter/solarflow-control/tree/master/doc

linuxstony, 2024-11-04:
> Für soc reicht das völlig Leistung wäre eher grenzwertig.
>
> Habt ihr schon raus gefunden, ob/wie sich Leistungsaufnahme und Abgabe über MQTT steuern lassen? Ich sehe command_topics im Broker, aber [...]

Leider noch nicht - ich hatte für dieses Side-Projekt das zu entkoppeln und zu recherchieren zu wenig Zeit :(

harrymayr, 2024-11-04:
> Habt ihr schon raus gefunden, ob/wie sich Leistungsaufnahme und Abgabe über MQTT steuern lassen? Ich sehe command_topics im Broker, aber die Nutzung ist unklar.

Ja, im IOBroker gibt es eine Section Control für den Adapter. Über den AC-Mode kann man Laden/Entladen steuern und über setInputLimit und setOutputLimit die entsprechende Leistung.
Ich habe es auch in Home-Assistant mit dem IOBroker nur für die Steuerung des Speichers implementiert.
[image]

andig, 2024-11-05:
Der Code erschliesst sich mir leider nicht eine Api Beschreibung wäre hilfreich.

linuxstony, 2025-04-03:
Moin - ich habe gerade gesehen, dass es eine Integration für Homeassistant gibt mit der auch die Steuerung des Hyper funktioniert:

https://github.com/fireson/zendure-ha

Es werden zwei Geräte Hinzugefügt - einmal der Account über den man entweder die Leistung manuell einstellen kann (mit positiven der negativen Werten) oder einen smart match modus, wo die Leistung dem Hausverbrauch angepasst wird. 

[image]

Und das zweite Gerät ist der Hyper mit allen möglichen Entitäten und Steuereinheiten. 

[image]

Ich denke nicht, dass das die offizielle Integration von Zendure ist - jedoch scheint die Steuerung auch jetzt schon zu funktionieren.
Eine lokale API wäre natürlich der Königsweg aber eine Steuerung durch EVCC sollte schon jetzt über die Cloud möglich sein. 

Ich wollte das hier nur einmal droppen als Option - nur für den Fall, dass die lokale API doch nicht kommt.

romestylez, 2025-06-03:
> So, ich habe jetzt wohl ein Lösung gefunden, die ganz gut funktioniert: Mit [diesem ioBroker Adapter](https://github.com/nograx/ioBroker.zendure-solarflow) lässt sich alles problemlos abrufen und [...]

Magst du mir ein Beispiel geben ? Ich nutze ebenfalls ioBroker mit dem Adapter (schon relativ lange) und würde nun gerne meine beiden Zendure HUB 20000 in EVCC einbinden. Vielleicht magst du ja deinen "Code" hier teilen :)

andig, 2025-06-03:
Meine Steuerung ist hier: https://github.com/andig/zendure. Sie läuft als autonomer Dockercontainer und bezieht die Messwerte aus evcc. Ziel ist Aussteuerung des Netzübergabepunktes. Funktioniert ganz gut. Gerne verwenden und anpassen, ohne jeden Support oder Funktionsgarantie.

andig, 2025-06-03:
Dazu kann man nichts sagen die dein Posting nicht formatiert ist. Yaml braucht Einrückungen- die sind nicht erkennbar.

romestylez, 2025-06-03:
Das ist aber nur für den Hyper nicht für den Hub2000 ;) Ja leider zerballert github die Formatierung

[image]

andig, 2025-06-03:
>  Ja leider zerballert github die Formatierung

Du musst es halt formatieren (als Code)- dafür gibts die Buttonleiste über dem Post.

harrymayr, 2025-06-03:
capacity: auf Höhe von soc:

[image]

chriuso, 2025-06-08:
>Andig
>Die Diskussion erübrigt sicher aber auch, denn die Unterstützung ist schon lange umgesetzt: https://docs.evcc.io/docs/devices/meters#zendure-hyper-2000

>romestylez
> Das ist aber nur für den Hyper nicht für den Hub2000 ;) Ja leider zerballert github die Formatierung

Möchte mir auch einen Zendure Speicher zulegen, bin aber verwirrt, was nun schon gut funktioniert und was nicht, und für welches Zendure Produkt. 

Ich möchte gerne EVCC als die **eine** Lösung zum Auslesen und Steuern der Batterie verwenden. EVCC läuft bei mir schon mit der vorhandenen Wallbox und PV-Anlage super. EVCC soll als die eine zentrale Instanz vorgeben, was wann geladen und entladen wird in Berücksichtigung der Dach-PV-Leistung und dem Hausverbrauch - kein homeAssistant / ioBroker oder anderes. Keep it simple. 

So wie ich es verstehe ist ein Zendure Hyper 2000 (Auslesen/Steuern) einfach einbinddbar?
https://docs.evcc.io/docs/devices/meters#zendure-hyper-2000
https://www.zendure.de/products/solarflow-hyper-2000?srsltid=AfmBOopp5LRi8-l1BmUeGqugIe3QHAfgWCqOEtr7CeVnMFHbOGKAmVt3

Warum aber dann hier die Diskussion zu den Hubs? Ist das diese Produkt?
https://www.zendure.de/products/solarflow-hub2000?srsltid=AfmBOoo5xj4vcdmSWjSQkQpB3veQWi43j5WyHKEumxfaC5OsruYK2lAn

Und: hat jemand schon EVCC-Erfahrung gesammelt mit dem neuen Solarflow 2400 AC?
https://www.zendure.de/products/zendure-solarflow-2400-ac?srsltid=AfmBOorLhngUXCclNPrCN6n6T4ve8H9n-lbSwVmz2dgSNQ4oYNLRu5YW

romestylez, 2025-06-08:
Der Hyper ist in evcc implementiert als Template. Ich habe meinen Hub2000 auch angebunden jetzt inkludive Batterien. Zwar per MQTT aber egal. Ich lese und steuer aber sowieso per ioBroker weil es da quasi unendliche Möglichkeiten gibt. evcc habe ich zum Testen installiert und schaue mal wie ich es final integriere.

harrymayr, 2025-06-08:
> Der Hyper ist in evcc implementiert als Template.

Das geht über die Zendure Cloud und da ist meines Wissens keine Steuerung möglich, nur lesen der Daten.

Insofern wird

> kein homeAssistant / ioBroker oder anderes. Keep it simple.

nicht funktionieren (bis jetzt).

chriuso, 2025-06-08:
OK - alles klar. Romestylez, Harrymayr, vielen Dank für die hilfreichen Erklärungen!
Werde mir mal den Marstek Venus C oder E ansehen (https://github.com/evcc-io/evcc/discussions/21037)