evcc-io/evcc #14801: Error Message max charge current 6A: 421 Misdirected Request: user out of region

Contrebass, 2024-07-09:
### Describe the bug

Trotz Sponsor Token und 3 minuten Refresh bekomme ich diese Fehlermeldunf wenn die Ladeleistung reduziert werden soll:
max charge current 6A: 421 Misdirected Request: user out of region, use base URL: https://fleet-api.prd.eu.vn.cloud.tesla.com, see https://developer.tesla.com/docs/fleet-api#regional-requirements

Die Installation regelt dann den Ladestrom über den Tesla Model S nicht runter.

Gleiches passiert wenn der min SoC von 50% erreicht ist, dann schaltet EVCC nicht ab und im log ist die gleiche Meldung

Kann es sein dass irgendwo noch die Geokoordinaten angegeben werden müssen?
Sponsortoken ist vorhanden.
Hardware RASPI 4 bullseye
Tesla Powerwall 2
TWC3 latest firmware

Mite dem Fehler erscheint in der UI ein Dreieck mit Ausrufezeichen neben dem Hamburge (Drei Striche)

### Steps to reproduce

1. Einstellen min charge <6A
2. Abwarten bis Wolke vor der Sonne und PV Leistung zu klein
3. Einstellen min SoC 50%
4. Min+ PV als Setup
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxxxx

interval: 180s # control cycle interval

sponsortoken: xxxxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: tesla-powerwall 
  usage: grid  
  host: 10.200.10.3  
  password: password
  name: grid1
- type: template
  template: tesla-powerwall 
  usage: pv  
  host: 10.200.10.3  
  password: password
  name: pv2
- type: template
  template: tesla-powerwall 
  usage: battery  
  host: 10.200.10.3  
  password: password
  name: battery3
- type: template
  template: eastron
  usage: charge
  modbus: rs485serial
  id: 1
  device: /dev/ttyUSB1
  baudrate: 9600
  comset: "8N1"
  name: charge1

chargers:
- type: template
  template: twc3 
  host: 192.168.101.58  
  name: wallbox7

vehicles:
- type: template
  template: tesla 
  title: Polarfuchs  
  accessToken: xxxxxxx 
  vin: 5YJSA7E40Kxxxxxxx
  capacity: 95  
  name: ev6

loadpoints:
- title: Garage
  charger: wallbox7
  mode: minpv
  meter: charge1

site:
  title: Site
  meters:
    grid: grid1
    pv: pv2
    battery:
    - battery3

tariffs:
   currency: EUR
   grid:
     type: fixed
     price: 0.3664
   feedin:
     type: fixed
     price: 0.1195
   co2:
     type: grünstromindex
     zip: zip
```

### Log details

```text
[lp-1  ] INFO 2024/07/09 16:25:06 vehicle updated: unknown -> Polarfuchs
[lp-1  ] DEBUG 2024/07/09 16:25:06 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[site  ] DEBUG 2024/07/09 16:26:30 ----
[lp-1  ] DEBUG 2024/07/09 16:26:30 charge power: 5397W
[lp-1  ] DEBUG 2024/07/09 16:26:30 charge currents: [7.93 7.89 7.86]A
[site  ] DEBUG 2024/07/09 16:26:31 pv power: 2339W
[site  ] DEBUG 2024/07/09 16:26:31 battery soc: 100%
[site  ] DEBUG 2024/07/09 16:26:31 battery power: 3480W
[site  ] DEBUG 2024/07/09 16:26:31 grid meter: -10W
[site  ] DEBUG 2024/07/09 16:26:31 site power: 3570W
[lp-1  ] DEBUG 2024/07/09 16:26:33 vehicle odometer: 117065km
[lp-1  ] DEBUG 2024/07/09 16:26:33 charge voltages: [231 228 228]V
[lp-1  ] DEBUG 2024/07/09 16:26:33 detected connected phases: 3p
[lp-1  ] DEBUG 2024/07/09 16:26:33 charge total import: 93.568kWh
[lp-1  ] DEBUG 2024/07/09 16:26:33 charger status: C
[lp-1  ] DEBUG 2024/07/09 16:26:33 charge start energy: 93.6kWh
[lp-1  ] INFO 2024/07/09 16:26:33 car connected
[lp-1  ] DEBUG 2024/07/09 16:26:33 vehicle api refresh
[lp-1  ] DEBUG 2024/07/09 16:26:33 pv timer elapse
[lp-1  ] DEBUG 2024/07/09 16:26:33 pv timer inactive
[lp-1  ] DEBUG 2024/07/09 16:26:33 charge total import: 93.568kWh
[lp-1  ] DEBUG 2024/07/09 16:26:33 charge start energy: 93.6kWh
[lp-1  ] INFO 2024/07/09 16:26:33 start charging ->
[lp-1  ] DEBUG 2024/07/09 16:26:33 wake-up timer: stop
[lp-1  ] DEBUG 2024/07/09 16:26:33 vehicle status: C
[lp-1  ] DEBUG 2024/07/09 16:26:33 vehicle soc: 73%
[lp-1  ] DEBUG 2024/07/09 16:26:33 vehicle soc limit: 90%
[lp-1  ] DEBUG 2024/07/09 16:26:33 vehicle range: 422km
[lp-1  ] WARN 2024/07/09 16:26:33 charger out of sync: expected disabled, got enabled
[lp-1  ] DEBUG 2024/07/09 16:26:33 charger enable
[lp-1  ] DEBUG 2024/07/09 16:26:33 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/07/09 16:26:33 pv charge current: min 1A > 0A (3570W @ 3p, battery: true)
[lp-1  ] ERROR 2024/07/09 16:26:36 max charge current 1A: 421 Misdirected Request: user out of region, use base URL: https://fleet-api.prd.eu.vn.cloud.tesla.com, see https://developer.tesla.com/docs/fleet-api#regional-requirements
[lp-1  ] INFO 2024/07/09 16:29:12 vehicle updated: Polarfuchs -> unknown
[lp-1  ] DEBUG 2024/07/09 16:29:12 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] INFO 2024/07/09 16:29:15 vehicle updated: unknown -> Polarfuchs
[lp-1  ] DEBUG 2024/07/09 16:29:15 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[site  ] DEBUG 2024/07/09 16:29:30 ----
[lp-1  ] DEBUG 2024/07/09 16:29:30 charge power: 5411W
[lp-1  ] DEBUG 2024/07/09 16:29:30 charge currents: [7.86 7.95 7.92]A
[site  ] DEBUG 2024/07/09 16:29:30 pv power: 2322W
[site  ] DEBUG 2024/07/09 16:29:31 battery soc: 97%
[site  ] DEBUG 2024/07/09 16:29:31 battery power: 3440W
[site  ] DEBUG 2024/07/09 16:29:31 grid meter: 4W
[site  ] DEBUG 2024/07/09 16:29:31 site power: 3544W
[lp-1  ] DEBUG 2024/07/09 16:29:31 vehicle odometer: 117065km
[lp-1  ] DEBUG 2024/07/09 16:29:31 charge voltages: [232 227 227]V
[lp-1  ] DEBUG 2024/07/09 16:29:31 detected connected phases: 3p
[lp-1  ] DEBUG 2024/07/09 16:29:31 detected active phases: 3p
[lp-1  ] DEBUG 2024/07/09 16:29:31 !! session: chargeRater.chargedEnergy=0.3 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/07/09 16:29:31 charge total import: 93.835kWh
[lp-1  ] DEBUG 2024/07/09 16:29:31 charger status: C
[lp-1  ] DEBUG 2024/07/09 16:29:31 vehicle status: C
[lp-1  ] DEBUG 2024/07/09 16:29:31 vehicle soc: 73%
[lp-1  ] DEBUG 2024/07/09 16:29:31 vehicle soc limit: 90%
[lp-1  ] DEBUG 2024/07/09 16:29:31 vehicle range: 422km
[lp-1  ] WARN 2024/07/09 16:29:31 charger logic error: current mismatch (got 7.95A measured, expected 0A)
[lp-1  ] DEBUG 2024/07/09 16:29:31 !! active phases: 3p = min(3p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/07/09 16:29:31 pv charge current: 2.82A = 7.95A + -5.14A (3544W @ 3p)
[lp-1  ] DEBUG 2024/07/09 16:29:31 pv timer reset
[lp-1  ] DEBUG 2024/07/09 16:29:31 pv timer inactive
[lp-1  ] ERROR 2024/07/09 16:29:32 max charge current 2A: 421 Misdirected Request: user out of region, use base URL: https://fleet-api.prd.eu.vn.cloud.tesla.com, see https://developer.tesla.com/docs/fleet-api#regional-requirements
```

### What type of operating system are you running?

Linux

### Version

0.128.1

andig, 2024-07-09:
Es braucht ein trace log für tesla.

Contrebass, 2024-07-10:
Trace log (Ausschnitt)
[tesla ] TRACE 2024/07/10 18:59:06 GET https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/users/region
[tesla ] TRACE 2024/07/10 18:59:07 {"response":{"region":"eu","fleet_api_base_url":"https://fleet-api.prd.eu.vn.cloud.tesla.com"}}
[tesla ] TRACE 2024/07/10 18:59:07 GET https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles
[tesla ] TRACE 2024/07/10 18:59:07 {"response":[{"id":xxxxxxxxxxx,"vehicle_id":xxxxxxxxxx,"vin":"5YJSA7Exxxxxxxxxxx,"color":null,"access_type":"DRIVER","display_name":"Polarfuchs","option_codes":null,"granular_access":{"hide_private":false},"tokens":null,"state":"online","in_service":false,"id_s":"xxxxxxxxxxxx","calendar_enabled":true,"api_version":76,"backseat_token":null,"backseat_token_updated_at":null,"share_type_s":"ADD_DRIVER_EXTERNAL","ble_autopair_enrolled":false}],"pagination":{"previous":null,"next":null,"current":1,"per_page":1,"count":1,"pages":1},"count":1}
[tesla ] TRACE 2024/07/10 18:59:10 GET https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles/5YJSA7E40xxxxxxxx/vehicle_data
[tesla ] TRACE 2024/07/10 18:59:11 {"response":{"langer Text mit Details}
[tesla ] TRACE 2024/07/10 18:59:11 POST https://tesla.evcc.io/api/1/vehicles/5YJSA7E40xxxxxxxx/command/set_charging_amps
[tesla ] TRACE 2024/07/10 18:59:12 {"charging_amps": 1}
--
{"response":null,"error":"user out of region, use base URL: https://fleet-api.prd.eu.vn.cloud.tesla.com, see https://developer.tesla.com/docs/fleet-api#regional-requirements","error_description":"","txid":"xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"}
[tesla ] TRACE 2024/07/10 19:02:09 POST https://tesla.evcc.io/api/1/vehicles/5YJSA7E40xxxxxxxxx/command/set_charging_amps
[tesla ] TRACE 2024/07/10 19:02:15 {"charging_amps": 1}
--
{"response":null,"error":"user out of region, use base URL: https://fleet-api.prd.eu.vn.cloud.tesla.com, see https://developer.tesla.com/docs/fleet-api#regional-requirements","error_description":"","txid":"xxxxxxxxxxxxxxxxxxxxx"}

Ich havbe um schnell einen Fehler zu bekommen den Ladestrom auf 1A gesetzt.
Der Fehler kommt aber auch bei 6A

Contrebass, 2024-07-10:
Der dazugehörige log, evcc Version nun die 0.182.2

[main ] INFO 2024/07/10 18:59:04 evcc 0.128.2
[main ] INFO 2024/07/10 18:59:04 using config file: /etc/evcc.yaml
[db ] INFO 2024/07/10 18:59:05 using sqlite database: /var/lib/evcc/evcc.db
[main ] INFO 2024/07/10 18:59:05 listening at :7070
[lp-1 ] DEBUG 2024/07/10 18:59:08 charge total import: 99.900kWh
[site ] WARN 2024/07/10 18:59:08 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site ] INFO 2024/07/10 18:59:08 site config:
[site ] INFO 2024/07/10 18:59:08 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2024/07/10 18:59:08 grid: power ✓ energy ✗ currents ✗
[site ] INFO 2024/07/10 18:59:08 pv 1: power ✓ energy ✓ currents ✗
[site ] INFO 2024/07/10 18:59:08 battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site ] INFO 2024/07/10 18:59:08 vehicles:
[site ] INFO 2024/07/10 18:59:08 vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1 ] INFO 2024/07/10 18:59:08 loadpoint 1:
[lp-1 ] INFO 2024/07/10 18:59:08 mode: minpv
[lp-1 ] INFO 2024/07/10 18:59:08 charger: power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1 ] INFO 2024/07/10 18:59:08 meters: charge ✓
[lp-1 ] INFO 2024/07/10 18:59:08 charge: power ✓ energy ✓ currents ✓
[site ] DEBUG 2024/07/10 18:59:08 set buffer soc: 50
[site ] DEBUG 2024/07/10 18:59:08 set buffer start soc: 80
[site ] DEBUG 2024/07/10 18:59:08 set priority soc: 50
[site ] DEBUG 2024/07/10 18:59:08 set residual power: 100
[lp-1 ] DEBUG 2024/07/10 18:59:08 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/07/10 18:59:08 phase timer inactive
[lp-1 ] DEBUG 2024/07/10 18:59:08 pv timer inactive
[site ] DEBUG 2024/07/10 18:59:08 ----
[lp-1 ] DEBUG 2024/07/10 18:59:08 charge power: 0W
[lp-1 ] DEBUG 2024/07/10 18:59:08 charge currents: [0 0 0]A
[site ] WARN 2024/07/10 18:59:08 smartCostActive: no matching rate
[site ] DEBUG 2024/07/10 18:59:09 pv power: 269W
[site ] DEBUG 2024/07/10 18:59:09 battery soc: 100%
[site ] DEBUG 2024/07/10 18:59:09 battery power: 190W
[site ] DEBUG 2024/07/10 18:59:10 grid meter: -1W
[site ] DEBUG 2024/07/10 18:59:10 site power: 289W
[lp-1 ] DEBUG 2024/07/10 18:59:10 charge voltages: [226 226 226]V
[lp-1 ] DEBUG 2024/07/10 18:59:10 detected connected phases: 3p
[lp-1 ] DEBUG 2024/07/10 18:59:10 charge total import: 99.900kWh
[lp-1 ] DEBUG 2024/07/10 18:59:10 charger status: B
[lp-1 ] DEBUG 2024/07/10 18:59:10 charge start energy: 99.9kWh
[lp-1 ] INFO 2024/07/10 18:59:10 car connected
[lp-1 ] DEBUG 2024/07/10 18:59:10 vehicle api refresh
[lp-1 ] DEBUG 2024/07/10 18:59:10 pv timer elapse
[lp-1 ] DEBUG 2024/07/10 18:59:10 pv timer inactive
[lp-1 ] DEBUG 2024/07/10 18:59:10 charge total import: 99.900kWh
[main ] DEBUG 2024/07/10 18:59:11 vehicle status: B (Polarfuchs)
[lp-1 ] INFO 2024/07/10 18:59:11 vehicle updated: unknown -> Polarfuchs
[lp-1 ] DEBUG 2024/07/10 18:59:11 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/07/10 18:59:11 vehicle status: B
[lp-1 ] DEBUG 2024/07/10 18:59:11 vehicle soc: 75%
[lp-1 ] DEBUG 2024/07/10 18:59:11 vehicle soc limit: 90%
[lp-1 ] DEBUG 2024/07/10 18:59:11 vehicle range: 434km
[lp-1 ] DEBUG 2024/07/10 18:59:11 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/07/10 18:59:11 pv charge current: min 1A > 0A (289W @ 3p, battery: true)
[lp-1 ] ERROR 2024/07/10 18:59:12 max charge current 1A: 421 Misdirected Request: user out of region, use base URL: https://fleet-api.prd.eu.vn.cloud.tesla.com, see https://developer.tesla.com/docs/fleet-api#regional-requirements
[site ] DEBUG 2024/07/10 19:02:08 ----
[lp-1 ] DEBUG 2024/07/10 19:02:08 charge power: 3892W
[lp-1 ] DEBUG 2024/07/10 19:02:09 charge currents: [7.87 7.92 1.6]A
[site ] DEBUG 2024/07/10 19:02:09 pv power: 166W
[site ] DEBUG 2024/07/10 19:02:09 battery soc: 100%
[site ] DEBUG 2024/07/10 19:02:09 battery power: 4090W
[site ] DEBUG 2024/07/10 19:02:09 grid meter: -14W
[site ] DEBUG 2024/07/10 19:02:09 site power: 4176W
[lp-1 ] DEBUG 2024/07/10 19:02:09 vehicle odometer: 117123km
[lp-1 ] DEBUG 2024/07/10 19:02:09 charge voltages: [228 225 226]V
[lp-1 ] DEBUG 2024/07/10 19:02:09 detected connected phases: 3p
[lp-1 ] DEBUG 2024/07/10 19:02:09 charge total import: 99.905kWh
[lp-1 ] DEBUG 2024/07/10 19:02:09 charger status: C
[lp-1 ] DEBUG 2024/07/10 19:02:09 charge start energy: 99.9kWh
[... 6 lines cut ...]
[lp-1 ] WARN 2024/07/10 19:02:09 charger out of sync: expected disabled, got enabled
[lp-1 ] DEBUG 2024/07/10 19:02:09 charger enable
[lp-1 ] DEBUG 2024/07/10 19:02:09 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/07/10 19:02:09 pv charge current: min 1A > 0A (4176W @ 3p, battery: true)
[lp-1 ] ERROR 2024/07/10 19:02:15 max charge current 1A: 421 Misdirected Request: user out of region, use base URL: https://fleet-api.prd.eu.vn.cloud.tesla.com, see https://developer.tesla.com/docs/fleet-api#regional-requirements
[site ] DEBUG 2024/07/10 19:05:08 ----
[lp-1 ] DEBUG 2024/07/10 19:05:08 charge power: 0W
[lp-1 ] DEBUG 2024/07/10 19:05:08 charge currents: [0 0 0]A
[site ] DEBUG 2024/07/10 19:05:09 pv power: 152W
[site ] DEBUG 2024/07/10 19:05:09 battery soc: 100%
[site ] DEBUG 2024/07/10 19:05:09 battery power: 140W
[site ] DEBUG 2024/07/10 19:05:09 grid meter: 10W
[site ] DEBUG 2024/07/10 19:05:09 site power: 250W
[... cut ...]
[lp-1 ] DEBUG 2024/07/10 19:05:09 final charge energy: 0kWh
[lp-1 ] INFO 2024/07/10 19:05:09 stop charging <-
[lp-1 ] DEBUG 2024/07/10 19:05:09 wake-up timer: start
[lp-1 ] DEBUG 2024/07/10 19:05:09 charge total import: 99.905kWh
[lp-1 ] DEBUG 2024/07/10 19:05:09 !! session: start=99.900 stop=99.905 chargedEnergy=0.000
[lp-1 ] INFO 2024/07/10 19:05:09 car disconnected
[lp-1 ] DEBUG 2024/07/10 19:05:09 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] INFO 2024/07/10 19:05:09 vehicle updated: Polarfuchs -> unknown

andig, 2024-07-11:
> Es braucht bitte ein **trace** log für tesla.

Bitte vollständig ab Start.

Contrebass, 2024-07-11:
Hallo Andreas,den hatte ich nur für "tesla" herunterladen und eingehängt (warum auch immer in der komischen Schriftart), nur die persönlichen Daten (...Details...) habe ich rausgenommen. Darunter war noch mal der aktualisierte Debug Log.Ich kann gerne den vollständigen trace dir persönlich schicken, aber nicht Public.VGChristian Dettmers

Contrebass, 2024-07-11:
```
Nun mit der V 0.128.2.
 Anmerkung trace tesla (weiter unten noch mal debug log wegen der Zeitstempel). 
tesla trace
[tesla ] TRACE 2024/07/11 17:06:15 GET https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/users/region
[tesla ] TRACE 2024/07/11 17:06:15 {"response":{"region":"eu","fleet_api_base_url":"https://fleet-api.prd.eu.vn.cloud.tesla.com"}}
[tesla ] TRACE 2024/07/11 17:06:15 GET https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles
[tesla ] TRACE 2024/07/11 17:06:16 {"response":[{"id":Tela_ID,"Tela_ID":157660611,"vin":"Tesla_VIN","color":null,"access_type":"DRIVER","display_name":"Polarfuchs","option_codes":null,"granular_access":{"hide_private":false},"tokens":null,"state":"online","in_service":false,"id_s":"Tela_ID","calendar_enabled":true,"api_version":76,"backseat_token":null,"backseat_token_updated_at":null,"share_type_s":"ADD_DRIVER_EXTERNAL","ble_autopair_enrolled":false}],"pagination":{"previous":null,"next":null,"current":1,"per_page":1,"count":1,"pages":1},"count":1}
[tesla ] TRACE 2024/07/11 17:06:19 GET https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles/Tesla_VIN/vehicle_data
[tesla ] TRACE 2024/07/11 17:06:19 {"response":{"id":Tela_ID,"user_id":1689271057669599,"Tela_ID":157660611,"vin":"Tesla_VIN","color":null,"access_type":"DRIVER","granular_access":{"hide_private":false},"tokens":null,"state":"online","in_service":false,"id_s":"Tela_ID","calendar_enabled":true,"api_version":76,"backseat_token":null,"backseat_token_updated_at":null,"share_type_s":"ADD_DRIVER_EXTERNAL","ble_autopair_enrolled":false,"charge_state":{"battery_heater_on":false,"battery_level":80,"battery_range":287.81,"charge_amps":7,"charge_current_request":7,"charge_current_request_max":9,"charge_enable_request":false,"charge_energy_added":1.37,"charge_limit_soc":90,"charge_limit_soc_max":100,"charge_limit_soc_min":50,"charge_limit_soc_std":90,"charge_miles_added_ideal":1368.0,"charge_miles_added_rated":5.0,"charge_port_cold_weather_mode":null,"charge_port_color":"Off","charge_port_door_open":true,"charge_port_latch":"Engaged","charge_rate":0.0,"charger_actual_current":0,"charger_phases":null,"charger_pilot_current":9,"charger_power":0,"charger_voltage":0,"charging_state":"Stopped","conn_charge_cable":"IEC","est_battery_range":250.56,"fast_charger_brand":"Tesla","fast_charger_present":false,"fast_charger_type":"ACSingleWireCAN","ideal_battery_range":999.0,"max_range_charge_counter":0,"minutes_to_full_charge":150,"not_enough_power_to_heat":false,"off_peak_charging_enabled":false,"off_peak_charging_times":"all_week","off_peak_hours_end_time":480,"preconditioning_enabled":false,"preconditioning_times":"all_week","scheduled_charging_mode":"Off","scheduled_charging_pending":false,"scheduled_charging_start_time":null,"scheduled_charging_start_time_app":990,"scheduled_departure_time":1720762200,"scheduled_departure_time_minutes":450,"supercharger_session_trip_planner":false,"time_to_full_charge":2.5,"timestamp":1720710379580,"trip_charging":false,"usable_battery_level":80,"user_charge_enable_request":false},"climate_state":{"allow_cabin_overheat_protection":false,"battery_heater":false,"battery_heater_no_power":false,"bioweapon_mode":false,"cabin_overheat_protection":"Off","climate_keeper_mode":"off","cop_activation_temperature":"High","defrost_mode":0,"driver_temp_setting":22.0,"fan_status":0,"hvac_auto_request":"On","inside_temp":37.5,"is_auto_conditioning_on":false,"is_climate_on":false,"is_front_defroster_on":false,"is_preconditioning":false,"is_rear_defroster_on":false,"left_temp_direction":-235,"max_avail_temp":28.0,"min_avail_temp":15.0,"outside_temp":30.2,"passenger_temp_setting":22.0,"remote_heater_control_enabled":false,"right_temp_direction":-235,"seat_fan_front_left":0,"seat_fan_front_right":0,"seat_heater_left":0,"seat_heater_rear_center":0,"seat_heater_rear_left":0,"seat_heater_rear_right":0,"seat_heater_right":0,"side_mirror_heaters":false,"steering_wheel_heater":false,"supports_fan_only_cabin_overheat_protection":false,"timestamp":1720710379580,"wiper_blade_heater":false},"drive_state":{"power":0,"shift_state":"P","speed":null,"timestamp":1720710379580},"gui_settings":{"gui_24_hour_time":true,"gui_charge_rate_units":"kW","gui_distance_units":"km/hr","gui_range_display":"Rated","gui_temperature_units":"C","gui_tirepressure_units":"Bar","show_range_units":false,"timestamp":1720710379580},"parked_accessory":{"can_accept_navigation_requests":true,"can_actuate_trunks":true,"car_special_type":"base","car_type":"models2","charge_port_type":"EU","cop_user_set_temp_supported":false,"dashcam_clip_save_supported":true,"default_charge_to_max":false,"driver_assist":"TeslaAP3","ece_restrictions":true,"efficiency_package":"S2","eu_vehicle":true,"exterior_color":"color","exterior_trim_override":"","front_drive_unit":"PermanentMagnet","has_air_suspension":true,"has_ludicrous_mode":true,"has_seat_cooling":true,"headlamp_type":"Led","interior_trim_type":"Color","motorized_charge_port":true,"paint_color_override":"","plg":true,"pws":false,"rear_drive_unit":"Large","rear_seat_heaters":1,"rear_seat_type":0,"rhd":false,"roof_color":"Glass","seat_type":2,"sentry_preview_supported":false,"spoiler_type":"Passive","sun_roof_installed":0,"supports_qr_pairing":false,"third_row_seats":"None","timestamp":1720710379580,"trim_badging":"p100d","use_range_badging":true,"utc_offset":7200,"webcam_selfie_supported":false,"webcam_supported":false,"wheel_type":"Arachnid21Silver"},"vehicle_config":{"can_accept_navigation_reques [... cut ...]
[site  ] DEBUG 2024/07/11 17:33:17 battery power: -590W
[site  ] DEBUG 2024/07/11 17:33:17 grid meter: -11W
[site  ] DEBUG 2024/07/11 17:33:17 site power: -501W
[lp-1  ] DEBUG 2024/07/11 17:33:17 charge voltages: [229 230 230]V
[lp-1  ] DEBUG 2024/07/11 17:33:17 detected connected phases: 3p
[lp-1  ] DEBUG 2024/07/11 17:33:17 !! session: chargeRater.chargedEnergy=1.5 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/07/11 17:33:17 charge total import: 105.205kWh
[lp-1  ] DEBUG 2024/07/11 17:33:18 charger status: A

```

andig, 2024-07-12:
Ich habe Service und Teslabibliothek dafür aktualisiert- bitte nochmal probieren. Wenn das nicht geht bitte Zugangsdaten für Test an info@evcc.io.

Contrebass, 2024-07-13:
Es geht leider nicht, Rest per Mail

Contrebass, 2024-07-13:
Nach etwas stöbern habe ich die Ursache für die "Kommunikationsfehler" gefunden:
Die evcc Instanz muss mit dem Token des Hauptaccounts bei Tesla laufen (siehe hier: https://github.com/evcc-io/evcc/discussions/11851).
Wäre was für die FAQ, oder?

Also - kein Bug.

closed 2024-07-14 by andig

Contrebass, 2024-07-14:
Hallo Andi, danke fürs Schließen.

PS.: Frage eines Newbies: was ist „Gerne PR fürs Tesla Template“ 

 

Closed #14801 <https://github.com/evcc-io/evcc/issues/14801>  as completed.

andig, 2024-07-14:
Die gewünschte Änderung müsste hier rein:https://github.com/evcc-io/evcc/blob/master/templates/definition/vehicle/tesla.yaml

Viele Grüße, Andreas