evcc-io/evcc #14803: Add  possibility to select 2 phases in evcc (instead of only 1 or 3)

Hermi87, 2024-07-09:
I wish to have the possibility to be able to select 2 phases in evcc. My system (and many others) works with 2 phases connected.
I have a Heidelberg (Amperified) Energy Control (Heidelberg works with only 2 instead of 3 phases connected).
My Tesla Model Y also works with 2 phases without any problems.
I have Photovoltaik power of 7 kW at home (former 10kWp limit still reduced to 70% like many others)
To be able to charge only with pv surplus power the selection of 2 phases in evcc would be perfect (in 16A 2 phases is up to 7.4 kW)
Unfortunately at the moment only 1 or 3 phases is possible to select, so calculation can not be correct unfortunately.
Thank you very much in advance.

premultiply, 2024-07-09:
IEC 62196 is only defined to support 1 or 3 phases at the connector.

closed 2024-07-09 by premultiply

typxxi, 2024-09-13:
> Unfortunately at the moment only 1 or 3 phases is possible to select, so calculation can not be correct unfortunately.

well, we face the same issue here even though not really and not right now due to enough surplus or too low EV consumption.

But our solution is a different one , not perfect, but better till someone finds whatever bypass here or somewhere else cause the demand for a better solution than the easy 1 phase or 3 phase is there .

We cut off the 2nd phase by a WiFi Switch. Our Pulsar Plus can then charge from what is available: 1 phase or 2 phases
The decision about surplus from 1 or 2 phases is driven by Home Assistant and manually which means the car is disconnected in the moment of that switch. Not the best solution but the Pulsar Plus itself can turn on/off phases for charging.

Sad but true, at the end it is a workaround for the evening or early morning.

Hermi87, 2024-09-13:
[image]

Thanks a lot for feedback. I understand, but to be able to calculate the loading current / power correctly from the surplus it is necessary to select either 1 or 3 phases. So how do you handle that? Otherwise the surplus --> loading current calculationbis not correct.