evcc-io/evcc #14804: Ocpp: localstart without localstop

andig, 2024-07-09:
Is there any way of having some middle ground between the default behaviour (transaction per charging session) and the `autoStart` behaviour in #14733 ? My EVSE (ZJ Beny) will not auto start, and if an RFID card is used to start the transaction it will then ignore OCPP altogether and charge at full rate. It does need to keep a single transaction, as if one is stopped it will not allow a new one to start until the car is physically disconnected.

_Originally posted by @ross-w in https://github.com/evcc-io/evcc/issues/12861#issuecomment-2216316607_

andig, 2024-07-09:
I just have to cry when I read about the plethora of ocpp quirks. What we could do:
- add `nostop`
- `localstart` implies `nostop` and is equivalent to not using `remotestart` or `remotestop`
- `nostop` standalone is equivalant to using `remotestart`, but not `remotestop`

andig, 2024-07-10:
@ross-w could you give https://github.com/evcc-io/evcc/pull/14824 a try? Totally untested.

ross-w, 2024-07-11:
@andig thanks heaps for this.

Am I missing something here? I am building from the branch `feat/ocpp-nostop`, however if I specify my charger like this:
```
chargers:
  - name: zjbeny
    type: template
    template: ocpp
    stationid: zjb
    getconfiguration: false
    timeout: 2m
    nostop: true
```
I get the following error on startup:
```
[main  ] FATAL 2024/07/11 20:13:51 cannot create charger 'zjbeny': cannot create charger type 'template': invalid key: nostop
```

Build is
```
[main  ] INFO 2024/07/11 20:14:35 evcc 0.128.2 (7e99a843)
```

ross-w, 2024-07-11:
Ah, I worked it out. I needed to add
```
  - name: autostart
    advanced: true
    type: bool
```
to templates/definition/charger/ocpp.yaml

Unfortunately with the above configuration it will still stop the transaction when charging is set to off:
```
[ocpp  ] TRACE 2024/07/11 20:39:24 recv zjb: [2,"ynb2s8y895n3fyi9s6ibpdy8","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Finishing","timestamp":"2024-07-11T10:39:23Z"}]
[ocpp  ] TRACE 2024/07/11 20:39:24 sent zjb: [3,"ynb2s8y895n3fyi9s6ibpdy8",{}]
[ocpp  ] TRACE 2024/07/11 20:39:25 recv zjb: [2,"bue28hyt0lkni5k4uaeiv108","StopTransaction",{"idTag":"evcc","meterStop":519258,"timestamp":"2024-07-11T10:39:24Z","transactionId":1,"reason":"Remote"}]
[ocpp  ] TRACE 2024/07/11 20:39:25 sent zjb: [3,"bue28hyt0lkni5k4uaeiv108",{"idTagInfo":{"status":"Accepted"}}]
```

andig, 2024-07-11:
> it will still stop the transaction

Who is "it"? What happens before?

ross-w, 2024-07-11:
Here's the full OCPP log
```
[ocpp  ] DEBUG 2024/07/11 20:38:40 charge point connected: zjb
[ocpp  ] TRACE 2024/07/11 20:38:40 recv zjb: [2,"oy9fg46tv6j57jzezzmpeg42","BootNotification",{"chargePointModel":"BCP-B2N-L","chargePointVendor":"ZJBENY","firmwareVersion":"1.0.23"}]
[ocpp  ] TRACE 2024/07/11 20:38:40 sent zjb: [3,"oy9fg46tv6j57jzezzmpeg42",{"currentTime":"2024-07-11T10:38:40Z","interval":60,"status":"Accepted"}]
[ocpp  ] TRACE 2024/07/11 20:38:41 recv zjb: [2,"9ts51elye5x0qooofix0pf13","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available","timestamp":"2024-07-11T10:38:40Z"}]
[ocpp  ] TRACE 2024/07/11 20:38:41 recv zjb: [2,"13y1rw4x2en6uusvl2t06242","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Preparing","timestamp":"2024-07-11T10:38:40Z"}]
[ocpp  ] TRACE 2024/07/11 20:38:41 sent zjb: [3,"13y1rw4x2en6uusvl2t06242",{}]
[ocpp  ] TRACE 2024/07/11 20:38:41 recv zjb: [2,"ynlkd5yyxzbu3rnhfi0zcgyx","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Preparing","timestamp":"2024-07-11T10:38:41Z"}]
[ocpp  ] TRACE 2024/07/11 20:38:41 sent zjb: [3,"ynlkd5yyxzbu3rnhfi0zcgyx",{}]
[ocpp  ] TRACE 2024/07/11 20:38:55 sent zjb: [2,"1058496282","RemoteStartTransaction",{"connectorId":1,"idTag":"evcc","chargingProfile":{"chargingProfileId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}]
[ocpp  ] TRACE 2024/07/11 20:38:56 recv zjb: [3,"1058496282",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/07/11 20:38:57 recv zjb: [2,"h7xm7i0o9h5xmko1iwrs43jw","StartTransaction",{"connectorId":1,"idTag":"evcc","meterStart":519253,"timestamp":"2024-07-11T10:38:56Z"}]
[ocpp  ] TRACE 2024/07/11 20:38:57 sent zjb: [3,"h7xm7i0o9h5xmko1iwrs43jw",{"idTagInfo":{"status":"Accepted"},"transactionId":1}]
[ocpp  ] TRACE 2024/07/11 20:38:58 recv zjb: [2,"5s1rwdl404buu73n5xncda0b","MeterValues",{"connectorId":1,"transactionId":1,"meterValue":[{"timestamp":"2024-07-11T10:38:57Z","sampledValue":[{"value":"519253","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","format":"Raw","unit":"Wh"},{"value":"0","measurand":"Power.Active.Import","format":"Raw","unit":"W"},{"value":"0","phase":"L1-N","measurand":"Current.Import","format":"Raw","unit":"A"},{"value":"6","measurand":"Current.Offered","format":"Raw","unit":"A"},{"value":"231","phase":"L1-N","measurand":"Voltage","format":"Raw","unit":"V"}]}]}]
[ocpp  ] TRACE 2024/07/11 20:38:58 sent zjb: [3,"5s1rwdl404buu73n5xncda0b",{}]
[ocpp  ] TRACE 2024/07/11 20:38:58 recv zjb: [2,"sohyifpdaqunw16jr78occ0m","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Charging","timestamp":"2024-07-11T10:38:58Z"}]
[ocpp  ] TRACE 2024/07/11 20:38:58 sent zjb: [3,"sohyifpdaqunw16jr78occ0m",{}]
[ocpp  ] TRACE 2024/07/11 20:39:19 recv zjb: [2,"pg4sxd0g3dvtjpuylc4ux4ko","MeterValues",{"connectorId":1,"transactionId":1,"meterValue":[{"timestamp":"2024-07-11T10:39:18Z","sampledValue":[{"value":"519257","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","format":"Raw","unit":"Wh"},{"value":"1170","measurand":"Power.Active.Import","format":"Raw","unit":"W"},{"value":"5","phase":"L1-N","measurand":"Current.Import","format":"Raw","unit":"A"},{"value":"6","measurand":"Current.Offered","format":"Raw","unit":"A"},{"value":"228","phase":"L1-N","measurand":"Voltage","format":"Raw","unit":"V"}]}]}]
[ocpp  ] TRACE 2024/07/11 20:39:19 sent zjb: [3,"pg4sxd0g3dvtjpuylc4ux4ko",{}]
[ocpp  ] TRACE 2024/07/11 20:39:22 sent zjb: [2,"104198208","RemoteStopTransaction",{"transactionId":1}]
[ocpp  ] TRACE 2024/07/11 20:39:23 recv zjb: [3,"104198208",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/07/11 20:39:24 recv zjb: [2,"ynb2s8y895n3fyi9s6ibpdy8","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Finishing","timestamp":"2024-07-11T10:39:23Z"}]
[ocpp  ] TRACE 2024/07/11 20:39:24 sent zjb: [3,"ynb2s8y895n3fyi9s6ibpdy8",{}]
[ocpp  ] TRACE 2024/07/11 20:39:25 recv zjb: [2,"bue28hyt0lkni5k4uaeiv108","StopTransaction",{"idTag":"evcc","meterStop":519258,"timestamp":"2024-07-11T10:39:24Z","transactionId":1,"reason":"Remote"}]
[ocpp  ] TRACE 2024/07/11 20:39:25 sent zjb: [3,"bue28hyt0lkni5k4uaeiv108",{"idTagInfo":{"status":"Accepted"}}]
[ocpp  ] TRACE 2024/07/11 20:39:41 sent zjb: [2,"2439188458","RemoteStartTransaction",{"connectorId":1,"idTag":"evcc","chargingProfile":{"chargingProfileId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}]
[ocpp  ] TRACE 2024/07/11 20:39:41 recv zjb: [3,"2439188458",{"status":"Rejected"}]
[ocpp  ] TRACE 2024/07/11 20:39:43 sent zjb: [2,"243495501","RemoteStartTransaction",{"connectorId":1,"idTag":"evcc","chargingProfile":{"chargingProfileId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}]
[ocpp  ] TRACE 2024/07/11 20:39:43 recv zjb: [3,"243495501",{"status":"Rejected"}]
[ocpp  ] TRACE 2024/07/11 20:39:53 sent zjb: [2,"3334999137","RemoteStartTransaction",{"connectorId":1,"idTag":"evcc","chargingProfile":{"chargingProfileId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}]
[ocpp  ] TRACE 2024/07/11 20:39:53 recv zjb: [3,"3334999137",{"status":"Rejected"}]
[ocpp  ] TRACE 2024/07/11 20:40:03 sent zjb: [2,"3699098649","RemoteStartTransaction",{"connectorId":1,"idTag":"evcc","chargingProfile":{"chargingProfileId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}]
```

andig, 2024-07-11:
And what's your config? Make sure to test latest commit (or use `type: ocpp`). https://github.com/evcc-io/evcc/issues/14804#issuecomment-2222603096 is not sufficient.

ross-w, 2024-07-11:
Ah, you've added a new commit. I've just rebuilt with b2c8ecddebb41079c2e7443e2564f1adbae3084f.

Unfortunately, same results. Config:
```
chargers:
  - name: zjbeny
    type: template
    template: ocpp
    stationid: zjb
    getconfiguration: false
    timeout: 2m
    nostop: true
  ```

ross-w, 2024-07-11:
I can confirm that with b2c8ecdd I am able to stop and start charging again. There is however a bit of a glitch.

Using the following sequence:
* Evcc set to "solar" (it's night now so there is no solar)
* Plug in car
* Set to Min+Solar (car will start charging)
* Set to "off" (car stops charging, charger does not beep indicating end of transaction, good!)
* Set to Min+Solar

The display will remain on "Ready. Waiting for vehicle…". If I instead repeat the sequence but change the last step to "Fast", it will once again charge.

I can repeat this without unplugging the car and happily switch between "Fast" and "off" successfully. But not Min+Solar

ross-w, 2024-07-11:
And to clarify, if I switch to `Min+Solar` /after/ switching to `Fast`, it will work as well. Just going from `Off` to `Min+Solar` triggers the issue.

ross-w, 2024-07-11:
OCPP log from above sequence:
```
[ocpp  ] DEBUG 2024/07/11 22:23:15 charge point connected: zjb
[ocpp  ] TRACE 2024/07/11 22:23:15 recv zjb: [2,"qa3d41m7ebxtzf523hk13egz","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available","timestamp":"2024-07-11T12:23:14Z"}]
[ocpp  ] TRACE 2024/07/11 22:23:15 recv zjb: [2,"t25z471w4d10ftw4on4a8iop","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Preparing","timestamp":"2024-07-11T12:23:14Z"}]
[ocpp  ] TRACE 2024/07/11 22:23:15 sent zjb: [3,"t25z471w4d10ftw4on4a8iop",{}]
[ocpp  ] TRACE 2024/07/11 22:23:36 sent zjb: [2,"1327262371","RemoteStartTransaction",{"connectorId":1,"idTag":"evcc","chargingProfile":{"chargingProfileId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}]
[ocpp  ] TRACE 2024/07/11 22:23:37 recv zjb: [3,"1327262371",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/07/11 22:23:38 recv zjb: [2,"maibibo52pgjtcgv4rc9lilg","StartTransaction",{"connectorId":1,"idTag":"evcc","meterStart":519576,"timestamp":"2024-07-11T12:23:37Z"}]
[ocpp  ] TRACE 2024/07/11 22:23:38 sent zjb: [3,"maibibo52pgjtcgv4rc9lilg",{"idTagInfo":{"status":"Accepted"},"transactionId":1}]
[ocpp  ] TRACE 2024/07/11 22:23:39 recv zjb: [2,"ms1sfkl03y9mwuk9d08lg66f","MeterValues",{"connectorId":1,"transactionId":1,"meterValue":[{"timestamp":"2024-07-11T12:23:38Z","sampledValue":[{"value":"519576","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","format":"Raw","unit":"Wh"},{"value":"0","measurand":"Power.Active.Import","format":"Raw","unit":"W"},{"value":"0","phase":"L1-N","measurand":"Current.Import","format":"Raw","unit":"A"},{"value":"6","measurand":"Current.Offered","format":"Raw","unit":"A"},{"value":"231","phase":"L1-N","measurand":"Voltage","format":"Raw","unit":"V"}]}]}]
[ocpp  ] TRACE 2024/07/11 22:23:39 sent zjb: [3,"ms1sfkl03y9mwuk9d08lg66f",{}]
[ocpp  ] TRACE 2024/07/11 22:23:39 recv zjb: [2,"itoxwwrwqo0y1krow4ewoq9g","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Charging","timestamp":"2024-07-11T12:23:39Z"}]
[ocpp  ] TRACE 2024/07/11 22:23:39 sent zjb: [3,"itoxwwrwqo0y1krow4ewoq9g",{}]
[ocpp  ] TRACE 2024/07/11 22:23:59 recv zjb: [2,"d5wq5u1ag280kw54qvcp9m9p","MeterValues",{"connectorId":1,"transactionId":1,"meterValue":[{"timestamp":"2024-07-11T12:23:58Z","sampledValue":[{"value":"519580","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","format":"Raw","unit":"Wh"},{"value":"1182","measurand":"Power.Active.Import","format":"Raw","unit":"W"},{"value":"5","phase":"L1-N","measurand":"Current.Import","format":"Raw","unit":"A"},{"value":"6","measurand":"Current.Offered","format":"Raw","unit":"A"},{"value":"229","phase":"L1-N","measurand":"Voltage","format":"Raw","unit":"V"}]}]}]
[ocpp  ] TRACE 2024/07/11 22:23:59 sent zjb: [3,"d5wq5u1ag280kw54qvcp9m9p",{}]
[ocpp  ] TRACE 2024/07/11 22:24:00 sent zjb: [2,"2918442925","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":1,"transactionId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":0}]}}}]
[ocpp  ] TRACE 2024/07/11 22:24:00 recv zjb: [3,"2918442925",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/07/11 22:24:02 recv zjb: [2,"x3mdxbmjnt6go7d9ccaoxx7t","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"SuspendedEVSE","timestamp":"2024-07-11T12:24:01Z"}]
[ocpp  ] TRACE 2024/07/11 22:24:02 sent zjb: [3,"x3mdxbmjnt6go7d9ccaoxx7t",{}]
[ocpp  ] TRACE 2024/07/11 22:24:13 recv zjb: [2,"i6ump4hk9ib8m3si5rzbvr98","MeterValues",{"connectorId":1,"transactionId":1,"meterValue":[{"timestamp":"2024-07-11T12:24:13Z","sampledValue":[{"value":"519581","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","format":"Raw","unit":"Wh"},{"value":"0","measurand":"Power.Active.Import","format":"Raw","unit":"W"},{"value":"0","phase":"L1-N","measurand":"Current.Import","format":"Raw","unit":"A"},{"value":"6","measurand":"Current.Offered","format":"Raw","unit":"A"},{"value":"231","phase":"L1-N","measurand":"Voltage","format":"Raw","unit":"V"}]}]}]
[ocpp  ] TRACE 2024/07/11 22:24:13 sent zjb: [3,"i6ump4hk9ib8m3si5rzbvr98",{}]
[ocpp  ] TRACE 2024/07/11 22:24:43 recv zjb: [2,"1m4s64bjfvt13a7a04euny9e","MeterValues",{"connectorId":1,"transactionId":1,"meterValue":[{"timestamp":"2024-07-11T12:24:43Z","sampledValue":[{"value":"519581","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","format":"Raw","unit":"Wh"},{"value":"0","measurand":"Power.Active.Import","format":"Raw","unit":"W"},{"value":"0","phase":"L1-N","measurand":"Current.Import","format":"Raw","unit":"A"},{"value":"6","measurand":"Current.Offered","format":"Raw","unit":"A"},{"value":"231","phase":"L1-N","measurand":"Voltage","format":"Raw","unit":"V"}]}]}]
[ocpp  ] TRACE 2024/07/11 22:24:43 sent zjb: [3,"1m4s64bjfvt13a7a04euny9e",{}]
```
Final step of setting to minpv happened at 22:24:12:
```
[lp-1  ] DEBUG 2024/07/11 22:24:12 charger enable
[cache ] TRACE 2024/07/11 22:24:12 lp-1/effectivePlanTime: 0001-01-01 00:00:00 +0000 UTC
[cache ] TRACE 2024/07/11 22:24:12 lp-1/effectivePlanSoc: 0
[cache ] TRACE 2024/07/11 22:24:12 lp-1/effectiveMinCurrent: 6
[cache ] TRACE 2024/07/11 22:24:12 lp-1/effectiveMaxCurrent: 32
[... cut ...]
[lp-1  ] DEBUG 2024/07/11 22:24:12 wake-up timer: start
[cache ] TRACE 2024/07/11 22:24:12 lp-1/mode: minpv
[cache ] TRACE 2024/07/11 22:24:12 lp-1/planProjectedStart: 0001-01-01 00:00:00 +0000 UTC
[cache ] TRACE 2024/07/11 22:24:12 lp-1/planProjectedEnd: 0001-01-01 00:00:00 +0000 UTC
[cache ] TRACE 2024/07/11 22:24:12 lp-1/planOverrun: 0s
[cache ] TRACE 2024/07/11 22:24:12 lp-1/enabled: true
```
But I don't see any corresponding OCPP messages following that timestamp (which is consistent with the charger not starting)

andig, 2024-07-11:
The mental load to interleave different logs is too much for me :(. Can we please look at a single log?

andig, 2024-07-11:
@ross-w  could your give https://github.com/evcc-io/evcc/pull/14824 a try? Fixed a bug that I could reproduce.

ross-w, 2024-07-12:
@andig sorry for the delay. The latest version does not start charging. When I switch to min+solar I get this repeatedly in the log:
```
[lp-1 ] ERROR 2024/07/12 22:31:31 max charge current 6A: no transaction running 
```

andig, 2024-07-12:
Do you have a log? Your charger should start the transaction and hence have one?

ross-w, 2024-07-13:
@andig the whole issue here is that the charger requires evcc to start the transaction (it just can't end.. hence this `noStop` solution)

With the latest commits, it does not appear that evcc even attempts to start the transaction. The previous one worked almost perfectly aside from the `minpv` issue. I wonder if that is to do with some retention of previous charge rate?

 log from latest commit follows (plugging in car, then set to minpv):
```
[lp-1 ] DEBUG 2024/07/13 20:10:33 charge power: 0W
[site ] DEBUG 2024/07/13 20:10:33 pv power: 0W
[site ] DEBUG 2024/07/13 20:10:33 grid meter: 1302W
[site ] DEBUG 2024/07/13 20:10:33 site power: 1302W
[lp-1 ] DEBUG 2024/07/13 20:10:33 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1 ] DEBUG 2024/07/13 20:10:33 charger status: A
[lp-1 ] DEBUG 2024/07/13 20:10:33 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[ocpp ] TRACE 2024/07/13 20:10:33 recv zjb: [2,"gx7icm4zyoofqtzgssxzlcz6","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Preparing","timestamp":"2024-07-13T10:10:32Z"}]
[ocpp ] TRACE 2024/07/13 20:10:33 sent zjb: [3,"gx7icm4zyoofqtzgssxzlcz6",{}]
[site ] DEBUG 2024/07/13 20:10:43 ----
[lp-1 ] DEBUG 2024/07/13 20:10:43 charge power: 0W
[site ] DEBUG 2024/07/13 20:10:43 pv power: 0W
[site ] DEBUG 2024/07/13 20:10:43 grid meter: 1289W
[site ] DEBUG 2024/07/13 20:10:43 site power: 1289W
[lp-1 ] DEBUG 2024/07/13 20:10:43 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1 ] DEBUG 2024/07/13 20:10:43 charger status: B
[lp-1 ] INFO 2024/07/13 20:10:43 car connected
[lp-1 ] DEBUG 2024/07/13 20:10:43 pv timer elapse
[lp-1 ] DEBUG 2024/07/13 20:10:43 pv timer inactive
[lp-1 ] DEBUG 2024/07/13 20:10:43 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1 ] DEBUG 2024/07/13 20:10:44 vehicle soc: 60%
[lp-1 ] DEBUG 2024/07/13 20:10:44 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1 ] DEBUG 2024/07/13 20:10:44 pv charge current: 0A = 0A + -5.6A (1289W @ 1p)
[lp-1 ] DEBUG 2024/07/13 20:10:44 pv enable timer reset
[lp-1 ] DEBUG 2024/07/13 20:10:44 pv timer inactive
[lp-1 ] DEBUG 2024/07/13 20:10:44 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[site ] DEBUG 2024/07/13 20:10:53 ----
[lp-1 ] DEBUG 2024/07/13 20:10:53 charge power: 0W
[site ] DEBUG 2024/07/13 20:10:53 pv power: 0W
[site ] DEBUG 2024/07/13 20:10:53 grid meter: 1286W
[site ] DEBUG 2024/07/13 20:10:53 site power: 1286W
[lp-1 ] DEBUG 2024/07/13 20:10:53 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1 ] DEBUG 2024/07/13 20:10:53 charger status: B
[lp-1 ] DEBUG 2024/07/13 20:10:53 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1 ] DEBUG 2024/07/13 20:10:53 pv charge current: 0A = 0A + -5.59A (1286W @ 1p)
[lp-1 ] DEBUG 2024/07/13 20:10:53 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1 ] DEBUG 2024/07/13 20:10:58 set charge mode: minpv
[site ] DEBUG 2024/07/13 20:10:58 ----
[lp-1 ] DEBUG 2024/07/13 20:10:58 charge power: 0W
[site ] DEBUG 2024/07/13 20:10:58 pv power: 0W
[site ] DEBUG 2024/07/13 20:10:58 grid meter: 1297W
[site ] DEBUG 2024/07/13 20:10:58 site power: 1297W
[lp-1 ] DEBUG 2024/07/13 20:10:58 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1 ] DEBUG 2024/07/13 20:10:58 charger status: B
[lp-1 ] DEBUG 2024/07/13 20:10:58 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1 ] DEBUG 2024/07/13 20:10:58 pv charge current: min 6A > 0A (1297W @ 1p, battery: false)
[lp-1 ] DEBUG 2024/07/13 20:10:58 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1 ] ERROR 2024/07/13 20:10:58 max charge current 6A: no transaction running
[site ] DEBUG 2024/07/13 20:11:03 ----
[lp-1 ] DEBUG 2024/07/13 20:11:03 charge power: 0W
[site ] DEBUG 2024/07/13 20:11:03 pv power: 0W
[site ] DEBUG 2024/07/13 20:11:03 grid meter: 1297W
[site ] DEBUG 2024/07/13 20:11:03 site power: 1297W
[lp-1 ] DEBUG 2024/07/13 20:11:03 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1 ] DEBUG 2024/07/13 20:11:03 charger status: B
[lp-1 ] DEBUG 2024/07/13 20:11:03 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1 ] DEBUG 2024/07/13 20:11:03 pv charge current: min 6A > 0A (1297W @ 1p, battery: false)
[lp-1 ] DEBUG 2024/07/13 20:11:03 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1 ] ERROR 2024/07/13 20:11:03 max charge current 6A: no transaction running
```

andig, 2024-07-13:
I've updated PR once more. Also: I need the trace log for OCPP.

ross-w, 2024-07-13:
Thanks @andig 

(Trace) log supplied below with latest commit. I've tried to filter out the noisy fluff that isn't relevant (DB, mqtt, etc)
```
[main  ] INFO 2024/07/13 20:35:50 evcc 0.128.2 (54ad0e28)
[main  ] INFO 2024/07/13 20:35:50 using config file: /Users/ross/Projects/evcc/evcc.yaml
[main  ] INFO 2024/07/13 20:35:51 listening at :7070
[zjb-1 ] DEBUG 2024/07/13 20:35:51 waiting for chargepoint: 5m0s
[ocpp  ] DEBUG 2024/07/13 20:35:53 charge point connected: zjb
[ocpp  ] TRACE 2024/07/13 20:35:53 recv zjb: [2,"oja8qfcpm0xdsiu31wken7ma","BootNotification",{"chargePointModel":"BCP-B2N-L","chargePointVendor":"ZJBENY","firmwareVersion":"1.0.23"}]
[ocpp  ] TRACE 2024/07/13 20:35:53 sent zjb: [3,"oja8qfcpm0xdsiu31wken7ma",{"currentTime":"2024-07-13T10:35:53Z","interval":60,"status":"Accepted"}]
[ocpp  ] TRACE 2024/07/13 20:35:53 recv zjb: [2,"b71jp58xjkbq0foux091viaj","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available","timestamp":"2024-07-13T10:35:53Z"}]
[ocpp  ] TRACE 2024/07/13 20:35:54 recv zjb: [2,"iogdwzb2m0uuabxd52pqv6ph","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Preparing","timestamp":"2024-07-13T10:35:53Z"}]
[ocpp  ] TRACE 2024/07/13 20:35:54 sent zjb: [3,"iogdwzb2m0uuabxd52pqv6ph",{}]
[lp-1  ] WARN 2024/07/13 20:35:54 deprecated: maxcurrent setting is ignored, please remove
[lp-1  ] WARN 2024/07/13 20:35:54 deprecated: phases setting is ignored, please remove
[ocpp  ] TRACE 2024/07/13 20:35:54 recv zjb: [2,"p5f3kbujrcz3pp6276o8t8h5","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Preparing","timestamp":"2024-07-13T10:35:54Z"}]
[ocpp  ] TRACE 2024/07/13 20:35:54 sent zjb: [3,"p5f3kbujrcz3pp6276o8t8h5",{}]
[site  ] INFO 2024/07/13 20:35:55 site config:
[site  ] INFO 2024/07/13 20:35:55   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2024/07/13 20:35:55     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/07/13 20:35:55     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/07/13 20:35:55   vehicles:
[site  ] INFO 2024/07/13 20:35:55     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/07/13 20:35:55 loadpoint 1:
[lp-1  ] INFO 2024/07/13 20:35:55   mode:        pv
[lp-1  ] INFO 2024/07/13 20:35:55   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/07/13 20:35:55   meters:      charge ✗
[main  ] TRACE 2024/07/13 20:35:55 GET https://api.github.com/repos/evcc-io/evcc/releases/latest
[lp-1  ] DEBUG 2024/07/13 20:35:55 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/13 20:35:55 phase timer inactive
[lp-1  ] DEBUG 2024/07/13 20:35:55 pv timer inactive
[lp-1  ] INFO 2024/07/13 20:35:55 vehicle updated: unknown -> Ioniq 5
[lp-1  ] DEBUG 2024/07/13 20:35:55 set charge mode: pv
[lp-1  ] DEBUG 2024/07/13 20:35:55 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[site  ] WARN 2024/07/13 20:35:55 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/07/13 20:35:55 ----
[lp-1  ] DEBUG 2024/07/13 20:35:55 charge power: 0W
[site  ] DEBUG 2024/07/13 20:35:55 pv power: 0W
[site  ] DEBUG 2024/07/13 20:35:55 grid meter: 3264W
[site  ] DEBUG 2024/07/13 20:35:55 site power: 3264W
[lp-1  ] DEBUG 2024/07/13 20:35:55 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/13 20:35:55 charger status: B
[lp-1  ] INFO 2024/07/13 20:35:55 car connected
[lp-1  ] DEBUG 2024/07/13 20:35:55 pv timer elapse
[lp-1  ] DEBUG 2024/07/13 20:35:55 pv timer inactive
[lp-1  ] DEBUG 2024/07/13 20:35:55 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[http  ] TRACE 2024/07/13 20:35:56 {"soc_display":60.5,"soc_bms":60.5,"last_soc":1720857946}
[lp-1  ] DEBUG 2024/07/13 20:35:56 vehicle soc: 60%
[lp-1  ] DEBUG 2024/07/13 20:35:56 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/13 20:35:56 pv charge current: 0A = 0A + -14.2A (3264W @ 1p)
[lp-1  ] DEBUG 2024/07/13 20:35:56 pv enable timer reset
[lp-1  ] DEBUG 2024/07/13 20:35:56 pv timer inactive
[lp-1  ] DEBUG 2024/07/13 20:35:56 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[site  ] DEBUG 2024/07/13 20:35:56 ----
[lp-1  ] DEBUG 2024/07/13 20:35:56 charge power: 0W
[site  ] DEBUG 2024/07/13 20:35:56 pv power: 0W
[site  ] DEBUG 2024/07/13 20:35:56 grid meter: 3264W
[site  ] DEBUG 2024/07/13 20:35:56 site power: 3264W
[lp-1  ] DEBUG 2024/07/13 20:35:56 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/13 20:35:56 charger status: B
[lp-1  ] DEBUG 2024/07/13 20:35:56 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/13 20:35:56 pv charge current: 0A = 0A + -14.2A (3264W @ 1p)
[lp-1  ] DEBUG 2024/07/13 20:35:56 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[site  ] DEBUG 2024/07/13 20:36:05 ----
[lp-1  ] DEBUG 2024/07/13 20:36:05 charge power: 0W
[site  ] DEBUG 2024/07/13 20:36:05 pv power: 0W
[site  ] DEBUG 2024/07/13 20:36:05 grid meter: 3610W
[site  ] DEBUG 2024/07/13 20:36:05 site power: 3610W
[lp-1  ] DEBUG 2024/07/13 20:36:05 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[... 104 lines cut ...]
[... cut ...]
[lp-1  ] DEBUG 2024/07/13 20:37:45 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/13 20:37:45 charger status: B
[lp-1  ] DEBUG 2024/07/13 20:37:45 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/13 20:37:45 pv charge current: min 6A > 0A (4906W @ 1p, battery: false)
[lp-1  ] DEBUG 2024/07/13 20:37:45 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] ERROR 2024/07/13 20:37:45 max charge current 6A: no transaction running
```

andig, 2024-07-13:
Pls once more- our internal testing is happy now :)

ross-w, 2024-07-13:
Thanks @andig ! This one is nearly working again. It has the same behaviour as before, but I've now realised it's not to do with `minpv` but rather whatever it was last set to. I can reproduce the issue with `fast` as well. It seems to be that if I set the charger to whatever the current was before I set it to off, it gets stuck on "Ready, waiting for vehicle". If I set something else, it works.

Eg: Off->Minpv->Off->Minpv fails, Off->Minpv->Off->Fast works. Off->Fast->Off->Fast fails, Off->Fast->Off->Minpv works.

Trace log:
```
[main  ] INFO 2024/07/13 21:04:43 evcc 0.128.2 (0ff9f38b)
[main  ] INFO 2024/07/13 21:04:43 using config file: /Users/ross/Projects/evcc/evcc.yaml
[main  ] INFO 2024/07/13 21:04:44 listening at :7070
[zjb-1 ] DEBUG 2024/07/13 21:04:44 waiting for chargepoint: 5m0s
[ocpp  ] DEBUG 2024/07/13 21:04:49 charge point connected: zjb
[ocpp  ] TRACE 2024/07/13 21:04:49 recv zjb: [2,"o30df15h518x0f8wej8yww1i","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available","timestamp":"2024-07-13T11:04:48Z"}]
[ocpp  ] TRACE 2024/07/13 21:04:50 recv zjb: [2,"e1ten2lo4u827xb9l8iows5h","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Preparing","timestamp":"2024-07-13T11:04:48Z"}]
[ocpp  ] TRACE 2024/07/13 21:04:50 send zjb: [3,"e1ten2lo4u827xb9l8iows5h",{}]
[lp-1  ] WARN 2024/07/13 21:04:50 deprecated: maxcurrent setting is ignored, please remove
[lp-1  ] WARN 2024/07/13 21:04:50 deprecated: phases setting is ignored, please remove
[lp-1  ] INFO 2024/07/13 21:04:52 loadpoint 1:
[lp-1  ] INFO 2024/07/13 21:04:52   mode:        pv
[lp-1  ] INFO 2024/07/13 21:04:52   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/07/13 21:04:52   meters:      charge ✗
[main  ] TRACE 2024/07/13 21:04:52 GET https://api.github.com/repos/evcc-io/evcc/releases/latest
[lp-1  ] DEBUG 2024/07/13 21:04:52 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/13 21:04:52 phase timer inactive
[lp-1  ] DEBUG 2024/07/13 21:04:52 pv timer inactive
[lp-1  ] INFO 2024/07/13 21:04:52 vehicle updated: unknown -> Ioniq 5
[lp-1  ] DEBUG 2024/07/13 21:04:52 set charge mode: pv
[lp-1  ] DEBUG 2024/07/13 21:04:52 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/13 21:04:52 charge power: 0W
[lp-1  ] DEBUG 2024/07/13 21:04:53 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/13 21:04:53 charger status: B
[lp-1  ] INFO 2024/07/13 21:04:53 car connected
[lp-1  ] DEBUG 2024/07/13 21:04:53 pv timer elapse
[lp-1  ] DEBUG 2024/07/13 21:04:53 pv timer inactive
[lp-1  ] DEBUG 2024/07/13 21:04:53 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/13 21:04:54 vehicle soc: 60%
[lp-1  ] DEBUG 2024/07/13 21:04:54 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/13 21:04:54 pv charge current: 0A = 0A + -6.09A (1400W @ 1p)
[lp-1  ] DEBUG 2024/07/13 21:04:54 pv enable timer reset
[lp-1  ] DEBUG 2024/07/13 21:04:54 pv timer inactive
[lp-1  ] DEBUG 2024/07/13 21:04:54 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/13 21:04:54 charge power: 0W
[lp-1  ] DEBUG 2024/07/13 21:04:54 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/13 21:04:54 charger status: B
[lp-1  ] DEBUG 2024/07/13 21:04:54 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/13 21:04:54 pv charge current: 0A = 0A + -6.09A (1400W @ 1p)
[lp-1  ] DEBUG 2024/07/13 21:04:54 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/13 21:05:02 charge power: 0W
[lp-1  ] DEBUG 2024/07/13 21:05:02 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/13 21:05:02 charger status: B
[lp-1  ] DEBUG 2024/07/13 21:05:02 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/13 21:05:02 pv charge current: 0A = 0A + -6.09A (1400W @ 1p)
[lp-1  ] DEBUG 2024/07/13 21:05:02 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/13 21:05:06 set charge mode: minpv
[lp-1  ] DEBUG 2024/07/13 21:05:06 charge power: 0W
[lp-1  ] DEBUG 2024/07/13 21:05:06 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/13 21:05:06 charger status: B
[lp-1  ] DEBUG 2024/07/13 21:05:06 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/13 21:05:06 pv charge current: min 6A > 0A (1351W @ 1p, battery: false)
[lp-1  ] DEBUG 2024/07/13 21:05:06 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/13 21:05:06 max charge current: 6A
[lp-1  ] DEBUG 2024/07/13 21:05:06 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[ocpp  ] TRACE 2024/07/13 21:05:06 send zjb: [2,"3172667729","RemoteStartTransaction",{"connectorId":1,"idTag":"evcc","chargingProfile":{"chargingProfileId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}]
[ocpp  ] TRACE 2024/07/13 21:05:06 recv zjb: [3,"3172667729",{"status":"Accepted"}]
[... cut ...]
[lp-1  ] INFO 2024/07/13 21:08:02 stop charging <-
[lp-1  ] DEBUG 2024/07/13 21:08:02 !! session: chargedEnergy=0.0 > chargedEnergy=0.0
[lp-1  ] DEBUG 2024/07/13 21:08:02 !! session: start=0.000 stop=0.000 chargedEnergy=0.048
[lp-1  ] DEBUG 2024/07/13 21:08:02 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/13 21:08:02 vehicle soc: 60%
[lp-1  ] DEBUG 2024/07/13 21:08:02 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)

```

andig, 2024-07-13:
@premultiply could you take a look at the test board?

premultiply, 2024-07-13:
From the log above it looks like in the cases where it does not start again no `SetChargingProfile` command is sent for unknown reason.

andig, 2024-07-13:
Indeed. Seems something is short-circuited but I don't see what:

```
[lp-1  ] DEBUG 2024/07/13 21:05:06 set charge mode: minpv
[lp-1  ] DEBUG 2024/07/13 21:05:06 charger status: B
[lp-1  ] DEBUG 2024/07/13 21:05:06 pv charge current: min 6A > 0A (1351W @ 1p, battery: false)
[lp-1  ] DEBUG 2024/07/13 21:05:06 max charge current: 6A
[ocpp  ] TRACE 2024/07/13 21:05:06 send zjb: [2,"3172667729","RemoteStartTransaction",{"connectorId":1,"idTag":"evcc","chargingProfile":{"chargingProfileId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}]
[ocpp  ] TRACE 2024/07/13 21:05:06 recv zjb: [3,"3172667729",{"status":"Accepted"}]
[lp-1  ] DEBUG 2024/07/13 21:05:06 charger enable
[lp-1  ] DEBUG 2024/07/13 21:05:06 wake-up timer: start
[ocpp  ] TRACE 2024/07/13 21:05:08 recv zjb: [2,"zfm1bukdpu774cisba8mqvzj","StartTransaction",{"connectorId":1,"idTag":"evcc","meterStart":547698,"timestamp":"2024-07-13T11:05:07Z"}]
[ocpp  ] TRACE 2024/07/13 21:05:08 send zjb: [3,"zfm1bukdpu774cisba8mqvzj",{"idTagInfo":{"status":"Accepted"},"transactionId":1}]
[ocpp  ] TRACE 2024/07/13 21:05:08 recv zjb: [2,"fv0qg09b0ehmhwum1wmoq1yt","MeterValues",{"connectorId":1,"transactionId":1,"meterValue":[{"timestamp":"2024-07-13T11:05:08Z","sampledValue":[{"value":"547698","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","format":"Raw","unit":"Wh"},{"value":"0","measurand":"Power.Active.Import","format":"Raw","unit":"W"},{"value":"0","phase":"L1-N","measurand":"Current.Import","format":"Raw","unit":"A"},{"value":"6","measurand":"Current.Offered","format":"Raw","unit":"A"},{"value":"232","phase":"L1-N","measurand":"Voltage","format":"Raw","unit":"V"}]}]}]
[ocpp  ] TRACE 2024/07/13 21:05:08 send zjb: [3,"fv0qg09b0ehmhwum1wmoq1yt",{}]
[ocpp  ] TRACE 2024/07/13 21:05:09 recv zjb: [2,"fin8jc72i20yls7qmwpaztr2","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Charging","timestamp":"2024-07-13T11:05:09Z"}]
[ocpp  ] TRACE 2024/07/13 21:05:09 send zjb: [3,"fin8jc72i20yls7qmwpaztr2",{}]
[lp-1  ] DEBUG 2024/07/13 21:05:12 charger status: C
[lp-1  ] INFO 2024/07/13 21:05:12 start charging ->
[lp-1  ] DEBUG 2024/07/13 21:05:12 wake-up timer: stop
[lp-1  ] DEBUG 2024/07/13 21:05:12 charger vehicle id: evcc
[lp-1  ] DEBUG 2024/07/13 21:05:12 pv charge current: min 6A > 0.126A (1351W @ 1p, battery: false)
[lp-1  ] DEBUG 2024/07/13 21:05:22 charger status: C
[lp-1  ] DEBUG 2024/07/13 21:05:22 pv charge current: min 6A > 0A (1401W @ 1p, battery: false)
[ocpp  ] TRACE 2024/07/13 21:05:31 recv zjb: [2,"7dbz1fe03613el6fklm55r9d","MeterValues",{"connectorId":1,"transactionId":1,"meterValue":[{"timestamp":"2024-07-13T11:05:30Z","sampledValue":[{"value":"547702","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","format":"Raw","unit":"Wh"},{"value":"1182","measurand":"Power.Active.Import","format":"Raw","unit":"W"},{"value":"5","phase":"L1-N","measurand":"Current.Import","format":"Raw","unit":"A"},{"value":"6","measurand":"Current.Offered","format":"Raw","unit":"A"},{"value":"230","phase":"L1-N","measurand":"Voltage","format":"Raw","unit":"V"}]}]}]
[ocpp  ] TRACE 2024/07/13 21:05:31 send zjb: [3,"7dbz1fe03613el6fklm55r9d",{}]
[lp-1  ] DEBUG 2024/07/13 21:05:32 charger status: C
[lp-1  ] DEBUG 2024/07/13 21:05:32 pv charge current: min 6A > 0A (2598W @ 1p, battery: false)
[lp-1  ] DEBUG 2024/07/13 21:05:37 set charge mode: off
[lp-1  ] DEBUG 2024/07/13 21:05:37 charger status: C
[ocpp  ] TRACE 2024/07/13 21:05:37 send zjb: [2,"232648269","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":1,"transactionId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":0}]}}}]
[ocpp  ] TRACE 2024/07/13 21:05:37 recv zjb: [3,"232648269",{"status":"Accepted"}]
[lp-1  ] DEBUG 2024/07/13 21:05:37 charger disable
[lp-1  ] DEBUG 2024/07/13 21:05:37 wake-up timer: stop
[ocpp  ] TRACE 2024/07/13 21:05:39 recv zjb: [2,"2wkj9mlfp714wawtemas1v5f","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"SuspendedEVSE","timestamp":"2024-07-13T11:05:38Z"}]
[ocpp  ] TRACE 2024/07/13 21:05:39 send zjb: [3,"2wkj9mlfp714wawtemas1v5f",{}]
[lp-1  ] DEBUG 2024/07/13 21:05:42 charger status: B
[lp-1  ] INFO 2024/07/13 21:05:42 stop charging <-
[lp-1  ] DEBUG 2024/07/13 21:05:49 set charge mode: minpv
[lp-1  ] DEBUG 2024/07/13 21:05:49 charger status: B
[lp-1  ] DEBUG 2024/07/13 21:05:49 pv charge current: min 6A > 0A (1348W @ 1p, battery: false)
[lp-1  ] DEBUG 2024/07/13 21:05:49 charger enable
[lp-1  ] DEBUG 2024/07/13 21:05:49 wake-up timer: start
[ocpp  ] TRACE 2024/07/13 21:05:50 recv zjb: [2,"gyfu6op1bdguzoz3bkc9zbzc","MeterValues",{"connectorId":1,"transactionId":1,"meterValue":[{"timestamp":"2024-07-13T11:05:50Z","sampledValue":[{"value":"547704","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","format":"Raw","unit":"Wh"},{"value":"0","measurand":"Power.Active.Import","format":"Raw","unit":"W"},{"value":"0","phase":"L1-N","measurand":"Current.Import","format":"Raw","unit":"A"},{"value":"6","measurand":"Current.Offered","format":"Raw","unit":"A"},{"value":"231","phase":"L1-N","measurand":"Voltage","format":"Raw","unit":"V"}]}]}]
[ocpp  ] TRACE 2024/07/13 21:05:50 send zjb: [3,"gyfu6op1bdguzoz3bkc9zbzc",{}]
```

andig, 2024-07-13:
Ahhh: found and fixed: https://github.com/evcc-io/evcc/pull/14824/commits/68edb170a0c5d3186dd49116bd8a5157a8722ebd. Please retry.

ross-w, 2024-07-13:
@andig you're a legend! It works perfectly now, can't fault it at all! Thanks so much for your patience, persistence, and expertise

andig, 2024-07-13:
Thank you for the patient testing and logging!

ross-w, 2024-07-15:
Did something change after this was merged in? This branch was working fine, but Nightly is failing to start for me. If there's a better issue for this let me know
```

[ocpp ] TRACE 2024/07/15 07:30:12 recv zjb: [2,"1bf2iedxpuwucdr0z6cb266b","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Available","timestamp":"2024-07-15T05:30:11Z"}]
[ocpp ] TRACE 2024/07/15 07:30:12 send zjb: [3,"1bf2iedxpuwucdr0z6cb266b",{}]
[ocpp ] TRACE 2024/07/15 07:30:14 recv zjb: [2,"o43cmejh5xs7lkl2yevk4uit","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Preparing","timestamp":"2024-07-15T05:30:13Z"}]
[ocpp ] TRACE 2024/07/15 07:30:14 send zjb: [3,"o43cmejh5xs7lkl2yevk4uit",{}]
[ocpp ] TRACE 2024/07/15 07:31:14 recv zjb: [2,"f04yooy4z73n8dh1wnsn5pk6","Heartbeat",{}]
[ocpp ] TRACE 2024/07/15 07:31:14 send zjb: [3,"f04yooy4z73n8dh1wnsn5pk6",{"currentTime":"2024-07-15T05:31:14Z"}]
[ocpp ] TRACE 2024/07/15 07:31:58 send zjb: [2,"969918320","RemoteStartTransaction",{"connectorId":1,"idTag":"evcc","chargingProfile":{"chargingProfileId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":31}]}}}]
[ocpp ] TRACE 2024/07/15 07:31:58 recv zjb: [3,"969918320",{"status":"Accepted"}]
[ocpp ] TRACE 2024/07/15 07:32:00 recv zjb: [2,"4u412y3jubcpikd7ccn3hdlw","StartTransaction",{"connectorId":1,"idTag":"evcc","meterStart":547890,"timestamp":"2024-07-15T05:31:59Z"}]
[ocpp ] TRACE 2024/07/15 07:32:00 send zjb: [3,"4u412y3jubcpikd7ccn3hdlw",{"idTagInfo":{"status":"Blocked"},"transactionId":0}]
[ocpp ] TRACE 2024/07/15 07:32:00 recv zjb: [2,"dk3dl3ohjnzz7upjh0amkfo3","StopTransaction",{"idTag":"evcc","meterStop":547890,"timestamp":"2024-07-15T05:32:00Z","transactionId":0,"reason":"DeAuthorized"}]
[ocpp ] TRACE 2024/07/15 07:32:00 send zjb: [3,"dk3dl3ohjnzz7upjh0amkfo3",{"idTagInfo":{"status":"Accepted"}}]
[ocpp ] TRACE 2024/07/15 07:32:02 recv zjb: [2,"e7p1mqkthmep35h4pemr6u0f","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Finishing","timestamp":"2024-07-15T05:32:01Z"}]
[ocpp ] TRACE 2024/07/15 07:32:02 send zjb: [3,"e7p1mqkthmep35h4pemr6u0f",{}]
```

andig, 2024-07-15:
@premultiply reverted #14861. The trace above shows that the `StartSession` only comes once `RemoteStart` has finished. Let's keep the local sessions then which has the additional advantage of always getting RFID if exists.

@ross-w new nightly in 15min

ross-w, 2024-07-15:
Thanks @andig , the new nightly is working :)

andig, 2024-07-16:
Good morning, could you give the nightly another try? @premultiply has done a bunch of simplifications that should further reduce issues (and I hope we didn't introduce new ones). Note: with https://github.com/evcc-io/evcc/pull/14888 `nostop` will likely vanish again and be no longer required.

ross-w, 2024-07-16:
Thanks @andig   latest nightly (dca36128) is still working fine. Did you want me to test #14888 as well? I can do it later tonight

andig, 2024-07-20:
@premultiply is PR ready for testing?

andig, 2024-07-28:
@ross-w would be great if you could test, only cosmetic changes to be done.

andig, 2024-08-06:
@ross-w currently nightly would be great to get another test- thank you!

ross-w, 2024-08-06:
@andig thanks for that. Testing with latest nightly, I needed to change my config to add `remoteStart` and remove `noStop`, but it does now work. It does have a long delay (about a minute) between selecting to charge and the charge starting (unlike the current released version using `noStop`), but it does work. Stopping is immediate.

Trace logs for ocpp below, with charge mode logs included.

```
[ocpp  ] DEBUG 2024/08/06 12:41:34 charge point connected: zjb
[ocpp  ] TRACE 2024/08/06 12:41:34 recv zjb: [2,"cxmrujzmlrcaszwx4en1ulz8","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available","timestamp":"2024-08-06T10:41:33Z"}]
[ocpp  ] TRACE 2024/08/06 12:41:34 recv zjb: [2,"4a7iiw2h6ayrfoq2nsu148mb","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Preparing","timestamp":"2024-08-06T10:41:33Z"}]
[ocpp  ] TRACE 2024/08/06 12:41:34 send zjb: [3,"4a7iiw2h6ayrfoq2nsu148mb",{}]
[ocpp  ] TRACE 2024/08/06 12:41:35 send zjb: [2,"753714930","GetConfiguration",{}]
[ocpp  ] TRACE 2024/08/06 12:41:35 recv zjb: [3,"753714930",{"configurationKey":[{"key":"AllowOfflineTxForUnknownId","readonly":false,"value":"true"},{"key":"AuthorizationCacheEnabled","readonly":false,"value":"true"},{"key":"AuthorizeRemoteTxRequests","readonly":false,"value":"false"},{"key":"ClockAlignedDataInterval","readonly":false,"value":"0"},{"key":"ConnectionTimeOut","readonly":false,"value":"120"},{"key":"GetConfigurationMaxKeys","readonly":true,"value":"10"},{"key":"HeartbeatInterval","readonly":false,"value":"60"},{"key":"LocalAuthorizeOffline","readonly":false,"value":"true"},{"key":"LocalPreAuthorize","readonly":false,"value":"true"},{"key":"MeterValueSampleInterval","readonly":false,"value":"10"},{"key":"MinimumStatusDuration","readonly":false,"value":"1"},{"key":"NumberOfConnectors","readonly":true,"value":"1"},{"key":"ResetRetries","readonly":false,"value":"1"},{"key":"StopTransactionOnEVSideDisconnect","readonly":false,"value":"true"},{"key":"StopTransactionOnInvalidId","readonly":false,"value":"true"},{"key":"TransactionMessageAttempts","readonly":false,"value":"3"},{"key":"TransactionMessageRetryInterval","readonly":false,"value":"5"},{"key":"UnlockConnectorOnEVSideDisconnect","readonly":false,"value":"true"},{"key":"WebSocketPingInterval","readonly":false,"value":"30"},{"key":"LocalAuthListEnabled","readonly":false,"value":"true"},{"key":"LocalAuthListMaxLength","readonly":true,"value":"200"},{"key":"SendLocalListMaxLength","readonly":true,"value":"10"},{"key":"ReserveConnectorZeroSupported","readonly":true,"value":"true"},{"key":"ChargeProfileMaxStackLevel","readonly":true,"value":"10"},{"key":"ChargingScheduleAllowedChargingRateUnit","readonly":true,"value":"Current,Power"},{"key":"ChargingScheduleMaxPeriods","readonly":true,"value":"10"},{"key":"ConnectorSwitch3to1PhaseSupported","readonly":true,"value":"false"},{"key":"MaxChargingProfilesInstalled","readonly":true,"value":"20"}]}]
[ocpp  ] TRACE 2024/08/06 12:41:35 send zjb: [2,"2332058160","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Import.Register"}]
[ocpp  ] TRACE 2024/08/06 12:41:35 recv zjb: [3,"2332058160",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/06 12:41:35 send zjb: [2,"1383255008","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Active.Import"}]
[ocpp  ] TRACE 2024/08/06 12:41:36 recv zjb: [3,"1383255008",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/06 12:41:36 send zjb: [2,"4102566411","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"SoC"}]
[ocpp  ] TRACE 2024/08/06 12:41:36 recv zjb: [3,"4102566411",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/06 12:41:36 send zjb: [2,"208787011","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Offered"}]
[ocpp  ] TRACE 2024/08/06 12:41:37 recv zjb: [3,"208787011",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/06 12:41:37 send zjb: [2,"2131284993","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Offered"}]
[ocpp  ] TRACE 2024/08/06 12:41:37 recv zjb: [3,"2131284993",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/06 12:41:37 send zjb: [2,"908884930","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Import"}]
[ocpp  ] TRACE 2024/08/06 12:41:37 recv zjb: [3,"908884930",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/06 12:41:37 send zjb: [2,"4135063379","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Voltage"}]
[ocpp  ] TRACE 2024/08/06 12:41:37 recv zjb: [3,"4135063379",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/06 12:41:37 send zjb: [2,"1108548328","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Import.Register"}]
[ocpp  ] TRACE 2024/08/06 12:41:37 recv zjb: [3,"1108548328",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/06 12:41:37 send zjb: [2,"3679269059","ChangeConfiguration",{"key":"MeterValueSampleInterval","value":"10"}]
[ocpp  ] TRACE 2024/08/06 12:41:37 recv zjb: [3,"3679269059",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/06 12:41:37 send zjb: [2,"3925561591","ChangeConfiguration",{"key":"WebSocketPingInterval","value":"30"}]
[ocpp  ] TRACE 2024/08/06 12:41:37 recv zjb: [3,"3925561591",{"status":"Accepted"}]
[lp-1  ] DEBUG 2024/08/06 12:41:40 set charge mode: pv
[ocpp  ] TRACE 2024/08/06 12:41:40 send zjb: [2,"3761197280","GetCompositeSchedule",{"connectorId":1,"duration":60}]
[... cut ...]
[ocpp  ] TRACE 2024/08/06 12:43:12 send zjb: [3,"4u83wvmgobmsk8nzgixxb0oj",{}]
[ocpp  ] TRACE 2024/08/06 12:43:13 recv zjb: [2,"nfu3hauavpaqsiqxjb6542ww","MeterValues",{"connectorId":1,"transactionId":1,"meterValue":[{"timestamp":"2024-08-06T10:43:12Z","sampledValue":[{"value":"714068","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","format":"Raw","unit":"Wh"},{"value":"1166","measurand":"Power.Active.Import","format":"Raw","unit":"W"}]}]}]
[ocpp  ] TRACE 2024/08/06 12:43:13 send zjb: [3,"nfu3hauavpaqsiqxjb6542ww",{}]
```

andig, 2024-08-06:
> t does have a long delay (about a minute) between selecting to charge and the charge starting (unlike the current released version using noStop)

Is that visible in the logs? /cc @premultiply

ross-w, 2024-08-06:
@andig you can see in the log above I selected `minpv` at 12:41:58, then it sent the `SetChargingProfile` request at 12:42:43

andig, 2024-08-06:
Hard to say- there's so little logging for the loadpoint?

ross-w, 2024-08-06:
@andig updated with more detail. I'm trying not to spam you will the full trace log which includes a lot of irrelevant stuff like MQTT, tariff, DB etc

andig, 2024-08-06:
Doesn't charge because there's simply not enough power:

    [lp-1  ] DEBUG 2024/08/06 12:41:42 pv charge current: 0A = 0A + -2.23A (514W @ 1p)

ross-w, 2024-08-06:
@andig sorry for the alarm. I've tried it again and it seems fine. So in short, latest nightly works! There must have been something strange about my MQTT feed right as I booted evcc (it should not be anywhere near its limit at this time of night)

ross-w, 2024-08-09:
Latest nightly appears to have an issue and is perpetually stuck on `waiting for local authentication`. I'll do some more testing tonight and grab some logs
`

ross-w, 2024-08-09:
@andig here's the behaviour on the latest nightly
```
[zjb-1 ] DEBUG 2024/08/09 12:52:12 waiting for chargepoint: 5m0s
[ocpp  ] DEBUG 2024/08/09 12:52:32 charge point connected: zjb
[ocpp  ] TRACE 2024/08/09 12:52:32 recv zjb: [2,"g97zdjqothtwp5bdkjntn2hf","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available","timestamp":"2024-08-09T10:52:33Z"}]
[ocpp  ] TRACE 2024/08/09 12:52:32 send zjb: [3,"g97zdjqothtwp5bdkjntn2hf",{}]
[ocpp  ] TRACE 2024/08/09 12:52:33 recv zjb: [2,"h79w3lkk1wtzmim8u978trbr","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Preparing","timestamp":"2024-08-09T10:52:33Z"}]
[ocpp  ] TRACE 2024/08/09 12:52:33 send zjb: [3,"h79w3lkk1wtzmim8u978trbr",{}]
[ocpp  ] TRACE 2024/08/09 12:52:33 send zjb: [2,"2942027420","GetConfiguration",{}]
[ocpp  ] TRACE 2024/08/09 12:52:33 recv zjb: [3,"2942027420",{"configurationKey":[{"key":"AllowOfflineTxForUnknownId","readonly":false,"value":"true"},{"key":"AuthorizationCacheEnabled","readonly":false,"value":"true"},{"key":"AuthorizeRemoteTxRequests","readonly":false,"value":"false"},{"key":"ClockAlignedDataInterval","readonly":false,"value":"0"},{"key":"ConnectionTimeOut","readonly":false,"value":"120"},{"key":"GetConfigurationMaxKeys","readonly":true,"value":"10"},{"key":"HeartbeatInterval","readonly":false,"value":"60"},{"key":"LocalAuthorizeOffline","readonly":false,"value":"true"},{"key":"LocalPreAuthorize","readonly":false,"value":"true"},{"key":"MeterValueSampleInterval","readonly":false,"value":"10"},{"key":"MinimumStatusDuration","readonly":false,"value":"1"},{"key":"NumberOfConnectors","readonly":true,"value":"1"},{"key":"ResetRetries","readonly":false,"value":"1"},{"key":"StopTransactionOnEVSideDisconnect","readonly":false,"value":"true"},{"key":"StopTransactionOnInvalidId","readonly":false,"value":"true"},{"key":"TransactionMessageAttempts","readonly":false,"value":"3"},{"key":"TransactionMessageRetryInterval","readonly":false,"value":"5"},{"key":"UnlockConnectorOnEVSideDisconnect","readonly":false,"value":"true"},{"key":"WebSocketPingInterval","readonly":false,"value":"30"},{"key":"LocalAuthListEnabled","readonly":false,"value":"true"},{"key":"LocalAuthListMaxLength","readonly":true,"value":"200"},{"key":"SendLocalListMaxLength","readonly":true,"value":"10"},{"key":"ReserveConnectorZeroSupported","readonly":true,"value":"true"},{"key":"ChargeProfileMaxStackLevel","readonly":true,"value":"10"},{"key":"ChargingScheduleAllowedChargingRateUnit","readonly":true,"value":"Current,Power"},{"key":"ChargingScheduleMaxPeriods","readonly":true,"value":"10"},{"key":"ConnectorSwitch3to1PhaseSupported","readonly":true,"value":"false"},{"key":"MaxChargingProfilesInstalled","readonly":true,"value":"20"}]}]
[ocpp  ] TRACE 2024/08/09 12:52:33 send zjb: [2,"549291329","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Import.Register"}]
[ocpp  ] TRACE 2024/08/09 12:52:33 recv zjb: [3,"549291329",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/09 12:52:33 send zjb: [2,"3786760549","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Active.Import"}]
[ocpp  ] TRACE 2024/08/09 12:52:33 recv zjb: [3,"3786760549",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/09 12:52:33 send zjb: [2,"2661204682","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"SoC"}]
[ocpp  ] TRACE 2024/08/09 12:52:34 recv zjb: [3,"2661204682",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/09 12:52:34 send zjb: [2,"3773003073","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Offered"}]
[ocpp  ] TRACE 2024/08/09 12:52:34 recv zjb: [3,"3773003073",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/09 12:52:34 send zjb: [2,"1301889724","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Offered"}]
[ocpp  ] TRACE 2024/08/09 12:52:34 recv zjb: [3,"1301889724",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/09 12:52:34 send zjb: [2,"1370134030","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Import"}]
[ocpp  ] TRACE 2024/08/09 12:52:34 recv zjb: [3,"1370134030",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/09 12:52:34 send zjb: [2,"4264633097","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Voltage"}]
[ocpp  ] TRACE 2024/08/09 12:52:34 recv zjb: [3,"4264633097",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/09 12:52:34 send zjb: [2,"3331998625","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Import.Register"}]
[ocpp  ] TRACE 2024/08/09 12:52:34 recv zjb: [3,"3331998625",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/09 12:52:34 send zjb: [2,"1021215764","ChangeConfiguration",{"key":"MeterValueSampleInterval","value":"10"}]
[ocpp  ] TRACE 2024/08/09 12:52:34 recv zjb: [3,"1021215764",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/09 12:52:34 send zjb: [2,"3840926526","ChangeConfiguration",{"key":"WebSocketPingInterval","value":"30"}]
[ocpp  ] TRACE 2024/08/09 12:52:34 recv zjb: [3,"3840926526",{"status":"Accepted"}]
[lp-1  ] WARN 2024/08/09 12:52:34 deprecated: maxcurrent setting is ignored, please remove
[lp-1  ] WARN 2024/08/09 12:52:34 deprecated: phases setting is ignored, please remove
[lp-1  ] INFO 2024/08/09 12:52:37 loadpoint 1:
[lp-1  ] INFO 2024/08/09 12:52:37   mode:        pv
[lp-1  ] INFO 2024/08/09 12:52:37   charger:     power ✗ energy ✓ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/08/09 12:52:37   meters:      charge ✗
[lp-1  ] DEBUG 2024/08/09 12:52:37 phase timer inactive
[... cut ...]
[lp-1  ] DEBUG 2024/08/09 12:55:47 charger status: B
[ocpp  ] TRACE 2024/08/09 12:55:47 send zjb: [2,"90321759","GetCompositeSchedule",{"connectorId":1,"duration":60}]
[ocpp  ] TRACE 2024/08/09 12:55:47 recv zjb: [3,"90321759",{"status":"Accepted","connectorId":1,"scheduleStart":"2024-08-09T10:55:47Z","chargingSchedule":{"duration":60,"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6.0}]}}]
[lp-1  ] DEBUG 2024/08/09 12:55:47 pv charge current: min 6A > 0A (441W @ 1p, battery: false)
```

premultiply, 2024-08-10:
There is no way to walk into the conditional path where this message is generated if `remotestart: true` is set.

```go
	if c.conn.NeedsAuthentication() {
		if c.remoteStart {
			// lock the cable by starting remote transaction after vehicle connected
			if err := c.initTransaction(); err != nil {
				c.log.WARN.Printf("failed to start remote transaction: %v", err)
			}
		} else {
			// TODO: bring this status to UI
			c.log.WARN.Printf("waiting for local authentication")
		}
	}
```

ross-w, 2024-08-11:
I'm at a loss as to how this is occurring. Looking at the code I don't see how this is possible either. Config for charger is:
```
  - name: zjbeny
    type: template
    template: ocpp
    stationid: zjb
    timeout: 2m
    remotestart: true
  ```
  
  And yet I'm still seeing "waiting for local authentication". Nightly a2e36a2e

andig, 2024-08-11:
While strictly not the same issue lets reopen here to not lose

andig, 2024-08-12:
@ross-w could you retry current nightly? There was a bad bug in https://github.com/evcc-io/evcc/pull/15351 that may have caused this and even worse, it only happened by chance.

ross-w, 2024-08-13:
@andig Unfortunately the behaviour is the same with nightly 9bf8dc7b. Just in case I tried `remotestart` and `remoteStart` but same results with both

andig, 2024-08-13:
@premultiply in der `ocpp.tpl` sehe ich kein `remotestart` mehr. Fehler beim Umbau?

closed 2024-08-13 by andig

premultiply, 2024-08-13:
Sieht ganz so aus, dass das beim auseinander ziehen verschütt' gegangen ist.

pdonovan, 2024-08-13:
As another ZJ Beny owner I'm following this issue closely. I don't have the ability to build nightlies, is there somewhere I can download one for an RPi3?

andig, 2024-08-13:
See docs. We offer nightly builds.

ross-w, 2024-08-16:
@andig did something change? latest nightly 3798d4ea is working again, no more "waiting for local authentication" :) no changes on my end. I'm still seeing weird delays on the first charge after plugging in, but I will try to troubleshoot that

andig, 2024-08-16:
There is https://github.com/evcc-io/evcc/commit/40b6d1b7f3963e4fea8ec9e291024730eaa9208f. Not sure if that makes sense?