evcc-io/evcc #14807: "request canceled" error with Tesla and wallbox gen 3

tiehfood, 2024-07-09:
### Describe the bug

[image]

### Steps to reproduce

controlling Tesla charging with callbox gen 3

### Configuration details

```yaml
network:
  schema: http
  host: ***
  port: ***

database:
  type: sqlite
  dsn: ***

sponsortoken: ***

log: debug
levels:
  cache: error

plant: ***

interval: ***

modbusproxy:
  - port: ***
    uri: ***

meters:
  - type: template
    template: alpha-ess-smile
    id: ***
    host: ***
    port: ***
    usage: ***
    modbus: ***
    name: ***
  - type: template
    template: alpha-ess-smile
    id: ***
    host: ***
    port: ***
    usage: ***
    modbus: ***
    name: ***
  - type: template
    template: alpha-ess-smile
    id: ***
    host: ***
    port: ***
    usage: ***
    modbus: ***
    name: ***

chargers:
  - name: ***
    type: template
    template: twc3
    host: ***

loadpoints:
  - title: ***
    charger: ***
    mode: ***
    phases: ***
    mincurrent: ***
    maxcurrent: ***
    guardduration: ***
    vehicle: ***

site:
  title: ***
  meters:
    grid: ***
    pv:
      - ***
    battery:
      - ***
  residualPower: ***

vehicles:
  - name: ***
    type: template
    template: tesla
    title: ***
    accessToken: ***
    refreshToken: ***
```

### Log details

```text
[lp-1 ] ERROR 2024/07/09 20:30:37 charger disable: Post "https://tesla.evcc.io/api/1/vehicles/***/command/charge_stop": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[site ] DEBUG 2024/07/09 20:30:37 ----
[lp-1 ] DEBUG 2024/07/09 20:30:37 charge power: 5687W
[lp-1 ] DEBUG 2024/07/09 20:30:37 charge currents: [8.1 8.1 8.1]A
[site ] DEBUG 2024/07/09 20:30:37 pv power: 0W
[site ] DEBUG 2024/07/09 20:30:37 battery soc: 78%
[site ] DEBUG 2024/07/09 20:30:37 battery power: 6553W
[site ] DEBUG 2024/07/09 20:30:37 grid meter: 48W
[site ] DEBUG 2024/07/09 20:30:37 grid currents: [1.5 2 -1.6]A
[site ] DEBUG 2024/07/09 20:30:37 site power: 6801W
[lp-1 ] DEBUG 2024/07/09 20:30:37 charge voltages: [235 233 234]V
[lp-1 ] DEBUG 2024/07/09 20:30:37 detected connected phases: 3p
[lp-1 ] DEBUG 2024/07/09 20:30:37 detected active phases: 3p
[lp-1 ] DEBUG 2024/07/09 20:30:37 !! session: chargeRater.chargedEnergy=0.2 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/07/09 20:30:37 charger status: C
[lp-1 ] DEBUG 2024/07/09 20:30:37 vehicle soc: 76%
[lp-1 ] DEBUG 2024/07/09 20:30:37 vehicle soc limit: 69%
[lp-1 ] DEBUG 2024/07/09 20:30:37 vehicle range: 337km
[lp-1 ] ERROR 2024/07/09 20:30:47 charger disable: Post "https://tesla.evcc.io/api/1/vehicles/***/command/charge_stop": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[site ] DEBUG 2024/07/09 20:30:47 ----
[lp-1 ] DEBUG 2024/07/09 20:30:47 charge power: 0W
[lp-1 ] DEBUG 2024/07/09 20:30:47 charge currents: [0.2 0.4 0.3]A
[site ] DEBUG 2024/07/09 20:30:47 pv power: 0W
[site ] DEBUG 2024/07/09 20:30:47 battery soc: 78%
[site ] DEBUG 2024/07/09 20:30:47 battery power: 5108W
[site ] DEBUG 2024/07/09 20:30:47 grid meter: -4185W
[site ] DEBUG 2024/07/09 20:30:47 grid currents: [-5.6 -5.7 -6.8]A
[site ] DEBUG 2024/07/09 20:30:47 site power: 1123W
[lp-1 ] DEBUG 2024/07/09 20:30:47 charge voltages: [7 0 0]V
[lp-1 ] DEBUG 2024/07/09 20:30:47 !! session: chargeRater.chargedEnergy=0.2 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/07/09 20:30:47 charger status: B
[lp-1 ] INFO 2024/07/09 20:30:47 stop charging <-
[lp-1 ] DEBUG 2024/07/09 20:30:47 wake-up timer: start
[lp-1 ] DEBUG 2024/07/09 20:30:47 !! session: chargedEnergy=0.2 > chargedEnergy=0.0
[lp-1 ] DEBUG 2024/07/09 20:30:47 !! session: start=0.000 stop=0.000 chargedEnergy=0.200
[lp-1 ] DEBUG 2024/07/09 20:30:48 vehicle soc: 76%
[lp-1 ] DEBUG 2024/07/09 20:30:48 vehicle soc limit: 69%
[lp-1 ] DEBUG 2024/07/09 20:30:48 vehicle range: 337km
[lp-1 ] WARN 2024/07/09 20:30:48 charger out of sync: expected enabled, got disabled
[lp-1 ] DEBUG 2024/07/09 20:30:48 charger disable
```

### What type of operating system are you running?

Linux

### Version

0.128.1

andig, 2024-07-09:
Duplicate https://github.com/evcc-io/evcc/issues/14226

closed 2024-07-09 by andig

tiehfood, 2024-07-09:
sorry for not looking in the existing issues :(